arkq/bluez-alsa #367: Can't see battery level in alsamixer anymore

alex14641, 2020-08-30:
When I first started messing with bluez alsa, I was able to see the battery level in alsamixer; now I can't make it appear. I'd been doing a lot of rebuilding to diagnose a AptX related crash. This is my current build configuration:

 ../configure --enable-aac --enable-ofono --enable-upower --enable-debug --sysconfdir=/etc --prefix=/usr --libdir=/usr/lib64

Here debug from bluealsa:

/usr/bin/bluealsa: D: ../../src/ba-rfcomm.c:1085: RFCOMM poll timeout
/usr/bin/bluealsa: D: ../../src/dbus.c:57: Called: org.bluez.MediaEndpoint1.ClearConfiguration() on /org/bluez/hci0/A2DP/AAC/Source/1
/usr/bin/bluealsa: D: ../../src/bluez.c:332: Disconnecting media endpoint: /org/bluez/hci0/A2DP/AAC/Source/1
/usr/bin/bluealsa: D: ../../src/ba-rfcomm.c:1212: RFCOMM disconnected: Connection reset by peer
/usr/bin/bluealsa: D: ../../src/ba-rfcomm.c:843: Closing RFCOMM: 9
/usr/bin/bluealsa: D: ../../src/ba-rfcomm.c:852: RFCOMM link lost quirk: Destroying SCO transport
/usr/bin/bluealsa: D: ../../src/bluez.c:1009: Signal: org.freedesktop.DBus.ObjectManager.InterfacesRemoved()
/usr/bin/bluealsa: D: ../../src/bluez.c:1009: Signal: org.freedesktop.DBus.ObjectManager.InterfacesRemoved()
/usr/bin/bluealsa: D: ../../src/bluez.c:1009: Signal: org.freedesktop.DBus.ObjectManager.InterfacesRemoved()
/usr/bin/bluealsa: D: ../../src/bluez.c:1009: Signal: org.freedesktop.DBus.ObjectManager.InterfacesRemoved()
/usr/bin/bluealsa: D: ../../src/ba-transport.c:1054: Exiting IO thread: HFP Audio Gateway (CVSD)
/usr/bin/bluealsa: D: ../../src/bluez.c:1009: Signal: org.freedesktop.DBus.ObjectManager.InterfacesRemoved()
/usr/bin/bluealsa: D: ../../src/bluez.c:1009: Signal: org.freedesktop.DBus.ObjectManager.InterfacesRemoved()
/usr/bin/bluealsa: D: ../../src/bluez.c:1009: Signal: org.freedesktop.DBus.ObjectManager.InterfacesRemoved()
/usr/bin/bluealsa: D: ../../src/dbus.c:57: Called: org.bluez.MediaEndpoint1.Release() on /org/bluez/hci1/A2DP/AAC/Source/1
/usr/bin/bluealsa: D: ../../src/bluez.c:362: Releasing media endpoint: /org/bluez/hci1/A2DP/AAC/Source/1
/usr/bin/bluealsa: D: ../../src/dbus.c:57: Called: org.bluez.MediaEndpoint1.Release() on /org/bluez/hci1/A2DP/AAC/Source/1
/usr/bin/bluealsa: D: ../../src/bluez.c:362: Releasing media endpoint: /org/bluez/hci1/A2DP/AAC/Source/1
/usr/bin/bluealsa: D: ../../src/dbus.c:57: Called: org.bluez.MediaEndpoint1.Release() on /org/bluez/hci1/A2DP/AAC/Source/2
/usr/bin/bluealsa: D: ../../src/bluez.c:362: Releasing media endpoint: /org/bluez/hci1/A2DP/AAC/Source/2
/usr/bin/bluealsa: D: ../../src/dbus.c:57: Called: org.bluez.MediaEndpoint1.Release() on /org/bluez/hci1/A2DP/AAC/Source/2
/usr/bin/bluealsa: D: ../../src/bluez.c:362: Releasing media endpoint: /org/bluez/hci1/A2DP/AAC/Source/2
/usr/bin/bluealsa: D: ../../src/dbus.c:57: Called: org.bluez.MediaEndpoint1.Release() on /org/bluez/hci1/A2DP/SBC/Source/1
/usr/bin/bluealsa: D: ../../src/bluez.c:362: Releasing media endpoint: /org/bluez/hci1/A2DP/SBC/Source/1
/usr/bin/bluealsa: D: ../../src/dbus.c:57: Called: org.bluez.MediaEndpoint1.Release() on /org/bluez/hci1/A2DP/SBC/Source/1
/usr/bin/bluealsa: D: ../../src/bluez.c:362: Releasing media endpoint: /org/bluez/hci1/A2DP/SBC/Source/1
/usr/bin/bluealsa: D: ../../src/dbus.c:57: Called: org.bluez.MediaEndpoint1.Release() on /org/bluez/hci1/A2DP/SBC/Source/2
/usr/bin/bluealsa: D: ../../src/bluez.c:362: Releasing media endpoint: /org/bluez/hci1/A2DP/SBC/Source/2
/usr/bin/bluealsa: D: ../../src/dbus.c:57: Called: org.bluez.MediaEndpoint1.Release() on /org/bluez/hci1/A2DP/SBC/Source/2
/usr/bin/bluealsa: D: ../../src/bluez.c:362: Releasing media endpoint: /org/bluez/hci1/A2DP/SBC/Source/2
/usr/bin/bluealsa: D: ../../src/bluez.c:1009: Signal: org.freedesktop.DBus.ObjectManager.InterfacesRemoved()
/usr/bin/bluealsa: D: ../../src/ba-adapter.c:142: Freeing adapter: hci1
/usr/bin/bluealsa: D: ../../src/bluez.c:915: Signal: org.freedesktop.DBus.ObjectManager.InterfacesAdded()
/usr/bin/bluealsa: D: ../../src/bluez.c:490: Creating media endpoint object: /org/bluez/hci1/A2DP/AAC/Source/1
/usr/bin/bluealsa: D: ../../src/bluez.c:406: Registering media endpoint: /org/bluez/hci1/A2DP/AAC/Source/1
/usr/bin/bluealsa: D: ../../src/bluez.c:490: Creating media endpoint object: /org/bluez/hci1/A2DP/AAC/Source/2
/usr/bin/bluealsa: D: ../../src/bluez.c:406: Registering media endpoint: /org/bluez/hci1/A2DP/AAC/Source/2
/usr/bin/bluealsa: D: ../../src/bluez.c:490: Creating media endpoint object: /org/bluez/hci1/A2DP/SBC/Source/1
/usr/bin/bluealsa: D: ../../src/bluez.c:406: Registering media endpoint: /org/bluez/hci1/A2DP/SBC/Source/1
/usr/bin/bluealsa: D: ../../src/bluez.c:490: Creating media endpoint object: /org/bluez/hci1/A2DP/SBC/Source/2
/usr/bin/bluealsa: D: ../../src/bluez.c:406: Registering media endpoint: /org/bluez/hci1/A2DP/SBC/Source/2
/usr/bin/bluealsa: D: ../../src/bluez.c:915: Signal: org.freedesktop.DBus.ObjectManager.InterfacesAdded()
/usr/bin/bluealsa: D: ../../src/dbus.c:57: Called: org.bluealsa.Manager1.GetPCMs() on /org/bluealsa
/usr/bin/bluealsa: D: ../../src/ba-transport.c:361: Freeing transport: A2DP Source (AAC)
/usr/bin/bluealsa: D: ../../src/ba-transport.c:361: Freeing transport: HFP Audio Gateway (CVSD)
/usr/bin/bluealsa: D: ../../src/ba-device.c:131: Freeing device: CC:98:8B:93:B8:30
/usr/bin/bluealsa: D: ../../src/dbus.c:57: Called: org.bluez.Profile1.NewConnection() on /org/bluez/HFP/AudioGateway
/usr/bin/bluealsa: D: ../../src/ba-rfcomm.c:880: Starting RFCOMM loop: HFP Audio Gateway (CVSD)
/usr/bin/bluealsa: D: ../../src/ba-rfcomm.c:1262: Created new RFCOMM thread [ba-rfcomm]: HFP Audio Gateway (CVSD)
/usr/bin/bluealsa: D: ../../src/bluez.c:609: HFP Audio Gateway (CVSD) configured for device CC:98:8B:93:B8:30
/usr/bin/bluealsa: D: ../../src/ba-transport.c:750: Starting transport: HFP Audio Gateway (CVSD)
/usr/bin/bluealsa: D: ../../src/ba-transport.c:1032: Created new thread [ba-sco]: HFP Audio Gateway (CVSD)
/usr/bin/bluealsa: D: ../../src/sco.c:241: Starting SCO loop: HFP Audio Gateway (CVSD)
/usr/bin/bluealsa: D: ../../src/at.c:161: AT message: SET: command:+BRSF, value:665
/usr/bin/bluealsa: D: ../../src/ba-rfcomm.c:107: Sending AT message: RESP: command:+BRSF, value:2272
/usr/bin/bluealsa: D: ../../src/ba-rfcomm.c:107: Sending AT message: RESP: command:(null), value:OK
/usr/bin/bluealsa: D: ../../src/ba-rfcomm.c:126: RFCOMM: HFP Audio Gateway (CVSD) state transition: 0 -> 2
/usr/bin/bluealsa: D: ../../src/bluez.c:915: Signal: org.freedesktop.DBus.ObjectManager.InterfacesAdded()
/usr/bin/bluealsa: D: ../../src/bluez.c:998: Adding new Stream End-Point: CC:98:8B:93:B8:30: SBC
/usr/bin/bluealsa: D: ../../src/bluez.c:915: Signal: org.freedesktop.DBus.ObjectManager.InterfacesAdded()
/usr/bin/bluealsa: D: ../../src/bluez.c:998: Adding new Stream End-Point: CC:98:8B:93:B8:30: LDAC
/usr/bin/bluealsa: D: ../../src/bluez.c:915: Signal: org.freedesktop.DBus.ObjectManager.InterfacesAdded()
/usr/bin/bluealsa: D: ../../src/bluez.c:998: Adding new Stream End-Point: CC:98:8B:93:B8:30: aptX-HD
/usr/bin/bluealsa: D: ../../src/bluez.c:915: Signal: org.freedesktop.DBus.ObjectManager.InterfacesAdded()
[... 119 lines cut ...]
/usr/bin/bluealsa: D: ../../src/ba-rfcomm.c:1085: RFCOMM poll timeout
/usr/bin/bluealsa: D: ../../src/ba-rfcomm.c:1085: RFCOMM poll timeout
/usr/bin/bluealsa: D: ../../src/ba-rfcomm.c:1085: RFCOMM poll timeout
/usr/bin/bluealsa: D: ../../src/ba-rfcomm.c:1085: RFCOMM poll timeout
/usr/bin/bluealsa: D: ../../src/ba-rfcomm.c:1085: RFCOMM poll timeout
/usr/bin/bluealsa: D: ../../src/ba-rfcomm.c:1085: RFCOMM poll timeout
/usr/bin/bluealsa: D: ../../src/ba-rfcomm.c:1085: RFCOMM poll timeout
/usr/bin/bluealsa: D: ../../src/ba-rfcomm.c:1085: RFCOMM poll timeout
/usr/bin/bluealsa: D: ../../src/ba-rfcomm.c:1085: RFCOMM poll timeout
/usr/bin/bluealsa: D: ../../src/ba-rfcomm.c:1085: RFCOMM poll timeout
/usr/bin/bluealsa: D: ../../src/ba-rfcomm.c:1085: RFCOMM poll timeout
/usr/bin/bluealsa: D: ../../src/ba-rfcomm.c:1085: RFCOMM poll timeout
/usr/bin/bluealsa: D: ../../src/ba-rfcomm.c:1085: RFCOMM poll timeout
/usr/bin/bluealsa: D: ../../src/ba-rfcomm.c:1085: RFCOMM poll timeout
/usr/bin/bluealsa: D: ../../src/ba-rfcomm.c:1085: RFCOMM poll timeout
/usr/bin/bluealsa: D: ../../src/ba-rfcomm.c:1085: RFCOMM poll timeout
/usr/bin/bluealsa: D: ../../src/ba-rfcomm.c:1085: RFCOMM poll timeout
/usr/bin/bluealsa: D: ../../src/ba-rfcomm.c:1085: RFCOMM poll timeout

Software versions:
bluez: 5.54
alsa-lib:  1.2.3.2
alsa-plugins: 1.2.2
alsa-utils: 1.2.3
Linux kernel: 5.4.60

Arkq, 2020-08-31:
The problem might be with repeating `/usr/bin/bluealsa: D: ../../src/ba-rfcomm.c:1085: RFCOMM poll timeout` message. It seems that HFP service level connection (SLC) is not established.

Your headset has sent `AT message: TEST: command:+CIND, value:(null)`, but just after TEST command, there should be CIND GET command - bluealsa is waiting for it. From the headset point of view, if SLC is not established, it might not send battery level status command (via Apple extension). Please try to use HSP instead of HFP. Since HSP does not have service level connection, if I'm right, your headset should send battery status just after HSP profile connection.

alex14641, 2020-08-31:
Tried several combinations of profiles, and what finally worked was any combination with the hfp-ag profile and  without the hsp-ag profile:
/usr/bin/bluealsa -p a2dp-source -p hfp-ag -p hfp-hf           # works
/usr/bin/bluealsa -p a2dp-source -p hfp-ag -p hfp-hf -p hsp-hs # works
/usr/bin/bluealsa -p a2dp-source -p hfp-ag -p hfp-hf -p hsp-hs # works
/usr/bin/bluealsa -p a2dp-source -p hfp-hf -p hsp-hs -p hfp-ag # works
/usr/bin/bluealsa -p a2dp-source -p hfp-hf -p hfp-ag -p hsp-hs # works

Here is debug from a working session:
/usr/bin/bluealsa: D: ../../src/main.c:402: Starting main dispatching loop
/usr/bin/bluealsa: D: ../../src/dbus.c:57: Called: org.bluez.Profile1.NewConnection() on /org/bluez/HFP/AudioGateway
/usr/bin/bluealsa: D: ../../src/ba-rfcomm.c:880: Starting RFCOMM loop: HFP Audio Gateway (CVSD)
/usr/bin/bluealsa: D: ../../src/ba-rfcomm.c:1262: Created new RFCOMM thread [ba-rfcomm]: HFP Audio Gateway (CVSD)
/usr/bin/bluealsa: D: ../../src/sco.c:166: Checking Broadcom internal SCO routing
/usr/bin/bluealsa: D: ../../src/sco.c:172: Current SCO interface setup: 1 2 0 1 1
/usr/bin/bluealsa: D: ../../src/sco.c:200: Created SCO dispatcher [ba-sco-dispatch]: hci0
/usr/bin/bluealsa: D: ../../src/bluez.c:609: HFP Audio Gateway (CVSD) configured for device CC:98:8B:93:B8:30
/usr/bin/bluealsa: D: ../../src/ba-transport.c:750: Starting transport: HFP Audio Gateway (CVSD)
/usr/bin/bluealsa: D: ../../src/sco.c:77: Starting SCO dispatcher loop: hci0
/usr/bin/bluealsa: D: ../../src/ba-transport.c:1032: Created new thread [ba-sco]: HFP Audio Gateway (CVSD)
/usr/bin/bluealsa: D: ../../src/sco.c:241: Starting SCO loop: HFP Audio Gateway (CVSD)
/usr/bin/bluealsa: D: ../../src/at.c:161: AT message: SET: command:+BRSF, value:665
/usr/bin/bluealsa: D: ../../src/ba-rfcomm.c:107: Sending AT message: RESP: command:+BRSF, value:2272
/usr/bin/bluealsa: D: ../../src/ba-rfcomm.c:107: Sending AT message: RESP: command:(null), value:OK
/usr/bin/bluealsa: D: ../../src/ba-rfcomm.c:126: RFCOMM: HFP Audio Gateway (CVSD) state transition: 0 -> 2
/usr/bin/bluealsa: D: ../../src/at.c:161: AT message: TEST: command:+CIND, value:(null)
/usr/bin/bluealsa: D: ../../src/ba-rfcomm.c:107: Sending AT message: RESP: command:+CIND, value:("service",(0-1)),("call",(0,1)),("callsetup",(0-3)),("callheld",(0-2)),("signal",(0-5)),("roam",(0-1)),("battchg",(0-5))
/usr/bin/bluealsa: D: ../../src/ba-rfcomm.c:107: Sending AT message: RESP: command:(null), value:OK
/usr/bin/bluealsa: D: ../../src/ba-rfcomm.c:126: RFCOMM: HFP Audio Gateway (CVSD) state transition: 2 -> 5
/usr/bin/bluealsa: D: ../../src/bluez.c:915: Signal: org.freedesktop.DBus.ObjectManager.InterfacesAdded()
/usr/bin/bluealsa: D: ../../src/bluez.c:998: Adding new Stream End-Point: CC:98:8B:93:B8:30: SBC
/usr/bin/bluealsa: D: ../../src/bluez.c:915: Signal: org.freedesktop.DBus.ObjectManager.InterfacesAdded()
/usr/bin/bluealsa: D: ../../src/bluez.c:998: Adding new Stream End-Point: CC:98:8B:93:B8:30: LDAC
/usr/bin/bluealsa: D: ../../src/bluez.c:915: Signal: org.freedesktop.DBus.ObjectManager.InterfacesAdded()
/usr/bin/bluealsa: D: ../../src/bluez.c:998: Adding new Stream End-Point: CC:98:8B:93:B8:30: aptX-HD
/usr/bin/bluealsa: D: ../../src/bluez.c:915: Signal: org.freedesktop.DBus.ObjectManager.InterfacesAdded()
/usr/bin/bluealsa: D: ../../src/bluez.c:998: Adding new Stream End-Point: CC:98:8B:93:B8:30: aptX
/usr/bin/bluealsa: D: ../../src/bluez.c:915: Signal: org.freedesktop.DBus.ObjectManager.InterfacesAdded()
/usr/bin/bluealsa: D: ../../src/bluez.c:998: Adding new Stream End-Point: CC:98:8B:93:B8:30: AAC
/usr/bin/bluealsa: D: ../../src/at.c:161: AT message: GET: command:+CIND, value:(null)
/usr/bin/bluealsa: D: ../../src/ba-rfcomm.c:107: Sending AT message: RESP: command:+CIND, value:0,0,0,0,0,0,5
/usr/bin/bluealsa: D: ../../src/ba-rfcomm.c:107: Sending AT message: RESP: command:(null), value:OK
/usr/bin/bluealsa: D: ../../src/ba-rfcomm.c:126: RFCOMM: HFP Audio Gateway (CVSD) state transition: 5 -> 7
/usr/bin/bluealsa: D: ../../src/at.c:161: AT message: SET: command:+CMER, value:3, 0, 0, 1
/usr/bin/bluealsa: D: ../../src/ba-rfcomm.c:107: Sending AT message: RESP: command:(null), value:OK
/usr/bin/bluealsa: D: ../../src/ba-rfcomm.c:126: RFCOMM: HFP Audio Gateway (CVSD) state transition: 7 -> 8
/usr/bin/bluealsa: D: ../../src/ba-rfcomm.c:126: RFCOMM: HFP Audio Gateway (CVSD) state transition: 8 -> 9
/usr/bin/bluealsa: D: ../../src/dbus.c:57: Called: org.bluez.MediaEndpoint1.SelectConfiguration() on /org/bluez/hci0/A2DP/LDAC/Source/1
/usr/bin/bluealsa: D: ../../src/at.c:161: AT message: SET: command:+XAPL, value:054C-0C93-0452,10
/usr/bin/bluealsa: D: ../../src/ba-rfcomm.c:107: Sending AT message: RESP: command:(null), value:+XAPL=BlueALSA,6
/usr/bin/bluealsa: D: ../../src/ba-rfcomm.c:107: Sending AT message: RESP: command:(null), value:OK
/usr/bin/bluealsa: D: ../../src/bluez.c:915: Signal: org.freedesktop.DBus.ObjectManager.InterfacesAdded()
/usr/bin/bluealsa: D: ../../src/dbus.c:57: Called: org.bluez.MediaEndpoint1.SetConfiguration() on /org/bluez/hci0/A2DP/LDAC/Source/1
/usr/bin/bluealsa: D: ../../src/bluez.c:291: A2DP Source (LDAC) configured for device CC:98:8B:93:B8:30
/usr/bin/bluealsa: D: ../../src/bluez.c:294: Configuration: channels: 2, sampling: 96000
/usr/bin/bluealsa: D: ../../src/bluez.c:490: Creating media endpoint object: /org/bluez/hci0/A2DP/LDAC/Source/3
[... cut ...]
/usr/bin/bluealsa: D: ../../src/at.c:161: AT message: SET: command:+IPHONEACCEV, value:1,1,9
/usr/bin/bluealsa: D: ../../src/ba-rfcomm.c:107: Sending AT message: RESP: command:(null), value:OK
/usr/bin/bluealsa: D: ../../src/at.c:161: AT message: GET: command:+APLSIRI, value:(null)
/usr/bin/bluealsa: W: Unsupported AT message: GET: command:+APLSIRI, value:(null)
/usr/bin/bluealsa: D: ../../src/ba-rfcomm.c:107: Sending AT message: RESP: command:(null), value:ERROR
/usr/bin/bluealsa: D: ../../src/dbus.c:57: Called: org.bluealsa.Manager1.GetPCMs() on /org/bluealsa

closed 2020-09-13 by Arkq