arkq/bluez-alsa #368: Duplicate Mixer Name Handling

hrc-amilcar, 2020-08-31:
Hello,

In my application, I'm displaying a list of available sound devices to the user. This includes built-in sound cards, USB, and Bluetooth devices. The user can then select a device to listen and record. The application uses ALSA to play, record, change volume (`snd_mixer_selem_set_playback_volume`), mute (`snd_mixer_selem_set_playback_switch`), and detect changes (`snd_mixer_poll_descriptors`).

I'm trying to automatically determine the proper mixer name for volume control so the user doesn't have to also pick a mixer name. It's relatively simple for built-in cards and USB headsets albeit somewhat arbitrary (e.g. PCM, Master, Headset).

In bluez-alsa, all devices are accessible via a single ALSA CTL device "bluealsa". When I get a `PCMAdded` I can append either **"- SCO"** or **"- A2DP"** to the Bluetooth alias name and control its volume. However, when there are multiple devices with the same name, bluealsa makes distinguishing names by appending a number to the end of the mixer name. To properly control the device volume, I have to recreate this name in my application.
```
# amixer -Dbluealsa
Simple mixer control 'Zebra HS3100 #1 - SCO',0
  Capabilities: pvolume pvolume-joined cvolume cvolume-joined pswitch pswitch-joined cswitch cswitch-joined
  Playback channels: Mono
  Capture channels: Mono
  Limits: Playback 0 - 15 Capture 0 - 15
  Mono: Playback 10 [67%] [on] Capture 15 [100%] [on]
Simple mixer control 'Zebra HS3100 #1 | Battery',0
  Capabilities: pvolume pvolume-joined
  Playback channels: Mono
  Limits: Playback 0 - 100
  Mono: Playback 255 [255%]
Simple mixer control 'Zebra HS3100 #2 - SCO',0
  Capabilities: pvolume pvolume-joined cvolume cvolume-joined pswitch pswitch-joined cswitch cswitch-joined
  Playback channels: Mono
  Capture channels: Mono
  Limits: Playback 0 - 15 Capture 0 - 15
  Mono: Playback 10 [67%] [on] Capture 15 [100%] [on]
Simple mixer control 'Zebra HS3100 #2 | Battery',0
  Capabilities: pvolume pvolume-joined
  Playback channels: Mono
  Limits: Playback 0 - 100
  Mono: Playback 255 [255%]
```
I managed to replicate this naming convention by mimicking the functionality in `bluealsa_elem_set_name`, including the numbering and truncation. However, it seems cumbersome and unwise to duplicate this code in my own application.

My two current options:
1. Continue to use the duplicate name handling in my own application and hope it doesn't change in the future in bluealsa.
2. Bypass ALSA for Bluetooth devices and instead make DBus calls to set/get the `Volume` property of the `org.bluealsa.PCM1` interface.

Option 1 is working for me, but I'd like to find a better way. Option 2 is probably fine, but I was hoping to continue using ALSA for setting volumes and detecting changes, in order to keep things unified with the non-Bluetooth devices.

What I'm wondering is if there's a third option. Would it be possible to add a **"MixerName"** to the PCM properties? Or should I stick to using DBus or building the mixer name myself?

borine, 2020-08-31:
a way of side-stepping the issue is to use bluetoothctl to set an alias for the device. By default, the alias is set to the name reported by the device (set by manufacturer), but local controller can change the alias.

Bluealsa uses the alias, not the name, to generate control element names. By defining unique aliases for each device, you can control the element name used by bluealsa.

This is not a "solution" to the originla problem here, but is a workaround that may be useful for may users.

hrc-amilcar, 2020-08-31:
@borine, I would still prefer a read-only method, but that's not a bad idea!

I could grab the alias (since it can be a prettier than the name) and change the alias to the Bluetooth address to get a unique mixer name (e.g. `AA:BB:CC:DD:EE:FF - SCO`). I would just have to make sure it works across reboots, pair/unpair, and that bluealsa doesn't miss the PropertiesChanged.

Technically, `bluealsa-ctl.c` could use that naming scheme instead and cut down the code quite a bit? And no truncation would be needed, since it fits within the 44 characters allotted by ALSA for mixer names.

Then I could do:
`amixer -Dbluealsa sset '00:1B:41:B7:31:E3 - SCO' 100%`
instead of
`amixer -Dbluealsa sset 'Zebra HS3100 #2 - SCO' 100%`

I would still have to distinguish between multiple devices with the same name in my UI. So there still has to be duplicate name handling code somewhere. But I think  `Zebra HS3100 (AA:BB:CC:DD:EE:FF)` could be better than `Zebra HS3100 #2` especially when connecting/disconnecting same name devices. They end up jumping around. That is, `Device #3` becomes `Device #2` when you disconnect some other device. It's not a big deal, but it could be confusing to the user.

borine, 2020-08-31:
I didn't explain that very well. What I meant was that bluealsa already uses the alias. You just have to change the alias using bluetoothctl to have a unique value, the bluealsa will automatically use that new, unique, value. 

Bluez stores the alias with the pairing information (in `/var/lib/bluetooth`), so once set, the alias remains the same until the device is unpaired.

For example, I have a bluetooth speaker, which is named, by the manufacturer, as "Libratone Zipp Mini", so when I first paired it, Bluez set its alias to "Libratone Zipp Mini", and bluealsa used the name `Libratone Zipp Mini - A2DP` for the volume control.

Then I used the bluetoothctl `set-alias` command top change its alias to `zipp`. The next time I connected it, bluealsa called the volume control `zipp - A2DP` and it has remained as that ever since.

hrc-amilcar, 2020-08-31:
Yes. I was trying to apply your solution to my application.

When I get a `org.bluez.Device1` added, I would check to see if the alias is different than the BD address. If it's different, I would set the alias to the address. Bluealsa would then regenerate the mixer name to be `AA:BB:CC:DD:EE:FF - SCO` and `AA:BB:CC:DD:EE:FF - A2DP`. And I can use that well-known unique name for changing and monitoring the volume.

borine, 2020-08-31:
In your original post, you said:

> However, when there are multiple devices with the same name, bluealsa makes distinguishing names by appending a number to the end of the mixer name.

I have shown you how to make sure that no two devices ever have the same name. So I don't understand why the user ever needs to see the bluetooth address. When a device is first paired, that is the time to give it a unique, meaningful, name (alias) that can be presented to the user.

Your program still has to generate the control element name itself, but that should be much easier when you know there is no numeric identifier required to distinguish two devices with the same make and model.

hrc-amilcar, 2020-08-31:
I see what you're saying. I can use your suggestion to my advantage. You suggest pushing the naming to the app by changing the alias upon pairing, such that I don't trigger the bluealsa mixer duplicate naming code. That really helps. I still have to apply the truncation rule that `bluealsa_elem_set_name` uses for long names. But as you say, it's much easier than dealing with matching the numbering scheme.

On the UI, I have to distinguish between multiple devices with the same name somehow. I'm currently applying the same naming scheme that bluealsa uses, which has some interesting side effects.

I mention displaying the BT address because it's already unique. If I use a plain integer, I have to make sure it doesn't clash with some other paired device. And I would prefer the device not change names on the UI when other devices are added or removed. For example, bluealsa sorts by alias alphabetically, then by BT address, then appends the number.

But I get into situations like this:
- First A is added. `List = { A }`.
- Second A is added. `List = { A1, A2 }`. But depending on the BT address, A2 may refer to the first device added.
- Third A is added. `List = { A1, A2, A3 }`.
- A2 is removed. `List = { A1, A2 }`. The previously named A3 is now A2.

This can be confusing to the user.

I can have devices keep their name either by appending the BT address, or scanning through the existing list of devices to determine a 'free' integer to append. e.g. In the case of A2 removed. The UI list would remain A1, A3. When some other A is added, it would take up the A2 slot.

Thanks for listening. I'll have to figure something out.

Arkq, 2020-08-31:
> When I get a PCMAdded I can append either "- SCO" or "- A2DP" to the Bluetooth alias name and control its volume. However, when there are multiple devices with the same name, bluealsa makes [...]

I think that the problem here is that you are mixing D-Bus API and ALSA API. When listening for PCMAdded it suggest that you are creating your own user interface for BlueALSA service. In such case you should use D-Bus API also for controlling volume. On the other hand, if you'd like to use ALSA for volume control via mixer API, you should also listen for new control elements (and for disappearing elements) via ALSA API, IMHO. There is however one problem when listening for new elements via ALSA API - when new element is added, all elements are removed and created once again (the element ID might change). I think, that there is an issue on github regarding this.

Maybe there should be an option for ALSA control plugin whether element name should be based on BT device name, or simply use BT MAC address?

hrc-amilcar, 2020-08-31:
@Arkq, yes. I listen to the BlueAlsa API for PCMAdded/Removed so that I can get available properties of the connected BT devices. I also listen to UDEV for USB devices added/removed. I then use the ALSA API to play, record, mute, and set volume, assuming I know the main mixer name of each device.

ALSA is uniquely naming the system sound cards and USB devices, and most headsets use "Master", "PCM", or "Headset" as their mixer name.
- `snd_ctl_card_info_get_id`: Unique.
- `snd_ctl_card_info_get_name`: Not unique.
- `snd_ctl_card_info_get_longname`: Unique.

... but BlueAlsa doesn't add a new card/CTL per device with a "Master" volume. That would be convenient, but I'm sure there are good reasons why it was all done as a single ALSA device.

> When listening for PCMAdded it suggest that you are creating your own user interface for BlueALSA service. In such case you should use D-Bus API also for controlling volume.

Yes. I could make another code path specifically for changing volumes of BT devices. But I was hoping not to have to do that, and keep everything as ALSA as possible. I may go down that path anyway and bypass the mixer.

> if you'd like to use ALSA for volume control via mixer API... There is however one problem when listening for new elements via ALSA API - when new element is added, all elements are removed and created once again (the element ID might change).

Yes, I've run into this. That's why I choose to use a known mixer name. I use `snd_mixer_selem_id_set_name` to `snd_mixer_find_selem` the element to mute or change the volume.

> Maybe there should be an option for ALSA control plugin whether element name should be based on BT device name, or simply use BT MAC address?

That would be nice.

When interacting with a BlueAlsa device via ALSA, I use the `bluealsa:DEV=AA:BB:CC:DD:EE:FF,PROFILE=sco` for opening PCMs, but for the mixer, I need to replicate the naming scheme. It would be convenient to assume `AA:BB:CC:DD:EE:FF - SCO` is the mixer name as well. Or a way to get the mixer name from the PCM properties at PCMAdded time.

borine, 2020-09-01:
> and most headsets use "Master", "PCM", or "Headset" as their mixer name.

Actually, that's not true. Many sound cards (eg some I2C cards using Texas Instruments chips) do not have "PCM", "Master" nor "Headset" controls. My IQAudio card with Texas Instruments PCM5122  DAC chip has volume control element called "Digital". Neither the Linux sound subsystem, not alsa-lib, define any particular standards or conventions for control element naming, so it is not, in general, possible to programatically work out what the appropriate control element name is.

> a way to get the mixer name from the PCM properties at PCMAdded time.

I think the problem with that is the mixer name is an ALSA-specific addtiion, so is implemented in the bluealsa ctl plugin, not the bleualsa daemon. The daemon is not an alsa application, it and deliberately avoids ALSA-specific details. You can control volume directly by using the bluealsa D-Bus API using only information provided by the PCMAdded signal - there is no need to involve any concept of "ctl" or "mixer name" which are both ALSA artifacts. In the D-Bus API, volume is simply a read/write property of the PCM itself, there is no concept of a separate "ctl" object.

hrc-amilcar, 2020-09-01:
Thanks, @borine.

I'll probably switch to using D-Bus to mute or change volume. I hope 'PropertiesChanged' comes through when the user changes headset volume with the physical buttons. I presume it will, since I'm already getting volume changes through the `snd_mixer_poll_descriptors` ALSA mixer API.

As far as duplicate naming. I'll have to handle that in my UI anyway with my own scheme.

closed 2020-09-01 by hrc-amilcar

graham8, 2020-09-01:
Sorry to hi-jack this, but how are you enumerating BT devices @hrc-amilcar?

We are working on a script that will listen to dbus org.bluez and generate entries in .asoundrc with the appropriate BT MACs.  Each device generates a pcm with a name made from the Name on dbus + the last two bytes of the MAC in case there are duplicates.  Are we doing it the hard way?  If not, could you use a similar idea to name ctls?

hrc-amilcar, 2020-09-01:
Hi, @graham8.

- I'm enumerating BT devices by listening for `org.bluez.Device1` interface added.
- I only consider devices with `ServiceClass=Audio` and `MajorDeviceClass=AudioVideo`.
- The user can then "connect" the device, if it hasn't already auto-connected. Using `org.bluez` API.
- Once connected, I receive a `PCMAdded` from BlueAlsa.
- I then open PCMs using `bluealsa:DEV=AA:BB:CC:DD:EE:FF,PROFILE=sco` (or a2dp).
- To change the volume, I build and use a mixer name that matches what BlueAlsa adds to its CTL.

Re: .asoundrc. The use-case for your script be different, but I did try the approach of updating the file when devices were added and removed. But rewriting the file proved tedious, especially when it already exists with other user entries. I also had to make sure the configuration was reloaded at runtime, so that the app didn't have to restart. i.e. using `snd_config_update`. I opted for doing everything through D-Bus BlueZ and BlueAlsa API in code.

"Name + Last Two Bytes of MAC" is another way to handle the duplicate device names. Thanks for the idea! I wonder how likely it is for someone to have two same-model devices that end in the same byte. At least it's less ugly than the whole MAC. :)
e.g. "Jabra Evolve 65 (FF)" vs. "Jabra Evolve 65 (AA:BB:CC:DD:EE:FF)"

graham8, 2020-09-01:
Thanks for the reply.  I deleted my post after realising my idea doesn't work - all ctls will still appear under one device whether it's called ctl.bluealsa or ctl.foobar.  As you explained (and wondered why) above.

We want to generate .asoundrc entries so that any application finds the devices through ALSA.  In production we won't be using .asoundrc itself. We thought **2** bytes (eg JBL-Flip-3-45-4f) should be enough to be unique.