arkq/bluez-alsa #369: ALSA: Cannot open PCM device alsa_pcm for playback when starting JACK server

Shamaoke, 2020-09-03:
After updating bluez-alsa from 2.0.0 to 2.1.0 my bluetooth audio device is no longer available in JACK, so I can't play sound using the JACK backend (ALSA is working fine).

I'm running a JACK server with the following script.

~~~~
jack_control ds alsa

jack_control dps device bluealsa

jack_control dps rate 48000

jack_control dps period 1024

jack_control dps nperiods 2

jack_control start
~~~~

The following is the output at `~/.log/jack/jackdbus.log`.

~~~~
Thu Sep  3 04:32:33 2020: Starting jack server...
Thu Sep  3 04:32:33 2020: JACK server starting in realtime mode with priority 10
Thu Sep  3 04:32:33 2020: self-connect-mode is "Don't restrict self connect requests"
Thu Sep  3 04:32:33 2020: Acquired audio card Audio0
Thu Sep  3 04:32:33 2020: creating alsa driver ... bluealsa|bluealsa|1024|2|48000|0|0|nomon|swmeter|-|32bit
Thu Sep  3 04:32:33 2020: ERROR: ALSA: Cannot open PCM device alsa_pcm for playback. Falling back to capture-only mode
Thu Sep  3 04:32:33 2020: Released audio card Audio0
Thu Sep  3 04:32:33 2020: ERROR: Cannot initialize driver
Thu Sep  3 04:32:33 2020: ERROR: JackServer::Open failed with -1
Thu Sep  3 04:32:33 2020: ERROR: Failed to open server
~~~~

Here's my ALSA configuration.

~~~~
defaults.bluealsa.service "org.bluealsa"
defaults.bluealsa.profile "a2dp"
defaults.bluealsa.device  "40:ED:98:19:9C:33"
defaults.bluealsa.delay   10000
defaults.bluealsa.battery "yes"

pcm.!default {
 type plug
 slave.pcm "bluealsa"
}
~~~~

What can be the reason JACK/ALSA can't open a device? How can I tackle the problem?

Thanks.

_Linux (Arch); jack2 1.9.14; bluez-alsa 2.1.0_

Arkq, 2020-09-03:
Please, check with current master. This project has not enough manpower to
maintain old releases. Only bugs occurring on fresh master branch will be
addressed (at least from my side). And there is a great chance, that this
bug does not appear on master, since recently there was a lot of work
related to stability.

borine, 2020-09-03:
There is no significant difference in the ALSA components of bluealsa between v2.0.0 and v2.1.0 as far as jack is concerned. Both versions work fine for me, as does the latest master branch.

I suspect something else on your system has changed, not just the bluealsa version.

Here is what I recommend

1. Delete your current jack config:
```
$ jack_control stop
--- stop
$ jack_control exit
--- exit
$ ps aux | grep jack | grep -v grep
$
$ rm -r ~/.config/jack
```

2. Don't use ALSA "plug" with jack. Instead use the following in ~/.asoundrc specifically for jack:
```
pcm.bluetooth {
	type "bluealsa"
	device "40:ED:98:19:9C:33"
	profile "a2dp"
}
```

3. Reboot - just to make sure we have a clean start for all services.

4. If not done automatically by systemd or whatever, start bluealsa and connect your bluetooth device
```
$ sudo bluealsa
$ echo  "connect 40:ED:98:19:9C:33" | bluetoothctl
```

5. Configure jack like this:
```
$ jack_control ds alsa
--- driver select "alsa"
$ jack_control dps device bluetooth
--- driver param set "device" -> "bluetooth"
$ jack_control dps playback bluetooth
--- driver param set "playback" -> "bluetooth"
$ jack_control dps duplex False
--- driver param set "duplex" -> "False"
```

6. Test - I used:
```
mpg123 -o jack music.mp3 
```

Shamaoke, 2020-09-03:
I reinstalled bluez-alsa from master.

I configured my `~/.asoundrc` as the following.

~~~~
pcm.bluetooth {
  type    "bluealsa"
  device  "40:ED:98:19:9C:33"
  profile "a2dp"
}
~~~~

No other configuration entires present at `~/.asoundrc`.

When I run the following commands to configure and start JACK.

~~~~
jack_control ds alsa

jack_control dps device bluetooth

jack_control dps playback bluetooth

jack_control dps duplex False

jack_control dps rate 48000

jack_control dps period 1024

jack_control dps nperiods 2

jack_control start
~~~~

I get the following output.

~~~~
--- driver select "alsa"
--- driver param set "device" -> "bluetooth"
--- driver param set "playback" -> "bluetooth"
--- driver param set "duplex" -> "False"
--- driver param set "rate" -> "48000"
--- driver param set "period" -> "1024"
--- driver param set "nperiods" -> "2"
--- start
~~~~

And then after a pause.

~~~~
DBus exception: org.freedesktop.DBus.Error.NoReply: Did not receive a reply. Possible causes include: the remote application did not send a reply, the message bus security policy blocked the reply, the reply timeout expired, or the network connection was broken.
~~~~

Did I configure `.asoundrc` properly? Does bluez-alsa have any configuration files or cache which could be cleared in order to restore initial program state?

borine, 2020-09-03:
> Did I configure .asoundrc properly?

Yes, I think so

> Does bluez-alsa have any configuration files or cache which could be cleared in order to restore initial program state?

No.

> `jack_control dps period 1024`
`jack_control dps nperiods 2`
 
I would recommend not to use that: allow jack to negotiate its own alsa params with bluealsa; I don't think that is causing this issue, but it is unneccessary anyway.

Did you delete your old jack config and reboot before creating new jack config? That appears to be the action that got master branch bluealsa working with jack for me. I don't know why, but its something to do with D-Bus.

Shamaoke, 2020-09-03:
I deleted old `~/.config/jack` and rebooted.

I created `~/.asoundrc` with the following.

~~~~
pcm.bluetooth {
  type    "bluealsa"
  device  "40:ED:98:19:9C:33"
  profile "a2dp"
}

pcm.!default bluetooth
~~~~

When using the ALSA backend with bluez-alsa, this configuration works perfectly fine — I get sound on my bluetooth device, I can control it through alsamixer, I see the battery charge level.

But again when I run the JACK server with the following.

~~~~
jack_control ds alsa

jack_control dps device bluetooth

jack_control dps rate 48000

jack_control start
~~~~

I get the following output at `~/.log/jack/jackdbus.log`.

~~~~
Thu Sep  3 20:34:27 2020: Starting jack server...
Thu Sep  3 20:34:27 2020: JACK server starting in realtime mode with priority 10
Thu Sep  3 20:34:27 2020: self-connect-mode is "Don't restrict self connect requests"
Thu Sep  3 20:34:27 2020: ERROR: control open "bluetooth" (No such file or directory)
Thu Sep  3 20:34:27 2020: ERROR: control open "bluetooth" (No such file or directory)
Thu Sep  3 20:34:27 2020: creating alsa driver ... bluetooth|bluetooth|1024|2|48000|0|0|nomon|swmeter|-|32bit
Thu Sep  3 20:34:27 2020: ERROR: control open "bluetooth" (No such file or directory)

Thu Sep  3 20:34:27 2020: ERROR: ALSA: Cannot open PCM device alsa_pcm for playback. Falling back to capture-only mode

Thu Sep  3 20:34:27 2020: ERROR: control open "bluetooth" (No such file or directory)
Thu Sep  3 20:34:27 2020: ERROR: control open "bluetooth" (No such file or directory)
Thu Sep  3 20:34:27 2020: ERROR: Cannot initialize driver
Thu Sep  3 20:34:27 2020: ERROR: JackServer::Open failed with -1
Thu Sep  3 20:34:27 2020: ERROR: Failed to open server
~~~~

On another device the JACK server runs without any problems. It also ran fine on the same devices before the update.

Is there any way to get more information about what's going on when the JACK server starts?

borine, 2020-09-03:
I've tried again, and now I get the same D-Bus error as you. This seems to be an intermittent error. Maybe a race condition somewhere in the system.

I really am out of time with this now, will get back to it in a week or so.

Arkq, 2020-09-03:
I'm not JACK user, but are there any logs from ALSA? @Shamaoke can you try to compile bluez-alsa with debug enabled `--enable-debug`, maybe there will be some more meaningful logs...

Shamaoke, 2020-09-03:
@Arkq Yeah, sure. But let's put it off till tomorrow.

Shamaoke, 2020-09-04:
I compiled bluez-alsa with debug enabled.

I started bluez-alsa with the `bluealsa` command.

I have the following in my `.asoundrc`.

~~~~
pcm.bluetooth {
  type    "bluealsa"
  device  "40:ED:98:19:9C:33"
  profile "a2dp"
}

pcm.!default bluetooth
~~~~

I configure and start the JACK server with the following commands.

~~~~
jack_control ds alsa

jack_control dps device bluetooth

jack_control dps rate 48000

jack_control start
~~~~

The last one outputs the following in the console.

~~~~
--- driver select "alsa"
--- driver param set "device" -> "bluetooth"
--- driver param set "rate" -> "48000"
--- start
DBus exception: org.jackaudio.Error.Generic: Failed to open server
~~~~

This results in the following record at `~/.log/jack/jackdbus.log`.

~~~~
Fri Sep  4 22:40:56 2020: Starting jack server...
Fri Sep  4 22:40:56 2020: JACK server starting in realtime mode with priority 10
Fri Sep  4 22:40:56 2020: self-connect-mode is "Don't restrict self connect requests"
Fri Sep  4 22:40:56 2020: ERROR: control open "bluetooth" (No such file or directory)
Fri Sep  4 22:40:56 2020: ERROR: control open "bluetooth" (No such file or directory)
Fri Sep  4 22:40:56 2020: creating alsa driver ... bluetooth|bluetooth|1024|2|48000|0|0|nomon|swmeter|-|32bit
Fri Sep  4 22:40:56 2020: ERROR: control open "bluetooth" (No such file or directory)

Fri Sep  4 22:40:56 2020: ERROR: ALSA: Cannot open PCM device alsa_pcm for playback. Falling back to capture-only mode

Fri Sep  4 22:40:56 2020: ERROR: control open "bluetooth" (No such file or directory)
Fri Sep  4 22:40:56 2020: ERROR: control open "bluetooth" (No such file or directory)
Fri Sep  4 22:40:56 2020: ERROR: Cannot initialize driver
Fri Sep  4 22:40:56 2020: ERROR: JackServer::Open failed with -1
Fri Sep  4 22:40:56 2020: ERROR: Failed to open server
~~~~

Bluez-alsa outpus the following line at the console when I run `jack_control start`.

~~~~
bluealsa: D: dbus.c:57: Called: org.bluealsa.Manager1.GetPCMs() on /org/bluealsa
~~~~

To recap:

I'm using bluez-alsa 2.1.0-2c441f6;

The JACK server is working fine with another sound device;

Bluez-alsa is working fine with the ALSA backend;

Bluez-alsa had been working fine until the update to 2.1.0.

Arkq, 2020-09-04:
If `bluealsa: D: dbus.c:57: Called: org.bluealsa.Manager1.GetPCMs() on /org/bluealsa` is printed but nothing else, it might mean that BT device is not connected, or bluealsa ALSA plug-in is trying to open wrong device (e.g. wrong direction, or wrong MAC).

Please, get the list of available BT devices with `bluealsa-aplay -l -L`. It this returns correct device, then it will be required to add some debug code to bluealsa ALSA plugin in order to get what configuration ALSA actually sees.

Shamaoke, 2020-09-04:
The `bluealsa: D: dbus.c:57: Called: org.bluealsa.Manager1.GetPCMs() on /org/bluealsa` line is the only line `bluealsa` outputs when I run `jack_control start`.

The `bluealsa-aplay -l -L` command returns the following.

~~~~
bluealsa:SRV=org.bluealsa,DEV=40:ED:98:19:9C:33,PROFILE=sco
    FiiO BTR3, trusted audio-card, playback
    SCO (mSBC): S16_LE 1 channel 16000 Hz
bluealsa:SRV=org.bluealsa,DEV=40:ED:98:19:9C:33,PROFILE=sco
    FiiO BTR3, trusted audio-card, capture
    SCO (mSBC): S16_LE 1 channel 16000 Hz
bluealsa:SRV=org.bluealsa,DEV=40:ED:98:19:9C:33,PROFILE=a2dp
    FiiO BTR3, trusted audio-card, playback
    A2DP (AAC): S16_LE 2 channels 48000 Hz

**** List of PLAYBACK Bluetooth Devices ****
hci0: 40:ED:98:19:9C:33 [FiiO BTR3], trusted audio-card
  SCO (mSBC): S16_LE 1 channel 16000 Hz
  A2DP (AAC): S16_LE 2 channels 48000 Hz

**** List of CAPTURE Bluetooth Devices ****
hci0: 40:ED:98:19:9C:33 [FiiO BTR3], trusted audio-card
  SCO (mSBC): S16_LE 1 channel 16000 Hz
~~~~

This is the correct data from my bluetooth soundcard (FiiO BTR3).

What should I do next?

Shamaoke, 2020-09-04:
I recompiled bluez-alsa from the `alsa-debug` branch.

When I use the `jack_control` utility, any of bluez-alsa debug data isn't output neither to the console nor to the `~/.log/jack/jackdbus.log` file.

However, when I start JACK with the `/usr/bin/jackd -dalsa -dbluetooth -r48000` command, I get the following output.

~~~~
jackdmp 1.9.14
Copyright 2001-2005 Paul Davis and others.
Copyright 2004-2016 Grame.
Copyright 2016-2019 Filipe Coelho.
jackdmp comes with ABSOLUTELY NO WARRANTY
This is free software, and you are welcome to redistribute it
under certain conditions; see the file COPYING for details
no message buffer overruns
no message buffer overruns
no message buffer overruns
JACK server starting in realtime mode with priority 10
self-connect-mode is "Don't restrict self connect requests"
ALSA lib control.c:1379:(snd_ctl_open_noupdate) Invalid CTL bluetooth
control open "bluetooth" (No such file or directory)
ALSA lib control.c:1379:(snd_ctl_open_noupdate) Invalid CTL bluetooth
control open "bluetooth" (No such file or directory)
creating alsa driver ... bluetooth|bluetooth|1024|2|48000|0|0|nomon|swmeter|-|32bit
ALSA lib control.c:1379:(snd_ctl_open_noupdate) Invalid CTL bluetooth
control open "bluetooth" (No such file or directory)

D: bluealsa-pcm.c:891: Getting BlueALSA PCM: PLAYBACK 40:ED:98:19:9C:33 a2dp
D: bluealsa-pcm.c:761: /org/bluealsa/hci0/dev_40_ED_98_19_9C_33/a2dpsrc/sink: Setting constraints
D: bluealsa-pcm.c:891: Getting BlueALSA PCM: CAPTURE 40:ED:98:19:9C:33 a2dp

ALSA lib bluealsa-pcm.c:893:(_snd_pcm_bluealsa_open) Couldn't get BlueALSA PCM: PCM not found

ALSA: Cannot open PCM device alsa_pcm for capture. Falling back to playback-only mode
configuring for 48000Hz, period = 1024 frames (21.3 ms), buffer = 2 periods
ALSA: final selected sample format for playback: 16bit little-endian
ALSA: use 10 periods for playback
D: bluealsa-pcm.c:397: /org/bluealsa/hci0/dev_40_ED_98_19_9C_33/a2dpsrc/sink: Initializing HW
D: bluealsa-pcm.c:417: /org/bluealsa/hci0/dev_40_ED_98_19_9C_33/a2dpsrc/sink: FIFO buffer size: 4096
D: bluealsa-pcm.c:423: /org/bluealsa/hci0/dev_40_ED_98_19_9C_33/a2dpsrc/sink: Selected HW buffer: 10 periods x 4096 bytes == 40960 bytes
D: bluealsa-pcm.c:441: /org/bluealsa/hci0/dev_40_ED_98_19_9C_33/a2dpsrc/sink: Initializing SW
D: bluealsa-pcm.c:480: /org/bluealsa/hci0/dev_40_ED_98_19_9C_33/a2dpsrc/sink: Prepared
D: bluealsa-pcm.c:441: /org/bluealsa/hci0/dev_40_ED_98_19_9C_33/a2dpsrc/sink: Initializing SW
D: bluealsa-pcm.c:450: /org/bluealsa/hci0/dev_40_ED_98_19_9C_33/a2dpsrc/sink: Changing SW avail min: 1024 -> 9216
D: bluealsa-pcm.c:480: /org/bluealsa/hci0/dev_40_ED_98_19_9C_33/a2dpsrc/sink: Prepared
D: bluealsa-pcm.c:317: /org/bluealsa/hci0/dev_40_ED_98_19_9C_33/a2dpsrc/sink: Starting
D: bluealsa-pcm.c:177: /org/bluealsa/hci0/dev_40_ED_98_19_9C_33/a2dpsrc/sink: Waiting for first period of frames
~~~~

And this is the corresponding part from the output of the `bluealsa` command.

~~~~
bluealsa: D: dbus.c:57: Called: org.bluealsa.Manager1.GetPCMs() on /org/bluealsa
bluealsa: D: dbus.c:57: Called: org.bluealsa.PCM1.Open() on /org/bluealsa/hci0/dev_40_ED_98_19_9C_33/a2dpsrc/sink
bluealsa: D: ba-transport.c:868: New transport: 20 (MTU: R:672 W:894)
bluealsa: D: bluez.c:1068: Signal: org.freedesktop.DBus.Properties.PropertiesChanged()
bluealsa: D: bluez.c:1104: Signal: PropertiesChanged: org.bluez.MediaTransport1: State
bluealsa: D: ba-transport.c:750: Starting transport: A2DP Source (AAC)
bluealsa: D: ba-transport.c:1032: Created new thread [ba-a2dp-aac]: A2DP Source (AAC)
bluealsa: D: a2dp-audio.c:1437: Starting IO loop: A2DP Source (AAC)
bluealsa: D: ba-transport.c:789: PCM resumed: 16
~~~~

It seems that the correct bluetooth device is selected, however the JACK still don't work. When a sound application trying to connect to it, `/usr/bin/jackd` outputs the following.

~~~~
JackPosixProcessSync::LockedTimedWait error usec = 5000000 err = Connection timed out
Driver is not running
Cannot create new client
~~~~

borine, 2020-09-10:
The failure of jackd to start with the lastest bluealsa code is actually a bug in jack, in the way it uses the avail_min sw parameter. Previous releases of bluealsa did not implement that parameter, so did not trigger this bug.

Jack requests a buffer size of two times the period size, which it gets with sound cards, but when using bluealsa, the ALSA ioplug creates a buffer size of ten periods. Because of the bug in jack, jack starts by filling 20% of the buffer with audio frames, then the bluealsa iothread (i.e. virtual hardware) waits for 90% of the buffer to be filled with audio frames, while the application is blocked waiting for 90% of the buffer to be empty. So both threads are left waiting forever.

The only workaround I can find that does not break other applications is to explicitly tell jackd to use a buffer of 10 periods; then it does the avail_min calculation correctly and the stream starts.

Having fixed that, I discovered a bug in the bluealsa pcm plugin, which also needs to be fixed before jack works correctly. Here is the patch:

```
diff --git a/src/asound/bluealsa-pcm.c b/src/asound/bluealsa-pcm.c
index f675c86..f857a12 100644
--- a/src/asound/bluealsa-pcm.c
+++ b/src/asound/bluealsa-pcm.c
@@ -668,7 +668,7 @@ static int bluealsa_poll_revents(snd_pcm_ioplug_t *io, struct pollfd *pfd,
                        case SND_PCM_STATE_XRUN:
                        case SND_PCM_STATE_PAUSED:
                        case SND_PCM_STATE_SUSPENDED:
-                               *revents = POLLERR;
+                               *revents |= POLLERR;
                                break;
                        case SND_PCM_STATE_DISCONNECTED:
                                *revents = POLLERR;
```

@Shamaoke are you able to test bluealsa master branch with this patch applied? Remember to add -n 10 to the jackd command line, or dps nperiods 10 to jack_control). To minimize the number of error messages I would recommend to add this to your .asoundrc:
```
ctl.bluetooth "bluealsa"
```

and to start jackd with:

```
jackd -r -v -d alsa -P bluetooth -D false -n 10
```

Shamaoke, 2020-09-10:
@borine OK. I'll be back in an hour.

borine, 2020-09-10:
Correction to last comment - I misunderstood the meaning of "--duplex" for jack alsa driver - it actually creates more error messages with bluealsa ! So I recommend to test with
```
jackd -r -v -d alsa -P bluetooth -n 10
```
... and if you've already set duplex with jack_control, unset it with
```
jack_control dpr duplex
```
My apologies for the confusion.

borine, 2020-09-10:
AAAGGGHHH !!! the patch above breaks alsaloop. I have another approach that I'm testing now.
@Shamaoke please wait until the new patch is ready.

borine, 2020-09-10:
Here is a patch against master branch which in testing so far is working with both jack and alsaloop
```
diff --git a/src/asound/bluealsa-pcm.c b/src/asound/bluealsa-pcm.c
index f675c86..ff15721 100644
--- a/src/asound/bluealsa-pcm.c
+++ b/src/asound/bluealsa-pcm.c
@@ -138,10 +138,10 @@ static void io_thread_cleanup(struct bluealsa_pcm *pcm) {
  * write the balance of the period. */
 static void io_thread_wait_first_period(snd_pcm_ioplug_t *io) {
        struct bluealsa_pcm *pcm = io->private_data;
-       while (io->appl_ptr < pcm->io_avail_min && (
+       while (io->appl_ptr <= io->period_size && (
                                io->state == SND_PCM_STATE_RUNNING ||
                                io->state == SND_PCM_STATE_PREPARED)) {
-               uint64_t nsec = (pcm->io_avail_min - io->appl_ptr) * 1000000000 / io->rate;
+               uint64_t nsec = (1 + io->period_size - io->appl_ptr) * 1000000000 / io->rate;
                struct timespec ts = {
                        .tv_sec = nsec / 1000000000,
                        .tv_nsec = nsec % 1000000000,
```
jack still needs the nperiods param set to 10

Shamaoke, 2020-09-10:
@borine npnp, I tested it.

When running JACK with `jackd -d alsa --device bluealsa --rate 48000 --nperiods 10` _and_ using the following in `.asoundrc`,

~~~~
defaults.bluealsa.service org.bluealsa
defaults.bluealsa.profile a2dp
defaults.bluealsa.device  40:ED:98:19:9C:33
defaults.bluealsa.delay   10000
defaults.bluealsa.battery yes

pcm.!default {
 type plug
 slave.pcm bluealsa
}
~~~~

the JACK server have begun to work. So, the `--nperiods 10` option affects the issue. But, if I change `.asoundrc` to 

~~~~
pcm.bluealsa {
   type    bluealsa
   device  40:ED:98:19:9C:33
   profile a2dp
 }

ctl.bluealsa bluealsa

 pcm.!default bluealsa
~~~~

and use the same command to start JACK, it works 50% of the time. That is, sometimes it works but sometimes it doesn't.

Strangely, when running JACK with the `jack_control` command

~~~~
jack_control ds alsa
jack_control dps device bluealsa
jack_control dps rate 48000
jack_control dps nperiods 10
~~~~

it still doesn't start as before, regardless of `.asoundrc` content (I removed `~/.config/jack`, and restarted the system).

I made the tests with your old patch and without it. The results are _the same_. I will test with your new patch in an hour.

borine, 2020-09-10:
>  But, if I change .asoundrc to
```
pcm.bluealsa {
   type    bluealsa
   device  40:ED:98:19:9C:33
   profile a2dp
 }

ctl.bluealsa bluealsa

 pcm.!default bluealsa
```

The problem there is using `pcm.bluealsa`. The bluez-alsa project installs a predefined pcm called `bluealsa`, which is interfering with your definition. I have a wiki article ready about that, which I intend to post once the next bluez-alsa release is ready.

What you need is
```
pcm.bluetooth {
   type    bluealsa
   device  40:ED:98:19:9C:33
   profile a2dp
   delay 10000
 }

ctl.bluetooth {
   type bluealsa
   battery yes
   profile a2dp
}

pcm.!default bluetooth
ctl.!default bluetooth
```
or any other pcm / ctl name that is not already defined if you don't like "bluetooth".

Also, I would recommend not to use
```
jack_control dps rate 48000
```
because bluealsa negotiates the frame rate with the bluetooth device when the device is connected, and cannot then be changed by an application. Bluealsa will tell jack what frame rate to use, and it will convert whatever the source stream rate is to the rate required by the bluetooth device. The jack `rate` parameter is only relevant for sound cards which generally offer a range of available rates, so the application has to choose which one to use.

Shamaoke, 2020-09-11:
@borine My `.asoundrc` contains only the settings you gave in your last comment, i. e.

~~~~
pcm.bluetooth {
  type    bluealsa
  device  40:ED:98:19:9C:33
  profile a2dp
  delay   10000
}

ctl.bluetooth {
  type    bluealsa
  battery yes
}

pcm.!default bluetooth
ctl.!default bluetooth
~~~~

It doesn't contain anything else.

The problem with these settings is the following:

1. The JACK server doesn't correctly start. You cannot use it as a sound backend no matter how you launch it (`jackd` or `jack_control`);

2. For some reason, it's impossible to output sound from the Firefox browser when using the ALSA backend. Media players on popular sites (e. g. YouTube, Bandcamp etc.) just don't work. When you push the play button in these players the playback doesn't start. Other application which output sound, e. g. MPV, MPD still work though.

Both of these issues are absent when I configure BA as a  plug, i. e

~~~~
defaults.bluealsa.service org.bluealsa
defaults.bluealsa.profile a2dp
defaults.bluealsa.device  40:ED:98:19:9C:33
defaults.bluealsa.delay   10000
defaults.bluealsa.battery yes

pcm.!default {
  type plug
  slave.pcm bluealsa
}
~~~~

With these, I can run JACK with `jackd -d alsa --device bluealsa --nperiods 10`; I can play sound in Firefox on media sites when using ALSA without any problems.

borine, 2020-09-11:
> I can run JACK with `jackd -d alsa --device bluealsa --nperiods 10`

but have you tried
```
jackd -d alsa --device bluetooth --nperiods 10
```
when using my example .asoundrc?
As I said earlier, the *pcm* called bluealsa is predefined by bluez-alsa, and is not the same thing as the *type* bluealsa. Perhaps i should have used a clearer pcm device name. Try .asoundrc
```
# For JACK
pcm.my_headphones {
  type    bluealsa
  device  40:ED:98:19:9C:33
  profile a2dp
  delay   10000
}

ctl.my_headphones {
  type    bluealsa
  battery yes
}

# For ALSA applications
defaults.bluealsa.service org.bluealsa
defaults.bluealsa.profile a2dp
defaults.bluealsa.device  40:ED:98:19:9C:33
defaults.bluealsa.delay   10000
defaults.bluealsa.battery yes

pcm.!default bluealsa
ctl.!default bluealsa
```
Then start jackd as:
```
jackd -d alsa --playback my_headphones --nperiods 10
```

Note that jack is using the pcm called "my_headphones", but ALSA appications use the pcm called "default", which happens to be an alias for the pre-deined pcm called "bluealsa".

Incidentally, not related to this issue, how did you get Firefox to work at all with ALSA? With Ubuntu it only works with Pulseaudio, nothing else?

Shamaoke, 2020-09-11:
@borine 

> As I said earlier, the pcm called bluealsa is predefined by bluez-alsa, and is not the same thing as the type bluealsa. Perhaps i should have used a clearer pcm device name.

I understand that. You explained the thing very clearly. When I use the PCM type `plug` I run JACK with `jackd ... --device bluealsa`, but when I use the PCM type `bluealsa` I run it with `jackd ... --device whatevernameiset`. But no, PCM type `bluealsa` just don't work with JACK on my system (with ALSA it's working fine).

>Incidentally, not related to this issue, how did you get Firefox to work at all with ALSA? With Ubuntu it only works with Pulseaudio, nothing else?

I don't know. In Arch it just works. I believe it's related to the compilation flags which is used by maintainers when building the browser. In Arch they also enable JACK, so I can use it as a backend when playing sound in Firefox. That's how things are.

borine, 2020-09-11:
> no, PCM type bluealsa just don't work with JACK on my system (with ALSA it's working fine).

ok, that's disappointing. I can't debug this anymore as it works fine for me and I am unable to reproduce the failure you are seeing. Clearly there is something else here that I do not yet understand.

> I believe it's related to the compilation flags which is used by maintainers when building the browser.

Thanks, that makes sense. I don't want to have to build Firefox myself, so I guess I just can't use bluealsa with it on Ubuntu.

borine, 2021-06-07:
JACK and BlueALSA appear to be working better together with bluez-alsa release 3.1.0 and recent JACK releases. See the wiki page for more info:
https://github.com/Arkq/bluez-alsa/wiki/Using-BlueALSA-with-the-JACK-Audio-Connection-Kit

closed 2021-06-07 by borine