arkq/bluez-alsa #370: cannot build on aarch64

gearhead, 2020-09-08:
Strange recent occurrence. I have been able to build this package with a PKGBUILD that I made for our project for quite a while for armv6l, armv7l,  aarch64, and x86_64 up till this weekend. Arch updated from gcc 9.3 to 10.2 and it builds fine with gcc 10.2.0 (current Arch Linux package) on armv6l and armv7l but with recent Arch Linux updates and now it will not build but only on aarch64. I attach the error listing when I try to build it.
[blue_alsa_build_error.txt](https://github.com/Arkq/bluez-alsa/files/5188988/blue_alsa_build_error.txt)
This is what gcc reports for aarch64
```
$ gcc -v
Using built-in specs.
COLLECT_GCC=gcc
COLLECT_LTO_WRAPPER=/usr/lib/gcc/aarch64-unknown-linux-gnu/10.2.0/lto-wrapper
Target: aarch64-unknown-linux-gnu
Configured with: /build/gcc/src/gcc/configure --prefix=/usr --libdir=/usr/lib --libexecdir=/usr/lib --mandir=/usr/share/man --infodir=/usr/share/info --with-bugurl=https://github.com/archlinuxarm/PKGBUILDs/issues --enable-languages=c,c++,fortran,go,lto,objc,obj-c++,d --with-isl --with-linker-hash-style=gnu --with-system-zlib --enable-__cxa_atexit --enable-checking=release --enable-clocale=gnu --enable-default-pie --enable-default-ssp --enable-gnu-indirect-function --enable-gnu-unique-object --enable-install-libiberty --enable-linker-build-id --enable-lto --enable-plugin --enable-shared --enable-threads=posix --disable-libssp --disable-libstdcxx-pch --disable-libunwind-exceptions --disable-multilib --disable-werror --host=aarch64-unknown-linux-gnu --build=aarch64-unknown-linux-gnu --with-arch=armv8-a --enable-fix-cortex-a53-835769 --enable-fix-cortex-a53-843419 gdc_include_dir=/usr/include/dlang/gdc
Thread model: posix
Supported LTO compression algorithms: zlib zstd
gcc version 10.2.0 (GCC)
```
I can check on the armv6/7l gcc -v response later today.
The successful build messages from armv7 are here:
[blue_alsa_build_armv7.txt](https://github.com/Arkq/bluez-alsa/files/5189045/blue_alsa_build_armv7.txt)
My build commands:
```
        autoreconf --install
        ./configure --prefix=/usr \
                    --sysconfdir=/etc \
                    --enable-aac
        make
```

Arkq, 2020-09-08:
You need a fix for external variables. I assume that you are not building
from master, because this issue should be already fixed.

gearhead, 2020-09-08:
Thanks. I am building v2.1.0. Is a new release imminent? I can adapt and build from git master, but have always built the 'release'. Why does it build on armv6/7 but not on aarch64?

Arkq, 2020-09-08:
> Is a new release imminent?

I was planing to do it in the previous week, however, there will be one more backward-incompatible change (in the D-Bus API), so I want to put it into the release 3.0.0 as well. I think it will be released in the next week or two.

> Why does it build on armv6/7 but not on aarch64?

I don't know... Maybe it is a different issue then... Maybe you should try to build master after all (if you don't want to change much, you can fork this repo to your github and make a release by yourself, then `sed -i 's/Arkq/gearhead/' some_pkg_file` and voila :D).

paul-1, 2020-09-08:
I'm running a fairly recent commit on aarch64.   Compiled clean.

gearhead, 2020-09-08:
I just built the git. I have not tried to see if it works, but it should.. the version is v2.1.0.r88.gdefe35c

closed 2020-09-13 by Arkq