arkq/bluez-alsa #371: Device or resource busy after reopening PCM

hrc-amilcar, 2020-09-10:
Hi, I'm running into an issue where `snd_pcm_hw_params` reports `"Device or resource busy"` if a PCM is closed and re-opened very quickly. The issue is alleviated if I insert a ~500ms sleep between `snd_pcm_close` and the subsequent `snd_pcm_open`. I'm on commit 7ff14a0.

Is there perhaps a synchronization issue when the same SCO is closed and re-opened immediately, that doesn't allow the `hci_sco_connect` to succeed after `shutdown`/`close`?

I will modify aplay to open -> close -> open to see if I can reproduce it outside of my application. I can't get it to happen on the command line with a while loop around aplay. I'm sure enough time passes between the process ending and restarting.

Here is the output without the sleep:
```
D: src/asound/bluealsa-pcm.c:761: /org/bluealsa/hci0/dev_00_1B_41_B2_86_9A/hfpag/source: Setting constraints
D: src/asound/bluealsa-pcm.c:397: /org/bluealsa/hci0/dev_00_1B_41_B2_86_9A/hfpag/source: Initializing HW
./bluealsa: D: src/dbus.c:57: Called: org.bluealsa.PCM1.Open() on /org/bluealsa/hci0/dev_00_1B_41_B2_86_9A/hfpag/source
./bluealsa: D: src/ba-transport.c:964: New SCO link: 00:1B:41:B2:86:9A: 22
./bluealsa: D: src/hci.c:132: SCO link socket MTU: 22: 180
D: src/asound/bluealsa-pcm.c:426: /org/bluealsa/hci0/dev_00_1B_41_B2_86_9A/hfpag/source: Selected HW buffer: 20 periods x 160 bytes == 3200 bytes
D: src/asound/bluealsa-pcm.c:441: /org/bluealsa/hci0/dev_00_1B_41_B2_86_9A/hfpag/source: Initializing SW
D: src/asound/bluealsa-pcm.c:441: /org/bluealsa/hci0/dev_00_1B_41_B2_86_9A/hfpag/source: Initializing SW
D: src/asound/bluealsa-pcm.c:480: /org/bluealsa/hci0/dev_00_1B_41_B2_86_9A/hfpag/source: Prepared
D: src/asound/bluealsa-pcm.c:441: /org/bluealsa/hci0/dev_00_1B_41_B2_86_9A/hfpag/source: Initializing SW
D: src/asound/bluealsa-pcm.c:480: /org/bluealsa/hci0/dev_00_1B_41_B2_86_9A/hfpag/source: Prepared
./bluealsa: D: src/dbus.c:57: Called: org.bluealsa.Manager1.GetPCMs() on /org/bluealsa
D: src/asound/bluealsa-pcm.c:761: /org/bluealsa/hci0/dev_00_1B_41_B2_86_9A/hfpag/sink: Setting constraints
D: src/asound/bluealsa-pcm.c:397: /org/bluealsa/hci0/dev_00_1B_41_B2_86_9A/hfpag/sink: Initializing HW
./bluealsa: D: src/dbus.c:57: Called: org.bluealsa.PCM1.Open() on /org/bluealsa/hci0/dev_00_1B_41_B2_86_9A/hfpag/sink
./bluealsa: D: src/ba-transport.c:949: Reusing SCO: 22
D: src/asound/bluealsa-pcm.c:417: /org/bluealsa/hci0/dev_00_1B_41_B2_86_9A/hfpag/sink: FIFO buffer size: 4096
D: src/asound/bluealsa-pcm.c:426: /org/bluealsa/hci0/dev_00_1B_41_B2_86_9A/hfpag/sink: Selected HW buffer: 20 periods x 160 bytes == 3200 bytes
D: src/asound/bluealsa-pcm.c:441: /org/bluealsa/hci0/dev_00_1B_41_B2_86_9A/hfpag/sink: Initializing SW
D: src/asound/bluealsa-pcm.c:441: /org/bluealsa/hci0/dev_00_1B_41_B2_86_9A/hfpag/sink: Initializing SW
D: src/asound/bluealsa-pcm.c:480: /org/bluealsa/hci0/dev_00_1B_41_B2_86_9A/hfpag/sink: Prepared
D: src/asound/bluealsa-pcm.c:441: /org/bluealsa/hci0/dev_00_1B_41_B2_86_9A/hfpag/sink: Initializing SW
D: src/asound/bluealsa-pcm.c:480: /org/bluealsa/hci0/dev_00_1B_41_B2_86_9A/hfpag/sink: Prepared
D: src/asound/bluealsa-pcm.c:317: /org/bluealsa/hci0/dev_00_1B_41_B2_86_9A/hfpag/sink: Starting
./bluealsa: D: src/ba-transport.c:789: PCM resumed: 18
./bluealsa: D: src/ba-transport.c:789: PCM resumed: 19
D: src/asound/bluealsa-pcm.c:317: /org/bluealsa/hci0/dev_00_1B_41_B2_86_9A/hfpag/source: Starting
D: src/asound/bluealsa-pcm.c:184: /org/bluealsa/hci0/dev_00_1B_41_B2_86_9A/hfpag/source: Starting IO loop: 14
D: src/asound/bluealsa-pcm.c:177: /org/bluealsa/hci0/dev_00_1B_41_B2_86_9A/hfpag/sink: Waiting for first period of frames
D: src/asound/bluealsa-pcm.c:184: /org/bluealsa/hci0/dev_00_1B_41_B2_86_9A/hfpag/sink: Starting IO loop: 19
D: src/asound/bluealsa-pcm.c:350: /org/bluealsa/hci0/dev_00_1B_41_B2_86_9A/hfpag/source: Stopping
D: src/asound/bluealsa-pcm.c:129: /org/bluealsa/hci0/dev_00_1B_41_B2_86_9A/hfpag/source: IO thread cleanup
./bluealsa: D: src/ba-transport.c:1006: Closing PCM: 19
D: src/asound/bluealsa-pcm.c:433: /org/bluealsa/hci0/dev_00_1B_41_B2_86_9A/hfpag/source: Freeing HW
D: src/asound/bluealsa-pcm.c:386: /org/bluealsa/hci0/dev_00_1B_41_B2_86_9A/hfpag/source: Closing
D: src/asound/bluealsa-pcm.c:350: /org/bluealsa/hci0/dev_00_1B_41_B2_86_9A/hfpag/sink: Stopping
D: src/asound/bluealsa-pcm.c:129: /org/bluealsa/hci0/dev_00_1B_41_B2_86_9A/hfpag/sink: IO thread cleanup
./bluealsa: D: src/ba-transport.c:823: PCM dropped: 18
./bluealsa: D: src/ba-transport.c:1006: Closing PCM: 18
D: src/asound/bluealsa-pcm.c:433: /org/bluealsa/hci0/dev_00_1B_41_B2_86_9A/hfpag/sink: Freeing HW
D: src/asound/bluealsa-pcm.c:386: /org/bluealsa/hci0/dev_00_1B_41_B2_86_9A/hfpag/sink: Closing
./bluealsa: D: src/sco.c:328: Releasing SCO due to PCM inactivity
./bluealsa: D: src/ba-transport.c:982: Closing SCO: 22
./bluealsa: D: src/sco.c:328: Releasing SCO due to PCM inactivity
./bluealsa: D: src/dbus.c:57: Called: org.bluealsa.Manager1.GetPCMs() on /org/bluealsa
D: src/asound/bluealsa-pcm.c:761: /org/bluealsa/hci0/dev_00_1B_41_B2_86_9A/hfpag/source: Setting constraints
D: src/asound/bluealsa-pcm.c:397: /org/bluealsa/hci0/dev_00_1B_41_B2_86_9A/hfpag/source: Initializing HW
./bluealsa: D: src/dbus.c:57: Called: org.bluealsa.PCM1.Open() on /org/bluealsa/hci0/dev_00_1B_41_B2_86_9A/hfpag/source
./bluealsa: E: Couldn't establish SCO link: Software caused connection abort
D: src/asound/bluealsa-pcm.c:404: /org/bluealsa/hci0/dev_00_1B_41_B2_86_9A/hfpag/source: Couldn't open PCM: Acquire transport: Software caused connection abort
```

Here is the output with the 500ms sleep:
```
D: src/asound/bluealsa-pcm.c:761: /org/bluealsa/hci0/dev_00_1B_41_B2_86_9A/hfpag/source: Setting constraints
D: src/asound/bluealsa-pcm.c:397: /org/bluealsa/hci0/dev_00_1B_41_B2_86_9A/hfpag/source: Initializing HW
./bluealsa: D: src/dbus.c:57: Called: org.bluealsa.PCM1.Open() on /org/bluealsa/hci0/dev_00_1B_41_B2_86_9A/hfpag/source
./bluealsa: D: src/ba-transport.c:964: New SCO link: 00:1B:41:B2:86:9A: 22
./bluealsa: D: src/hci.c:132: SCO link socket MTU: 22: 180
D: src/asound/bluealsa-pcm.c:426: /org/bluealsa/hci0/dev_00_1B_41_B2_86_9A/hfpag/source: Selected HW buffer: 20 periods x 160 bytes == 3200 bytes
D: src/asound/bluealsa-pcm.c:441: /org/bluealsa/hci0/dev_00_1B_41_B2_86_9A/hfpag/source: Initializing SW
D: src/asound/bluealsa-pcm.c:441: /org/bluealsa/hci0/dev_00_1B_41_B2_86_9A/hfpag/source: Initializing SW
[... 44 lines cut ...]
D: src/asound/bluealsa-pcm.c:441: /org/bluealsa/hci0/dev_00_1B_41_B2_86_9A/hfpag/source: Initializing SW
D: src/asound/bluealsa-pcm.c:441: /org/bluealsa/hci0/dev_00_1B_41_B2_86_9A/hfpag/source: Initializing SW
D: src/asound/bluealsa-pcm.c:480: /org/bluealsa/hci0/dev_00_1B_41_B2_86_9A/hfpag/source: Prepared
D: src/asound/bluealsa-pcm.c:441: /org/bluealsa/hci0/dev_00_1B_41_B2_86_9A/hfpag/source: Initializing SW
D: src/asound/bluealsa-pcm.c:480: /org/bluealsa/hci0/dev_00_1B_41_B2_86_9A/hfpag/source: Prepared
./bluealsa: D: src/dbus.c:57: Called: org.bluealsa.Manager1.GetPCMs() on /org/bluealsa
D: src/asound/bluealsa-pcm.c:761: /org/bluealsa/hci0/dev_00_1B_41_B2_86_9A/hfpag/sink: Setting constraints
D: src/asound/bluealsa-pcm.c:397: /org/bluealsa/hci0/dev_00_1B_41_B2_86_9A/hfpag/sink: Initializing HW
./bluealsa: D: src/dbus.c:57: Called: org.bluealsa.PCM1.Open() on /org/bluealsa/hci0/dev_00_1B_41_B2_86_9A/hfpag/sink
./bluealsa: D: src/ba-transport.c:949: Reusing SCO: 22
D: src/asound/bluealsa-pcm.c:417: /org/bluealsa/hci0/dev_00_1B_41_B2_86_9A/hfpag/sink: FIFO buffer size: 4096
D: src/asound/bluealsa-pcm.c:426: /org/bluealsa/hci0/dev_00_1B_41_B2_86_9A/hfpag/sink: Selected HW buffer: 20 periods x 160 bytes == 3200 bytes
D: src/asound/bluealsa-pcm.c:441: /org/bluealsa/hci0/dev_00_1B_41_B2_86_9A/hfpag/sink: Initializing SW
D: src/asound/bluealsa-pcm.c:441: /org/bluealsa/hci0/dev_00_1B_41_B2_86_9A/hfpag/sink: Initializing SW
D: src/asound/bluealsa-pcm.c:480: /org/bluealsa/hci0/dev_00_1B_41_B2_86_9A/hfpag/sink: Prepared
D: src/asound/bluealsa-pcm.c:441: /org/bluealsa/hci0/dev_00_1B_41_B2_86_9A/hfpag/sink: Initializing SW
D: src/asound/bluealsa-pcm.c:480: /org/bluealsa/hci0/dev_00_1B_41_B2_86_9A/hfpag/sink: Prepared
D: src/asound/bluealsa-pcm.c:317: /org/bluealsa/hci0/dev_00_1B_41_B2_86_9A/hfpag/sink: Starting
./bluealsa: D: src/ba-transport.c:789: PCM resumed: 18
./bluealsa: D: src/ba-transport.c:789: PCM resumed: 19
D: src/asound/bluealsa-pcm.c:317: /org/bluealsa/hci0/dev_00_1B_41_B2_86_9A/hfpag/source: Starting
D: src/asound/bluealsa-pcm.c:184: /org/bluealsa/hci0/dev_00_1B_41_B2_86_9A/hfpag/source: Starting IO loop: 14
D: src/asound/bluealsa-pcm.c:177: /org/bluealsa/hci0/dev_00_1B_41_B2_86_9A/hfpag/sink: Waiting for first period of frames
D: src/asound/bluealsa-pcm.c:184: /org/bluealsa/hci0/dev_00_1B_41_B2_86_9A/hfpag/sink: Starting IO loop: 19
```

hrc-amilcar, 2020-09-10:
No luck reproducing with aplay/arecord yet, but I did manage to get it to work without the sleep by retrying `snd_pcm_hw_params` after getting the busy error (-16).

Is there something not flushed properly or the previous thread still using the PCM?

```
D: src/asound/bluealsa-pcm.c:350: /org/bluealsa/hci0/dev_00_1B_41_B2_86_9A/hfpag/source: Stopping
D: src/asound/bluealsa-pcm.c:129: /org/bluealsa/hci0/dev_00_1B_41_B2_86_9A/hfpag/source: IO thread cleanup
./bluealsa: D: src/ba-transport.c:1006: Closing PCM: 19
./bluealsa: D: src/ba-transport.c:823: PCM dropped: 18
D: src/asound/bluealsa-pcm.c:433: /org/bluealsa/hci0/dev_00_1B_41_B2_86_9A/hfpag/source: Freeing HW
D: src/asound/bluealsa-pcm.c:386: /org/bluealsa/hci0/dev_00_1B_41_B2_86_9A/hfpag/source: Closing
D: src/asound/bluealsa-pcm.c:350: /org/bluealsa/hci0/dev_00_1B_41_B2_86_9A/hfpag/sink: Stopping
D: src/asound/bluealsa-pcm.c:129: /org/bluealsa/hci0/dev_00_1B_41_B2_86_9A/hfpag/sink: IO thread cleanup
./bluealsa: D: src/ba-transport.c:1006: Closing PCM: 18
D: src/asound/bluealsa-pcm.c:433: /org/bluealsa/hci0/dev_00_1B_41_B2_86_9A/hfpag/sink: Freeing HW
D: src/asound/bluealsa-pcm.c:386: /org/bluealsa/hci0/dev_00_1B_41_B2_86_9A/hfpag/sink: Closing
./bluealsa: D: src/sco.c:328: Releasing SCO due to PCM inactivity
./bluealsa: D: src/ba-transport.c:982: Closing SCO: 22
./bluealsa: D: src/sco.c:328: Releasing SCO due to PCM inactivity
./bluealsa: D: src/dbus.c:57: Called: org.bluealsa.Manager1.GetPCMs() on /org/bluealsa
D: src/asound/bluealsa-pcm.c:761: /org/bluealsa/hci0/dev_00_1B_41_B2_86_9A/hfpag/source: Setting constraints
D: src/asound/bluealsa-pcm.c:397: /org/bluealsa/hci0/dev_00_1B_41_B2_86_9A/hfpag/source: Initializing HW
./bluealsa: D: src/dbus.c:57: Called: org.bluealsa.PCM1.Open() on /org/bluealsa/hci0/dev_00_1B_41_B2_86_9A/hfpag/source
./bluealsa: E: Couldn't establish SCO link: Software caused connection abort
D: src/asound/bluealsa-pcm.c:404: /org/bluealsa/hci0/dev_00_1B_41_B2_86_9A/hfpag/source: Couldn't open PCM: Acquire transport: Software caused connection abort
***** Unable to set hw params: -16 = Device or resource busy. Trying again...
D: src/asound/bluealsa-pcm.c:397: /org/bluealsa/hci0/dev_00_1B_41_B2_86_9A/hfpag/source: Initializing HW
./bluealsa: D: src/dbus.c:57: Called: org.bluealsa.PCM1.Open() on /org/bluealsa/hci0/dev_00_1B_41_B2_86_9A/hfpag/source
./bluealsa: D: src/ba-transport.c:964: New SCO link: 00:1B:41:B2:86:9A: 22
./bluealsa: D: src/hci.c:132: SCO link socket MTU: 22: 180
D: src/asound/bluealsa-pcm.c:426: /org/bluealsa/hci0/dev_00_1B_41_B2_86_9A/hfpag/source: Selected HW buffer: 20 periods x 160 bytes == 3200 bytes
D: src/asound/bluealsa-pcm.c:441: /org/bluealsa/hci0/dev_00_1B_41_B2_86_9A/hfpag/source: Initializing SW
D: src/asound/bluealsa-pcm.c:441: /org/bluealsa/hci0/dev_00_1B_41_B2_86_9A/hfpag/source: Initializing SW
D: src/asound/bluealsa-pcm.c:480: /org/bluealsa/hci0/dev_00_1B_41_B2_86_9A/hfpag/source: Prepared
D: src/asound/bluealsa-pcm.c:441: /org/bluealsa/hci0/dev_00_1B_41_B2_86_9A/hfpag/source: Initializing SW
D: src/asound/bluealsa-pcm.c:480: /org/bluealsa/hci0/dev_00_1B_41_B2_86_9A/hfpag/source: Prepared
./bluealsa: D: src/dbus.c:57: Called: org.bluealsa.Manager1.GetPCMs() on /org/bluealsa
D: src/asound/bluealsa-pcm.c:761: /org/bluealsa/hci0/dev_00_1B_41_B2_86_9A/hfpag/sink: Setting constraints
D: src/asound/bluealsa-pcm.c:397: /org/bluealsa/hci0/dev_00_1B_41_B2_86_9A/hfpag/sink: Initializing HW
./bluealsa: D: src/dbus.c:57: Called: org.bluealsa.PCM1.Open() on /org/bluealsa/hci0/dev_00_1B_41_B2_86_9A/hfpag/sink
./bluealsa: D: src/ba-transport.c:949: Reusing SCO: 22
D: src/asound/bluealsa-pcm.c:417: /org/bluealsa/hci0/dev_00_1B_41_B2_86_9A/hfpag/sink: FIFO buffer size: 4096
D: src/asound/bluealsa-pcm.c:426: /org/bluealsa/hci0/dev_00_1B_41_B2_86_9A/hfpag/sink: Selected HW buffer: 20 periods x 160 bytes == 3200 bytes
D: src/asound/bluealsa-pcm.c:441: /org/bluealsa/hci0/dev_00_1B_41_B2_86_9A/hfpag/sink: Initializing SW
D: src/asound/bluealsa-pcm.c:441: /org/bluealsa/hci0/dev_00_1B_41_B2_86_9A/hfpag/sink: Initializing SW
D: src/asound/bluealsa-pcm.c:480: /org/bluealsa/hci0/dev_00_1B_41_B2_86_9A/hfpag/sink: Prepared
D: src/asound/bluealsa-pcm.c:441: /org/bluealsa/hci0/dev_00_1B_41_B2_86_9A/hfpag/sink: Initializing SW
D: src/asound/bluealsa-pcm.c:480: /org/bluealsa/hci0/dev_00_1B_41_B2_86_9A/hfpag/sink: Prepared
D: src/asound/bluealsa-pcm.c:317: /org/bluealsa/hci0/dev_00_1B_41_B2_86_9A/hfpag/sink: Starting
./bluealsa: D: src/ba-transport.c:789: PCM resumed: 18
./bluealsa: D: src/ba-transport.c:789: PCM resumed: 19
```

hrc-amilcar, 2020-09-10:
Adding a retry to `transport_acquire_bt_sco` if `hci_sco_connect` fails with ECONNABORTED also fixes it.
i.e.:
```
hci_sco_open;
if( hci_sco_connect fails with ECONNABORTED )
{
    close socket;
    hci_sco_open;
    hci_sco_connect; <-- doesn't fail
}
```

hrc-amilcar, 2020-09-10:
Perhaps this is a red herring but... are there multiple threads accessing the transport bt_fd while it's being shutdown/closed?

I added some logs and sleeps in `transport_release_bt_sco` between the `shutdown` and `close` of the socket, and it makes things even worse:
i.e.:
```
debug("Shutting down SCO: %d", t->bt_fd);
shutdown(t->bt_fd, SHUT_RDWR);
usleep( 1000 * 1000 );
debug("Closing SCO: %d", t->bt_fd);
close(t->bt_fd);
t->bt_fd = -1;
usleep( 1000 * 1000 );
debug("Closed SCO");
```

Here are the logs with thread IDs added:
```
D: 76FA7F80 src/ba-transport.c:1020: Closing PCM: 19
D: 76FA7F80 src/ba-transport.c:823: PCM dropped: 12
D: 745FD410 src/sco.c:507: PCM poll error status: 0x10
D: 745FD410 src/ba-transport.c:1020: Closing PCM: 12
D: 745FD410 src/sco.c:328: Releasing SCO due to PCM inactivity
D: 745FD410 src/ba-transport.c:993: Shutting down SCO: 22
D: 76FA7F80 src/dbus.c:57: Called: org.bluealsa.Manager1.GetPCMs() on /org/bluealsa
D: 76FA7F80 src/dbus.c:57: Called: org.bluealsa.PCM1.Open() on /org/bluealsa/hci0/dev_00_1B_41_B2_86_9A/hfpag/source
D: 73DFC410 src/ba-transport.c:949: Reusing SCO: 22
D: 76FA7F80 src/dbus.c:57: Called: org.bluealsa.Manager1.GetPCMs() on /org/bluealsa
D: 76FA7F80 src/dbus.c:57: Called: org.bluealsa.PCM1.Open() on /org/bluealsa/hci0/dev_00_1B_41_B2_86_9A/hfpag/sink
D: 73DFC410 src/ba-transport.c:949: Reusing SCO: 22
D: 76FA7F80 src/ba-transport.c:789: PCM resumed: 12
D: 76FA7F80 src/ba-transport.c:789: PCM resumed: 19
D: 745FD410 src/ba-transport.c:996: Closing SCO: 22
D: 745FD410 src/ba-transport.c:1000: Closed SCO
```
It seems like thread 73DFC410 is attempting to "reuse" SCO 22 that is in the middle of being closed by thread 745FD410. I wouldn't think sleeping in the close would break things. I'm trying to figure out what thread 73DFC410 is, perhaps with the `callstackdump` mechanism.

hrc-amilcar, 2020-09-10:
Adding lock/unlock to the `sco_thread` when the transport close signal comes in, allows the next PCM to open successfully, with sleeps in the close:
```
ba_transport_pthread_cleanup_lock(t);
t->release(t);
ba_transport_pthread_cleanup_unlock(t);
```
Not sure if that's the proper way to protect the `bt_fd` from being accessed while it's shutting down, or if there are other places in the code where locking is needed.

But even with this, `ECONNABORTED` still happens. I still need at least a 500ms sleep somewhere in the `transport_release_bt_sco` or the retry in `transport_acquire_bt_sco` if the connect fails.

Arkq, 2020-09-13:
> Hi, I'm running into an issue where snd_pcm_hw_params reports "Device or resource busy" if a PCM is closed and re-opened very quickly. The issue is alleviated if I insert a ~500ms sleep between snd_pcm_close and the subsequent snd_pcm_open.

Yeeessss.... that might happen with current bluealsa design. Closing BlueALSA PCM is made asynchronously, and there is a constrain of only one simultaneous client. So, in the end there might be a race in there. Before D-Bus API, closing bluealsa PCM was made synchronously, but in order to simplify things I've decided not to create dedicated "Close" API call. In the long run, I've got plan to create a mixer in bluelasa server, which will allow to connect not one client to particular PCM but many.

The solution for your issue, for now, would be to create a dedicated Close D-Bus API call and add some locking. Please note, that I will not merge such feature into the master, since this "bug" is unfortunately by design.

> Perhaps this is a red herring but... are there multiple threads accessing the transport bt_fd while it's being shutdown/closed?

Hmm... you might be right that some synchronization is missing in here :/ I will look into that, but not now. Maybe I will have some spare time for that in a month or two...

hrc-amilcar, 2020-09-14:
Thanks for your reponse, @Arkq.

A multi-client mixing feature sounds great, and it may indirectly fix what I'm seeing, but I'm currently only opening a single ALSA PCM for one BT device at a time. I just happen to run into this issue because I closed the PCM and re-opened it to the same device within 50ms or so. If I understood correctly, I see what you're saying about adding a Close D-Bus call that has locking while maintaining the single client constraint.

I look forward to your findings of missing synchronization as well as any new features and improvements! Thank you.

borine, 2020-12-24:
@hrc-amilcar I realize much time has passed since you raised this issue, but if you are still working on your application I would be very interested to know if the changes in PR #406 fix this issue. I have chosen timeout values that work for me, but have no way of knowing if they are generally appropriate. If you get time to experiment with this after Christmas your comments would be most helpful.

hrc-amilcar, 2021-01-10:
Hi, @borine I'm still working on my application but haven't made any changes related to Bluetooth handling in a long while. I ended up leaving in the artificial delay between close and re-open that has mitigated the problem for now.

I can try the PR in #406 to see if it helps. Is there a particular commit hash you would prefer me to try, or simply pull the borine:sco-state-model branch?

borine, 2021-01-11:
#406 does not apply cleanly any more to the master branch sources, so it's probably easiest to just use the https://github.com/borine/bluez-alsa/tree/sco-state-model branch for now.

From https://github.com/Arkq/bluez-alsa/pull/406#issuecomment-757448422, it seems @Arkq is considering a revised thread model for SCO, in which case my solution may no longer be appropriate. It may still be useful to know if my branch works for you, in case that helps with implementing the new model, so if you have time to test it that would be great!

hrc-amilcar, 2021-01-26:
Hi @borine, I restored my app to a previous state and verified that I was once again getting the "Software caused connection abort" error after rapid close + open.

I then pulled your sco-state-model branch and rebuilt. However, I immediately got "Invalid Arguments" from `snd_pcm_hw_params_get_period_time`. I swapped the order of `snd_pcm_hw_params_set_buffer_size_near` and `snd_pcm_hw_params_set_period_size_near` and I was back in business. I remember I ran into this before and reported #376 . Perhaps this can be made more robust without changing the order in the user code?

The good news is that I no longer get the "connection abort" error with your latest code. I can keep testing with this for a while.

borine, 2021-01-26:
> The good news is that I no longer get the "connection abort" error

Great, I think that validates the approach of using a timer rather than a sleep to delay re-connecting a SCO socket. Thank you for testing this.
 
> Perhaps this can be made more robust without changing the order in the user code?

No, the bluealsa hw params code is correct. The previous algorithm (before commit 3b474ae) was too restrictive and prevented applications such as alsaloop, jackd etc from using their default hardware settings. Your issue is that you appear to be calling `snd_pcm_hw_params_get_period_time()` *before* you have committed the config with `snd_pcm_hw_params()`. At that time there is not enough info in the config space to calculate the period time (I guess alsa-lib must fill in some default values when the config space is committed). That is not documented by the alsa project, so you can only discover it by seeing this failure in you own code. Alsa documentation is frustratingly lacking. For your informational logging I recommend that you read all the hardware param values after calling `snd_pcm_hw_params()` so that you report the final values calculated and commited by alsa-lib.

It is also better practice to set the period size before setting the buffer size - that way alsa-lib can ensure that constraints on period and buffer are all met. It is possible to set them in the reverse order, but then you have to be prepared to retry different values for period size if the call to snd_pcm_hw_params_set_period_size_near() fails.

Note that these restrictions are all the result of alsa-lib code, and you can hit the same issues with other devices, not just bluealsa. There is nothing bluealsa can do to improve the situation for all applications - changes to help your application would only result in breaking other applications.

hrc-amilcar, 2021-01-26:
@borine

I'm happy to read the parameters after `snd_pcm_hw_params`. I just hadn't seen that documented.

The [example on the alsa-project website](https://www.alsa-project.org/alsa-doc/alsa-lib/_2test_2pcm_8c-example.html) must be ancient or untested, as they read the some of the values before committing the parameters. The only call that seems to be allowed before committing is `snd_pcm_hw_params_get_buffer_size_max` and perhaps other similar fixed parameters, as in aplay.c and PulseAudio source.

I will fix that in my own code. I also use a third party library that reads values in the incorrect order. I'll ask them to fix that.

Thank you for the info.

borine, 2021-01-26:
> The example on the alsa-project website must be ancient or untested

Not untested - that code explicilty sets the period time then reads the resulting period size; it just happens that that sequence does work. To fully understand all this, it is necessary to read through the alsa-lib code. But be warned it is very complicated, and not well commented. Programming to the alsa-lib api is, I think, very painful and unnecessarily hard work. The ALSA project would benefit from some help from professional documentation writers, but it seems that even after more than 20 years they have received little interest in improving their docs.

closed 2021-07-09 by Arkq

Arkq, 2021-07-09:
@hrc-amilcar I've just pushed to master a fix for SCO link close/open issue. There is a `HCI_SCO_CLOSE_CONNECT_QUIRK_DELAY` definition which defines a delay between close() and connect(). If you will experience any connection issues, please increase this value (it will work up to 1 second) and report back here how much delay works for you. Also, with current master branch you can specify connection keep-alive for SCO (and A2DP) connections. So, if your use case is that you often try to open PCM just after closing it, you can start bluealsa as follows, e.g: `bluealsa --keep-alive=0.5` - it will add 0.5 seconds keep alive for BT connections.

hrc-amilcar, 2021-07-09:
Thank you @Arkq ! I will take out my own delay and give this a try soon.

commit 0de2f8cb (2021-07-09): Delay between close/connect calls for SCO links

It seems that there is some kind of an anomaly during SCO socket
shutdown. As a workaround this commit adds forced delay between
close(2) and connect(2) for SCO links.

This workaround is based on a pull request submitted by @borine.

Fixes #371 and closes #406
commit 67734420 (2021-07-09): Delay between close/connect calls for SCO links

It seems that there is some kind of an anomaly during SCO socket
shutdown. As a workaround this commit adds forced delay between
close(2) and connect(2) for SCO links.

This workaround is based on a pull request submitted by @borine.

Fixes #371 and closes #406