arkq/bluez-alsa #373: bluetooth no longer asks to access aplay when paired

gearhead, 2020-09-12:
I am building master. This is on Arch Linux aarch64 wiith bluez 5.55-1. I am using both bluealsa service files from the wiki.

Previously (with v2.1.0), I would get 2 questions at pair time if I wanted to use the bluealsa aplay device. I cannot remember what they were but thet were long numbers and in order for me to play from my phone to the RPi, I had to answer yes to both other wise it would not work. 

Now with the 'master' git based package build, I ensure that the bluealsa-aplay is up and running:
```
● bluealsa-aplay.service - Bluealsa audio player
     Loaded: loaded (/etc/systemd/system/bluealsa-aplay.service; enabled; vendor preset: disabled)
     Active: active (running) since Sat 2020-09-12 12:51:04 CDT; 15min ago
       Docs: https://github.com/Arkq/bluez-alsa/
   Main PID: 4084 (bluealsa-aplay)
      Tasks: 1 (limit: 1067)
     CGroup: /system.slice/bluealsa-aplay.service
             └─4084 /usr/bin/bluealsa-aplay -Dhw:0,1 00:00:00:00:00:00

Sep 12 12:51:04 rune64 systemd[1]: Started Bluealsa audio player.
```
My /etc/default/bluealsa-aplay is:
`OPTIONS="-Dhw:0,1"`
But now when I connect, All I can do it share my contact list from my phone. No longer any audio devices show up.
```
[bluetooth]# show
Controller 43:45:C0:00:1F:AC (public)
        Name: rune64
        Alias: rune64
        Class: 0x00480000
        Powered: yes
        Discoverable: no
        DiscoverableTimeout: 0x000000b4
        Pairable: yes
        UUID: A/V Remote Control        (0000110e-0000-1000-8000-00805f9b34fb)
        UUID: Audio Source              (0000110a-0000-1000-8000-00805f9b34fb)
        UUID: PnP Information           (00001200-0000-1000-8000-00805f9b34fb)
        UUID: Headset AG                (00001112-0000-1000-8000-00805f9b34fb)
        UUID: Handsfree Audio Gateway   (0000111f-0000-1000-8000-00805f9b34fb)
        UUID: A/V Remote Control Target (0000110c-0000-1000-8000-00805f9b34fb)
        UUID: Generic Access Profile    (00001800-0000-1000-8000-00805f9b34fb)
        UUID: Generic Attribute Profile (00001801-0000-1000-8000-00805f9b34fb)
        UUID: Device Information        (0000180a-0000-1000-8000-00805f9b34fb)
        Modalias: usb:v1D6Bp0246d0537
        Discovering: no
        Roles: central
        Roles: peripheral
Advertising Features:
        ActiveInstances: 0x00 (0)
        SupportedInstances: 0x05 (5)
        SupportedIncludes: tx-power
        SupportedIncludes: appearance
        SupportedIncludes: local-name
```
isn't headset AG the device? Why does it no longer ask for me to allow a connection?
this is with bluez 5.55-1 on Arch I no longer have bluez-libs 5.54 to back date it.

borine, 2020-09-12:
> Isn't headset AG the device

No

If you want to play A2DP streams from your phone through bluealsa-aplay, you need to start bluealsa with `-p a2dp-sink`
then bluetoothctl will show
```
UUID: Audio Sink                (0000110b-0000-1000-8000-00805f9b34fb)
```
If you want HFP voice connection from the phone, you need to start bluealsa with `-p hfp-hf`
then bluetoothctl will show
```
UUID: Handsfree                 (0000111e-0000-1000-8000-00805f9b34fb)
```
You can , of course, start bluealsa with both options (and others)

gearhead, 2020-09-12:
oops, sorry for the confusion on my part. 
I'm back in business. Will upgrade back to a release version when available.

gearhead, 2020-12-12:
this was a user error and has been resolved.

closed 2020-12-12 by gearhead