arkq/bluez-alsa #376: snd_pcm_hw_params_get_period_time invalid argument

hrc-amilcar, 2020-09-14:
Hi, I'm getting "Invalid Argument" from `snd_pcm_hw_params_get_period_time` since commit 3b474ae.

Before commit 3b474ae:
```
D: src/asound/bluealsa-pcm.c:900: Getting BlueALSA PCM: CAPTURE 00:1B:41:B2:86:9A sco
./bluealsa: D: src/dbus.c:57: Called: org.bluealsa.Manager1.GetPCMs() on /org/bluealsa
D: src/asound/bluealsa-pcm.c:769: /org/bluealsa/hci0/dev_00_1B_41_B2_86_9A/hfpag/source: Setting constraints
D: src/asound/bluealsa-pcm.c:396: /org/bluealsa/hci0/dev_00_1B_41_B2_86_9A/hfpag/source: Initializing HW
./bluealsa: D: src/dbus.c:57: Called: org.bluealsa.PCM1.Open() on /org/bluealsa/hci0/dev_00_1B_41_B2_86_9A/hfpag/source
./bluealsa: D: src/dbus.c:57: Called: org.bluealsa.Manager1.GetPCMs() on /org/bluealsa
./bluealsa: D: src/ba-transport.c:964: New SCO link: 00:1B:41:B2:86:9A: 22
./bluealsa: D: src/hci.c:132: SCO link socket MTU: 22: 180
D: src/asound/bluealsa-pcm.c:425: /org/bluealsa/hci0/dev_00_1B_41_B2_86_9A/hfpag/source: Selected HW buffer: 20 periods x 160 bytes == 3200 bytes
D: src/asound/bluealsa-pcm.c:440: /org/bluealsa/hci0/dev_00_1B_41_B2_86_9A/hfpag/source: Initializing SW
D: src/asound/bluealsa-pcm.c:440: /org/bluealsa/hci0/dev_00_1B_41_B2_86_9A/hfpag/source: Initializing SW
D: src/asound/bluealsa-pcm.c:477: /org/bluealsa/hci0/dev_00_1B_41_B2_86_9A/hfpag/source: Prepared
Open capture successful: buffer_size=1600, buffer_time=200000us, period_size=80, period_time=10000us
D: src/asound/bluealsa-pcm.c:440: /org/bluealsa/hci0/dev_00_1B_41_B2_86_9A/hfpag/source: Initializing SW
D: src/asound/bluealsa-pcm.c:477: /org/bluealsa/hci0/dev_00_1B_41_B2_86_9A/hfpag/source: Prepared
D: src/asound/bluealsa-pcm.c:900: Getting BlueALSA PCM: PLAYBACK 00:1B:41:B2:86:9A sco
./bluealsa: D: src/dbus.c:57: Called: org.bluealsa.Manager1.GetPCMs() on /org/bluealsa
D: src/asound/bluealsa-pcm.c:769: /org/bluealsa/hci0/dev_00_1B_41_B2_86_9A/hfpag/sink: Setting constraints
D: src/asound/bluealsa-pcm.c:396: /org/bluealsa/hci0/dev_00_1B_41_B2_86_9A/hfpag/sink: Initializing HW
./bluealsa: D: src/dbus.c:57: Called: org.bluealsa.PCM1.Open() on /org/bluealsa/hci0/dev_00_1B_41_B2_86_9A/hfpag/sink
./bluealsa: D: src/ba-transport.c:949: Reusing SCO: 22
D: src/asound/bluealsa-pcm.c:416: /org/bluealsa/hci0/dev_00_1B_41_B2_86_9A/hfpag/sink: FIFO buffer size: 4096
D: src/asound/bluealsa-pcm.c:425: /org/bluealsa/hci0/dev_00_1B_41_B2_86_9A/hfpag/sink: Selected HW buffer: 20 periods x 160 bytes == 3200 bytes
D: src/asound/bluealsa-pcm.c:440: /org/bluealsa/hci0/dev_00_1B_41_B2_86_9A/hfpag/sink: Initializing SW
D: src/asound/bluealsa-pcm.c:440: /org/bluealsa/hci0/dev_00_1B_41_B2_86_9A/hfpag/sink: Initializing SW
D: src/asound/bluealsa-pcm.c:477: /org/bluealsa/hci0/dev_00_1B_41_B2_86_9A/hfpag/sink: Prepared
Open playback successful: buffer_size=1600, buffer_time=200000us, period_size=80, period_time=10000us
```

After commit 3b474ae:
```
D: src/asound/bluealsa-pcm.c:900: Getting BlueALSA PCM: CAPTURE 00:1B:41:B2:86:9A sco
./bluealsa: D: src/dbus.c:57: Called: org.bluealsa.Manager1.GetPCMs() on /org/bluealsa
D: src/asound/bluealsa-pcm.c:769: /org/bluealsa/hci0/dev_00_1B_41_B2_86_9A/hfpag/source: Setting constraints
snd_pcm_hw_params_get_period_time failed: "Invalid argument"
```

The sequence of calls is:
```
snd_pcm_hw_params_set_format( S16LE );
snd_pcm_hw_params_set_channels( 1 );
snd_pcm_hw_params_set_rate_near( 8000 );
snd_pcm_hw_params_set_buffer_size_near( 1536 );
buffer_size = snd_pcm_hw_params_get_buffer_size();
buffer_time = snd_pcm_hw_params_get_buffer_time();
snd_pcm_hw_params_set_period_size_near( 64 );
period_size = snd_pcm_hw_params_get_period_size();
period_time = snd_pcm_hw_params_get_period_time();
```
If I don't call `snd_pcm_hw_params_get_period_time` then it proceeds successfully:
```
D: src/asound/bluealsa-pcm.c:900: Getting BlueALSA PCM: CAPTURE 00:1B:41:B2:86:9A sco
./bluealsa: D: src/dbus.c:57: Called: org.bluealsa.Manager1.GetPCMs() on /org/bluealsa
D: src/asound/bluealsa-pcm.c:769: /org/bluealsa/hci0/dev_00_1B_41_B2_86_9A/hfpag/source: Setting constraints
D: src/asound/bluealsa-pcm.c:396: /org/bluealsa/hci0/dev_00_1B_41_B2_86_9A/hfpag/source: Initializing HW
./bluealsa: D: src/dbus.c:57: Called: org.bluealsa.PCM1.Open() on /org/bluealsa/hci0/dev_00_1B_41_B2_86_9A/hfpag/source
./bluealsa: D: src/dbus.c:57: Called: org.bluealsa.Manager1.GetPCMs() on /org/bluealsa
./bluealsa: D: src/ba-transport.c:964: New SCO link: 00:1B:41:B2:86:9A: 22
./bluealsa: D: src/hci.c:132: SCO link socket MTU: 22: 180
D: src/asound/bluealsa-pcm.c:425: /org/bluealsa/hci0/dev_00_1B_41_B2_86_9A/hfpag/source: Selected HW buffer: 19 periods x 160 bytes <= 3072 bytes
D: src/asound/bluealsa-pcm.c:440: /org/bluealsa/hci0/dev_00_1B_41_B2_86_9A/hfpag/source: Initializing SW
D: src/asound/bluealsa-pcm.c:440: /org/bluealsa/hci0/dev_00_1B_41_B2_86_9A/hfpag/source: Initializing SW
D: src/asound/bluealsa-pcm.c:477: /org/bluealsa/hci0/dev_00_1B_41_B2_86_9A/hfpag/source: Prepared
Open capture successful: buffer_size = 1536, buffer_time = 192000us, period_size = 80, period_time = 0us
D: src/asound/bluealsa-pcm.c:440: /org/bluealsa/hci0/dev_00_1B_41_B2_86_9A/hfpag/source: Initializing SW
D: src/asound/bluealsa-pcm.c:477: /org/bluealsa/hci0/dev_00_1B_41_B2_86_9A/hfpag/source: Prepared
D: src/asound/bluealsa-pcm.c:900: Getting BlueALSA PCM: PLAYBACK 00:1B:41:B2:86:9A sco
./bluealsa: D: src/dbus.c:57: Called: org.bluealsa.Manager1.GetPCMs() on /org/bluealsa
D: src/asound/bluealsa-pcm.c:769: /org/bluealsa/hci0/dev_00_1B_41_B2_86_9A/hfpag/sink: Setting constraints
D: src/asound/bluealsa-pcm.c:396: /org/bluealsa/hci0/dev_00_1B_41_B2_86_9A/hfpag/sink: Initializing HW
./bluealsa: D: src/dbus.c:57: Called: org.bluealsa.PCM1.Open() on /org/bluealsa/hci0/dev_00_1B_41_B2_86_9A/hfpag/sink
./bluealsa: D: src/ba-transport.c:949: Reusing SCO: 22
D: src/asound/bluealsa-pcm.c:416: /org/bluealsa/hci0/dev_00_1B_41_B2_86_9A/hfpag/sink: FIFO buffer size: 4096
D: src/asound/bluealsa-pcm.c:425: /org/bluealsa/hci0/dev_00_1B_41_B2_86_9A/hfpag/sink: Selected HW buffer: 19 periods x 160 bytes <= 3072 bytes
D: src/asound/bluealsa-pcm.c:440: /org/bluealsa/hci0/dev_00_1B_41_B2_86_9A/hfpag/sink: Initializing SW
D: src/asound/bluealsa-pcm.c:440: /org/bluealsa/hci0/dev_00_1B_41_B2_86_9A/hfpag/sink: Initializing SW
D: src/asound/bluealsa-pcm.c:477: /org/bluealsa/hci0/dev_00_1B_41_B2_86_9A/hfpag/sink: Prepared
Open playback successful: buffer_size = 1536, buffer_time = 192000us, period_size = 80, period_time = 0us
```
I'm not really using the period_time for anything, just informational logging. So I could technically get rid of the call, but just wanted to let you know in case it's an unwanted side-effect. Or perhaps get_period_time is not meant to be called unless you first call `snd_pcm_hw_params_set_period_time_near` ?

borine, 2020-09-14:
Its not well documented, but when configuring the hardware params you should set the period size *before* setiing the buffer size. So, (hopefully) if you sequence your calls:
```
snd_pcm_hw_params_set_period_size_near( 64 );
snd_pcm_hw_params_set_buffer_size_near( 1536 );
buffer_size = snd_pcm_hw_params_get_buffer_size();
buffer_time = snd_pcm_hw_params_get_buffer_time();
period_size = snd_pcm_hw_params_get_period_size();
period_time = snd_pcm_hw_params_get_period_time();
```
then everything should be ok.

borine, 2020-09-14:
@Arkq This issue has made me think it might be better to explicitly limit the minimum buffer size to 2 periods, rather than a number of bytes. It doesn't completely avoid this issue because the `plug` plugin can also suffer from it, but it seems more logical. Here is a simple patch:
```diff
diff --git a/src/asound/bluealsa-pcm.c b/src/asound/bluealsa-pcm.c
index 0503a3a..23dc90e 100644
--- a/src/asound/bluealsa-pcm.c
+++ b/src/asound/bluealsa-pcm.c
@@ -786,14 +786,13 @@ static int bluealsa_set_hw_constraint(struct bluealsa_pcm *pcm) {
 	 * period size is 10ms and the minimum buffer size is 2 periods.
 	 * Upper limits are not constrained. */
 	unsigned int min_p = 10 * pcm->ba_pcm.sampling / 1000 * pcm->ba_pcm.channels * 2;
-	unsigned int min_b = 2 * min_p;
 
 	if ((err = snd_pcm_ioplug_set_param_minmax(io, SND_PCM_IOPLUG_HW_PERIOD_BYTES,
 					min_p, 1024 * 16)) < 0)
 		return err;
 
-	if ((err = snd_pcm_ioplug_set_param_minmax(io, SND_PCM_IOPLUG_HW_BUFFER_BYTES,
-					min_b, 1024 * 1024 * 16)) < 0)
+	if ((err = snd_pcm_ioplug_set_param_minmax(io, SND_PCM_IOPLUG_HW_PERIODS,
+					2, 1024)) < 0)
 		return err;
 
 	if ((err = snd_pcm_ioplug_set_param_minmax(io, SND_PCM_IOPLUG_HW_CHANNELS,
```
Also, looking at this, the period size constraint calculation appears to assume that sample width is always 16 bits, but is that true any longer?

Arkq, 2020-09-14:
> This issue has made me think it might be better to explicitly limit the minimum buffer size to 2 periods, rather than a number of bytes.

Yes, that seems to be a better approach.

> Also, looking at this, the period size constraint calculation appears to assume that sample width is always 16 bits, but is that true any longer?

Yes, I've missed that place... I was grepping for ´sizeof(int16_t)´...

hrc-amilcar, 2021-01-26:
Issue addressed by reordering to set_period_size, set_buffer_size, install hw_params, then read values.

```
snd_pcm_hw_params_set_period_size_near( ... );
snd_pcm_hw_params_set_buffer_size_near( ... );
snd_pcm_hw_params( ... );
buffer_size = snd_pcm_hw_params_get_buffer_size();
buffer_time = snd_pcm_hw_params_get_buffer_time();
period_size = snd_pcm_hw_params_get_period_size();
period_time = snd_pcm_hw_params_get_period_time();
```

closed 2021-01-26 by hrc-amilcar