arkq/bluez-alsa #379: Use bluealsa device within a multi in asound.conf

carlyd95, 2020-09-19:
Hey I’m sorry if this is repetitive or not relevant but I have looked everywhere to no avail to get this to work, I figured you might be able to point me in the right direction. Essentially I want to use [Librespot](https://github.com/librespot-org/librespot)  and send the stream to a multi containing both my Bluetooth speaker and an ALSA loopback (to turn the stream into a mic/line in for [MLSC](https://github.com/TobKra96/music_led_strip_control)) 

I was trying to do it all in the ALSA But the asound.conf I have is still not working. I get the following error. 

Error: aplay -D multi piano.wav 
Playing WAVE 'piano.wav' : Signed 16 bit Little Endian, Rate 48000 Hz, Stereo
aplay: set_params:1345: Channels count non available

My asound.conf:
`pcm.multi {
    type route;
    slave.pcm {
        type multi;
        slaves.a.pcm "jbl";
        slaves.b.pcm "loopin";
        slaves.a.channels 2;
        slaves.b.channels 2;
        bindings.0.slave a;
        bindings.0.channel 0;
        bindings.1.slave a;
        bindings.1.channel 1;
        bindings.2.slave b;
        bindings.2.channel 0;
        bindings.3.slave b;
        bindings.3.channel 1;
    }

    ttable.0.0 1;
    ttable.1.1 1;
    ttable.0.2 1;
    ttable.1.3 1;
}

pcm.!default {
	type plug
	slave.pcm "multi"
} 

#carlyd95
pcm.loopin {
	type plug
	slave.pcm "plughw:Loopback,0,0"
}
pcm.loopout {
        type plug
        slave.pcm "plughw:Loopback,1,0"
}
pcm.jbl {
        type plug
        slave {
            pcm {
                type bluealsa
                device 98:52:3D:BB:82:F3
                profile "a2dp"
            }
        }
        hint {
            show on
            description "JBL Charge 4"
        }
     }

pcm.!default {
	type plug
	slave.pcm jbl
}`

Initially I got this to work with snapcast and ran two clients simultaneously on my raspi but id like a more simple solution.  
I would like to get this to work in ALSA but if it would be more practical to do this in pulse, fifo or any other method I’ll take that too. Thanks for any help you may offer! 

[What the project actually is if interested (but with Snapcast) ](https://carltons-karoo.com/ledboard.mov)

borine, 2020-09-19:
A couple of issues with your config:
- You have re-defined the pcm `default` twice: only the second definition will apply
- If you want to use the PCM called "multi" as a target for `aplay`, you need to wrap it in "plug". You do this for the first `pcm.!default` but as just noted, you then throw that definition away !

So I think the main thing is to add a pcm of type "plug" that wraps "multi" and use that with aplay.

There is a similar example in a previous issue here: https://github.com/Arkq/bluez-alsa/issues/352#issuecomment-656115275

carlyd95, 2020-09-19:
Ah, yes I added the latter messing around with it. Thank you for pointing that out. It appears whenever I add the default statement to asound.conf it makes mlsc mess up ill have to contact the owner of that git and see what's going on there. Thank you!

aplay seems to play to it now, or at least I hear it from the speaker however when running:
`aplay piano.wav | arecord -D loopout test.wav`
`aplay test.wav`
nothing plays, so im afraid its not sending audio to the loopback with the multi

on the other hand the following does play output back:
`aplay -D loopin piano.wav | arecord -D loopout test.wav`
`aplay test.wav`

also pyaudio used in [MLSC](https://github.com/TobKra96/music_led_strip_control) doesn't seem to like the declared default in asound.conf

ugh! I'm gonna reach out to the owner of that git. if you have any suggestions let me know, if not feel free to close the issue. im only keeping it open in case you have any ideas. Thanks

Arkq, 2020-10-12:
Here is my setup for `multi` plugin, which outputs audio to two BT headsets. I know it work's because I'm using it from time to time (just for the record, it works fine with alsa-lib 1.2.2):

```
pcm.jabra-1 {
	type plug
	slave.pcm {
		type bluealsa
		device "1C:48:F9:9D:81:5C"
		profile "a2dp"
	}
}
pcm.jabra-2 {
	type plug
	slave.pcm {
		type bluealsa
		device "74:5C:4B:2B:7D:98"
		profile "a2dp"
	}
}
pcm.jabra-multi {
  type route;
  slave.pcm {
    type multi;
    slaves.a.pcm "jabra-1";
    slaves.a.channels 2;
    slaves.b.pcm "jabra-2";
    slaves.b.channels 2;
    bindings.0 { slave a; channel 0; }
    bindings.1 { slave a; channel 1; }
    bindings.2 { slave b; channel 0; }
    bindings.3 { slave b; channel 1; }
  }
  ttable.0.0 1
  ttable.1.1 1
  ttable.0.2 1
  ttable.1.3 1
}
```

borine, 2020-10-13:
@Arkq - I think a possible issue with your `pcm.jabra-multi` definition is that it requires the source stream to have _at least_ two channels (more than two means channel 2 and above will be discarded). So it will fail with `Channels count not available` error message if you try to play a mono source stream. Simply changing the type from `route` to `plug`, leaving everything else the same, should fix that.

@carlyd95 - it is possible you are having issues with the parameters for the Loopback device. I would recommend to explicity set the required params for arecord, eg:
`arecord -c2 -r 48000 -f s16_le -D loopout test.wav`

gribouk, 2021-02-22:
> @Arkq - I think a possible issue with your `pcm.jabra-multi` definition is that it requires the source stream to have _at least_ two channels (more than two means channel 2 and above will be [...]

I am trying to reproduce working configuration, but fall into problem that I get:
" aplay: pcm_write:2053: write error: Invalid argument" if use plug, and
"aplay: set_params:1345: Channels count non available" if use route... Below is my config:
```
pcm.output {
  type hw
  card 0
}

pcm.bt_speaker {
        type plug
        slave.pcm {
                type bluealsa
                device "88:C6:26:AF:03:A2"
                profile "a2dp"
        }
}

ctl.bt {
type bluealsa
}

pcm.monitored {
    type route;
    slave.pcm {
        type multi;
        slaves.a.pcm "output";
        slaves.a.channels 2;
        slaves.b.pcm "bt_speaker"
        slaves.b.channels 2;
        bindings.0 { slave a; channel 0; }
        bindings.1 { slave a; channel 1; }
        bindings.2 { slave b; channel 0; }
        bindings.3 { slave b; channel 1; }
    }
    ttable.0.0 1
    ttable.1.1 1
    ttable.0.2 1
    ttable.1.3 1
}
```
Do you have any idea why? Thanks!

borine, 2021-02-22:
Try making the bluetooth device the "master slave" of the multi (ugh - horrible confusing terminology - blame the alsa project!).

```
pcm.monitored {
    type route;
    slave.pcm {
        type multi;
        slaves.a.pcm "output";
        slaves.a.channels 2;
        slaves.b.pcm "bt_speaker"
        slaves.b.channels 2;
        bindings.0 { slave a; channel 0; }
        bindings.1 { slave a; channel 1; }
        bindings.2 { slave b; channel 0; }
        bindings.3 { slave b; channel 1; }
        master 1
    }
    ttable.0.0 1
    ttable.1.1 1
    ttable.0.2 1
    ttable.1.3 1
}
```

In this case "master" determines which "slave" supplies the clock for the multi as a whole. A bluealsa device cannot be driven by an external clock, so when used in combination with a hardware device we have to allow bluealsa to drive the interrupts for the multi. Alternatively, make bluealsa "slave a" and the hw "slave b" as the default "master slave" is the first slave.

gribouk, 2021-02-22:
Thanks for your reply!
Looks likw I am on the right path but quite there yet, I tried both variants and got following error:
pi@raspberrypi:/etc $ aplay -D monitored test.wav
Playing WAVE 'test.wav' : Signed 16 bit Little Endian, Rate 44100 Hz, Stereo
aplay: xrun:1664: read/write error, state = PREPARED

what could it be to deal with it? Thanks!

graham8, 2021-02-22:
@borine Would your comments about clocks and interrupts explain why trying to send audio to two or three BT devices with a multi fails?

gribouk, 2021-02-22:
UPD: To provide more info I post the output with -v flag - maybe this would give some info for the solution:
```
pi@raspberrypi:/etc $ aplay -v -D monitored test.wav
Playing WAVE 'test.wav' : Signed 16 bit Little Endian, Rate 44100 Hz, Stereo
Plug PCM: Route conversion PCM (sformat=S16_LE)
  Transformation table:
    0 <- 0
    1 <- 1
    2 <- 0
    3 <- 1
Its setup is:
  stream       : PLAYBACK
  access       : RW_INTERLEAVED
  format       : S16_LE
  subformat    : STD
  channels     : 2
  rate         : 44100
  exact rate   : 44100 (44100/1)
  msbits       : 16
  buffer_size  : 22558
  period_size  : 3760
  period_time  : 85260
  tstamp_mode  : NONE
  tstamp_type  : GETTIMEOFDAY
  period_step  : 1
  avail_min    : 3760
  period_event : 0
  start_threshold  : 22558
  stop_threshold   : 22558
  silence_threshold: 0
  silence_size : 0
  boundary     : 1478361088
Slave: Multi PCM
  Channel bindings:
    0: slave 0, channel 0
    1: slave 0, channel 1
    2: slave 1, channel 0
    3: slave 1, channel 1
Its setup is:
  stream       : PLAYBACK
  access       : MMAP_COMPLEX
  format       : S16_LE
  subformat    : STD
  channels     : 4
  rate         : 44100
  exact rate   : 44100 (44100/1)
  msbits       : 16
  buffer_size  : 22558
  period_size  : 3760
  period_time  : 85260
  tstamp_mode  : NONE
  tstamp_type  : GETTIMEOFDAY
  period_step  : 1
  avail_min    : 3760
  period_event : 0
  start_threshold  : 22558
  stop_threshold   : 22558
  silence_threshold: 0
  silence_size : 0
  boundary     : 1478361088
Slave #0: Plug PCM: Rate conversion PCM (48000, sformat=S16_LE)
Converter: libspeex (builtin)
Protocol version: 10002
Its setup is:
  stream       : PLAYBACK
  access       : MMAP_INTERLEAVED
  format       : S16_LE
  subformat    : STD
  channels     : 2
  rate         : 44100
  exact rate   : 44100 (44100/1)
  msbits       : 16
  buffer_size  : 22558
  period_size  : 3760
  period_time  : 85260
  tstamp_mode  : NONE
  tstamp_type  : GETTIMEOFDAY
  period_step  : 1
  avail_min    : 3760
  period_event : 0
  start_threshold  : 22558
  stop_threshold   : 22558
  silence_threshold: 0
  silence_size : 0
  boundary     : 1478361088
Slave: BlueALSA PCM: /org/bluealsa/hci0/dev_88_C6_26_AF_03_A2/a2dp
BlueALSA BlueZ device: /org/bluez/hci0/dev_88_C6_26_AF_03_A2
BlueALSA Bluetooth codec: 0
Slave #1: Hardware PCM card 0 'bcm2835 Headphones' device 0 subdevice 0
Its setup is:
  stream       : PLAYBACK
  access       : MMAP_INTERLEAVED
  format       : S16_LE
  subformat    : STD
  channels     : 2
  rate         : 44100
  exact rate   : 44100 (44100/1)
  msbits       : 16
  buffer_size  : 22558
  period_size  : 3760
  period_time  : 85260
  tstamp_mode  : NONE
  tstamp_type  : GETTIMEOFDAY
  period_step  : 1
  avail_min    : 3760
  period_event : 0
  start_threshold  : 22558
  stop_threshold   : 22558
  silence_threshold: 0
  silence_size : 0
  boundary     : 1478361088
  appl_ptr     : 0
  hw_ptr       : 0
Status(R/W):
  state       : PREPARED
  trigger_time: 0.000000
  tstamp      : 0.000000
  delay       : 0
  avail       : 18798
  avail_max   : 22562
aplay: xrun:1664: read/write error, state = PREPARED

```

borine, 2021-02-22:
@gribouk - what version of bluealsa are you using (`bluealsa --version`) and version of libasound (`aplay --version`)

@graham8 - "clocks" was probably an over simplification - its to do with snd_pcm_wait() and when poll() reports that the file descriptor is ready for I/O. I haven't tried multiple bluetooth outputs, but @Arkq 's comment above suggests that it can be done.

gribouk, 2021-02-22:
pi@raspberrypi:/etc $ bluealsa --version
1.4.0

pi@raspberrypi:/etc $ aplay --version
aplay: version 1.1.8 by Jaroslav Kysela <perex@perex.cz>

borine, 2021-02-22:
@gribouk bluealsa 1.4.0 is now very old, and much work has been done on improving the alsa plugin, especially I/O readiness notification which is important for the multi plugin. I would suggest you update your bluealsa installation to v3.0.0 or the latest "master" branch, that may work better for you.

gribouk, 2021-02-22:
Now I am really stuck - when I try to update it I get this:
root@raspberrypi:~# sudo apt install bluealsa
Reading package lists... Done
Building dependency tree
Reading state information... Done
bluealsa is already the newest version (0.13).
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.

Why does it say it is the latest version of bluealsa? How to update it to v3?

graham8, 2021-02-22:
@borine FWIW we had a user trying to connect three BT headphones at once.  I did [try two BT devices](https://discourse.osmc.tv/t/no-pulseaudio-in-kodi-alsa-sink-in-mono-only-after-buster-upgrade/87262/25?u=grahamh) but without much joy.  In the end, it was no use because if one device disconnected neither worked.  The user [resorted to Pulse](https://discourse.osmc.tv/t/howto-connect-3-bluetooth-headsets-to-osmc-rpi-mediacenter/87780?u=grahamh) and some heroic d-bus configs.

Arkq, 2021-02-22:
@graham8 from time to time I'm using two BT headsets connected at once (streaming to booth audio from e.g. a movie) and it works quite well (my ALSA setup is somewhere above in comments). Booth headsets are of the same brand (Jabra MOVE) but different revisions. Peculiar thing is that the newer one - Jabra MOVE v 2.5.0 - does not support AAC, only SBC is available... maybe dues to licensing issue, dunno :D. So, one headset is using AAC and the other SBC and everything works great. I think that if I've got three headsets with AAC support it would be possible to stream to three at the same time, because with current setup I'm at the limit of Bluetooth bandwidth - AAC VBR can significantly reduce bandwidth.

>  In the end, it was no use because if one device disconnected neither worked.

Yes, that's the issue one has to be aware of, unfortunately.

borine, 2021-02-22:
> if one device disconnected neither worked

That is to be expected - the multi plugin requires all devices or it fails.

gribouk, 2021-02-22:
I tried to build it from: https://github.com/Arkq/bluez-alsa/wiki/Installation-from-source
It did update to v3, but everything stopped working
I no longer can access to BT devices, and I cannot readd these...
Is there a repository for the v3 Installation?

borine, 2021-02-22:
> Why does it say it is the latest version of bluealsa

That is the raspberry pi repository - it has a very old bluealsa, with its own unhelpful version identifier.  I fear they do not have a maintainer for it any more so best advice has to be just don't install bluealsa from the RPi repository.

> Is there a repository for the v3 Installation?

Not for Raspbian or Raspios, you will have to build from source

> I no longer can access to BT devices, and I cannot readd these...

The latest version is not compatible with older versions, so you have to uninstall the old version first. 

I think you may have more work here than you realized. Raspberry Pi recently introduced Pulseaudio into their desktop and full SD images, so if you are using either of those images you may just have to accept the use of pulseaudio and give up on bluealsa. Only their Lite image is now suitable for use with bluealsa. So it depends on how you intend to use this Pi.

If the Lite image is good enough for you (no desktop, intended for embedded and headless server use only) then I may be able to help you get bluealsa up and running. However if you need a typical RPi user experience then I'm afraid bluealsa will be just too hard to get working without breaking so many other things, so just use pulseaudio which is already installed in the image. Its a shame, but that's how I see it.

gribouk, 2021-02-22:
Thanks for the info! This is veru usefull for me to understand the situation.
I use RPi for one purpose only - play sound via Snapcast. I plan to do that via bluetoth speakers. So I do not need the desktop and I am not using one. And I use Lite version now.

What I need to do is to:
1. Join several speakers into a single unit - the task I am stuck with now.
2. Broudcast sound from multiple sources to this unit.

I have already tried pulse audio. I used Full PiOs version for that. The problem with pulseaudio was that I could not make it working with Snapcast. It does not allow any uses exept pi to connect to pulse server. The recepee to set the default server = 127.0.0.1 did not work for me - just nothing changed, so I tried to launch it in a system mode and the sound quality dropped - interference and noise. So I decided that pulse audio is not an option after several days of sleepless nights... One person from their forum advised me to go back to ALSA which works well with Snapcast, but use it with demix plugin - as he said this should solve my problem. I guess I have no choise but to make the ALSA working finally.

I am not affraid of a hard work, so I would go with bluealsa. But what exactly would that break? As I said this miniserver will run:
1. Docker
2. Rhasspy in a docker as a satelite.
3. HomeAssistant in a docker 
4. Snapclients (3 instances)
5. 2-3 BT and wire speackers 
6. Mic to pass to rhasspy the commands.
7. USB Zigbee stick

No other programs or devices will be disturbing it. Is it ok to work with bluealsa?

borine, 2021-02-22:
I'm not familiar with any of the components you list above, so can only give some general guidance.

The first is to be aware of the scope and limits of bluealsa, and of ALSA devices in general. In particular, only one application can access a PCM device at a time. For hardware cards only, ALSA provides a plugin called "dmix" that permits multiple connections, but that specifically targets the hardware API and will reject any attempt to use a plugin such as bluealsa. As far as I know, there is no equivalent plugin available for bluealsa. So for example, if your 3 snapclients (I don't know what they are, so this is hypothetical) all need to connect to the same bluetooth speaker at the same time then you are out of luck.

The second is that ALSA configuration is mostly static. By which I mean, you cannot redefine any configuration item while it is in use. So for example see comment  https://github.com/Arkq/bluez-alsa/issues/379#issuecomment-783300147 above - the "slaves" of a multi must all exist at the time it is used, and if any one "disappears" then the whole multi fails. That is not normally an issue with soundcards (unless you unplug a USB card while in use), but is very common for bluetooth when devices such as headphones may go out of range etc.

It is precisely this need for a more "dynamic" audio system that led to the development of sound servers such as esound, pulseaudio or pipewire which sit between the applications and the audio (ALSA) devices.

That said, if a "raw" ALSA interface is good enough for your applications, then bluealsa is an excellent bluetooth solution. It will not break anything, when I said that I was referring to the removal of pulseaudio which would break everything that depends on it. As long as you have no pulseaudio on your system that is fine.

Assuming you are starting from the latest RaspiOS Lite image, then I would expect that simply following the guidance from the wiki https://github.com/Arkq/bluez-alsa/wiki/Installation-from-source and https://github.com/Arkq/bluez-alsa/wiki/Systemd-integration should get you a working installation - that is the way that I do it.

gribouk, 2021-02-23:
Thank you for a very cpmprehansive explanation!

 So, to put it in a simpler form, suppose I want to play 3 wav files via aplay with bluetooth speaker - I will not be able to do that using bluealsa and dmix would not help me with that?

Actually I was expecting the chain to be source ->dmix->alsa->speaker, where dmix merges a sound sources and alsa passes it the the hardware, so I was expecting to have a chain like that with bluetooth: source->dmix->alsa->bluealsa->speaker where bluealsa is in charge for passing the data to the physical device. Thus, it is a bit of a surprice to me...

borine, 2021-02-23:
> Thus, it is a bit of a surprice to me.

Yes, ALSA do not document things very clearly, you have to look at their source code to see that the "slave" device for dmix has to be type "hw" - any other type just returns an error.

cool-delete, 2021-05-24:
> That is the raspberry pi repository - it has a very old bluealsa, with its own unhelpful version identifier. I fear they do not have a maintainer for it any more so best advice has to be just don't [...]

你好 我的版本也是1.4.0 但是我的需求很简单 
`aplay  1.wav`
There is sound from the speakers!

But aplay can only be used in wav format, I hope any application can play music

When I use` mpg123 -C --list list.lst `
to play an mp3

Will get an error

```
[src/libout123/modules/alsa.c:129] error: initialize_device(): cannot set hw params

main: [src/mpg123.c:309] error: out123 error 7: failed to open device
```

borine, 2021-06-22:
> The original version 1.4.0 is so old. How to upgrade the version quickly

To get the latest version of bluez-alsa you will have to build from source. See the wiki guide here: https://github.com/Arkq/bluez-alsa/wiki/Installation-from-source

closed 2021-07-01 by borine