arkq/bluez-alsa #380: help with asound config to play sound on multiple cards

gavsiu, 2020-09-24:
I've tried stitching multiple guides together to get this to work and am still stuck. This asound config thing is confusing to me.

My laptop has internal speakers that I generally don't use because I have a set of 2 channel external speakers plugged into the headphone jack. I want to be able to connect my bluetooth headphones at night. I would prefer it to automatically mute my "headphones"/internal speakers when the bluetooth headphones are connected, but I'm guessing that's not possible. Therefore, I'm trying to output audio to both speakers and bluetooth at the same time which, as I understand, is possible and turn off the speakers manually.

I've managed to connect to my headphones and play an audio file using 

    aplay -D bluealsa:SRV=org.bluealsa,DEV=98:52:3D:7E:E5:55,PROFILE=a2dp /usr/share/sounds/alsa/Front_Center.wav

by adding a plug for the headset.

I think the bluetooth headset is supposed to have it's own dmix and I am to combine them by routing or something. I tested using

    aplay /usr/share/sounds/alsa/Front_Center.wav

but it threw an error:

    ALSA lib pcm_direct.c:1825:(_snd_pcm_direct_get_slave_ipc_offset) Invalid type 'bluetooth' for slave PCM

    aplay: main:830: audio open error: Invalid argument

Here is my asound.conf

```
# ALSA
# /etc/asound.conf

# ipc_key  	  - must be unique, it has no other meaning.
# ipc_key_add_uid - allows multi user audio when false.
# ipc_perm 	  - sets permissions, may fix issues on multi user or tty playback.

# smaller buffer/period sizes can generally reduce latency and hence give better performance,
# but going too small can cause crackling or other sound issues.

# Better resampling
defaults.pcm.rate_converter "samplerate_best"

# 44100Hz
# Lossless CD quality

# Sound card
pcm.snd_card {
    type hw
    card PCH
}

# Audio out
pcm.aout {
    type dmix
    ipc_key 1
    ipc_key_add_uid false
    ipc_perm 0666
    slave {
        pcm "snd_card"
        rate 44100
        channels 2

        # Buffer settings
        period_time 0
        period_size 1024
        buffer_time 0
        buffer_size 8192
    }
    bindings {
        0 0
        1 1
    }
}

# Audio in
pcm.ain {
    type dsnoop
    ipc_key 2
    ipc_key_add_uid false
    ipc_perm 0660
    slave.pcm "snd_card"
}

# Set playback routes
pcm.asymed {
    type asym
    playback.pcm "aout"
    capture.pcm  "ain"  # Input won't work with aout settings ;)
}

# Surround 5.1 -> Stereo
pcm.!surround51 {
    type vdownmix
    slave.pcm "aout"
}

# Surround 4.0 -> Stereo
pcm.!surround40 {
    type vdownmix
    slave.pcm "aout"
}

# OSS compatibility
pcm.dsp0 {
    type plug
    slave.pcm "asymed"
}

# Bluetooth headset
pcm.btheadset {
    type plug
    slave {
        pcm {
            type bluetooth
            device 98:52:3D:7E:E5:55
            profile "auto"
        }
    }
    hint {
        show on
        description "Soundcore Liberty 2 Pro"
    }
}

# Bluetooth headset audio out
pcm.btheadsetaout {
    type dmix
    ipc_key 2
    ipc_key_add_uid false
    ipc_perm 0666
    slave {
        pcm "btheadset"
        rate 44100
        channels 2

        # Buffer settings
        period_time 0
        period_size 1024
        buffer_time 0
        buffer_size 8192
    }
    bindings {
        0 0
        1 1
    }
}

# Bluetooth headset control
ctl.btheadset {
    type bluetooth
}

# Multiple outputs on multiple cards
pcm.multi {
    type route;
    slave.pcm {
        type multi;
        slaves.a.pcm "btheadsetaout";
        slaves.b.pcm "asymed";
        slaves.a.channels 2;
        slaves.b.channels 2;
        bindings.0.slave a;
        bindings.0.channel 0;
        bindings.1.slave a;
        bindings.1.channel 1;

        bindings.2.slave b;
        bindings.2.channel 0;
        bindings.3.slave b;
        bindings.3.channel 1;
    }

    # btheadsetaout
    ttable.0.0 1;
    ttable.1.1 1;

    # asymed
    ttable.0.2 1;
    ttable.1.3 1;
}

# Default PCM
pcm.!default {
    type plug
    slave.pcm "asymed"
}

# Default control
ctl.!default {
    type hw
    card PCH
}
```

graham8, 2020-09-24:
I'm confused why you would keep the internal sound card activated and then turn off the speakers.  Can't you just choose to output audio to BT only in your audio settings?

gavsiu, 2020-09-24:
I will only be using the BT headphones sometimes. I'll be using the internal sound card most of the time. The audio input is supposed to be for the built in mic on my laptop, although I've never really tested it. Unless I misunderstood what you meant by capture device.

graham8, 2020-09-24:
My bad, I thought you were setting up some sort of loop-back.

FWIW, this is how I've been playing audio to two devices. I'm having to apply a delay to the default output so using an upmix plugin.  With that, you don't need any of the ttable stuff.  The ctl element is taken care of by the release 20-bluealsa.conf file.

```
pcm.jblflip {
        type bluealsa
        device "FC:A8:9A:B3:42:4F"
        profile "a2dp"
        hint { show on description "JBL Flip 3"}
}

pcm.both {
        type upmix
        slave "pairup"
        channels 4
        delay 170
        hint {
                show {
                        @func refer
                        name defaults.namehint.extended
                }
                description "BT & default"
        }
}

pcm_slave.pairup.pcm {
        type plug
        slave.pcm {
                type multi
                        slaves {
                                a { pcm "jblflip" channels 2 }
                                b { pcm "default" channels 2 }
                        }
                bindings [
                { slave a channel 0 }
                { slave a channel 1 }
                { slave b channel 0 }
                { slave b channel 1 }
                ]
        }
}
```

graham8, 2020-09-24:
Just noticed, it's simpler than that!  Just change `type bluetooth` to `type bluealsa`.

gavsiu, 2020-09-24:
I'll have to try that tomorrow, thanks. I was following [this](https://askubuntu.com/questions/713145/how-do-i-connect-bluez-alsa-to-an-audio-device) for that section.

graham8, 2020-09-24:
That's waaay out of date.

gavsiu, 2020-09-24:
So... I'm still getting an error.

```
ALSA lib pcm_direct.c:1825:(_snd_pcm_direct_get_slave_ipc_offset) Invalid type 'bluealsa' for slave PCM

aplay: main:830: audio open error: Invalid argument
```

Is there some missing requirement I didn't install other than `bluez-alsa`?

I also changed out the `type route` to `type plug` like yours when I followed [this](https://unix.stackexchange.com/questions/194547/how-can-i-use-alsa-dmix-and-multi-plugins-together) after posting.

graham8, 2020-09-24:
I confess the conf I posted above was patched together from a device where I'm actually using a USB device and 'default' (HDMI) in parallel and I didn't have time to test exactly that with a BT device.  I'll have a go this evening to check it properly.

I'm not using dmix as I never have more than one app with audio output.  There are a few threads in here about problems with dmix.

borine, 2020-09-24:
dmix only works with "hw" devices (ie sound cards attached to the computer using a kernel driver). That is a limitation of dmix itself. So it cannot be used with bluealsa. There is nothing that bluealsa, or any other software plugin, can do about that.

gavsiu, 2020-09-25:
Alright, I pretty much changed my config to this:

```
# ALSA
# /etc/asound.conf

# ipc_key  	  - must be unique, it has no other meaning.
# ipc_key_add_uid - allows multi user audio when false.
# ipc_perm 	  - sets permissions, may fix issues on multi user or tty playback.

# smaller buffer/period sizes can generally reduce latency and hence give better performance,
# but going too small can cause crackling or other sound issues.

# Better resampling
defaults.pcm.rate_converter "samplerate_best"

# 44100Hz
# Lossless CD quality

# Sound card
pcm.snd_card {
    type hw
    card PCH
}

# Audio out
pcm.aout {
    type dmix
    ipc_key 1
    ipc_key_add_uid false
    ipc_perm 0666
    slave {
        pcm "snd_card"
        rate 44100
        channels 2

        # Buffer settings
        period_time 0
        period_size 1024
        buffer_time 0
        buffer_size 8192
    }
    bindings {
        0 0
        1 1
    }
}

# Audio in
pcm.ain {
    type dsnoop
    ipc_key 2
    ipc_key_add_uid false
    ipc_perm 0660
    slave.pcm "snd_card"
}

# Set playback routes
pcm.asymed {
    type asym
    playback.pcm "aout"
    capture.pcm  "ain"  # Input won't work with aout settings ;)
}

# Surround 5.1 -> Stereo
pcm.!surround51 {
    type vdownmix
    slave.pcm "aout"
}

# Surround 4.0 -> Stereo
pcm.!surround40 {
    type vdownmix
    slave.pcm "aout"
}

# OSS compatibility
pcm.dsp0 {
    type plug
    slave.pcm "asymed"
}

# Bluetooth headset
pcm.btheadset {
    type plug
    slave {
        pcm {
            type bluealsa
            device "98:52:3D:7E:E5:55"
            profile "a2dp"
        }
    }
    hint {
        show on
        description "Soundcore Liberty 2 Pro"
    }
}

# Bluetooth headset control
ctl.btheadset {
    type bluetooth
}

# Multiple outputs on multiple cards
pcm.multi {
    type plug
    slave.pcm {
        type multi
        slaves {
            a {
                channels 2
                pcm "btheadset"
            }
            b {
                channels 2
                pcm "asymed"
            }
        }
        bindings {
            0 {
                slave a
                channel 0
            }
            1 {
                slave a
                channel 1
            }
            2 {
                slave b
                channel 0
            }
            3 {
                slave b
                channel 1
            }
        }
    }
    ttable [
        [ 1 0 1 0 ] # left  -> a.left,  b.left
        [ 0 1 0 1 ] # right -> a.right, b.right
    ]
}

# Default PCM
pcm.!default {
    type plug
    slave.pcm "multi"
}

# Default control
ctl.!default {
    type hw
    card PCH
}
```

And `aplay` gives me this:

```
aplay /usr/share/sounds/alsa/Front_Center.wav

Playing WAVE '/usr/share/sounds/alsa/Front_Center.wav' : Signed 16 bit Little Endian, Rate 48000 Hz, Mono
aplay: set_params:1407: Unable to install hw params:
ACCESS:  RW_INTERLEAVED
FORMAT:  S16_LE
SUBFORMAT:  STD
SAMPLE_BITS: 16
FRAME_BITS: 16
CHANNELS: 1
RATE: 48000
PERIOD_TIME: (23219 23220)
PERIOD_SIZE: (1114 1115)
PERIOD_BYTES: (2228 2230)
PERIODS: (7 9)
BUFFER_TIME: 185750
BUFFER_SIZE: 8916
BUFFER_BYTES: 17832
TICK_TIME: 0
```

But I don't hear anything with my bluetooth headphones. How do I go from here? I'm sorry for failing to understand what's going on.

Using like your conf

```
pcm.btheadset {
    type bluealsa
    device "98:52:3D:7E:E5:55"
    profile "a2dp"
    hint {
        show on
        description "Soundcore Liberty 2 Pro"
    }
}
```

Gives me

```
aplay /usr/share/sounds/alsa/Front_Center.wav

Playing WAVE '/usr/share/sounds/alsa/Front_Center.wav' : Signed 16 bit Little Endian, Rate 48000 Hz, Mono
ALSA lib pcm_params.c:2226:(snd1_pcm_hw_refine_slave) Slave PCM not usable
ALSA lib pcm_params.c:2226:(snd1_pcm_hw_refine_slave) Slave PCM not usable
aplay: set_params:1314: Broken configuration for this PCM: no configurations available
```

borine, 2020-09-25:
try:
```
pcm.asymed {
    type asym
    playback.pcm "plug:aout"
    capture.pcm  "plug:ain"  # Input won't work with aout settings ;)
}
```
also, should be:
```
ctl.btheadset {
    type bluetooth
}
```
Be aware that this config will not work when the bt headset is not connected, so its probably not the solution you are looking for. You need some "active" management component that detects the connection of the headset and switches the default on that event, or even better redirects running streams on that event. That is the sort of feature that pulseaudio is designed for. If you refuse to use pulseaudio, then you need an alternative audio manager; ALSA alone does not provide that.

gavsiu, 2020-09-25:
Still doesn't work. I give up. I was hoping it was possible, but I guess that's why there aren't any guides to get this to work. I'll have to finally install Pulseaudio when I really want to use my bluetooth headset. Thanks for all your help.

closed 2020-09-25 by gavsiu