arkq/bluez-alsa #382: Issues starting service with runit on void linux

runrin, 2020-09-30:
I'm running Void Linux, on kernel version 5.8.12_1, and I cannot seem to get the `bluez-alsa` (v. 2.1.0_1) service to start with `runit` (v. 2.1.2_11).

The `bluetoothd`, `d-bus`, and `alsa` services are all running, and working correctly.

If I do `sudo sv start bluez-alsa` I just get `timeout: down: bluez-alsa: 0s, normally up, want up`.  

If I do `sudo bluealsa` I get `bluealsa: E: Couldn't acquire D-Bus name: org.bluealsa`.  

`sudo ps -e | grep -i blue` shows only `bluetoothd` running. `sudo killall bluealsa` returns `bluealsa: no process found`. So we know the issue isn't that it's already running.

Using `sudo bluealsa --syslog` does not seem to provide any additional information, nor could I find anything else related in my syslogs.

Perhaps this is a d-bus policy issue, but from what I can tell, if I run `bluealsa` as root, it should not be an issue.

Let me know if I can provide any additional information. :)

amak79, 2020-09-30:
Try the following as root:
`# chpst -u _bluez_alsa:audio bluealsa`
This is how the runit service starts bluez-alsa. It could give you information on why the service isn't running.

runrin, 2020-09-30:
```
bluealsa: W: Couldn't get HCI version: Network is down
bluealsa: D: bluez.c:764: Creating media endpoint object: /org/bluez/hci0/A2DP/MPEG24/Source/1
bluealsa: D: bluez.c:793: Registering media endpoint: /org/bluez/hci0/A2DP/MPEG24/Source/1
bluealsa: D: bluez.c:764: Creating media endpoint object: /org/bluez/hci0/A2DP/MPEG24/Source/2
bluealsa: D: bluez.c:793: Registering media endpoint: /org/bluez/hci0/A2DP/MPEG24/Source/2
bluealsa: D: bluez.c:764: Creating media endpoint object: /org/bluez/hci0/A2DP/SBC/Source/1
bluealsa: D: bluez.c:793: Registering media endpoint: /org/bluez/hci0/A2DP/SBC/Source/1
bluealsa: D: bluez.c:764: Creating media endpoint object: /org/bluez/hci0/A2DP/SBC/Source/2
bluealsa: D: bluez.c:793: Registering media endpoint: /org/bluez/hci0/A2DP/SBC/Source/2
bluealsa: D: bluez.c:1078: Creating hands-free profile object: /org/bluez/HSP/AudioGateway
bluealsa: D: bluez.c:1106: Registering hands-free profile: /org/bluez/HSP/AudioGateway
bluealsa: D: bluez.c:1078: Creating hands-free profile object: /org/bluez/HFP/AudioGateway
bluealsa: D: bluez.c:1106: Registering hands-free profile: /org/bluez/HFP/AudioGateway
bluealsa: D: main.c:378: Acquiring D-Bus service name: org.bluealsa
bluealsa: D: main.c:383: Starting main dispatching loop

```

runrin, 2020-09-30:
it actually starts this way, and i can see it with `ps -e`, but i still cannot start the service

amak79, 2020-09-30:
Enable [logging](https://docs.voidlinux.org/config/services/logging.html) if you haven't already done so. Then add the following to `/etc/sv/bluez-alsa/conf`:
`OPTS="--syslog"`
Restart the bluez-alsa service. Hopefully the syslog will show why the service isn't running.

runrin, 2020-09-30:
this appears to be the only relevant info
```
2020-09-30T07:58:21.27739 daemon.info: Sep 30 00:58:21 bluetoothd[1666]: Endpoint unregistered: sender=:1.38 path=/org/bluez/hci0/A2DP/MPEG24/Source/1
2020-09-30T07:58:21.27748 daemon.info: Sep 30 00:58:21 bluetoothd[1666]: Endpoint unregistered: sender=:1.38 path=/org/bluez/hci0/A2DP/MPEG24/Source/2
2020-09-30T07:58:21.27751 daemon.info: Sep 30 00:58:21 bluetoothd[1666]: Endpoint unregistered: sender=:1.38 path=/org/bluez/hci0/A2DP/SBC/Source/1
2020-09-30T07:58:21.27755 daemon.info: Sep 30 00:58:21 bluetoothd[1666]: Endpoint unregistered: sender=:1.38 path=/org/bluez/hci0/A2DP/SBC/Source/2
2020-09-30T07:58:23.38112 daemon.info: Sep 30 00:58:23 bluetoothd[1666]: Endpoint registered: sender=:1.42 path=/org/bluez/hci0/A2DP/MPEG24/Source/1
2020-09-30T07:58:23.38177 daemon.info: Sep 30 00:58:23 bluetoothd[1666]: Endpoint registered: sender=:1.42 path=/org/bluez/hci0/A2DP/MPEG24/Source/2
2020-09-30T07:58:23.38281 daemon.info: Sep 30 00:58:23 bluetoothd[1666]: Endpoint registered: sender=:1.42 path=/org/bluez/hci0/A2DP/SBC/Source/1
2020-09-30T07:58:23.38389 daemon.info: Sep 30 00:58:23 bluetoothd[1666]: Endpoint registered: sender=:1.42 path=/org/bluez/hci0/A2DP/SBC/Source/2

```

amak79, 2020-09-30:
You should have `bluealsa` entries in the log. Can you post `/etc/sv/bluez-alsa/conf`, and did you modify the bluez-alsa service file?

runrin, 2020-09-30:
`/etc/sv/bluez-alsa/conf` did not exist until I added the single `OPTS="--syslog"` line to it. that is all that it contains.

amak79, 2020-09-30:
Try force re-installing the bluez-alsa package and try again.

For my purposes I have `OPTS="--syslog --profile=a2dp-sink"` in `/etc/sv/bluez-alsa/conf`. The `bluetoothd` and `dbus` services are enabled and running. If I stop the `dbus` service and start the `bluez-alsa` service, the `bluez-alsa` service will still be marked as `up`, although bluealsa will log errors. So I can't reproduce your issue.

runrin, 2020-09-30:
Huh, I had to manually remove `/etc/sv/bluez-alsa` because it wouldn't write over it even with `xbps-install --force`, but that seemed to fix things. I will have to do some testing to be sure, but after rebooting the `bluez-alsa` service is running, and I can see `bluealsa` with `ps -e`

amak79, 2020-09-30:
Nice to know you have it working.

I've opened a PR to update the bluez-alsa package so version 3.0.0 should be available soon.

runrin, 2020-09-30:
Alright, I was able to do more testing. After connecting the device with `bluetoothctl` I tried the following:

`$ alsamixer -D bluealsa` shows only `This sound device does not have any controls.`

```
$ aplay -D bluealsa:DEV=4C:87:5D:A8:43:3A,PROFILE=a2dp Move.wav

ALSA lib bluealsa-pcm.c:755:(_snd_pcm_bluealsa_open) Couldn't get BlueALSA PCM: PCM not found
aplay: main:830: audio open error: No such device
```
`svlogtail` shows the following repeated a large number of times:
```
2020-09-30T17:51:43.34763 user.debug: Sep 30 10:51:43 bluealsa: bluez.c:1264: Signal: org.freedesktop.DBus.ObjectManager.InterfacesAdded()
```
and then this:
```
2020-09-30T17:55:17.59667 user.debug: Sep 30 10:55:17 bluealsa: bluealsa-dbus.c:155: Called: org.bluealsa.Manager1.GetPCMs()
```

amak79, 2020-10-01:
Use `amixer -D bluealsa` to show the controls. They will only be available after the device has been connected.

It appears that you're using bluez-alsa to send audio (a2dp-source). Unfortunately I don't have any Bluetooth speakers to test with, so I don't know what could be causing your issue. I've only used bluez-alsa as an a2dp-sink.

The only thing I can suggest is to ensure your user is in the `audio` group. The Void Linux bluez-alsa dbus policy (`/etc/dbus-1/system.d/bluealsa.conf`) restricts access to the `_bluez_alsa:audio` user/group.

runrin, 2020-10-23:
this is still not working for me. the device is correctly paired, bluealsa is running fine, i'm in the `audio` group, but `amixer -D bluealsa` shows nothing. if it do `sudo amixer -D bluealsa` it shows this:

```
ALSA lib bluealsa-ctl.c:901:(_snd_ctl_bluealsa_open) Couldn't get BlueALSA PCM list: Rejected send message, 1 matched rules; type="method_call", sender=":1.53" (uid=0 pid=321
91 comm="alsamixer -D bluealsa ") interface="org.bluealsa.Manager1" member="GetPCMs" error name="(unset)" requested_reply="0" destination="org.bluealsa" (uid=991 pid=1466 com
m="bluealsa --syslog --profile=a2dp-sink ")
cannot open mixer: No such device
```

any thoughts?

borine, 2020-10-23:
That error message is from D-Bus. Please can you post the contents of the file `/etc/dbus-1/system.d/bluealsa.conf`

runrin, 2020-10-23:
```
<!-- This configuration file specifies the required security policies
     for BlueALSA core daemon to work. -->

<!DOCTYPE busconfig PUBLIC "-//freedesktop//DTD D-BUS Bus Configuration 1.0//EN"
 "http://www.freedesktop.org/standards/dbus/1.0/busconfig.dtd">
<busconfig>

  <!-- ../system.conf have denied everything, so we just punch some holes -->

  <policy user="_bluez_alsa">
    <allow own_prefix="org.bluealsa"/>
    <allow send_destination="org.bluealsa"/>
  </policy>

  <policy group="audio">
    <allow send_destination="org.bluealsa"/>
  </policy>

</busconfig>

```

amak79, 2020-10-24:
Can you also run `groups` as your user and post the output.

borine, 2020-10-24:
assuming your account is in the audio group, it looks like either:

1. D-Bus has not reloaded its configuration files since bluealsa was installed, or
2. D-Bus has not been re-started since your account was added to the audio  group.

If that is correct, then a re-boot is the simplest fix.

runrin, 2020-10-24:
i have rebooted multiple times since installing bluealsa and dbus. i am also in the audio group.

Arkq, 2020-10-24:
> if it do `sudo amixer -D bluealsa` it shows this: [...] Rejected send message [...]

Please do not remove "root" user from the D-Bus configuration. Keep it like here: https://github.com/Arkq/bluez-alsa/blob/master/src/bluealsa.conf and add new policy for your "_bluez_alsa" user. With sudo your EUID is `root`, which is not allowed by D-Bus in your configuration.

> but `amixer -D bluealsa` shows nothing

Does `bluealsa-aplay -l` show anything? Also, please compile bluez-alsa with debug enabled (if not enabled right now) and look at `bluealsa` service logs whether your device is indeed registered within bluealsa service (during BT device connection, there should be some logs printed).

runrin, 2020-10-24:
i tried `bluealsa-aplay -l` earlier, and there was no output.i should have mentioned it in the last post.

i don't think its compiled with debug enabled in my package manager, so i'll try to compile it myself and report back. thanks.

borine, 2020-10-24:
The error message you see in comment https://github.com/Arkq/bluez-alsa/issues/382#issuecomment-715489751 is generated by D-Bus when a user that does not have permission tries to send a message to the service` org.bluealsa`.

Compare it with the result I get when running
`sudo -u games amixer -D bluealsa`

```
ALSA lib ../../../src/asound/bluealsa-ctl.c:927:(_snd_ctl_bluealsa_open) Couldn't get BlueALSA PCM list: Rejected send message, 1 matched rules; type="method_call", sender=":1.837" (uid=5 pid=30057 comm="amixer -D bluealsa " label="unconfined") interface="org.bluealsa.Manager1" member="GetPCMs" error name="(unset)" requested_reply="0" destination="org.bluealsa" (uid=0 pid=30040 comm="bluealsa " label="unconfined")
amixer: Mixer attach bluealsa error: No such device
```

As your `/etc/dbus-1/system.d/bluealsa.conf` clearly allows members of the group `audio` to communicate with `org.bluealsa` I still believe that D-Bus thinks that your user account is not a member of that group. You never posted the output of `groups` as requested by @amak79 (or alternatively the ouput of `id`)

runrin, 2020-10-24:
output of groups is this:
```runrin : runrin wheel tty floppy lp audio video cdrom optical kvm xbuilder libvirt bluetooth socklog```

borine, 2020-10-25:
ok, my bad, I only just noticed you used `sudo` when you got the "Rejected send message" error. As @Arkq said, that's because root does not (by default) have permission on Void Linux.

Going back to this comment https://github.com/Arkq/bluez-alsa/issues/382#issuecomment-701551560 , do you still see the same result? If the bluetooth device is connected, but amixer / alsamixer say `This sound device does not have any controls`, that suggests to me that the device is _not_ connected to bluealsa, but to some other service(s). With bluealsa running and the device connected, what is the output of
```
/bin/echo -e 'show\ninfo C4:67:B5:37:26:FC' | bluetoothctl
```

Arkq, 2020-10-26:
> i tried bluealsa-aplay -l earlier, and there was no output. i should have mentioned it in the last post.
> output of groups is this:
> `runrin : runrin wheel tty floppy lp audio video cdrom optical kvm xbuilder libvirt bluetooth socklog`

OK, so if there is no output from `bluealsa-aplay -l` (not even error message), that means that connection with bluealsa service works as expected from your normal user. Error message when using `sudo` is also a correct behavior with your setup. All this means that with 99% accuracy bluealsa service does not have any BT device registered. There are two most likely reasons for that:
1. no BT device is connected with your PC (please check with command mentioned by @borine: `bluetoothctl`)
2. some other service has intercepted BT connection (do you by any chance have `pulseaudio` running on your PC?)

runrin, 2020-11-25:
this issue is still not resolved for me.

running `info XX:XX:XX:XX` in `bluetoothctl` shows this (along with a bunch of details about vendor, UUID, etc):
```
...
Paired: yes
Trusted: yes
Blocked: no
Connected: yes
...
```
`bluealsa-aplay -l` shows this:
```
**** List of PLAYBACK Bluetooth Devices ****
**** List of CAPTURE Bluetooth Devices ****
```

`alsamixer -D bluealsa` shows this:
```
This sound device does not have any controls.
```

borine, 2020-11-25:
> running info XX:XX:XX:XX in bluetoothctl shows this (along with a bunch of details about vendor, UUID, etc):

Please can you run
```shell
/bin/echo -e 'show\ninfo 4C:87:5D:A8:43:3A' | bluetoothctl
```

as requested and post the full output. Delete the MAC addresses if you like, but the rest is useful data that might help understand which profiles / services are available and which are connected. Simply saying "its connected" is not useful without knowing what services are available on that connection.

amak79, 2020-11-26:
> Please can you run
> 
> ```shell
> /bin/echo -e 'show\ninfo 4C:87:5D:A8:43:3A' | bluetoothctl
> ```
Recent versions of BlueZ accept command line arguments, so now you can just run:
```
bluetoothctl info 4C:87:5D:A8:43:3A
```

runrin, 2020-11-26:
@borine i ran `/bin/echo -e 'show\ninfo 4C:87:5D:A8:43:3A' | bluetoothctl` when i ran `info` in `bluetoothctl` and it had the same output. sorry, i should have included the entire thing.

today, i am getting this message every time i try to `connect 4C:87:5D:A8:43:3A` in `bluetoothctl`:

```
Failed to connect: org.bluez.Error.Failed
```

yet, `info 4C:87:5D:A8:43:3A` in `bluetoothctl` claims to be connected, and gives this output:

```
Device 4C:87:5D:A8:43:3A (public)
        Name: LE-mydevice
        Alias: LE-mydevice
        Paired: yes
        Trusted: yes
        Blocked: no
        Connected: yes
        LegacyPairing: no
        UUID: Vendor specific           (00000000-deca-fade-deca-deafdecacaff)
        UUID: Serial Port               (00001101-0000-1000-8000-00805f9b34fb)
        UUID: Headset                   (00001108-0000-1000-8000-00805f9b34fb)
        UUID: Audio Sink                (0000110b-0000-1000-8000-00805f9b34fb)
        UUID: A/V Remote Control Target (0000110c-0000-1000-8000-00805f9b34fb)
        UUID: A/V Remote Control        (0000110e-0000-1000-8000-00805f9b34fb)
        UUID: Handsfree                 (0000111e-0000-1000-8000-00805f9b34fb)
        UUID: PnP Information           (00001200-0000-1000-8000-00805f9b34fb)
        UUID: Vendor specific           (81c2e72a-0591-443e-a1ff-05f988593351)
        UUID: Vendor specific           (85dbf2f9-73e3-43f5-a129-971b91c72f1e)
        UUID: Vendor specific           (931c7e8a-540f-4686-b798-e8df0a2ad9f7)
        UUID: Vendor specific           (9b26d8c0-a8ed-440b-95b0-c4714a518bcc)
        UUID: Vendor specific           (f8d1fbe4-7966-4334-8024-ff96c9330e15)
        Modalias: bluetooth:v009Ep4024d0182
        ManufacturerData Key: 0x0f03
        ManufacturerData Value:
  43 12 75 09 53 77 bf 19                          C.u.Sw..
```

`/bin/echo -e 'show\ninfo 4C:87:5D:A8:43:3A' | bluetoothctl` obviously gives the exact same output.

borine, 2020-11-26:
So, the service available on that connection is A/V Remote Control. You have no A2DP, HSP or HFP services because both the device and the controller are trying to be in the Audio Sink, Handsfree and Headset roles.

You need to run bluealsa in the Audio Source and Audio Gateway roles in order for your headset to connect. To do that try:
```shell
bluealsa -p a2dp-source -p hsp-ag
```
or
```shell
bluealsa -p a2dp-source -p hfp-ag
```

Read the home page of this project and consult the bluealsa manual page for more information.

runrin, 2020-11-26:
well, i was able to get it working by changing the profile settings in my conf.  

i can now get it to play audio using:  

```aplay -D bluealsa:DEV=XX:XX:XX:XX:XX:XX,PROFILE=a2dp Bourree_in_E_minor.wav```  

-----

it seems there isn't much documentation about getting the device integrated into alsamixer. i can change volume with `amixer -D bluealsa` but,

**how do i get all audio to use the bluealsa device without having to specify?  
how do i get the bluetooth device to show up in alsamixer by default?**

i've read the man page, and the project page on github, and there is no information about this, simply "look at the alsa docs."

i've added this to my ~/.asoundrc: 
```
defaults.bluealsa.service "org.bluealsa"
defaults.bluealsa.device "XX:XX:XX:XX:XX:XX"
defaults.bluealsa.profile "a2dp"
defaults.bluealsa.delay 10000
```
and the device seems to connect correctly, but it is not listed in alsamixer unless i use `-D bluealsa`, and clearly i cannot use that option with every application i use with sound.

runrin, 2020-11-27:
it pains me to say, i decided to switch to pulseaudio. thanks for all your help though.

closed 2020-11-27 by runrin

HerrBulles, 2022-12-28:
Those commands,  finnaly  ,made  my  bluetooth  headsets , to  become  usable , in void Linux  ,  hope  will help  someone  :

" sudo ln -s /etc/runit/sv/bluetoothd /run/runit/service
sudo sv start bluetooothd
sudo sv restart bluetoothd
"