arkq/bluez-alsa #383: Frame reassembly errors on Raspberry Pi Zero W

spietika, 2020-10-03:
I have a project that provides Yocto layer to build Bluetooth DAC from Raspberry Pi Zero W. The project uses bluez-alsa as the audio backend. Everything works for me when connecting from Android (OnePlus 7), but @kspeakman reported that using iPhone 6S the audio only plays for few seconds and then stops.

Originally bluez-alsa 1.4.0 was used, and it was then updated to 3.0.0 but that had no real impact on the issue. Moreover, the problem seems to be somehow related to AAC. Using an earlier version of the DAC project that did not have AAC enabled seems to work also on iPhone.

Here is the issue with attached logs from the device: https://github.com/spietika/meta-rpi-bt-dac/issues/6#issuecomment-702721626

I'm not familiar with bluez-alsa internals, so I was hoping someone here could give an insight what might be the problem.

borine, 2020-10-03:
As I understand it, "Frame reassembly error" is a low-level network protocol Rx error , and as such it is unreasonable to expect any higher level protocol to fix or compensate for it. Perhaps the RPi UART is unable to handle the required data rate. I really don't think there is anything that bluealsa, or bluez, can do about it. I think this is an issue for the Broadcom drivers or the RpI hardware.

spietika, 2020-10-03:
Thanks for the quick reply! 

I looked into the RPi BT UART and it seems that while some of the earlier RPi versions did not have flow control pins available, the Zero W does have them. So I enabled the flow control to see if that resolves the issue (waiting for comments on that).

Though, it would be interesting to understand why the Android is working and iPhone is not (both cases use the same HW for the sink). Presumably iPhone is pushing the audio with higher bitrate? Is this something that could be easily checked on the RPi from bluealsa when the music is playing?

spietika, 2020-10-04:
Adding the flow control to BT UART on the RPi solved the issue.

closed 2020-10-04 by spietika