arkq/bluez-alsa #384: Couldn't get BlueALSA PCM: PCM not found

ho246, 2020-10-03:
Hi All, sorry for bothering you with this, but it seems my new Sony WF-1000XM3 just won't talk to bluez-alsa.

For aplay I get
`aplay -v -D bluealsa:DEV=MY:MA:CA:DD:RE:SS,PROFILE=a2dp
bluealsa-pcm.c:895: Getting BlueALSA PCM: PLAYBACK 94:DB:56:19:48:14 a2dp
bluealsa-pcm.c:897:(_snd_pcm_bluealsa_open) Couldn't get BlueALSA PCM: PCM not found
aplay: main:830: audio open error: No such device`
It is of course correct, because GetPCMs() does indeed return empty when I check it in dbus-monitor

bluetoothctl says that the device is connected, paired and trusted. 
The controller shows the audio source and sink
'UUID: Audio Sink                (0000110b-0000-1000-8000-00805f9b34fb)
UUID: Audio Source              (0000110a-0000-1000-8000-00805f9b34fb)'

Yet, when I lock at the bluealsa output it ends with
'Acquiring D-Bus service name: org.bluealsa
Starting main dispatching loop'
but does not seem to be notified of the connection of the headphones. It does show the calls to GetPCMs, though.

The system is a gentoo linux:

- Linux version 5.8.11
- alsa-lib 1.2.3.2
- bluez 5.55
- dbus 1.12.18
- bluez-alsa built from live-git (but 3.0.0 does the same)
- no pulseaudio
- no systemd (openrc 0.42.1 instead)
- eudev 3.2.9

any hints would be greatly appreciated.
Thanks,
Harald

borine, 2020-10-03:
with the Sony WF-1000XM3 connected, please post the output of:
```sh
/bin/echo -e "show\ninfo MY:MA:CA:DD:RE:SS" | bluetoothctl
```

ho246, 2020-10-03:
Thank you so much for the quick reply, here it is.

```
Controller XX:XX:XX:XX:XX:XX (public)
        Name: BlueZ 5.55
        Alias: BlueZ 5.55
        Class: 0x001c010c
        Powered: yes
        Discoverable: yes
        DiscoverableTimeout: 0x00000000
        Pairable: yes
        UUID: Message Notification Se.. (00001133-0000-1000-8000-00805f9b34fb)
        UUID: A/V Remote Control        (0000110e-0000-1000-8000-00805f9b34fb)
        UUID: OBEX Object Push          (00001105-0000-1000-8000-00805f9b34fb)
        UUID: Message Access Server     (00001132-0000-1000-8000-00805f9b34fb)
        UUID: PnP Information           (00001200-0000-1000-8000-00805f9b34fb)
        UUID: IrMC Sync                 (00001104-0000-1000-8000-00805f9b34fb)
        UUID: Vendor specific           (00005005-0000-1000-8000-0002ee000001)
        UUID: A/V Remote Control Target (0000110c-0000-1000-8000-00805f9b34fb)
        UUID: Generic Attribute Profile (00001801-0000-1000-8000-00805f9b34fb)
        UUID: Phonebook Access Server   (0000112f-0000-1000-8000-00805f9b34fb)
        UUID: Device Information        (0000180a-0000-1000-8000-00805f9b34fb)
        UUID: Audio Sink                (0000110b-0000-1000-8000-00805f9b34fb)
        UUID: Generic Access Profile    (00001800-0000-1000-8000-00805f9b34fb)
        UUID: Audio Source              (0000110a-0000-1000-8000-00805f9b34fb)
        UUID: Vendor specific           (185f3df4-3268-4e3f-9fca-d4d5059915bd)
        UUID: OBEX File Transfer        (00001106-0000-1000-8000-00805f9b34fb)
        Modalias: usb:v1D6Bp0246d0537
        Discovering: no
        Roles: central
        Roles: peripheral
Advertising Features:
        ActiveInstances: 0x00 (0)
        SupportedInstances: 0x05 (5)
        SupportedIncludes: tx-power
        SupportedIncludes: appearance
        SupportedIncludes: local-name
[LE_WF-1000XM3]# info MY:MA:CA:DD:RE:SS
Device MY:MA:CA:DD:RE:SS (public)
        Name: LE_WF-1000XM3
        Alias: LE_WF-1000XM3
        Class: 0x00240404
        Icon: audio-card
        Paired: yes
        Trusted: yes
        Blocked: no
        Connected: yes
        LegacyPairing: no
        UUID: Unknown                   (00000709-0000-1000-8000-00805f9b34fb)
        UUID: Generic Access Profile    (00001800-0000-1000-8000-00805f9b34fb)
        UUID: Amazon.com Services, Inc. (0000fe03-0000-1000-8000-00805f9b34fb)
        UUID: Vendor specific           (55f80aef-d89f-41a4-9e36-0ffc88dc81ce)
        UUID: Vendor specific           (5b833e05-6bc7-4802-8e9a-723ceca4bd8f)
        UUID: Vendor specific           (5b833e06-6bc7-4802-8e9a-723ceca4bd8f)
        UUID: Vendor specific           (67a846ad-de3e-451b-a6d8-7b2899ca2370)
        UUID: Vendor specific           (91c10d9c-aaef-42bd-b6d6-8a648c19213d)
        UUID: Vendor specific           (dc405470-a351-4a59-97d8-2e2e3b207fbb)
        UUID: Vendor specific           (fe59bfa8-7fe3-4a05-9d94-99fadc69faff)
        Modalias: usb:v054Cp0CE0d0300
```

borine, 2020-10-04:
So the device `LE_WF-1000XM3` does not advertise support for A2DP, or HFP/HSP. That is why bluealsa does not respond when the device connects. I think the only service that is connected is the `Generic Access Profile`.

I suspect some software provided by the vendor (possibly Windows specific ?) is required for the `Vendor specific` services that probably provide the audio streams.

borine, 2020-10-04:
Hmm, I just looked the specs for this device, and found:
```
Bluetooth® Specification
      Bluetooth® version:            Version 5.0
      Effective Range:               Line of sight approx.30ft (10m)
      Frequency Range:               2.4GHz band (2.4000GHz-2.4835GHz)
      Profile:                       A2DP, AVRCP, HFP, HSP
      Supported Audio Format(s):     SBC, AAC
      Supported Content Protection:  SCMS-T
```

So it does claim to support A2DP, but for some reason is not adverting that on the Service Discovery Protocol. I don't know what can be done about that (but would like to learn if anyone has a solution?)

Arkq, 2020-10-04:
If in the device name "LE_WF-1000XM3" the LE stands for Low Energy (Bluetooth Audio 5.0), than it will not work with Linux (BlueZ does not support Low Energy Audio). Also, it might be possible that if you install proprietary mobile app for this earbuds, you will be able to switch to Bluetooth 4.0 for audio profile (non low energy). Similar case is with Sony WH1000XM3 where you can select between quality and stability (from proprietary mobile app), and in case where stability is selected you've got only SBC codec (no aptX/HD, no LDAC).

ho246, 2020-10-04:
Hi, by putting them into pairing mode I can disable the low energy mode, then they identify as WF-1000XM3. Also thanks for the hint with the app, they were in a mode called stability, I tried switching them to quality but that didn't change matters at all. Seems I am out of luck for now. Thanks for your help, though.

ho246, 2020-10-04:
Ok one more part of the puzzle, in the app I deactivated something called dsee-hx. Then I deleted the device in bluetoothctl and started the pairing procedure and lo, it works now. Sound output works just like it does on the phone. Have not tried the Mike yet. Anyway thank you both a lot and sorry for bothering you with that.

closed 2020-10-05 by Arkq