arkq/bluez-alsa #385: Profile HFP Ofono no output and no input

xavi-b, 2020-10-03:
Hi,

I am using my Raspberry Pi as a "bluetooth speaker, music controller and call controller" for my phone.

I installed debian ofono package.
I installed bluealsa from master using :
```
git clone git@github.com:raspberrypi-ui/bluealsa.git && cd bluealsa/
autoreconf --install
mkdir build && cd build
../configure --enable-ofono --enable-debug --enable-aplay
make -j4
sudo make install
```
I launch bluealsa in a service file :
```
/usr/bin/bluealsa --profile=a2dp-sink --profile=hfp-ofono
```
~/.asoundrc
```
pcm.!default {
	type asym
	playback.pcm {
		type plug
		slave.pcm "dmixer"
	}
	capture.pcm {
		type plug
		slave.pcm "input"
	}
}

pcm.bluealsadmix {
    type plug
    slave.pcm "dmixer"
}

pcm.dmixer {
    type dmix
    ipc_key 1024
    slave.pcm "output"
}

pcm.output {
	type hw
	card 1
}

ctl.!default {
	type hw
	card 1
}

pcm.input {
	type hw
	card 1
}
```
bluealsa journal is the following one :
```
Oct 03 13:20:04 rpi4 bluealsa[438]: /usr/bin/bluealsa: ../../src/bluez.c:1330: Signal: org.freedesktop.DBus.Properties.PropertiesChanged()
Oct 03 13:20:04 rpi4 bluealsa[438]: /usr/bin/bluealsa: ../../src/bluez.c:1364: Signal: PropertiesChanged: org.bluez.MediaTransport1: State
Oct 03 13:20:04 rpi4 bluealsa[438]: /usr/bin/bluealsa: ../../src/ba-transport.c:601: State transition: 2 -> 0
Oct 03 13:20:04 rpi4 bluealsa[438]: /usr/bin/bluealsa: ../../src/ba-transport.c:749: Releasing transport: A2DP Sink (SBC)
Oct 03 13:20:04 rpi4 bluealsa[438]: /usr/bin/bluealsa: ../../src/ba-transport.c:778: Closing BT: 12
Oct 03 13:20:04 rpi4 bluealsa[438]: /usr/bin/bluealsa: ../../src/ba-transport.c:928: Exiting IO thread: A2DP Sink (SBC)
Oct 03 13:20:10 rpi4 bluealsa[438]: /usr/bin/bluealsa: ../../src/ofono.c:428: Called: org.ofono.HandsfreeAudioAgent.NewConnection()
Oct 03 13:20:10 rpi4 bluealsa[438]: /usr/bin/bluealsa: ../../src/ofono.c:384: New oFono SCO connection (codec: 0x1): 15
Oct 03 13:20:41 rpi4 bluealsa[438]: /usr/bin/bluealsa: ../../src/io.c:2248: SCO poll error status: 0x18
Oct 03 13:20:41 rpi4 bluealsa[438]: /usr/bin/bluealsa: ../../src/ofono.c:136: Closing oFono SCO: 15
Oct 03 13:20:41 rpi4 bluealsa[438]: /usr/bin/bluealsa: ../../src/bluez.c:1330: Signal: org.freedesktop.DBus.Properties.PropertiesChanged()
Oct 03 13:20:41 rpi4 bluealsa[438]: /usr/bin/bluealsa: ../../src/bluez.c:1364: Signal: PropertiesChanged: org.bluez.MediaTransport1: State
Oct 03 13:20:41 rpi4 bluealsa[438]: /usr/bin/bluealsa: ../../src/ba-transport.c:601: State transition: 0 -> 1
Oct 03 13:20:41 rpi4 bluealsa[438]: /usr/bin/bluealsa: ../../src/ba-transport.c:722: New transport: 12 (MTU: R:672 W:1008)
Oct 03 13:20:41 rpi4 bluealsa[438]: /usr/bin/bluealsa: ../../src/bluez.c:1330: Signal: org.freedesktop.DBus.Properties.PropertiesChanged()
Oct 03 13:20:41 rpi4 bluealsa[438]: /usr/bin/bluealsa: ../../src/bluez.c:1364: Signal: PropertiesChanged: org.bluez.MediaTransport1: State
Oct 03 13:20:41 rpi4 bluealsa[438]: /usr/bin/bluealsa: ../../src/ba-transport.c:601: State transition: 1 -> 2
Oct 03 13:20:41 rpi4 bluealsa[438]: /usr/bin/bluealsa: ../../src/io.c:2576: Created new IO thread: A2DP Sink (SBC)
Oct 03 13:20:41 rpi4 bluealsa[438]: /usr/bin/bluealsa: ../../src/io.c:305: Starting IO loop: A2DP Sink (SBC)
```
Bluetooth music with AD2P Sink works perfectly but, unfortunatly, phone calls do not produce any sound or hear anything.
Is there something I am missing in configurations file ?

xavi-b, 2020-10-04:
I partialy solved my issue.

Launching a second "bluealsa-aplay service" with --profile-sco :
```
[Unit]
Description=BlueALSA Ofono aplay service
After=bluealsa.service
Requires=bluealsa.service

[Service]
Type=simple
User=pi
ExecStartPre=/bin/sleep 20
ExecStart=/usr/bin/bluealsa-aplay -vv --profile-sco -d ofonoalsadmix 00:00:00:00:00:00

[Install]
WantedBy=multi-user.target
```
I added to my .asoundrc :
```
pcm.ofonoalsadmix {
    type plug
    slave.pcm "dmixer"
}
```

However, I can't hear the sound of the microphone of my Raspberry PI in the caller's phone.
Am I missing something ?

dr-ni, 2020-10-25:
sudo su
export LIBASOUND_THREAD_SAFE=0
bluealsa -p hfp-ofono

and in another terminal:
arecord -D bluealsa | aplay
to receive stream

and in another terminal:
arecord | aplay -D bluealsa
to send stream

sudo ofono-1.xx/test/dial-number 0815xyz
works, but huge delay 20s :-(

only using incomming stream the delay is ok –1s

cat ~/.asoundrc
defaults.bluealsa.service "org.bluealsa"
defaults.bluealsa.device "XX:XX:XX:XX:XX:XX"
defaults.bluealsa.profile "sco"
defaults.bluealsa.delay 10000

borine, 2020-10-26:
@dr-ni I can't explain the delay (I don't use ofono), but I advise that `arecord` by default always converts the stream to unsigned 8-bit samples. For use with HFP you should tell it to use signed 16-bit little-endian samples, so:
```
arecord -f s16_le -D bluealsa | aplay
```

and
```
arecord -f s16_le | aplay -D bluealsa
```

Also you could try to reduce the delay by setting explicit buffer and period sizes and start delay, I'm not sure how successful that will be though (see `aplay --help` and `man aplay`).

xavi-b, 2020-10-26:
@borine @dr-ni 
Given my .asoundrc
```
pcm.!default {
	type asym
	playback.pcm {
		type plug
		slave.pcm "dmixer"
	}
	capture.pcm {
		type plug
		slave.pcm "input"
	}
}

pcm.bluealsadmix {
    type plug
    slave.pcm "dmixer"
}

pcm.ofonoalsadmix {
    type plug
    slave.pcm "dmixer"
}

pcm.dmixer {
    type dmix
    ipc_key 1024
    slave.pcm "output"
}

pcm.output {
	type hw
	card 0
}

ctl.!default {
	type hw
	card 0
}

pcm.input {
    type hw
    card 0
}
```
How should I launch `bluealsa-aplay` in my service file to avoid doing:
```
arecord -f s16_le | aplay -D bluealsa
```

borine, 2020-10-26:
@xavi-b 
> Am I missing something ?

Your description does not say how you intend to capture sound from the microphone to send to bluealsa sco device. The `bluealsa-aplay` application does not do that for you. So `arecord -f s16_le | aplay -D bluealsa` is one way. If you have bluealsa v3.0.0 or later, you could try as an alternative:

 `alsaloop -f s16_le -c 1 -r 8000 -P bluealsa -t 100000`

Note that both those commands assume that you have 
```
defaults.bluealsa.device "XX:XX:XX:XX:XX:XX"
defaults.bluealsa.profile "sco"
```
in your ,asoundrc file

xavi-b, 2020-10-26:
So the command `/usr/bin/bluealsa-aplay -vv --profile-sco -d ofonoalsadmix 00:00:00:00:00:00` only works for the audio sent from the phone to the Linux system ?

I have to admit I expected it to also work for the audio sent from (the microphone of) the Linux system to the phone when setting a ..

Will it work with any connected device if I replace `"XX:XX:XX:XX:XX:XX"` with `00:00:00:00:00:00` ?

> ```
> defaults.bluealsa.device "XX:XX:XX:XX:XX:XX"
> defaults.bluealsa.profile "sco"
> ```
Won't adding those lines affect all my configured PCM devices ?

borine, 2020-10-26:
> I have to admit I expected it to also work for the audio sent from (the microphone of) the Linux system

No. The man page for bluealsa-aplay is available on-line here: https://github.com/Arkq/bluez-alsa/blob/master/doc/bluealsa-aplay.1.rst . There is no equivalent utility from the bluez-alsa project that sends audio in the other direction.

> Will it work with any connected device if I replace "XX:XX:XX:XX:XX:XX" with 00:00:00:00:00:00 ?

No, unfortunately not. To use alsa utilities such as arecord or alsaloop you have to tell them a specific PCM device to capture from. And also, that device must be connected before the alsa utility is started otherwise it will fail with "No such device" error. So you cannot simply start arecord or alsaloop at boot, you have to wait until the device is connected. I don't use ofono, so I have limited experience here, if there are any other applications available that handle the outgoing audio side of the connection maybe someone else can advise. 

> Won't adding those lines affect all my configured PCM devices ?

They only apply to bluealsa devices, but yes it will apply to all of those. You can instead specify the device id on the application command line, e.g:

```
arecord -f s16_le -D plughw:0,0 | aplay -D bluealsa:DEV=XX:XX:XX:XX:XX:XX,PROFILE=sco
```

or

```
alsaloop -C plughw:0,0 -P bluealsa:DEV=XX:XX:XX:XX:XX:XX,PROFILE=sco,DELAY=0 -t 100000
```

That way there is no need for .asoundrc. See the wiki page https://github.com/Arkq/bluez-alsa/wiki/Using-the-bluealsa-ALSA-pcm-plugin for more information on using the bluealsa pcm plugin with alsa applications,

dr-ni, 2020-10-26:
@borine 
still not working correct with your suggestion
arecord -f s16_le -D bluealsa | aplay
and
arecord -f s16_le | aplay -D bluealsa

>Also you could try to reduce the delay by setting explicit buffer and period sizes and start delay, 
what do you think - what are good values to start testing?

pi@raspberrypi_zero_w:~$ sudo su 
root@pilot:/home/pilot# export LIBASOUND_THREAD_SAFE=0
root@pilot:/home/pilot# bluealsa -p hfp-ofono 
bluealsa: D: ofono.c:471: Registering oFono audio agent: /org/bluez/HFP/oFono
bluealsa: D: ofono.c:247: Adding new oFono card: /card_1
bluealsa: D: ba-transport.c:765: Starting transport: HFP Hands-Free (CVSD)
bluealsa: D: ba-transport.c:1047: Created new thread [ba-sco]: HFP Hands-Free (CVSD)
bluealsa: D: main.c:390: Acquiring D-Bus service name: org.bluealsa
bluealsa: D: main.c:395: Starting main dispatching loop
bluealsa: D: sco.c:241: Starting SCO loop: HFP Hands-Free (CVSD)
bluealsa: D: bluez.c:1009: Signal: org.freedesktop.DBus.ObjectManager.InterfacesRemoved()
bluealsa: D: bluez.c:1009: Signal: org.freedesktop.DBus.ObjectManager.InterfacesRemoved()
bluealsa: D: bluez.c:1009: Signal: org.freedesktop.DBus.ObjectManager.InterfacesRemoved()
bluealsa: D: bluez.c:1009: Signal: org.freedesktop.DBus.ObjectManager.InterfacesRemoved()
bluealsa: D: bluez.c:1009: Signal: org.freedesktop.DBus.ObjectManager.InterfacesRemoved()
bluealsa: D: bluez.c:1009: Signal: org.freedesktop.DBus.ObjectManager.InterfacesRemoved()
bluealsa: D: bluez.c:1009: Signal: org.freedesktop.DBus.ObjectManager.InterfacesRemoved()
bluealsa: D: bluez.c:1009: Signal: org.freedesktop.DBus.ObjectManager.InterfacesRemoved()
bluealsa: D: bluez.c:1009: Signal: org.freedesktop.DBus.ObjectManager.InterfacesRemoved()
bluealsa: D: dbus.c:59: Called: org.bluealsa.Manager1.GetPCMs() on /org/bluealsa
bluealsa: D: dbus.c:59: Called: org.bluealsa.PCM1.Open() on /org/bluealsa/hci0/dev_xx_xx_xx_xx_xx_xx/hfphf/sink
bluealsa: D: ba-transport.c:804: PCM resumed: 9
bluealsa: D: dbus.c:59: Called: org.bluealsa.Manager1.GetPCMs() on /org/bluealsa
bluealsa: D: dbus.c:59: Called: org.bluealsa.PCM1.Open() on /org/bluealsa/hci0/dev_xx_xx_xx_xx_xx_xx/hfphf/source
bluealsa: D: ba-transport.c:804: PCM resumed: 13
bluealsa: D: dbus.c:59: Called: org.ofono.HandsfreeAudioAgent.NewConnection() on /org/bluez/HFP/oFono
bluealsa: D: ofono.c:398: New oFono SCO connection (codec: 0x1): 15
bluealsa: D: hci.c:132: SCO link socket MTU: 15: 64
bluealsa: D: ba-transport.c:1021: Closing PCM: 13
bluealsa: D: sco.c:410: SCO poll error status: 0x18
bluealsa: D: ofono.c:137: Closing oFono SCO: 15
bluealsa: E: SCO write error: Bad file descriptor
bluealsa: D: dbus.c:59: Called: org.ofono.HandsfreeAudioAgent.NewConnection() on /org/bluez/HFP/oFono
bluealsa: D: ofono.c:398: New oFono SCO connection (codec: 0x1): 13
bluealsa: D: hci.c:132: SCO link socket MTU: 13: 64
bluealsa: D: sco.c:410: SCO poll error status: 0x18
bluealsa: D: ofono.c:137: Closing oFono SCO: 13
bluealsa: E: SCO write error: Bad file descriptor

pi@raspberrypi_zero_w:~$ arecord -f s16_le | aplay -D bluealsa 
D: bluealsa-pcm.c:1004: Getting BlueALSA PCM: PLAYBACK xx_xx_xx_xx_xx_xx sco
D: bluealsa-pcm.c:876: /org/bluealsa/hci0/dev_xx_xx_xx_xx_xx_xx/hfphf/sink: Setting constraints
Aufnahme: WAVE 'stdin' : Signed 16 bit Little Endian, Rate: 8000 Hz, mono
Wiedergabe: WAVE 'stdin' : Signed 16 bit Little Endian, Rate: 8000 Hz, mono
D: bluealsa-pcm.c:443: /org/bluealsa/hci0/dev_xx_xx_xx_xx_xx_xx/hfphf/sink: Initializing HW
D: bluealsa-pcm.c:466: /org/bluealsa/hci0/dev_xx_xx_xx_xx_xx_xx/hfphf/sink: FIFO buffer size: 2048 frames
D: bluealsa-pcm.c:474: /org/bluealsa/hci0/dev_xx_xx_xx_xx_xx_xx/hfphf/sink: Selected HW buffer: 4 periods x 2000 bytes == 8000 bytes
D: bluealsa-pcm.c:489: /org/bluealsa/hci0/dev_xx_xx_xx_xx_xx_xx/hfphf/sink: Initializing SW
D: bluealsa-pcm.c:489: /org/bluealsa/hci0/dev_xx_xx_xx_xx_xx_xx/hfphf/sink: Initializing SW
D: bluealsa-pcm.c:526: /org/bluealsa/hci0/dev_xx_xx_xx_xx_xx_xx/hfphf/sink: Prepared
D: bluealsa-pcm.c:489: /org/bluealsa/hci0/dev_xx_xx_xx_xx_xx_xx/hfphf/sink: Initializing SW
D: bluealsa-pcm.c:356: /org/bluealsa/hci0/dev_xx_xx_xx_xx_xx_xx/hfphf/sink: Starting
D: bluealsa-pcm.c:213: /org/bluealsa/hci0/dev_xx_xx_xx_xx_xx_xx/hfphf/sink: Starting IO loop: 7
Überlauf!!! (mindestens 13,358 ms)
Überlauf!!! (mindestens 12,476 ms)
D: bluealsa-pcm.c:218: /org/bluealsa/hci0/dev_xx_xx_xx_xx_xx_xx/hfphf/sink: Pausing IO thread: -1
Unterlauf!!! (mindestens 244415,380 ms)
D: bluealsa-pcm.c:526: /org/bluealsa/hci0/dev_xx_xx_xx_xx_xx_xx/hfphf/sink: Prepared
D: bluealsa-pcm.c:356: /org/bluealsa/hci0/dev_xx_xx_xx_xx_xx_xx/hfphf/sink: Starting
D: bluealsa-pcm.c:232: /org/bluealsa/hci0/dev_xx_xx_xx_xx_xx_xx/hfphf/sink: IO thread resumed

pi@raspberrypi_zero_w:~$ arecord -f s16_le -D bluealsa | aplay
D: bluealsa-pcm.c:1004: Getting BlueALSA PCM: CAPTURE xx_xx_xx_xx_xx_xx sco
D: bluealsa-pcm.c:876: /org/bluealsa/hci0/dev_xx_xx_xx_xx_xx_xx/hfphf/source: Setting constraints
Aufnahme: WAVE 'stdin' : Signed 16 bit Little Endian, Rate: 8000 Hz, mono
D: bluealsa-pcm.c:443: /org/bluealsa/hci0/dev_xx_xx_xx_xx_xx_xx/hfphf/source: Initializing HW
D: bluealsa-pcm.c:466: /org/bluealsa/hci0/dev_xx_xx_xx_xx_xx_xx/hfphf/source: FIFO buffer size: 32768 frames
D: bluealsa-pcm.c:474: /org/bluealsa/hci0/dev_xx_xx_xx_xx_xx_xx/hfphf/source: Selected HW buffer: 4 periods x 2000 bytes == 8000 bytes
D: bluealsa-pcm.c:489: /org/bluealsa/hci0/dev_xx_xx_xx_xx_xx_xx/hfphf/source: Initializing SW
[... cut ...]
D: bluealsa-pcm.c:526: /org/bluealsa/hci0/dev_xx_xx_xx_xx_xx_xx/hfphf/source: Prepared
D: bluealsa-pcm.c:489: /org/bluealsa/hci0/dev_xx_xx_xx_xx_xx_xx/hfphf/source: Initializing SW
Wiedergabe: WAVE 'stdin' : Signed 16 bit Little Endian, Rate: 8000 Hz, mono
D: bluealsa-pcm.c:356: /org/bluealsa/hci0/dev_xx_xx_xx_xx_xx_xx/hfphf/source: Starting
D: bluealsa-pcm.c:213: /org/bluealsa/hci0/dev_xx_xx_xx_xx_xx_xx/hfphf/source: Starting IO loop: 7
aplay: xrun:1624: Lese-/Schreibfehler, Status = RUNNING
pi@raspberrypi_zero_w:~$

borine, 2020-10-27:
> Unterlauf!!! (mindestens 244415,380 ms)

An underrun of over 4 minutes ? something seriously broken, i thiink!

Please can you post the versions of:
bluealsa (`bluealsa --version`)
alsa (`aplay --version`)
kernel (`uname -vr`)

also what kind of bluetooth adapter are you using: `hciconfig -a`

dr-ni, 2020-10-27:
$ bluealsa --version
v3.0.0-9-g63e2eea
$ aplay --version
aplay: version 1.1.3 by Jaroslav Kysela <perex@perex.cz>
$ uname -vr
4.19.66+ #1253 Thu Aug 15 11:37:30 BST 2019
$ hciconfig -a
hci0:	Type: Primary  Bus: UART
	BD Address: xx:xx:xx:xx:xx:xx  ACL MTU: 1021:8  SCO MTU: 64:1
	UP RUNNING 
	RX bytes:1596195 acl:114 sco:24726 events:427 errors:0
	TX bytes:1681966 acl:69 sco:32256 commands:150 errors:0
	Features: 0xbf 0xfe 0xcf 0xfe 0xdb 0xff 0x7b 0x87
	Packet type: DM1 DM3 DM5 DH1 DH3 DH5 HV1 HV2 HV3 
	Link policy: RSWITCH SNIFF 
	Link mode: SLAVE ACCEPT 
	Name: 'pilot'
	Class: 0x200000
	Service Classes: Audio
	Device Class: Miscellaneous, 
	HCI Version: 4.1 (0x7)  Revision: 0x168
	LMP Version: 4.1 (0x7)  Subversion: 0x2209
	Manufacturer: Broadcom Corporation (15)

borine, 2020-10-27:
with bluez-alsa 3.0.0 and later, you can try alsaloop instead of arecord/aplay to get rid of the delay. You'll need to disable alsa thread blocking for this. Try:
```
LIBASOUND_THREAD_SAFE=0 alsaloop  -P bluealsa:DELAY=0 -t 100000
```

Also, it might be helpful to know which plugins are used by the `default` capture device. Please can you post the output from:
```shell
arecord -v -f s16_le -d 1 /dev/null
```

dr-ni, 2020-10-27:
>LIBASOUND_THREAD_SAFE=0 alsaloop  -P bluealsa:DELAY=0 -t 100000

I'll check this later...

$ arecord -v -f s16_le -d 1 /dev/null
Aufnahme: WAVE '/dev/null' : Signed 16 bit Little Endian, Rate: 8000 Hz, mono
Plug PCM: Rate conversion PCM (48000, sformat=S16_LE)
Converter: libsamplerate
Protocol version: 10002
Its setup is:
  stream       : CAPTURE
  access       : RW_INTERLEAVED
  format       : S16_LE
  subformat    : STD
  channels     : 1
  rate         : 8000
  exact rate   : 8000 (8000/1)
  msbits       : 16
  buffer_size  : 3000
  period_size  : 1000
  period_time  : 125000
  tstamp_mode  : NONE
  tstamp_type  : MONOTONIC
  period_step  : 1
  avail_min    : 1000
  period_event : 0
  start_threshold  : 1
  stop_threshold   : 3000
  silence_threshold: 0
  silence_size : 0
  boundary     : 196608000
Slave: Route conversion PCM (sformat=S16_LE)
  Transformation table:
    0 <- 0*0,5 + 1*0,5
Its setup is:
  stream       : CAPTURE
  access       : MMAP_INTERLEAVED
  format       : S16_LE
  subformat    : STD
  channels     : 1
  rate         : 48000
  exact rate   : 48000 (48000/1)
  msbits       : 16
  buffer_size  : 18000
  period_size  : 6000
  period_time  : 125000
  tstamp_mode  : NONE
  tstamp_type  : MONOTONIC
  period_step  : 1
  avail_min    : 6000
  period_event : 0
  start_threshold  : 6
  stop_threshold   : 18000
  silence_threshold: 0
  silence_size : 0
  boundary     : 1179648000
Slave: Direct Snoop PCM
Its setup is:
  stream       : CAPTURE
  access       : MMAP_INTERLEAVED
  format       : S16_LE
  subformat    : STD
  channels     : 2
  rate         : 48000
  exact rate   : 48000 (48000/1)
  msbits       : 16
  buffer_size  : 18000
  period_size  : 6000
  period_time  : 125000
  tstamp_mode  : NONE
  tstamp_type  : MONOTONIC
  period_step  : 1
  avail_min    : 6000
  period_event : 0
  start_threshold  : 6
  stop_threshold   : 18000
  silence_threshold: 0
  silence_size : 0
  boundary     : 1179648000
Hardware PCM card 0 'wm8960-soundcard' device 0 subdevice 0
Its setup is:
  stream       : CAPTURE
  access       : MMAP_INTERLEAVED
  format       : S16_LE
  subformat    : STD
  channels     : 2
  rate         : 48000
  exact rate   : 48000 (48000/1)
  msbits       : 16
  buffer_size  : 18000
  period_size  : 6000
  period_time  : 125000
  tstamp_mode  : ENABLE
  tstamp_type  : MONOTONIC
  period_step  : 1
  avail_min    : 6000
  period_event : 0
  start_threshold  : 1
  stop_threshold   : 1179648000
  silence_threshold: 0
  silence_size : 0
  boundary     : 1179648000
  appl_ptr     : 0
  hw_ptr       : 0

dr-ni, 2020-10-27:
here now more - but only digital noise in the remote phone - no sound local

$ LIBASOUND_THREAD_SAFE=0 bluealsa -p hfp-ofono
bluealsa: D: ofono.c:471: Registering oFono audio agent: /org/bluez/HFP/oFono
bluealsa: D: ofono.c:247: Adding new oFono card: /card_1
bluealsa: D: ba-transport.c:765: Starting transport: HFP Hands-Free (CVSD)
bluealsa: D: ba-transport.c:1047: Created new thread [ba-sco]: HFP Hands-Free (CVSD)
bluealsa: D: main.c:390: Acquiring D-Bus service name: org.bluealsa
bluealsa: D: sco.c:241: Starting SCO loop: HFP Hands-Free (CVSD)
bluealsa: D: main.c:395: Starting main dispatching loop
bluealsa: D: dbus.c:59: Called: org.bluealsa.Manager1.GetPCMs() on /org/bluealsa
bluealsa: D: dbus.c:59: Called: org.bluealsa.Manager1.GetPCMs() on /org/bluealsa
bluealsa: D: dbus.c:59: Called: org.bluealsa.PCM1.Open() on /org/bluealsa/hci0/dev_xx_xx_xx_xx_xx_xx/hfphf/sink
bluealsa: D: ba-transport.c:804: PCM resumed: 9
bluealsa: D: ba-transport.c:838: PCM dropped: 9
bluealsa: D: ba-transport.c:1021: Closing PCM: 9
bluealsa: D: dbus.c:59: Called: org.bluealsa.Manager1.GetPCMs() on /org/bluealsa
bluealsa: D: dbus.c:59: Called: org.bluealsa.PCM1.Open() on /org/bluealsa/hci0/dev_xx_xx_xx_xx_xx_xx/hfphf/sink
bluealsa: D: ba-transport.c:804: PCM resumed: 9
bluealsa: D: dbus.c:59: Called: org.ofono.HandsfreeAudioAgent.NewConnection() on /org/bluez/HFP/oFono
bluealsa: D: ofono.c:398: New oFono SCO connection (codec: 0x1): 13
bluealsa: D: hci.c:132: SCO link socket MTU: 13: 64
bluealsa: D: ofono.c:137: Closing oFono SCO: 13
bluealsa: D: ba-transport.c:838: PCM dropped: 9
bluealsa: D: ba-transport.c:1021: Closing PCM: 9
^C
bluealsa: D: main.c:398: Exiting main loop

$ LIBASOUND_THREAD_SAFE=0 alsaloop -P bluealsa:DELAY=0 -t 100000
D: bluealsa-pcm.c:1004: Getting BlueALSA PCM: PLAYBACK xx:xx:xx:xx:xx:xx sco
D: bluealsa-pcm.c:876: /org/bluealsa/hci0/dev_xx_xx_xx_xx_xx_xx/hfphf/sink: Setting constraints
D: bluealsa-pcm.c:443: /org/bluealsa/hci0/dev_xx_xx_xx_xx_xx_xx/hfphf/sink: Initializing HW
D: bluealsa-pcm.c:466: /org/bluealsa/hci0/dev_xx_xx_xx_xx_xx_xx/hfphf/sink: FIFO buffer size: 2048 frames
D: bluealsa-pcm.c:474: /org/bluealsa/hci0/dev_xx_xx_xx_xx_xx_xx/hfphf/sink: Selected HW buffer: 8 periods x 800 bytes == 6400 bytes
D: bluealsa-pcm.c:489: /org/bluealsa/hci0/dev_xx_xx_xx_xx_xx_xx/hfphf/sink: Initializing SW
D: bluealsa-pcm.c:489: /org/bluealsa/hci0/dev_xx_xx_xx_xx_xx_xx/hfphf/sink: Initializing SW
D: bluealsa-pcm.c:489: /org/bluealsa/hci0/dev_xx_xx_xx_xx_xx_xx/hfphf/sink: Initializing SW
D: bluealsa-pcm.c:526: /org/bluealsa/hci0/dev_xx_xx_xx_xx_xx_xx/hfphf/sink: Prepared
D: bluealsa-pcm.c:489: /org/bluealsa/hci0/dev_xx_xx_xx_xx_xx_xx/hfphf/sink: Initializing SW
D: bluealsa-pcm.c:496: /org/bluealsa/hci0/dev_xx_xx_xx_xx_xx_xx/hfphf/sink: Changing SW avail min: 400 -> 2600
D: bluealsa-pcm.c:526: /org/bluealsa/hci0/dev_xx_xx_xx_xx_xx_xx/hfphf/sink: Prepared
D: bluealsa-pcm.c:356: /org/bluealsa/hci0/dev_xx_xx_xx_xx_xx_xx/hfphf/sink: Starting
D: bluealsa-pcm.c:213: /org/bluealsa/hci0/dev_xx_xx_xx_xx_xx_xx/hfphf/sink: Starting IO loop: 9
overrun for capture hw:0,0
overrun for capture hw:0,0
D: bluealsa-pcm.c:218: /org/bluealsa/hci0/dev_xx_xx_xx_xx_xx_xx/hfphf/sink: Pausing IO thread: -1
underrun for playback bluealsa:DELAY=0
D: bluealsa-pcm.c:526: /org/bluealsa/hci0/dev_xx_xx_xx_xx_xx_xx/hfphf/sink: Prepared
D: bluealsa-pcm.c:526: /org/bluealsa/hci0/dev_xx_xx_xx_xx_xx_xx/hfphf/sink: Prepared
D: bluealsa-pcm.c:356: /org/bluealsa/hci0/dev_xx_xx_xx_xx_xx_xx/hfphf/sink: Starting
D: bluealsa-pcm.c:232: /org/bluealsa/hci0/dev_xx_xx_xx_xx_xx_xx/hfphf/sink: IO thread resumed
D: bluealsa-pcm.c:218: /org/bluealsa/hci0/dev_xx_xx_xx_xx_xx_xx/hfphf/sink: Pausing IO thread: -1
underrun for playback bluealsa:DELAY=0
D: bluealsa-pcm.c:526: /org/bluealsa/hci0/dev_xx_xx_xx_xx_xx_xx/hfphf/sink: Prepared
D: bluealsa-pcm.c:526: /org/bluealsa/hci0/dev_xx_xx_xx_xx_xx_xx/hfphf/sink: Prepared
D: bluealsa-pcm.c:356: /org/bluealsa/hci0/dev_xx_xx_xx_xx_xx_xx/hfphf/sink: Starting
D: bluealsa-pcm.c:232: /org/bluealsa/hci0/dev_xx_xx_xx_xx_xx_xx/hfphf/sink: IO thread resumed
overrun for capture hw:0,0
overrun for capture hw:0,0
overrun for capture hw:0,0
^C
D: bluealsa-pcm.c:392: /org/bluealsa/hci0/dev_xx_xx_xx_xx_xx_xx/hfphf/sink: Stopping
D: bluealsa-pcm.c:145: /org/bluealsa/hci0/dev_xx_xx_xx_xx_xx_xx/hfphf/sink: IO thread cleanup
D: bluealsa-pcm.c:481: /org/bluealsa/hci0/dev_xx_xx_xx_xx_xx_xx/hfphf/sink: Freeing HW
D: bluealsa-pcm.c:431: /org/bluealsa/hci0/dev_xx_xx_xx_xx_xx_xx/hfphf/sink: Closing

borine, 2020-10-27:
> no sound local

the alsaloop command I gave was only for outgoing audio. You still need a separate stream for incoming. If arecord | aplay was acceptable before, stick with that until you have outgoing audio working too.

> only digital noise in the remote phone

Two other things to try:
1. increase the target latency for alsaloop:
```
LIBASOUND_THREAD_SAFE=0 alsaloop  -P bluealsa:DELAY=0 -t 500000
```

2. Avoid all plugins on the capture side by using the native format of the card:
```
LIBASOUND_THREAD_SAFE=0 alsaloop  -C hw:0,0 -f s16_le -c 2 -r 48000 -P bluealsa:DELAY=0 -t 500000
```

xavi-b, 2020-10-27:
On my Raspberry PI 4, I have a 1 sec delay using
```LIBASOUND_THREAD_SAFE=0 arecord -f s16_le | aplay -D bluealsa:DEV=XX:XX:XX:XX:XX:XX,PROFILE=sco```
which is perfectly fine for me.

I will use my C++/Qt project and the Ofono-Qt lib signals to run this command each time a call is started with a specific device.

Thanks @borine !

dr-ni, 2020-10-28:
sorry for the delay - no time yesterday

still very long delay while only sending :-(  but no errors:

root@raspberrypi_zero_w:/home/pi# LIBASOUND_THREAD_SAFE=0 bluealsa -p hfp-ofono
bluealsa: D: ofono.c:471: Registering oFono audio agent: /org/bluez/HFP/oFono
bluealsa: D: ofono.c:247: Adding new oFono card: /card_1
bluealsa: D: ba-transport.c:765: Starting transport: HFP Hands-Free (CVSD)
bluealsa: D: ba-transport.c:1047: Created new thread [ba-sco]: HFP Hands-Free (CVSD)
bluealsa: D: main.c:390: Acquiring D-Bus service name: org.bluealsa
bluealsa: D: main.c:395: Starting main dispatching loop
bluealsa: D: sco.c:241: Starting SCO loop: HFP Hands-Free (CVSD)
bluealsa: D: dbus.c:59: Called: org.bluealsa.Manager1.GetPCMs() on /org/bluealsa
bluealsa: D: dbus.c:59: Called: org.bluealsa.PCM1.Open() on /org/bluealsa/hci0/dev_xx_xx_xx_xx_xx_xx/hfphf/sink
bluealsa: D: ba-transport.c:804: PCM resumed: 9
bluealsa: D: dbus.c:59: Called: org.ofono.HandsfreeAudioAgent.NewConnection() on /org/bluez/HFP/oFono
bluealsa: D: ofono.c:398: New oFono SCO connection (codec: 0x1): 13
bluealsa: D: hci.c:132: SCO link socket MTU: 13: 64
bluealsa: D: sco.c:410: SCO poll error status: 0x18
bluealsa: D: ofono.c:137: Closing oFono SCO: 13
bluealsa: E: SCO write error: Bad file descriptor
^C
bluealsa: D: main.c:398: Exiting main loop

pi@raspberrypi_zero_w:~$ LIBASOUND_THREAD_SAFE=0 alsaloop  -P bluealsa:DELAY=0 -t 500000
D: bluealsa-pcm.c:1004: Getting BlueALSA PCM: PLAYBACK xx:xx:xx:xx:xx:xx sco
D: bluealsa-pcm.c:876: /org/bluealsa/hci0/dev_xx_xx_xx_xx_xx_xx/hfphf/sink: Setting constraints
D: bluealsa-pcm.c:443: /org/bluealsa/hci0/dev_xx_xx_xx_xx_xx_xx/hfphf/sink: Initializing HW
D: bluealsa-pcm.c:466: /org/bluealsa/hci0/dev_xx_xx_xx_xx_xx_xx/hfphf/sink: FIFO buffer size: 2048 frames
D: bluealsa-pcm.c:474: /org/bluealsa/hci0/dev_xx_xx_xx_xx_xx_xx/hfphf/sink: Selected HW buffer: 8 periods x 4000 bytes == 32000 bytes
D: bluealsa-pcm.c:489: /org/bluealsa/hci0/dev_xx_xx_xx_xx_xx_xx/hfphf/sink: Initializing SW
D: bluealsa-pcm.c:489: /org/bluealsa/hci0/dev_xx_xx_xx_xx_xx_xx/hfphf/sink: Initializing SW
D: bluealsa-pcm.c:489: /org/bluealsa/hci0/dev_xx_xx_xx_xx_xx_xx/hfphf/sink: Initializing SW
D: bluealsa-pcm.c:526: /org/bluealsa/hci0/dev_xx_xx_xx_xx_xx_xx/hfphf/sink: Prepared
D: bluealsa-pcm.c:489: /org/bluealsa/hci0/dev_xx_xx_xx_xx_xx_xx/hfphf/sink: Initializing SW
D: bluealsa-pcm.c:496: /org/bluealsa/hci0/dev_xx_xx_xx_xx_xx_xx/hfphf/sink: Changing SW avail min: 2000 -> 13000
D: bluealsa-pcm.c:526: /org/bluealsa/hci0/dev_xx_xx_xx_xx_xx_xx/hfphf/sink: Prepared
D: bluealsa-pcm.c:356: /org/bluealsa/hci0/dev_xx_xx_xx_xx_xx_xx/hfphf/sink: Starting
D: bluealsa-pcm.c:213: /org/bluealsa/hci0/dev_xx_xx_xx_xx_xx_xx/hfphf/sink: Starting IO loop: 9
D: bluealsa-pcm.c:145: /org/bluealsa/hci0/dev_xx_xx_xx_xx_xx_xx/hfphf/sink: IO thread cleanup
pcmjob failed

root@raspberrypi_zero_w:/home/pi# LIBASOUND_THREAD_SAFE=0 bluealsa -p hfp-ofono
bluealsa: D: ofono.c:471: Registering oFono audio agent: /org/bluez/HFP/oFono
bluealsa: D: ofono.c:247: Adding new oFono card: /card_1
bluealsa: D: ba-transport.c:765: Starting transport: HFP Hands-Free (CVSD)
bluealsa: D: ba-transport.c:1047: Created new thread [ba-sco]: HFP Hands-Free (CVSD)
bluealsa: D: main.c:390: Acquiring D-Bus service name: org.bluealsa
bluealsa: D: main.c:395: Starting main dispatching loop
bluealsa: D: sco.c:241: Starting SCO loop: HFP Hands-Free (CVSD)
bluealsa: D: dbus.c:59: Called: org.bluealsa.Manager1.GetPCMs() on /org/bluealsa
bluealsa: D: dbus.c:59: Called: org.bluealsa.PCM1.Open() on /org/bluealsa/hci0/dev_xx_xx_xx_xx_xx_xx/hfphf/sink
bluealsa: D: ba-transport.c:804: PCM resumed: 9
bluealsa: D: dbus.c:59: Called: org.ofono.HandsfreeAudioAgent.NewConnection() on /org/bluez/HFP/oFono
bluealsa: D: ofono.c:398: New oFono SCO connection (codec: 0x1): 13
bluealsa: D: hci.c:132: SCO link socket MTU: 13: 64
bluealsa: D: ba-transport.c:838: PCM dropped: 9
bluealsa: D: ba-transport.c:1021: Closing PCM: 9
bluealsa: D: sco.c:410: SCO poll error status: 0x18
bluealsa: D: ofono.c:137: Closing oFono SCO: 13

pi@raspberrypi_zero_w:~$ LIBASOUND_THREAD_SAFE=0 alsaloop  -C hw:0,0 -f s16_le -c 2 -r 48000 -P bluealsa:DELAY=0 -t 500000
D: bluealsa-pcm.c:1004: Getting BlueALSA PCM: PLAYBACK xx:xx:xx:xx:xx:xx sco
D: bluealsa-pcm.c:876: /org/bluealsa/hci0/dev_xx_xx_xx_xx_xx_xx/hfphf/sink: Setting constraints
D: bluealsa-pcm.c:443: /org/bluealsa/hci0/dev_xx_xx_xx_xx_xx_xx/hfphf/sink: Initializing HW
D: bluealsa-pcm.c:466: /org/bluealsa/hci0/dev_xx_xx_xx_xx_xx_xx/hfphf/sink: FIFO buffer size: 2048 frames
D: bluealsa-pcm.c:474: /org/bluealsa/hci0/dev_xx_xx_xx_xx_xx_xx/hfphf/sink: Selected HW buffer: 8 periods x 4000 bytes == 32000 bytes
D: bluealsa-pcm.c:489: /org/bluealsa/hci0/dev_xx_xx_xx_xx_xx_xx/hfphf/sink: Initializing SW
D: bluealsa-pcm.c:489: /org/bluealsa/hci0/dev_xx_xx_xx_xx_xx_xx/hfphf/sink: Initializing SW
D: bluealsa-pcm.c:489: /org/bluealsa/hci0/dev_xx_xx_xx_xx_xx_xx/hfphf/sink: Initializing SW
D: bluealsa-pcm.c:526: /org/bluealsa/hci0/dev_xx_xx_xx_xx_xx_xx/hfphf/sink: Prepared
D: bluealsa-pcm.c:489: /org/bluealsa/hci0/dev_xx_xx_xx_xx_xx_xx/hfphf/sink: Initializing SW
D: bluealsa-pcm.c:496: /org/bluealsa/hci0/dev_xx_xx_xx_xx_xx_xx/hfphf/sink: Changing SW avail min: 2000 -> 13000
D: bluealsa-pcm.c:526: /org/bluealsa/hci0/dev_xx_xx_xx_xx_xx_xx/hfphf/sink: Prepared
D: bluealsa-pcm.c:356: /org/bluealsa/hci0/dev_xx_xx_xx_xx_xx_xx/hfphf/sink: Starting
D: bluealsa-pcm.c:213: /org/bluealsa/hci0/dev_xx_xx_xx_xx_xx_xx/hfphf/sink: Starting IO loop: 9
^CD: bluealsa-pcm.c:392: /org/bluealsa/hci0/dev_xx_xx_xx_xx_xx_xx/hfphf/sink: Stopping
D: bluealsa-pcm.c:145: /org/bluealsa/hci0/dev_xx_xx_xx_xx_xx_xx/hfphf/sink: IO thread cleanup
D: bluealsa-pcm.c:481: /org/bluealsa/hci0/dev_xx_xx_xx_xx_xx_xx/hfphf/sink: Freeing HW
D: bluealsa-pcm.c:431: /org/bluealsa/hci0/dev_xx_xx_xx_xx_xx_xx/hfphf/sink: Closing

borine, 2020-10-29:
@dr-ni is the delay still around 20 seconds with alsaloop, or is it now a more understandable 1 or 2 seconds ? Have you tested your local microphone to send to a locally-connected speaker instead of sending to bluealsa? I really cannot understand where such a long delay is coning from.

dr-ni, 2020-10-30:
>@dr-ni is the delay still around 20 seconds with alsaloop, or is it now a more understandable 1 or 2 seconds ?

still around 7 to 10s

>Have you tested your local microphone to send to a locally-connected speaker instead of sending to bluealsa? 

arecord -f s16_le | aplay works pretty fast

>If possible, you could also try with USB bluetooth dongle instead of the on-board UART device. If your board is a Raspberry Pi then there may be issues with the UART configuration, or with Wifi interference.

in the past bluetooth was working good in combination with wifi, may be I'll buy a dongle for testing

borine, 2020-10-30:
> why is then only receiving working ok?

I don't know either. Clearly a similar setup is working for @xavi-b who is also using an RPi, but doesn't say which model. A delay of 7s needs buffering of  112kB, so that is definitely not in any bluealsa code, nor in the bluetooth stack (HFP latency should not exceed 7.5ms within the bluetooth-specified components). So the HCI link (ie UART serial device on the RPi) is the only other place I can think of. You may well be right though, I can think of no explanation why or how the UART could be at fault.

vs1785, 2020-11-26:
If i am using 3.0, can i run two instances of alsaloop, one for playback and another for capture ?

borine, 2021-10-30:
This issue has been idle for a year now. There have been some changes to the BlueALSA server and plugins since then, and there are now also some notes on using the SCO profiles in the wiki here: https://github.com/Arkq/bluez-alsa/wiki/Using-BlueALSA-with-HFP-and-HSP-Devices.

If anyone has issues with the latest BlueALSA code with HFP or HSP after reading the wiki article, please raise a new issue with debug output from the latest code.

closed 2021-10-30 by borine