arkq/bluez-alsa #386: Can bluez-alsa provides pre-generated man-pages on release?

sgn, 2020-10-05:
First of all, thank you very much for your work on `bluez-alsa`.

Since `pandoc` is built by Haskell stack and is usually hard to be cross-built, and costly, too.
Could you please provide pre-generated man-page on each release?

It's something nice to have. But, if it's generated more work to you, feel free to dismiss this request.

borine, 2020-10-05:
`pandoc` is available as a pre-built binary package for most platforms - see https://pandoc.org/installing.html
so there should be no need for Haskell.

sgn, 2020-10-05:
Yes, but we're building it for our distribution.
There're some platforms that doesn't have `pandoc` in our platform:
- `ppc64` big endian
- We couldn't cross build `pandoc` for arm platform, so, anyone want to have manpages need to build the whole haskell stack in our distribution (yes, that's our problems)

Please see https://github.com/void-linux/void-packages/pull/25203

Arkq, 2020-10-05:
I see that it might be inconvenient to use `pandoc` on some platforms (I'm facing the same issue with Gentoo, because I don't feel like building whole Haskell for one program, so I'm using pre-built). However, I'd like not to include generated files into the source code repository.... Same thing applies to autotools and the `configure` script (some project ships auto-generated scripts for user convenience). The case is that (imho) the normal end user should use package from of choice distribution repository and if somehow one is building from source, than it is already a power-user so he/she should be able to gather all required tools.

But back to the `pandoc` issue. Maybe there is some other tool (available on most distributions) which will be able to convert markdown file to man page? Also, there is a possibility to change man page source format, but I'd rather not - the easier it is to edit the more chances that someone will create patch (pull request) in case of error or missing information. So in the end I think that it is better to have markdown+pandoc, than no man pages at all...

sgn, 2020-10-05:
Thanks for your answer. I guess the answer is no, then. I'll close this issue.

sgn, 2020-10-06:
Anyway, I think `ronn` can generate man-pages from markdown, too.
I've tried a bit, and the generated man-pages looks OK, with some minor different to output of `pandoc`.
`ronn` and `pandoc` requires different input, though.

Would it is OK to you to accept a change into project that allow using either `ronn` or `pandoc` to build man-pages?

If yes, I can take the work.

borine, 2020-10-06:
> allow using either ronn or pandoc to build man-pages

I can't see any benefit from supporting two text conversion methods. I would recommend to choose just one.

When I first hacked together the initial PR for bluealsa man pages, I looked at both `pandoc` and `ronn`, and selected `pandoc` simply because it is actively maintained whereas the `ronn` source code seems to have been abandoned  7 years ago. But if `pandoc` is problematic for distributions then I agree that `ronn` may have been a better choice, so long as github renders `ronn-format` markdown in a readable way.

sgn, 2020-10-06:
> I can't see any benefit from supporting two text conversion methods. I would recommend to choose just one.

yay, while I agree with your comment.

I've hacked 2 patches to support both methods, and it seems not too ugly, I think it's OK to support both methods.

> so long as github renders `ronn-format` markdown in a readable way.

I think if we decided to go with only `ronn`, it will look OK, `ronn` use level 1 header to `.SH NAME` and level 2 header for section. Other element looks the same to me.

Arkq, 2020-10-06:
I've made a little research, how most projects generate man pages. And (based on Gentoo) most projects use `scdoc` (it has it's own dedicated man page syntax), then there are `doxygen` (not applicable here), `kdoctools`(for KDE projects), `docutils` (Python-based, capable of converting rst to man page, where rst is also rendered by github), `sphinx`, `asciidoc`, `help2man`, and `go-md2man`.

@sgn how about using `docutils`? I think that Python is the least problematic to compile for esoteric architectures. Also the rst format is rather easy to deal with and it is rendered by github.

@borine what is your opinion about such potential change?

PS.
I would rather keep only one tool for building man pages.

sgn, 2020-10-07:
> I've made a little research, how most projects generate man pages. And (based on Gentoo) most projects use `scdoc` (it has it's own dedicated man page syntax), then there are `doxygen` (not [...]

I would say `help2man` is mostly useless because it requires running the binary, would be hard for cross-compiling. `go` is harder to port than `ruby`, last I check, it isn't available to `ppc64` (BE).

> @sgn how about using `docutils`? I think that Python is the least problematic to compile for esoteric architectures. Also the rst format is rather easy to deal with and it is rendered by github.

Yes, I agree. I'll look into porting to `docutils`.

> PS.
> I would rather keep only one tool for building man pages.

Agree, too.

borine, 2020-10-07:
> @borine what is your opinion about such potential change?

The one important thing is that there is clear documentation on how to use these programs, readily available. By adding the markdown pages to the Github project I think that obligation has been fulfilled. Any more than that is really just a matter of what interests you, @Arkq, and what you consider to be a productive use of your time.

> Also the rst format is rather easy to deal with and it is rendered by github

Unfortunately, github rendering of rst suffers the same fundamental limitation as its rendering of markdown; i.e. it does not understand any syntax for definition (description) lists. That makes it hard to represent manpage format in github. Other than that I think its only slightly less convenient that markdown to author, so probably acceptable in terms of future maintenance burden.

Arkq, 2020-10-08:
@sgn, if you could prepare PR proposal for docutils integration, it would be great :)

sgn, 2020-10-09:
> @sgn, if you could prepare PR proposal for docutils integration, it would be great :)

Yes, I can propose something today, after `$dayjob`.

sgn, 2020-10-09:
Arkq, I've updated the existing PR to use `rst2man`.

commit 93853e51 (2020-10-11): Build man-pages with rst2man from docutils

Pandoc might be very hard to port to some platforms. Let's use something
easier to be ported to exotic platforms, namely rst2man.

Fixes #386 and closes #387
closed 2020-10-11 by Arkq