arkq/bluez-alsa #391: bluez-alsa with zoom (in glibc container): no headphones

dkwo, 2020-10-27:
This is probably not an issue with bluez-alsa, but perhaps someone here can help.
I'm running bluez-alsa on my Void Linux (x86_64-musl) host, with
```
$ cat .asoundrc
pcm.!default "bluealsa"
ctl.!default "bluealsa"
defaults.bluealsa.interface "hci0"
defaults.bluealsa.device "..."
defaults.bluealsa.profile "a2dp"
```
and then I use a glibc container to run Zoom, as described e.g. [here](https://git.disroot.org/tuxliban/tutoriales_void/src/branch/master/Container%20Void/Guide.md).
While I can get audio to work (tested via `mpv`) both in the host and the container, when I run zoom in the container it cannot access my bluetooth headset.
As suggested on the AUR, I tried to change in `~/.config/zoomus.conf` the line `system.audio.type=default` to `system.audio.type=alsa`, but no luck.

Anyone has an idea how to make it work?

borine, 2020-10-27:
Is there any log info from zoom relating to audio that might give a clue where to start looking? If mpv works but zoom does not, then its likely that zoom needs something that mpv does not. For example (just speculation) zoom needs both audio capture and playback, mpv needs just playback. Please also state the version of alsalib and bluealsa that you are using.

dkwo, 2020-10-27:
Thanks.
There's a line that says `sh: 1: pactl: not found`, but I'm not sure it is relevant.
I'm using `alsa-lib-1.2.3.2_1` and `bluez-alsa-3.0.0_2`.

The full `.zoom/logs/zoom_stdout_stderr.log` is
```
ZoomLauncher started.
Zoom path is: /opt/zoom
cmd line: 
CreateReportChannel bp_server_fd=4
$HOME = /home/dkwo 
export SSB_HOME=/home/dkwo/.zoom; export QSG_INFO=1; export QT_AUTO_SCREEN_SCALE_FACTOR=1; export LD_LIBRARY_PATH=/opt/zoom; export BREAKPAD_CLIENT_FD=3; /opt/zoom/zoom "" 
sh: 1: pactl: not found
zoom started.
QStandardPaths: XDG_RUNTIME_DIR not set, defaulting to '/tmp/runtime-dkwo'
sh: 1: pactl: not found
[CZPClientLogMgr::LogClientEnvironment] [MacAddr: D2:23:1C:3E:D0:97][client: Linux][OS: void][Hardware: CPU Core:2 Frenquency:2.4 G Memory size:19751MB CPU Brand:Intel(R) Core(TM) i5-6300U CPU @ 2.40GHz GPU Brand:][Req ID: ]
Linux Client Version is 5.3.465578.0920
QSG_RENDER_LOOP is 
XDG_CURRENT_DESKTOP = ;   GDMSESSION = 
Graphics Card Info:: 00:02.0 VGA compatible controller: Intel Corporation Skylake GT2 [HD Graphics 520] (rev 07)
Zoom package arch is 64bit, runing OS arch is x86_64
Client: Breakpad is using Client-Server Mode! client fd = 3
AppIconMgr::systemDesktopName log Desktop Name:  
error was 0
qt.scenegraph.general: QSG: basic render loop
qt.scenegraph.general: Using sg animation driver
qt.svg: link image0 hasn't been detected!
qt.svg: :/images/wechat.svg:10:6: Could not resolve property: pattern0
qt.scenegraph.general: texture atlas dimensions: 1024x512
qt.scenegraph.general: R/G/B/A Buffers:    8 8 8 8
qt.scenegraph.general: Depth Buffer:       24
qt.scenegraph.general: Stencil Buffer:     8
qt.scenegraph.general: Samples:            -1
qt.scenegraph.general: GL_VENDOR:          Intel
qt.scenegraph.general: GL_RENDERER:        Mesa Intel(R) HD Graphics 520 (SKL GT2)
qt.scenegraph.general: GL_VERSION:         4.6 (Compatibility Profile) Mesa 20.2.1
qt.scenegraph.general: GL_EXTENSIONS:      GL_EXT_shader_integer_mix GL_ATI_fragment_shader GL_EXT_texture_shadow_lod GL_ARB_framebuffer_no_attachments GL_ARB_indirect_parameters GL_ARB_compressed_texture_pixel_storage GL_ARB_sync GL_ARB_window_pos GL_EXT_compiled_vertex_array GL_EXT_texture_sRGB_decode GL_ARB_vertex_type_2_10_10_10_rev GL_EXT_texture_compression_s3tc GL_ARB_enhanced_layouts GL_ARB_texture_compression_bptc GL_ARB_query_buffer_object GL_ARB_provoking_vertex GL_MESA_window_pos GL_INTEL_blackhole_render GL_ARB_texture_border_clamp GL_ARB_seamless_cubemap_per_texture GL_ARB_fragment_program GL_EXT_texture_compression_rgtc GL_ARB_texture_swizzle GL_EXT_texture_env_dot3 GL_ARB_shader_storage_buffer_object GL_ATI_draw_buffers GL_ARB_fragment_coord_conventions GL_ARB_texture_env_add GL_EXT_timer_query GL_ARB_shadow GL_ARB_polygon_offset_clamp GL_ARB_shader_clock GL_EXT_stencil_two_side GL_EXT_pixel_buffer_object GL_EXT_texture_cube_map GL_ARB_texture_compression GL_ARB_shader_group_vote GL_ARB_shader_bit_encoding GL_ARB_texture_mirror_clamp_to_edge GL_NV_copy_image GL_EXT_packed_float GL_NV_blend_square GL_ARB_map_buffer_range GL_ARB_get_program_binary GL_ARB_pipeline_statistics_query GL_EXT_framebuffer_multisample GL_ARB_transform_feedback_instanced GL_INTEL_shader_atomic_float_minmax GL_NV_primitive_restart GL_SUN_multi_draw_arrays GL_ARB_vertex_attrib_binding GL_ARB_shader_atomic_counter_ops GL_ARB_texture_mirrored_repeat GL_KHR_no_error GL_ARB_texture_buffer_object_rgb32 GL_MESA_shader_integer_functions GL_ARB_texture_cube_map_array GL_NV_fragment_shader_interlock GL_ARB_shading_language_packing GL_EXT_point_parameters GL_EXT_framebuffer_object GL_EXT_shader_framebuffer_fetch_non_coherent GL_ARB_transpose_matrix GL_ARB_half_float_vertex GL_EXT_framebuffer_multisample_blit_scaled GL_EXT_EGL_image_storage GL_ARB_robustness GL_ARB_clear_buffer_object GL_AMD_query_buffer_object GL_ARB_draw_buffers GL_ARB_draw_elements_base_vertex GL_ARB_explicit_attrib_location GL_ARB_shader_draw_parameters GL_EXT_direct_state_access GL_KHR_robustness GL_ARB_gpu_shader_int64 GL_SGIS_texture_edge_clamp GL_EXT_texture_sRGB GL_ARB_ES3_1_compatibility GL_ARB_vertex_shader GL_EXT_texture_sRGB_R8 GL_AMD_conservative_depth GL_ARB_depth_texture GL_ATI_texture_float GL_ARB_shader_image_size GL_ARB_tessellation_shader GL_EXT_texture_snorm GL_ARB_explicit_uniform_location GL_NV_texture_barrier GL_ARB_transform_feedback3 GL_EXT_shadow_funcs GL_ARB_point_sprite GL_EXT_blend_subtract GL_OES_EGL_image GL_ARB_sampler_objects GL_ARB_fragment_shader GL_EXT_multi_draw_arrays GL_ARB_shader_ballot GL_ARB_framebuffer_object GL_ARB_texture_gather GL_ARB_texture_cube_map GL_ARB_conditional_render_inverted GL_AMD_pinned_memory GL_MESA_texture_signed_rgba GL_ARB_sample_shading GL_KHR_blend_equation_advanced GL_ARB_base_instance GL_EXT_draw_instanced GL_ARB_transform_feedback2 GL_ARB_timer_query GL_ARB_clip_control GL_ARB_shader_image_load_store GL_NV_texture_rectangle GL_KHR_robust_buffer_access_behavior GL_SGIS_texture_lod GL_KHR_texture_compression_astc_ldr GL_IBM_texture_mirrored_repeat GL_ARB_compatibility GL_ARB_texture_barrier GL_ARB_draw_buffers_blend GL_ARB_depth_clamp GL_ARB_invalidate_subdata GL_NV_packed_depth_stencil GL_ARB_robust_buffer_access_behavior GL_ARB_vertex_array_object GL_ARB_arrays_of_arrays GL_ARB_texture_view GL_ARB_fragment_program_shadow GL_ARB_half_float_pixel GL_ARB_shader_texture_lod GL_ARB_debug_output GL_ARB_shader_texture_image_samples GL_EXT_texture_buffer_object GL_EXT_texture_lod_bias GL_ARB_shader_objects GL_ARB_texture_ste [... cut ...]
WorkerBase::loop() normal exit, fds[0].revents & POLLIN = true
WorkerBase::loop() normal exit, fds[0].revents & POLLIN = true
WorkerBase::loop() normal exit, fds[0].revents & POLLIN = true
WorkerBase::loop() normal exit, fds[0].revents & POLLIN = true
WorkerBase::loop() normal exit, fds[0].revents & POLLIN = true
WorkerBase::loop() normal exit, fds[0].revents & POLLIN = true
sh: 1: pactl: not found
success to create child process,status is 0.
zoom exited normally.
run zoom successfully,exit code is 0.
ZoomLauncher exit.
```

borine, 2020-10-27:
I see no mention of ALSA in the log. In fact the only line relating to audio is the `sh: 1: pactl: not found` error. So I suspect zoom is not even loading alsalib. Do you have on board speakers or sound card you can use for testing zoom first before introducing bluetooth? (make sure pulseaudio is not running and zoom really is using alsalib directly).

Also in your .asoundrc:
```
pcm.!default "bluealsa"
...
defaults.bluealsa.profile "a2dp"
```
Tells alsa applications to use bluealsa for both capture and playback - but I think you need to use profile `sco` to achieve that (I could be wrong, but AFAIK headset microphones require HSP or HFP, and do not allow A2DP at the same time as HFP/HSP). If you connect the headphones then run
```
bluealsa-aplay -L
```
That will list the playback and capture PCMs available. Is there an A2DP capture PCM device?

dkwo, 2020-10-27:
Zoom works fine with onboard speakers/mic and also with headphones/mic connected to audio jack.
I have no pulseaudio installed, just alsa.
```
$ bluealsa-aplay -L
bluealsa:SRV=org.bluealsa,DEV=...,PROFILE=sco
    HD1 M2 AEBT, trusted audio-card, playback
    SCO (CVSD): S16_LE 1 channel 8000 Hz
bluealsa:SRV=org.bluealsa,DEV=...,PROFILE=sco
    HD1 M2 AEBT, trusted audio-card, capture
    SCO (CVSD): S16_LE 1 channel 8000 Hz
bluealsa:SRV=org.bluealsa,DEV=...,PROFILE=a2dp
    HD1 M2 AEBT, trusted audio-card, playback
    A2DP (SBC): S16_LE 2 channels 48000 Hz
```
If I replace `a2dp` with `sco` in that line of .asoundrc, I still cannot get bluetooth headphones to work in zoom.

borine, 2020-10-27:
what is your command line for starting bluealsa ?

dkwo, 2020-10-28:
I start it on the host by
```
sudo sv start bluetoothd
sudo sv start bluez-alsa
```
and in the host I get
```
echo 'info ...' | bluetoothctl
Agent registered
[CHG] Controller ... Pairable: yes
[HD1 M2 AEBT]# info ...
Device ... (public)
        Name: HD1 M2 AEBT
        Alias: HD1 M2 AEBT
        Class: 0x00240404
        Icon: audio-card
        Paired: yes
        Trusted: yes
        Blocked: no
        Connected: yes
        LegacyPairing: no
        UUID: Headset                   (00001108-0000-1000-8000-00805f9b34fb)
        UUID: Audio Sink                (0000110b-0000-1000-8000-00805f9b34fb)
        UUID: A/V Remote Control Target (0000110c-0000-1000-8000-00805f9b34fb)
        UUID: A/V Remote Control        (0000110e-0000-1000-8000-00805f9b34fb)
        UUID: Handsfree                 (0000111e-0000-1000-8000-00805f9b34fb)
        UUID: PnP Information           (00001200-0000-1000-8000-00805f9b34fb)
        UUID: Vendor specific           (1ddce62a-ecb1-4455-8153-0743c87aec9f)
        Modalias: bluetooth:v0082p004Bd0103
```
I noticed in my journal (probably irrelevant, as everything works outsize of Zoom)
`bluetoothd[4787]: src/profile.c:ext_io_disconnected() Unable to get io data for Hands-Free Voice gateway: getpeername: Socket not connected (107)`

borine, 2020-10-28:
Looking at the bluetoothctl output, I see that your headset microphone in indeed only for SCO (HSP/HFP) profiles. So If you want to use the headset microphone with bluealsa you will have to set
```
defaults.bluealsa.profile sco
```

To test this I would recommend to try HSP first, as this is the simpler protocol. HFP is more closely tied to 'phone usage, and some devices require to use `ofono` to provide complete protocol implementation, while others work happily with bluealsa's partial implementation. So start bluealsa with:
```shell
bluealsa -p hsp-ag
```

There is a limitation of bluealsa's SCO implementation that may still prevent Zoom from working correctly. Some applications open and close the ALSA PCM device multiple times in quick succession as part of their initialization. While bluealsa can handle this for A2DP streams, for SCO it can cause the bluealsa service to fail. See https://github.com/Arkq/bluez-alsa/issues/371#issuecomment-691707829

If Zoom is such an application, then SCO is not an option. The best you could do is to use A2DP for the headphone speakers, and use the on-board microphone for audio capture. To set up the default PCM for this, use something like the following .asoundrc:

```
pcm.!default {
	type asym
	playback "bluealsa"
	capture "plughw:0,0"
}
```

dkwo, 2020-10-30:
I'm slightly confused: if I change `.asoundrc` as suggested, start `bluetoothd` and then (as my user, who is in `audio` group) do
`bluealsa -p hsp-ag`
I get
```
bluealsa -p hsp-ag
bluealsa: D: bluez.c:785: Creating hands-free profile object: /org/bluez/HSP/AudioGateway
bluealsa: D: bluez.c:719: Registering hands-free profile: /org/bluez/HSP/AudioGateway
bluealsa: D: main.c:390: Acquiring D-Bus service name: org.bluealsa
bluealsa: D: main.c:395: Starting main dispatching loop
bluealsa: E: Couldn't acquire D-Bus name: org.bluealsa
bluealsa: D: main.c:398: Exiting main loop
```
and I cannot even use `bluetoothctl` to connect to the headphones, since `bluealsa` is not started.
What am I doing wrong?

borine, 2020-10-30:
The bluealsa daemon does not need to be in group audio. It needs permission to take ownership of the D-Bus service name "org.bluealsa" - permission for that is granted in the file `/etc/dbus-1/system.d/bluealsa.conf`. The bluez-alsa project default  is to use `root`, but some distributions create a service account specifically for bluealsa to use. According to https://github.com/Arkq/bluez-alsa/issues/382#issuecomment-701227353 you can set the bluealsa command line on Void Linux by editing the file `/etc/sv/bluez-alsa/conf`. Try `"OPTS=-p hsp-ag"` that way you do not need to run it manually.

dkwo, 2020-10-30:
Thanks, I should have known that..
With `OPTS="-p hsp-ag"`, the SCO profile gets configured correctly,  but Zoom still seems unable to see it.
Probably not relevant, but in the journal I see
```
bluetoothd[2884]: src/service.c:btd_service_connect() a2dp-sink profile connect failed for ...: Protocol not available
```
And if I play music with it (which works), I see many instances of
```
kernel: [ 1711.592723] Bluetooth: hci0: SCO packet for unknown connection handle
0
```

As for the second option: is this what you mean?
```
pcm.!default {
        type asym
        playback "bluealsa"
        capture "plughw:0,0"
        defaults.bluealsa.interface "hci0"
        defaults.bluealsa.device "..."
        defaults.bluealsa.profile "a2dp"
}
```
Cause I cannot make it work, even outside Zoom, and I see
```bluetoothd[2884]: src/profile.c:ext_io_disconnected() Unable to get io data for Headset Voice gateway: getpeername: Socket not connected (107)```

borine, 2020-10-30:
> kernel: [ 1711.592723] Bluetooth: hci0: SCO packet for unknown connection handle

That looks like a kernel driver bug - searching for that error message shows lots of hits, but on a quick read of the first few I didn't see any definitive solution. If your bluetooth adapter uses a USB Bus for HCI (type  the command `hciconfig` to check) then you may have an issue with SCO routing in the hardware.

> As for the second option: is this what you mean?

No, sorry I should have been more careful when writing that. Like this:

```
defaults.bluealsa.device "..."
defaults.bluealsa.profile "a2dp"

pcm.!default {
        type asym
        playback.pcm "bluealsa"
        capture.pcm "plughw:0,0"
}
```

Make sure you return `/etc/sv/bluez-alsa/conf` to its original state to go back to a2dp playback support.

dkwo, 2020-10-30:
The second option works as expected with mpv and others, but not with Zoom, where the headphones do not appear.
I get in journal (probably irrelevant)
```
bluetoothd[22981]: src/profile.c:ext_io_disconnected() Unable to get io data for Hands-Free Voice gateway: getpeername: Socket not connected (107)
```
Thanks for helping, though it seems a problem specific to the way Zoom uses alsa.

Arkq, 2020-11-01:
> Is there an A2DP capture PCM device?

@borine, yes, there is, e.g. FastStream codec which has back channel for "high" quality microphone. There is also aptX Low Latency with optional bidirectional streaming. None of these are currently supported by bluez-alsa master branch.

And regarding this issue. I will try to test bluez-alsa with e.g. Skype or something similar, because I've never used bluez-alsa with voice communicators. Ofono is also out of my testing scope...

dkwo, 2021-07-16:
I think this is not an issue with bluez-alsa, so I'm closing it.
Thanks for helping, I learned a few things from your comments.

closed 2021-07-16 by dkwo