arkq/bluez-alsa #392: list dependencies

NikZak, 2020-10-28:
Thanks for the great product! 

The readme file contains compilation instructions but does not have dependencies. Compilation was quite tedious without knowing necessary dependencies. Could you please add them to readme file? Here is a list that helped me for Debian Buster:
```
sudo apt install bluez libbluetooth-dev dbus libdbus-glib-1-dev libsbc-dev libfdk-aac-dev
```
The last one was especially painful as it is located in the multiverse repo. In case anyone needs it for Buster:
```
echo "deb http://www.deb-multimedia.org buster main non-free" >> /etc/apt/sources.list \ 
&& apt-key adv --keyserver hkp://keyserver.ubuntu.com:80 --recv-keys 5C808C2B65558117 \
&& apt-get update 
```

borine, 2020-10-28:
There is a list of dependencies, with package names for selected distributions, on the wiki here:
https://github.com/Arkq/bluez-alsa/wiki/Installation-from-source

Feel free to add extra detail - it is a public wiki.

NikZak, 2020-10-28:
Missed the wiki. Thanks

closed 2020-10-28 by NikZak