arkq/bluez-alsa #394: Help with "Couldn't get BlueALSA PCM: PCM not found" error

geoffr98, 2020-11-15:
When I try to play audio I am getting a "Couldn't get BlueALSA PCM: PCM not found" error. Can someone point me in the right direction.

1. installed from the latest git
2. `sudo cp -r /usr/lib/arm-linux-gnueabihf/alsa-lib/libasound_module_pcm_bluealsa.* /usr/lib/aarch64-linux-gnu/alsa-lib/.`
3. ` cat .asoundrc 
pcm.mid {
        type plug
        slave {
            pcm {
                type bluealsa
                device BB:F8:BE:79:40:C0 
                profile "a2dp"
            }
        }
        hint {
            show on
            description "MID"
        }
     }
`

geoffr98, 2020-11-15:
```
$ aplay -D mid /usr/share/sounds/alsa/Front_Center.wav
ALSA lib ../../../src/asound/bluealsa-pcm.c:1006:(_snd_pcm_bluealsa_open) Couldn't get BlueALSA PCM: PCM not found
aplay: main:788: audio open error: No such device
```

borine, 2020-11-16:
What is the output of
```bluealsa-aplay -l```

geoffr98, 2020-11-16:
$ sudo bluealsa -p a2dp-sink -p hfp-hf -p hfp-ag -p hsp-hs &

$ bluealsa-aplay -l
**** List of PLAYBACK Bluetooth Devices ****
hci0: B8:F8:BE:79:40:C0 [BCS-700], trusted audio-card
  SCO (CVSD): S16_LE 1 channel 8000 Hz
**** List of CAPTURE Bluetooth Devices ****
hci0: B8:F8:BE:79:40:C0 [BCS-700], trusted audio-card
  SCO (CVSD): S16_LE 1 channel 8000 Hz

$ cat .asoundrc
pcm.mid {
        type plug
        slave {
            pcm {
                type bluealsa
                device B8:F8:BE:79:40:C0
                profile "a2dp"
            }
        }
        hint {
            show on
            description "Marshall MID"
        }
     }
$ aplay -D mid /usr/share/sounds/alsa/Front_Center.wav
ALSA lib ../../../src/asound/bluealsa-pcm.c:1006:(_snd_pcm_bluealsa_open) Couldn't get BlueALSA PCM: PCM not found
aplay: main:788: audio open error: No such device

borine, 2020-11-16:
> **** List of PLAYBACK Bluetooth Devices ****
hci0: B8:F8:BE:79:40:C0 [BCS-700], trusted audio-card
SCO (CVSD): S16_LE 1 channel 8000 Hz

You have no A2DP playback devices - only SCO. You need to run the bluealsa daemon with `-p a2dp-source` to get A2DP playback function.

geoffr98, 2020-11-16:
Great - I can now play audio, but I no longer have a record device - how do I also get a record device as well?

How should I set the device in the bluetooth manager?  (Headset / Handsfree / Audiosink)

borine, 2020-11-16:
> how do I also get a record device as well?

Just add all the profiles you need to the bluealsa command line. It depends on what you are trying to build. Do you need to record A2DP? if so, use:
```
bluealsa -p a2dp-sink -p a2dp-source
```

If you just want to connect to a bluetooth headset, try:
```
bluealsa -p a2dp-source -p hsp-ag
```

It rarely makes sense to use both hsp-ag and hsp-hs (or hfp-ag and hfp-hf) on the same machine. Use `-ag` to connect to headphones, use `-hs (or -hf`) to connect to the voice profile of a mobile 'phone. Note that hsp-hs is now deprecated by bluez (and the Bluetooth Consortium) so you may need to patch bluez sources to get it to work. hsp-ag should work fine.

geoffr98, 2020-11-16:
Thanks for all the help - I think I'm getting closer...

Using the following command I was able to see a playback and record device:
sudo bluealsa -p a2dp-source -p hsp-ag &
(Using -p a2dp-sink -p a2dp-source) did not give me both playback and record
I have been able to play audio, but am not sure how to record audio - what I have tried gives errors...
do I need to update my .asoundrc?

$ arecord -D bluealsa test.wav
ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find definition 'defaults.bluealsa.device'
ALSA lib conf.c:4568:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:5036:(snd_config_expand) Args evaluate error: No such file or directory
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM bluealsa
arecord: main:788: audio open error: No such file or directory

$ arecord -f S16_LE -D bluealsa:HCI=hci0,DEV=B8:F8:BE:79:40:C0,PROFILE=sco mic.wav
ALSA lib conf.c:4898:(parse_args) Unknown parameter HCI
ALSA lib conf.c:5031:(snd_config_expand) Parse arguments error: No such file or directory
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM bluealsa:HCI=hci0,DEV=B8:F8:BE:79:40:C0,PROFILE=sco
arecord: main:788: audio open error: No such file or directory

borine, 2020-11-16:
Have a look at the wiki page here: https://github.com/Arkq/bluez-alsa/wiki/Using-the-bluealsa-ALSA-pcm-plugin
Hopefully that will help you get started with using the bluealsa plugin.
>(Using -p a2dp-sink -p a2dp-source) did not give me both playback and record

I'm guessing that whatever you are recording from does not provide an A2DP capture stream - probably its a headset/headphones or BT speaker ? 
```
bluealsa-aplay -L
```
will tell you all the available PCM devices.
With most devices it is not possible to use both A2DP and HSP/HFP at the same time (a limitation of the hardware, nothing to do with bluealsa or bluez). If you want to record from the microphone you will most likely have to play back using the SCO device as well.

geoffr98, 2020-11-16:
Thanks for the link to the wiki.  I have tried a number of things, but still don't have things working - I must be missing something. Yes this is a BT headset .

$ sudo bluealsa -p a2dp-source -p hsp-ag &
$ bluealsa-aplay -L
bluealsa:SRV=org.bluealsa,DEV=B8:F8:BE:79:40:C0,PROFILE=a2dp
    BCS-700, trusted audio-card, playback
    A2DP (SBC): S16_LE 2 channels 48000 Hz
bluealsa:SRV=org.bluealsa,DEV=B8:F8:BE:79:40:C0,PROFILE=sco
    BCS-700, trusted audio-card, playback
    SCO (CVSD): S16_LE 1 channel 8000 Hz
bluealsa:SRV=org.bluealsa,DEV=B8:F8:BE:79:40:C0,PROFILE=sco
    BCS-700, trusted audio-card, capture
    SCO (CVSD): S16_LE 1 channel 8000 Hz

~$ aplay  -D bluealsa:B8:F8:BE:79:40:C0 Front_Center.wav       << works 
~$ aplay  -D bluealsa:B8:F8:BE:79:40:C0,a2dp Front_Center.wav    << works (as expected)

$ aplay  -D bluealsa:B8:F8:BE:79:40:C0,sco Front_Center.wav   << does not work
Playing WAVE 'Front_Center.wav' : Signed 16 bit Little Endian, Rate 48000 Hz, Mono

Since the record only seems to have SCO (as you had mentioned I have tried the following):
arecord -D bluealsa:B8:F8:BE:79:40:C0,sco -d 3 test.wav
however my terminal crashes/hangs  when i do this and I need to disconnect and reconnect the headset in order to play audio again.

geoffr98, 2020-11-17:
I think this may explain why the recoding isn't working, but I still don't know why..

$ sudo systemctl status bluetoo*
● bluetooth.target - Bluetooth
   Loaded: loaded (/lib/systemd/system/bluetooth.target; static; vendor preset: enabled)
   Active: active since Thu 2017-12-14 22:10:01 UTC; 2 years 11 months ago
     Docs: man:systemd.special(7)

● bluetooth.service - Bluetooth service
   Loaded: loaded (/lib/systemd/system/bluetooth.service; enabled; vendor preset: enabled)
   Active: active (running) since Thu 2017-12-14 22:10:01 UTC; 2 years 11 months ago
     Docs: man:bluetoothd(8)
 Main PID: 1938 (bluetoothd)
   Status: "Running"
    Tasks: 1 (limit: 4915)
   CGroup: /system.slice/bluetooth.service
           └─1938 /usr/lib/bluetooth/bluetoothd

Nov 16 23:50:08 linaro-alip bluetoothd[1938]: Can't open input device: No such file or directory (2)
Nov 16 23:50:08 linaro-alip bluetoothd[1938]: AVRCP: failed to init uinput for B8:F8:BE:79:40:C0
Nov 16 23:50:14 linaro-alip bluetoothd[1938]: /org/bluez/hci0/dev_B8_F8_BE_79_40_C0/fd6: fd(35) ready
Nov 16 23:52:30 linaro-alip bluetoothd[1938]: Unable to get io data for Headset Voice gateway: getpeername: Transport endpoint is not connected (1
Nov 16 23:52:37 linaro-alip bluetoothd[1938]: Can't open input device: No such file or directory (2)
Nov 16 23:52:37 linaro-alip bluetoothd[1938]: AVRCP: failed to init uinput for B8:F8:BE:79:40:C0
Nov 17 02:46:28 linaro-alip bluetoothd[1938]: /org/bluez/hci0/dev_B8_F8_BE_79_40_C0/fd7: fd(35) ready
Nov 17 02:47:23 linaro-alip bluetoothd[1938]: Can't open input device: No such file or directory (2)
Nov 17 02:47:23 linaro-alip bluetoothd[1938]: AVRCP: failed to init uinput for B8:F8:BE:79:40:C0
Nov 17 02:47:25 linaro-alip bluetoothd[1938]: Unable to get io data for Headset Voice gateway: getpeername: Transport endpoint is not connected (1
l

borine, 2020-11-17:
It looks like your bluetooth adapter has a problem with SCO transports -what is the output of
`hciconfig -a`

Arkq, 2020-11-17:
@geoffr98 , @borine do you think, that a brief note in the troubleshooting section from PR #395 might be helpful for others in a case of "PCM not found error" (it's not the first "PCM not found issue" and probably not the last :D)?

geoffr98, 2020-11-17:
I rebooted the board this morning and it still shows the BT service running for 3 years.. not sure why..

Here is the output from this morning:

$ sudo bluealsa -p a2dp-source -p hsp-ag &
[1] 2344

$ export LIBASOUND_THREAD_SAFE=0

$ bluealsa-aplay -L
bluealsa:SRV=org.bluealsa,DEV=B8:F8:BE:79:40:C0,PROFILE=sco
    BCS-700, trusted audio-card, playback
    SCO (CVSD): S16_LE 1 channel 8000 Hz
bluealsa:SRV=org.bluealsa,DEV=B8:F8:BE:79:40:C0,PROFILE=sco
    BCS-700, trusted audio-card, capture
    SCO (CVSD): S16_LE 1 channel 8000 Hz
bluealsa:SRV=org.bluealsa,DEV=B8:F8:BE:79:40:C0,PROFILE=a2dp
    BCS-700, trusted audio-card, playback
    A2DP (SBC): S16_LE 2 channels 48000 Hz

$ aplay -D bluealsa:B8:F8:BE:79:40:C0,a2dp Front_Center.wav          << worked as expected
Playing WAVE 'Front_Center.wav' : Signed 16 bit Little Endian, Rate 48000 Hz, Mono

$ date
Tue 17 Nov 2020 01:37:23 PM UTC

$ hciconfig -a
hci0:   Type: Primary  Bus: SMD
        BD Address: 02:00:0C:03:86:CA  ACL MTU: 1024:7  SCO MTU: 50:8
        UP RUNNING PSCAN 
        RX bytes:1717 acl:54 sco:0 events:444 errors:0
        TX bytes:66024 acl:171 sco:0 commands:260 errors:0
        Features: 0xff 0xfe 0x8f 0xfe 0xd8 0x3f 0x5b 0x87
        Packet type: DM1 DM3 DM5 DH1 DH3 DH5 HV1 HV2 HV3 
        Link policy: RSWITCH HOLD SNIFF 
        Link mode: SLAVE ACCEPT 
        Name: 'linaro-alip'
        Class: 0x180000
        Service Classes: Capturing, Object Transfer
        Device Class: Miscellaneous, 
        HCI Version: 4.1 (0x7)  Revision: 0x0
        LMP Version: 4.1 (0x7)  Subversion: 0x7d3
        Manufacturer: Qualcomm (29)

$ sudo systemctl status bluetoo*
● bluetooth.service - Bluetooth service
   Loaded: loaded (/lib/systemd/system/bluetooth.service; enabled; vendor preset: enabled)
   Active: active (running) since Thu 2017-12-14 22:10:02 UTC; 2 years 11 months ago
     Docs: man:bluetoothd(8)
 Main PID: 1989 (bluetoothd)
   Status: "Running"
    Tasks: 1 (limit: 4915)
   CGroup: /system.slice/bluetooth.service
           └─1989 /usr/lib/bluetooth/bluetoothd

Dec 14 22:10:02 linaro-alip bluetoothd[1989]: Bluetooth management interface 1.14 initialized
Dec 14 22:10:02 linaro-alip bluetoothd[1989]: Sap driver initialization failed.
Dec 14 22:10:02 linaro-alip bluetoothd[1989]: sap-server: Operation not permitted (1)
Dec 14 22:10:54 linaro-alip bluetoothd[1989]: a2dp-sink profile connect failed for B8:F8:BE:79:40:C0: Protocol not available
Nov 17 13:36:00 linaro-alip bluetoothd[1989]: Endpoint registered: sender=:1.27 path=/org/bluez/hci0/A2DP/SBC/Source/1
Nov 17 13:36:00 linaro-alip bluetoothd[1989]: Endpoint registered: sender=:1.27 path=/org/bluez/hci0/A2DP/SBC/Source/2
Nov 17 13:36:31 linaro-alip bluetoothd[1989]: Endpoint registered: sender=:1.27 path=/org/bluez/hci0/A2DP/SBC/Source/3
Nov 17 13:36:31 linaro-alip bluetoothd[1989]: Can't open input device: No such file or directory (2)
Nov 17 13:36:31 linaro-alip bluetoothd[1989]: AVRCP: failed to init uinput for B8:F8:BE:79:40:C0
Nov 17 13:37:01 linaro-alip bluetoothd[1989]: /org/bluez/hci0/dev_B8_F8_BE_79_40_C0/fd0: fd(35) ready

● bluetooth.target - Bluetooth
   Loaded: loaded (/lib/systemd/system/bluetooth.target; static; vendor preset: enabled)
   Active: active since Thu 2017-12-14 22:10:02 UTC; 2 years 11 months ago
     Docs: man:systemd.special(7)
 

$ aplay -D bluealsa:B8:F8:BE:79:40:C0,sco Front_Center.wav   << did not work
Playing WAVE 'Front_Center.wav' : Signed 16 bit Little Endian, Rate 48000 Hz, Mono

$

borine, 2020-11-17:
> hci0: Type: Primary Bus: SMD

I'm not familiar with bus type SMD, so don't know is SCO is ever an issue with these adapter types. Its possible that the problem lies with firmware / driver(s) as many issues relating to SCO have eventually been fixed by updating the system. Are you using a recent kernel, and have you researched issues with this Qualcomm adapter? If possible, you could try with a USB adapter to see if the system will deliver SCO traffic with that - if so then your symptoms are definitely caused by the SMD adapter.

@Arkq 
> do you think, that a brief note in the troubleshooting section from PR #395 might be helpful for others

Yes :smile:

I have considered writing some notes as a wiki page based on previous issues, but never got round to it and am unlikely to have any free time now.

geoffr98, 2020-11-18:
I just tried with a USB adapter and it is working well with SCO now.  I'll continue with this USB adapter rather than banging my head with the onboard BT. Thanks for your help (even if we didn't find the root cause I still have a solution).

closed 2020-11-18 by geoffr98

shrinivas510, 2025-02-04:
Hi,
I have bluealsa service running on yocto linux and able to connect Bluetooth earphone. I can able to play and record using below mentioned commands,

```
aplay -D bluealsa:SRV=org.bluealsa,DEV=xx:xx:xx:xx:xx:xx,PROFILE=a2dp /usr/share/sounds/alsa/Front_Center.wav
arecord -D bluealsa:xx:xx:xx:xx:xx:xx,sco -d 5 test.wav
```

While recording/playing audio I'm not getting any signals on PCM lines, but on uart.  How can I test PCM (want to check signals on PCM lines)?

Note:
this is what /usr/lib/systemd/system/bluealsa.service have (logs are trimmed)
```
~#cat /usr/lib/systemd/system/bluealsa.service
.
.
.
ExecStart=/usr/bin/bluealsa -p a2dp-source -p a2dp-sink -p hfp-hf -p hfp-ag -p hsp-hs
.
.
.
```

and this is output of bluealsa-aplay -l
```
~#bluealsa-aplay -l
**** List of PLAYBACK Bluetooth Devices ****
hci0: xx:xx:xx:xx:xx:xx [realme Buds Wireless 2], trusted audio-headset
  A2DP (SBC): S16_LE 2 channels 48000 Hz
  SCO (CVSD): S16_LE 1 channel 8000 Hz
**** List of CAPTURE Bluetooth Devices ****
hci0: xx:xx:xx:xx:xx:xx [realme Buds Wireless 2], trusted audio-headset
  SCO (CVSD): S16_LE 1 channel 8000 Hz
```

arkq, 2025-02-04:
> While recording/playing audio I'm not getting any signals on PCM lines, but on uart. How can I test PCM (want to check signals on PCM lines)?

The is no support for reading/writing PCM data to BT chip's PCM lines in bluez-alsa master. This might be very SoC specific so you have to do it by yourself according to your's PCB wiring.