arkq/bluez-alsa #398: bluez-alsa exits with any attempt to use aplay or arecord

xyzzy42, 2020-11-28:
Device pairs ok, detected by bluealsa, lists ok in bluealsa-aplay -l, but attempted to play causes bluealsa to exit.  Using:
bluealsa v3.0.0-15-gbbcfae4
alsa-lib 1.2.3.2
bluez 5.55
Linux kernel 5.9.8

I've tried a different BT speaker, in addition to the headphones show here, and other than different bt features on the device the result is the same.  arecord is basically the same too.

With pulseaudio-bluetooth (disabled for this test) a2dp does work.  But HFP doesn't.

Here is aplay command and result:
```
[~]$ aplay -D "bluealsa:SRV=org.bluealsa,DEV=2C:41:A1:03:4A:DF,PROFILE=a2dp" ~/test.wav
D: bluealsa-pcm.c:1053: Getting BlueALSA PCM: PLAYBACK 2C:41:A1:03:4A:DF a2dp
D: bluealsa-pcm.c:921: /org/bluealsa/hci0/dev_2C_41_A1_03_4A_DF/a2dpsrc/sink: Setting constraints
Playing WAVE '/home/tpiepho/atsc/test.wav' : Signed 16 bit Little Endian, Rate 48000 Hz, Stereo
D: bluealsa-pcm.c:445: /org/bluealsa/hci0/dev_2C_41_A1_03_4A_DF/a2dpsrc/sink: Initializing HW
D: bluealsa-pcm.c:452: /org/bluealsa/hci0/dev_2C_41_A1_03_4A_DF/a2dpsrc/sink: Couldn't open PCM: Remote peer disconnected
aplay: set_params:1407: Unable to install hw params:
ACCESS:  RW_INTERLEAVED
FORMAT:  S16_LE
SUBFORMAT:  STD
SAMPLE_BITS: 16
FRAME_BITS: 32
CHANNELS: 2
RATE: 48000
PERIOD_TIME: [85333 85334)
PERIOD_SIZE: (4095 4096)
PERIOD_BYTES: (16380 16381)
PERIODS: 6
BUFFER_TIME: (511895 511896)
BUFFER_SIZE: 24571
BUFFER_BYTES: 98284
TICK_TIME: 0
D: bluealsa-pcm.c:433: /org/bluealsa/hci0/dev_2C_41_A1_03_4A_DF/a2dpsrc/sink: Closing
```
Here is the output from bluealsa after starting aplay:
```
bluealsa: D: dbus.c:59: Called: org.bluealsa.Manager1.GetPCMs() on /org/bluealsa
bluealsa: D: dbus.c:59: Called: org.bluealsa.PCM1.Open() on /org/bluealsa/hci0/dev_2C_41_A1_03_4A_DF/a2dpsrc/sink
bluealsa: D: ba-transport.c:880: New transport: 20 (MTU: R:672 W:895)
bluealsa: D: bluez.c:1110: Signal: org.freedesktop.DBus.Properties.PropertiesChanged(): org.bluez.MediaTransport1: State
bluealsa: D: ba-transport.c:699: Starting transport: A2DP Source (AAC)
bluealsa: D: ba-transport.c:1044: Created new thread [ba-a2dp-aac]: A2DP Source (AAC)
bluealsa: D: ba-transport.c:1018: Closing PCM: 10
bluealsa: E: Couldn't acquire D-Bus name: org.bluealsa
bluealsa: D: main.c:398: Exiting main loop
```

Here is the output from bluealsa before aplay is started:
```
[~]# bluealsa
bluealsa: D: ofono.c:628: Checking oFono service presence
bluealsa: D: bluez.c:497: Creating media endpoint object: /org/bluez/hci0/A2DP/LDAC/Source/1
bluealsa: D: bluez.c:413: Registering media endpoint: /org/bluez/hci0/A2DP/LDAC/Source/1
bluealsa: D: bluez.c:497: Creating media endpoint object: /org/bluez/hci0/A2DP/LDAC/Source/2
bluealsa: D: bluez.c:413: Registering media endpoint: /org/bluez/hci0/A2DP/LDAC/Source/2
bluealsa: D: bluez.c:497: Creating media endpoint object: /org/bluez/hci0/A2DP/AAC/Source/1
bluealsa: D: bluez.c:413: Registering media endpoint: /org/bluez/hci0/A2DP/AAC/Source/1
bluealsa: D: bluez.c:497: Creating media endpoint object: /org/bluez/hci0/A2DP/AAC/Source/2
bluealsa: D: bluez.c:413: Registering media endpoint: /org/bluez/hci0/A2DP/AAC/Source/2
bluealsa: D: bluez.c:497: Creating media endpoint object: /org/bluez/hci0/A2DP/MPEG/Source/1
bluealsa: D: bluez.c:413: Registering media endpoint: /org/bluez/hci0/A2DP/MPEG/Source/1
bluealsa: D: bluez.c:497: Creating media endpoint object: /org/bluez/hci0/A2DP/MPEG/Source/2
bluealsa: D: bluez.c:413: Registering media endpoint: /org/bluez/hci0/A2DP/MPEG/Source/2
bluealsa: D: bluez.c:497: Creating media endpoint object: /org/bluez/hci0/A2DP/SBC/Source/1
bluealsa: D: bluez.c:413: Registering media endpoint: /org/bluez/hci0/A2DP/SBC/Source/1
bluealsa: D: bluez.c:497: Creating media endpoint object: /org/bluez/hci0/A2DP/SBC/Source/2
bluealsa: D: bluez.c:413: Registering media endpoint: /org/bluez/hci0/A2DP/SBC/Source/2
bluealsa: D: bluez.c:792: Creating hands-free profile object: /org/bluez/HSP/AudioGateway
[... 82 lines cut ...]
bluealsa: D: ba-rfcomm.c:736: RFCOMM: HFP Audio Gateway setting codec: mSBC
bluealsa: D: ba-rfcomm.c:107: Sending AT message: RESP: command:+BCS, value:2
bluealsa: D: at.c:161: AT message: SET: command:+BCS, value:2
bluealsa: D: ba-rfcomm.c:107: Sending AT message: RESP: command:(null), value:OK
```
And for good measure, `bluealsa-aplay -Ll`:
```
[~]$ bluealsa-aplay -Ll
bluealsa:SRV=org.bluealsa,DEV=2C:41:A1:03:4A:DF,PROFILE=sco
    Bose QC35 II, trusted audio-card, playback
    SCO (mSBC): S16_LE 1 channel 16000 Hz
bluealsa:SRV=org.bluealsa,DEV=2C:41:A1:03:4A:DF,PROFILE=sco
    Bose QC35 II, trusted audio-card, capture
    SCO (mSBC): S16_LE 1 channel 16000 Hz
bluealsa:SRV=org.bluealsa,DEV=2C:41:A1:03:4A:DF,PROFILE=a2dp
    Bose QC35 II, trusted audio-card, playback
    A2DP (AAC): S16_LE 2 channels 48000 Hz
**** List of PLAYBACK Bluetooth Devices ****
hci0: 2C:41:A1:03:4A:DF [Bose QC35 II], trusted audio-card
  SCO (mSBC): S16_LE 1 channel 16000 Hz
  A2DP (AAC): S16_LE 2 channels 48000 Hz
**** List of CAPTURE Bluetooth Devices ****
hci0: 2C:41:A1:03:4A:DF [Bose QC35 II], trusted audio-card
  SCO (mSBC): S16_LE 1 channel 16000 Hz
```

Arkq, 2020-11-28:
> bluealsa: D: bluez.c:1110: Signal: org.freedesktop.DBus.Properties.PropertiesChanged(): org.bluez.MediaTransport1: State
> bluealsa: D: ba-transport.c:699: Starting transport: A2DP Source (AAC)
> bluealsa: D: ba-transport.c:1044: Created new thread [ba-a2dp-aac]: A2DP Source (AAC)
> bluealsa: D: ba-transport.c:1018: Closing PCM: 10
> bluealsa: E: Couldn't acquire D-Bus name: org.bluealsa   <---- this line
> bluealsa: D: main.c:398: Exiting main loop

There was/is a similar issue (unresolved though). If you've got some time during this weekend, we could try to investigate it further. But I'm afraid it will be required to communicate in a more efficient way than github issue comments. There is a #bluealsa channel on Freenode (you can use web IRC client, e.g. https://webchat.freenode.net). I should be available for debugging around 7:00pm GMT. If such time is not covenant for you please let me know the time range of your availability.

xyzzy42, 2020-11-28:
Looks like I was too late for IRC.  I can join tomorrow some time too.

I tried the same version of bluealsa on a pinebook pro.  And after eventually getting the same headphones paired, it worked there.  Different BT adapter, aarch64 instead of x86_64, alsa-lib 1.2.4 instead of 1.2.3.2, same bluez.

The point where the bluealsa log diverges is the line above the dbus error.  When working, I get `PCM resumed: 10`, `PCM drained: 10`, `PCM dropped: 10`, `PCM dropped: 10` (again), `PCM has been closed: 10`, and then finally `Closing PCM: 10`.

xyzzy42, 2020-11-29:
Moved the USB adapter to the pinebook pro and it works there too.

xyzzy42, 2020-11-29:
Here is a dbus-monitor log of aplay starting.  Seems like it worked fine up to to the point bluealsa was told it lost the name org.bluealsa for no apparent cause.
bluealsa: 1.341
aplay: 1.343
bluetoothd: 1.176

[monitor-aplay.log](https://github.com/Arkq/bluez-alsa/files/5611797/monitor-aplay.log)

xyzzy42, 2020-11-29:
The cause has been determined.  It's selinux preventing the passing though D-Bus of the FDs used for the PCM stream from bluealsa to aplay.  I suspect the policies that are currently shipping with Fedora have a flaw w.r.t. dbus-broker and FD passing, but I'm not sure about that.  This is probably the same problem as in #361.

A simple test is to put selinux into permissive mode, `sudo setenforce 0`, and then it should work.

To continue to use selinux, a policy can be added to allow the access needed:
```sh
audit2allow -M dbusbroker-unix-stream <<EOF
type=AVC msg=audit(1606641094.896:1635): avc:  denied  { read write } for comm="dbus-broker" dev="sockfs" scontext=system_u:system_r:system_dbusd_t:s0-s0:c0.c1023 tcontext=unconfined_u:unconfined_r:unconfined_t:s0-s0:c0.c1023 tclass=unix_stream_socket permissive=0
EOF

sudo semodule -X 300 -i dbusbroker-unix-stream.pp
```

moozhub, 2020-11-30:
@xyzzy42 Thanks for the info on this, I felt earlier it may be related to selinux, but there's nothing in my audit log complaining of any denials and the solution you gave didn't correct this for me, only after switching to permissive was I able to progress. Did you have to add a new audit rule to make this audit message pop up in ausearch / audit.log?

xyzzy42, 2020-11-30:
I also go not audit log entries, which is something I checked long before the actual solution was found.  It turns out they were getting filtered and didn't show up in the log.  Turn off the dontaudit rules with `sudo semodule -DB`.  Re-enable with `-B`.  You should be able to get the correct audit log entry this way.  The important event will happen just after aplay starts.

It could be that your distro is not using dbus-broker, but dbus-daemon, so that is why my rule doesn't work for you.

moozhub, 2020-11-30:
Yep.. that was it..albeit a little different from yours, but here's a working solution for Fedora 32.. not sure it will work with all F32 systems..

```
audit2allow -M dbusbroker-unix-stream <<EOF
type=AVC msg=audit(1606708289.655:998): avc:  denied  { write } for  pid=933 comm="dbus-broker" path="pipe:[161184]" dev="pipefs" ino=161184 scontext=system_u:system_r:system_dbusd_t:s0-s0:c0.c1023 tcontext=system_u:system_r:unconfined_service_t:s0 tclass=fifo_file permissive=0
type=AVC msg=audit(1606708507.687:1038): avc:  denied  { read write } for  pid=933 comm="dbus-broker" path="socket:[168088]" dev="sockfs" ino=168088 scontext=system_u:system_r:system_dbusd_t:s0-s0:c0.c1023 tcontext=system_u:system_r:unconfined_service_t:s0 tclass=unix_stream_socket permissive=0
EOF
```

xyzzy42, 2020-11-30:
Interesting, I'm using Fedora 32 also, selinux-policy-3.14.5-45.  The second audit looks like it should create the same rule, but the first one is new.  I wonder why I don't seem to need that.

moozhub, 2020-11-30:
Not entirely sure either, might have something to do with differences in configuration in /etc/alsa or /etc/dbus-1/system.d/bluealsa.conf in accordance to the walkthrough's in the wiki... There's also no path for sockfs in your policy, makes me wonder if mine would survive a reboot..

Arkq, 2021-07-14:
The issue seems to be resolved. I'm closing it, then.

closed 2021-07-14 by Arkq