arkq/bluez-alsa #402: getting core dump with bluealsa-aplay -L

gearhead, 2020-12-12:
This is on a RPi B3+ running Arch Linux aarch64. I have not yet tried it with armv7, but this definitely repeats on aarch64. 
I paired the headset and tested it with aplay, and it works, 
```
~# aplay -D bluealsa:DEV=70:BF:92:D8:82:84,PROFILE=a2dp /usr/share/sounds/alsa/Front_Center.wav
Playing WAVE '/usr/share/sounds/alsa/Front_Center.wav' : Signed 16 bit Little Endian, Rate 48000 Hz, Mono
```
but when I try to list, I get an error.
```
# bluealsa-aplay -L
bluealsa:SRV=org.bluealsa,DEV=70:BF:92:D8:82:84,PROFILE=a2dp
    Jabra Evolve2 65, trusted audio-card, playback
    A2DP (SBC): S16_LE 2 channels 48000 Hz
*** stack smashing detected ***: terminated
Aborted (core dumped)
```
This is with kernel 5.4.79-ARCH and this bluetooth stack
```
# pacman -Q | grep blue
bluez 5.55-1
bluez-alsa 3.0.0-1
bluez-libs 5.55-1
bluez-tools 0.2.0-5
bluez-utils 5.55-1
```

Arkq, 2020-12-12:
Please, try with the *latest* *master* branch. It's hard to tell from the
log you've posted, but I think that the latest commit fixes this issue.

closed 2020-12-31 by Arkq

aharth, 2021-12-20:
With the current Debian testing/Bookworm from https://raspi.debian.net/ on a Raspberry Pi 4, I get the same error message:

````
$ bluealsa-aplay --list-pcms
bluealsa:SRV=org.bluealsa,DEV=F8:DF:15:4D:AA:4A,PROFILE=sco
    SRS-XB10, trusted audio-card, playback
    SCO (CVSD): S16_LE 1 channel 8000 Hz
bluealsa:SRV=org.bluealsa,DEV=F8:DF:15:4D:AA:4A,PROFILE=sco
    SRS-XB10, trusted audio-card, capture
    SCO (CVSD): S16_LE 1 channel 8000 Hz
bluealsa:SRV=org.bluealsa,DEV=F8:DF:15:4D:AA:4A,PROFILE=a2dp
    SRS-XB10, trusted audio-card, playback
    A2DP (SBC): S16_LE 2 channels 48000 Hz
*** stack smashing detected ***: terminated
Aborted
pi@one:~$ bluealsa-aplay --list-devices
**** List of PLAYBACK Bluetooth Devices ****
hci0: F8:DF:15:4D:AA:4A [SRS-XB10], trusted audio-card
  SCO (CVSD): S16_LE 1 channel 8000 Hz
  A2DP (SBC): S16_LE 2 channels 48000 Hz
**** List of CAPTURE Bluetooth Devices ****
hci0: F8:DF:15:4D:AA:4A [SRS-XB10], trusted audio-card
  SCO (CVSD): S16_LE 1 channel 8000 Hz
*** stack smashing detected ***: terminated
Aborted
pi@one:~$ uname -a
Linux one 5.15.0-2-arm64 #1 SMP Debian 5.15.5-1 (2021-11-26) aarch64 GNU/Linux
pi@one:~$ dpkg -l | grep -e blue
ii  bluez                             5.61-1                            arm64        Bluetooth tools and daemons
ii  bluez-alsa-utils                  3.0.0-2                           arm64        Bluetooth Audio ALSA Backend (utils)
ii  bluez-firmware                    1.2-7                             all          Firmware for Bluetooth devices
ii  libasound2-plugin-bluez:arm64     3.0.0-2                           arm64        Bluetooth Audio ALSA Backend (plugins)
ii  libbluetooth-dev:arm64            5.61-1                            arm64        Development files for using the BlueZ Linux Bluetooth library
ii  libbluetooth3:arm64               5.61-1                            arm64        Library to use the BlueZ Linux Bluetooth stack
pi@one:~$
````

Sound output with both the A2DP and SCO profiles work using aplay.

aharth, 2021-12-21:
Thank you! I have filed a bug report for the Debian package, see https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=1002068.