arkq/bluez-alsa #403: bluealsa devices disappear with ofono

speedyseal, 2020-12-16:
If ofono is stopped and bluealsa is run with these params:
```
sudo /usr/bin/bluealsa --profile=a2dp-source --profile=a2dp-sink -p hfp-ag
```
I can see these devices
```
$ bluealsa-aplay  -L
bluealsa:SRV=org.bluealsa,DEV=...,PROFILE=sco
    headphones, trusted audio-card, playback
    SCO (mSBC): S16_LE 1 channel 16000 Hz
bluealsa:SRV=org.bluealsa,DEV=...,PROFILE=sco
    headphones, trusted audio-card, capture
    SCO (mSBC): S16_LE 1 channel 16000 Hz
bluealsa:SRV=org.bluealsa,DEV=...,PROFILE=a2dp
    headphones, trusted audio-card, playback
    A2DP (SBC): S16_LE 2 channels 48000 Hz
```

If ofono is running and I configure bluealsa like this:
```
sudo /usr/bin/bluealsa --profile=a2dp-source --profile=a2dp-sink -p hfp-ag
```
The SCO devices disappear. Is it supposed to work like this?
```
$ bluealsa-aplay  -L
bluealsa:SRV=org.bluealsa,DEV=...,PROFILE=a2dp
    headphones, trusted audio-card, playback
    A2DP (SBC): S16_LE 2 channels 48000 Hz
```

```
/usr/bin/bluealsa: D: ../../src/ofono.c:628: Checking oFono service presence
/usr/bin/bluealsa: W: Disabling native HFP support due to oFono service presence
/usr/bin/bluealsa: D: ../../src/bluez.c:490: Creating media endpoint object: /org/bluez/hci0/A2DP/SBC/Source/1
/usr/bin/bluealsa: D: ../../src/bluez.c:406: Registering media endpoint: /org/bluez/hci0/A2DP/SBC/Source/1
/usr/bin/bluealsa: D: ../../src/bluez.c:490: Creating media endpoint object: /org/bluez/hci0/A2DP/SBC/Source/2
/usr/bin/bluealsa: D: ../../src/bluez.c:406: Registering media endpoint: /org/bluez/hci0/A2DP/SBC/Source/2
/usr/bin/bluealsa: D: ../../src/bluez.c:490: Creating media endpoint object: /org/bluez/hci0/A2DP/SBC/Sink/1
/usr/bin/bluealsa: D: ../../src/bluez.c:406: Registering media endpoint: /org/bluez/hci0/A2DP/SBC/Sink/1
/usr/bin/bluealsa: D: ../../src/bluez.c:490: Creating media endpoint object: /org/bluez/hci0/A2DP/SBC/Sink/2
/usr/bin/bluealsa: D: ../../src/bluez.c:406: Registering media endpoint: /org/bluez/hci0/A2DP/SBC/Sink/2
/usr/bin/bluealsa: D: ../../src/main.c:390: Acquiring D-Bus service name: org.bluealsa
/usr/bin/bluealsa: D: ../../src/main.c:395: Starting main dispatching loop
/usr/bin/bluealsa: D: ../../src/dbus.c:59: Called: org.bluealsa.Manager1.GetPCMs() on /org/bluealsa
```

If I configure bluealsa to use ofono with ofono running
```
sudo /usr/bin/bluealsa --profile=a2dp-source --profile=a2dp-sink --profile=hfp-ofono
```
I get the sco profiles, but the codec is null -- is ofono not configured correctly?
```
$ bluealsa-aplay  -L
bluealsa:SRV=org.bluealsa,DEV=...,PROFILE=a2dp
    LE-headphones, trusted audio-card, playback
    A2DP (SBC): S16_LE 2 channels 44100 Hz
bluealsa:SRV=org.bluealsa,DEV=...,PROFILE=sco
    LE-headphones, trusted audio-card, playback
    SCO (<null>): S16_LE 1 channel 0 Hz
bluealsa:SRV=org.bluealsa,DEV=...,PROFILE=sco
    LE-headphones, trusted audio-card, capture
    SCO (<null>): S16_LE 1 channel 0 Hz
```

```
/usr/bin/bluealsa: D: ../../src/bluez.c:490: Creating media endpoint object: /org/bluez/hci0/A2DP/SBC/Source/1
/usr/bin/bluealsa: D: ../../src/bluez.c:406: Registering media endpoint: /org/bluez/hci0/A2DP/SBC/Source/1
/usr/bin/bluealsa: D: ../../src/bluez.c:490: Creating media endpoint object: /org/bluez/hci0/A2DP/SBC/Source/2
/usr/bin/bluealsa: D: ../../src/bluez.c:406: Registering media endpoint: /org/bluez/hci0/A2DP/SBC/Source/2
/usr/bin/bluealsa: D: ../../src/bluez.c:490: Creating media endpoint object: /org/bluez/hci0/A2DP/SBC/Sink/1
/usr/bin/bluealsa: D: ../../src/bluez.c:406: Registering media endpoint: /org/bluez/hci0/A2DP/SBC/Sink/1
/usr/bin/bluealsa: D: ../../src/bluez.c:490: Creating media endpoint object: /org/bluez/hci0/A2DP/SBC/Sink/2
/usr/bin/bluealsa: D: ../../src/bluez.c:406: Registering media endpoint: /org/bluez/hci0/A2DP/SBC/Sink/2
/usr/bin/bluealsa: D: ../../src/ofono.c:471: Registering oFono audio agent: /org/bluez/HFP/oFono
/usr/bin/bluealsa: D: ../../src/main.c:390: Acquiring D-Bus service name: org.bluealsa
/usr/bin/bluealsa: D: ../../src/main.c:395: Starting main dispatching loop
/usr/bin/bluealsa: D: ../../src/bluez.c:915: Signal: org.freedesktop.DBus.ObjectManager.InterfacesAdded()
/usr/bin/bluealsa: D: ../../src/dbus.c:59: Called: org.bluez.MediaEndpoint1.SetConfiguration() on /org/bluez/hci0/A2DP/SBC/Source/1
```

Arkq, 2020-12-16:
> 1. If ofono is stopped and bluealsa is run with these params
> but the system can't respond to call initiation from the hands free kit.

Built-in HFP support if very limited. BlueALSA does not implement whole HFP specification. There is only a vary basic audio support which may not work with all HFP-capable devices.

> 2. If ofono is running and I configure bluealsa like this [...] The SCO devices disappear.

In the bluealsa log you can see line "W: Disabling native HFP support due to oFono service presence". Because of BlueZ design, it is not possible to run more than one applications which implements HFP or HSP. Since oFono implements HFP, BlueALSA disables internal HFP support if oFono is running (detection during BlueALSA startup, so if you start BlueALSA, and *THEN* oFono, you will get two applications "stealing" SCO connection for each other). Otherwise neither BlueALSA nor oFono will have functional HFP support.

> 3. If I configure bluealsa to use ofono with ofono running
> I get the sco profiles, but the codec is null -- is ofono not configured correctly?

To be honest, I'm not sure... I'm not oFono user. But as far as I remember, I was able to use oFono only with phone, not with headset. When running with internal HFP, you will get codec selected immediately, because BlueALSA selects codec just after service level connection establishment. But HFP specification allows to select codec just before the audio transfer (afterwards it shall be persistent, though), so it is possible that oFono works in such a way - you will get codec selection prior to audio transfer.

speedyseal, 2020-12-16:
> To be honest, I'm not sure... I'm not oFono user. But as far as I remember, I was able to use oFono only with phone, not with headset. When running with internal HFP, you will get codec selected [...]

I need HFP control to run my audio path - I guess that means using ofono. So in order to play/record the audio - presumably using bluealsa - the 3rd option is what I should use? The system is unable to load the right codec (it seems)

```
aplay -D bluealsa:DEV=...,PROFILE=sco ~/testsco16.wav
D: ../../../src/asound/bluealsa-pcm.c:895: Getting BlueALSA PCM: PLAYBACK ... sco
D: ../../../src/asound/bluealsa-pcm.c:769: /org/bluealsa/hci0/dev_.../hfpag/sink: Setting constraints
Playing WAVE '/home/pi/testsco16.wav' : Signed 16 bit Little Endian, Rate 16000 Hz, Mono
D: ../../../src/asound/bluealsa-pcm.c:396: /org/bluealsa/hci0/dev_.../hfpag/sink: Initializing HW
D: ../../../src/asound/bluealsa-pcm.c:403: /org/bluealsa/hci0/dev_.../hfpag/sink: Couldn't **open PCM: HFP audio codec not selected
aplay: set_params:1403: Unable to install hw params:**
ACCESS:  RW_INTERLEAVED
FORMAT:  S16_LE
SUBFORMAT:  STD
SAMPLE_BITS: 16
FRAME_BITS: 16
CHANNELS: 1
RATE: 16000
PERIOD_TIME: 125000
PERIOD_SIZE: 2000
PERIOD_BYTES: 4000
PERIODS: 4
BUFFER_TIME: 500000
BUFFER_SIZE: 8000
BUFFER_BYTES: 16000
TICK_TIME: 0
D: ../../../src/asound/bluealsa-pcm.c:385: /org/bluealsa/hci0/dev_.../hfpag/sink: Closing
```

Arkq, 2020-12-17:
> The system is unable to load the right codec (it seems)

The coded has to be selected *before* you can use PCM exposed by BlueALSA service. That's why you've got "HFP audio codec not selected" error. When using with oFono, BlueALSA does not handle RFCOMM communication at all. All that is left for oFono, and probably some other service which uses oFono for connection establishment. I think, that firstly you have to figure out how to use oFono as a gateway (sorry, but I will not help you with that, like I've said, I'm not using oFono that much). Then, when you will figure out how to select codec via oFono, you can use BlueALSA PCM endpoint.

borine, 2021-10-30:
After researching oFono's HFP support, I've realized that it does not, and is not intended to, implement the HFP-AG role. It uses HFP only to implement a virtual modem using a mobile 'phone as the gateway to the 'phone network. So the device connected via BlueALSA *must* be a mobile 'phone, not headphones - i.e. it is using BlueALSA in the HFP-HF role only.

So I am now closing this issue. I have updated the HFP wiki article here https://github.com/Arkq/bluez-alsa/wiki/Using-BlueALSA-with-HFP-and-HSP-Devices with this information.

closed 2021-10-30 by borine