arkq/bluez-alsa #408: a2dp-source profile connect failed: Protocol not available

reedstonefood, 2021-01-02:
I have two pi's, and I'm trying to send audio from one to the other. I want to sort it out all via console so I can automate things. Let's call the one I'm trying to get the sound from `SourcePi` and the one I'm trying to send output to `SpeakerPi`.

I start by setting up `SpeakerPi`, after some wrangling it all works (including adding `profile=a2dp-sink` in the service file). I can connect my Android device via bluetooth to `SpeakerPi` and the sound comes out as expected. Great!

I go to set up `SourcePi` but I'm having problems. I've been to so many web pages (like [this](https://www.sigmdel.ca/michel/ha/rpi/bluetooth_n_buster_01_en.html) and this[https://www.element14.com/community/thread/73647/l/route-all-pi-audio-to-bluetooth-speaker] - I deactivated onboard audio entirely as it suggested) to try and find the answer to my problem, but no success.

I've change my `/lib/systemd/system/bluealsa.service` file to read this on `SourcePi`:

```
ExecStart=/usr/bin/bluealsa --profile=a2dp-source
```

So it should have the a2dp-source available.

However when I connect the two devices via `bluetoothctl`, I run `sudo systemctl status bluetooth.service` and see the following lines, the last of which is in bold and red...

```
Jan 02 15:46:30 SourcePi bluetoothd[759]: Bluetooth management interface 1.14 initialized
Jan 02 15:46:30 SourcePi bluetoothd[759]: Endpoint registered: sender=:1.23 path=/org/bluez/hci0/A2DP/SBC/Source/1
Jan 02 15:46:30 SourcePi bluetoothd[759]: Endpoint registered: sender=:1.23 path=/org/bluez/hci0/A2DP/SBC/Source/2
Jan 02 15:46:30 SourcePi bluetoothd[759]: Endpoint registered: sender=:1.23 path=/org/bluez/hci0/A2DP/SBC/Source/3
Jan 02 15:56:31 SourcePi bluetoothd[759]: a2dp-source profile connect failed for 00:E0:4C:6E:AC:03: Protocol not available
```

I can use `aplay` to successfully send sound from `SourcePi` to `SpeakerPi`, by running this:

```aplay -D bluealsa:DEV=AB:CD:EF:01:02:03,PROFILE=a2dp sound/test.mp3```

The sound comes out of `SpeakerPi` as I would want. However what I want is all audio from `SourcePi` to be sent to `SpeakerPi`. I think the reason why that's not working is the `Protocol not available` message above.

I've tried all sorts of restarting the services, minor tweaks, rebooting but to no avail. 

The only potential fixes I've found refer to enabling pulseaudio but my impression from all the reading I've done is that alsa is the better way to do things & it's best to leave pulseaudio out...?

Arkq, 2021-01-02:
> The sound comes out of SpeakerPi as I would want.

If using `aplay` works, then it means that bluez-alsa on both devices is configured correctly.

>  However what I want is all audio from SourcePi to be sent to SpeakerPi.

Hmm, I'm not sure what do you mean by that. But assuming that you want to set bluez-alsa device as a default ALSA device on your system you will have to add something like this to your `.asundrc`:

```
pcm.!default {
 	slave.pcm "name-of-your-bluealsa-device";
}
```

where the `name-of-your-bluealsa-device` shall be the PCM name of remote BT device also set up in your `.asoundrc`. For more info see this: https://github.com/Arkq/bluez-alsa/wiki/Using-the-bluealsa-ALSA-pcm-plugin

There is also one limitation. With current master it is not possible to connect more than one client to the bluealsa PCM device. So, even though you will set bluealsa PCM as a default ALSA device you will be able to send only one audio from `SourcePi` to `SpeakerPi`. But, you can try this PR: #407 which might be a solution for your setup :)

borine, 2021-01-02:
> all the reading I've done is that alsa is the better way to do things & it's best to leave pulseaudio out...?

Well, that depends. pulseaudio is a general-purpose sound server, bluealsa provides a bluetooth-specific ALSA plugin. So, for example with desktops or multi-function systems, pulseaudio is generally the way to go. For dedicated sound servers, particularly embedded systems and small boards such as the Pi, bluealsa is best, IMHO. But you need to be clear that bluealsa is *only* the bluetooth audio interface - for everything else like mixing and monitoring of PCM streams you will need some application that does those things for you. 

So, do you need multiple applications accessing the same bluetooth device at the same time? If so, save yourself a lot of pain and go with pulseaudio. If one application at a time is all you need, go with bluealsa.

> what I want is all audio from SourcePi to be sent to SpeakerPi

Sounds like you are using multiple applications? Remember with ALSA that each application opens its own connection directly with the device. So each application needs to be told which device to use. If not told which device, then the application will look for a definition of "default" in the ALSA config and connect to that. You can redefine "default" to use a bluealsa device if you are sure that you only ever want to connect to that one device.

graham8, 2021-01-02:
You don't say which OS you are running, but here's my 2d.  I have successfully sent BT audio to/from devices (including Pi) running OSMC (Debian derivative).  I can start both devices with both source and sink profiles.  BT decides which device is source and which sink depending on which device initiates the connection.

Apart from that, you need to check your .asoundrc as @Arkq says.

reedstonefood, 2021-01-02:
Thanks for the responses everyone! :+1: A few quick answers for now without booting up my Pi again.

>>    However what I want is all audio from SourcePi to be sent to SpeakerPi.

> Hmm, I'm not sure what do you mean by that.

I want what the poster on this [link](https://www.element14.com/community/thread/73647/l/route-all-pi-audio-to-bluetooth-speaker) achieved - all audio from Raspbian sent to a bluetooth speaker - in my case the "bluetooth speaker" is SpeakerPi. The poster appears to have managed it using a mixture of the raspbian GUI & bluetoothctl.

Interestingly, when I try the "GUI" way I can't get it to work. I can connect from SourcePi to SpeakerPi, but SpeakerPi does not show up as an audio device which is what the [Raspbian instructions](https://magpi.raspberrypi.org/articles/bluetooth-audio-raspberry-pi-3) suggest.

> You don't say which OS you are running

Sorry, it's Raspbian (or Raspberry Pi OS as it's called now)

> So, do you need multiple applications accessing the same bluetooth device at the same time? If so, save yourself a lot of pain and go with pulseaudio. If one application at a time is all you need, go with bluealsa.

I imagine most, if not all of the output will be from Chromium. But ideally I wouldn't be limited to that.

I naïvely assumed that, just as my phone could send all its audio to SpeakerPi, that SourcePi would be able to do likewise. If I need to use pulseaudio, could I just use that on SourcePi and leave SpeakerPi with the current bluealsa setup?

> your .asoundrc

I found that file would disappear upon reboot. Couldn't figure it out. So I read some docs and put it in `/etc/asound.conf` instead, and it does seem to use that file, as I can now leave out the MAC address of SpeakerPi, ie `aplay -D  sound/test.mp3` successfully outputs the audio. No idea why that is. :man_shrugging: 

I'll look at @Arkq 's suggestion for that file tomorrow & if that doesn't work I'll post the contents of my `asound.conf` file here.

graham8, 2021-01-02:
Glad you're making progress, but I thought RaspberryPiOS has abandoned bluez-alsa for pulseaudio. Although the announcement implied they were abandoning alsa itself which can't be true so I'm not sure what's going on there.

Haven't tried the latest RaspberryPiOS yet.

borine, 2021-01-03:
The RaspberryPiOS full install is a _desktop_ distribution, and has just started using pulseaudio _for good reason_. That is nothing to do with bluetooth, but everything to do with integrated audio services, such as mixing, sharing, routing and re-routing PCM streams while applications are still running, and a centralized control panel to manage it all. It just happens that pulseaudio also provides a bluetooth interface, so bluealsa is no longer required there. 

If you are building a server with no desktop, eg a headless audio server, then bluealsa is the most efficient way to obtain bluetooth audio. Use the RaspberryPiOS _Lite_ image for that. If you are building a user desktop machine, then use pulseaudio - the RaspberryPiOS full image is good for that.

As for files disappearing etc, the older Raspbian images from the RPi foundation used a number of clumsy tricks to try to compensate for the lack of an audio server. Maybe some of those are still there on the new image and now doing more harm than good. When pulseaudio is properly configured then it takes over all ALSA management and users and applications should just use the pulse API and utilities - do not use .asoundrc, /etc/asound.conf    and such with pulseaudio. Web posts that suggest such things were all written before raspios adopted pulseaudio for its desktop.

A kind of "halfway" between desktop and headless is a so-called "kiosk" machine, with a single application such as Chromium or Kodi as its entire user interface. In such scenarios, most of the function of pulseaudio is redundant. It is those setups I had in mind when proposing PR #407

reedstonefood, 2021-01-03:
> If you are building a user desktop machine, then use pulseaudio - the RaspberryPiOS full image is good for that.

I think I'd rather do this. But, would I be right in saying `SpeakerPi` should be fine using bluealsa?

On `SourcePi` I've ran `sudo apt-get install pulseaudio-module-bluetooth`. This also uninstalls `bluealsa`. I'm not sure really where to go from here? The [GUI way](https://magpi.raspberrypi.org/articles/bluetooth-audio-raspberry-pi-3) still doesn't work as it still doesn't show up as a sound device.

When I select a bluetooth device to connect to, the icon is a ❓ rather than a speaker symbol, as is shown on the linked page. Is there perhaps some way to properly advertise `SpeakerPi` as an audio device, that raspbian is not picking up on?

reedstonefood, 2021-01-03:
> Glad you're making progress, but I thought RaspberryPiOS has abandoned bluez-alsa for pulseaudio. 
> When pulseaudio is properly configured then it takes over all ALSA management and users and applications should just use the pulse API and utilities

Are there any good links available explaining the pros/cons of installing pulseaudio over using alsa?

Most recent posts I can find recommend against pulseaudio. For example [this post](https://www.raspberrypi.org/forums/viewtopic.php?t=286373#p1732922) from less than 4 months ago that says:

> DO NOT INSERT PULSEAUDIO - it will disrupt the sound.

graham8, 2021-01-03:
I think 4 months ago RaspberryPiOS had not announced it was using pulse.  It sounds like you need to install the full RPiOS on SourcePi and use the GUI to set things up.

On SpeakerPi, bulez-alsa will do what you want.  It needs no .asoundrc files.  But I believe you will need to initiate the BT connection from SourcePi to SpeakerPi to make each live up to its name.  Just make sure bluez-alsa is running on SpeakerPi before you try to connect.

reedstonefood, 2021-01-03:
> It sounds like you need to install the full RPiOS on SourcePi and use the GUI to set things up.

This unfortunately doesn't work, as I've described already.

graham8, 2021-01-04:
> This unfortunately doesn't work, as I've described already.

Sorry, it wasn't clear you had switched to the full OS, and as noted above all howtos on the net are going to be suspect.  Pulseaudio has been in Raspberry Pi OS for only a month.  Can you connect from SourcePi to a BT headphone or speaker?

borine, 2021-01-04:
> I note also pulseaudio is in Raspberry Pi OS Lite. So I guess you should remove it if using bluez-alsa.

The lite image file does not have pulseaudio installed - at least the 20-12-02-raspios-buster-armhf-lite image does not.

graham8, 2021-01-04:
Hmmm. The release notes imply pulseaudio is there.

XECDesign, 2021-01-04:
Release notes are the same for all images, so they're a bit misleading. Pulseaudio is not included in the lite image because it drags in too many dependencies.

graham8, 2021-01-04:
> because it drags in too many dependencies.

Exactly why we don't like it at OSMC.

borine, 2021-01-04:
@reedstonefood - There are possibly thousands of different combinations of setups of audio sub-systems with linux, so it is impossible to recommend any one "best" solution from a simple exchange of comments on a forum. So instead, let me imagine a requirement which hopefully has some similarities to yours.

There is a RPi with attached screen and keyboard on which is run a desktop (eg the _*latest*_ Raspios image) and a number of different applications each requiring audio output. There is a USB DAC connecting to an amplifier and speakers, and also a need to send audio over bluetooth. This machine is called SourcePi.

There is a second RPi which is headless, and has an I2C HAT digital amplifier connected to a pair of wired speakers. This machine is to be used to play audio from bluetooth sources, and has a minimal OS installed (eg the _*latest*_ Raspios Lite). It is called SpeakerPi.

The requirement for multiple sound applications and multiple sound outputs on SourcePi makes pulseaudio the simplest option for audio management. It is also a *good* option, in spite of the many negative posts by people who basically complain about anything and everything. The lastest Raspios full image is already configured with desktop, Chromium browser, and pulseaudio with bluetooth - ready to go (I think, not having actually tried it myself :smiley:).

SpeakerPi has the Raspios Lite image. This has the bluetooth subsystem configured, but with no audio component. So bluealsa and bluealsa-aplay are installed (either from the repository or self-built from sources; if using self-build, the guide on the bluez-alsa project wiki has been followed).

SpeakerPi has its attached sound card configured as `default` using `/etc/asound.conf`, and has bluealsa started as:
```sh
bluealsa -p a2dp-sink
```
and bluealsa-aplay started as:
```sh
bluealsa-aplay
```

SourcePi and SpeakerPi are paired using bluetoothctl in the usual way. SpeakerPi is set to trust SourcePi so that it automatically accepts connect requests from SourcePi.

On SourcePi, using the desktop audio control panel, SpeakerPi is selected as the default audio sink. Chromium browser is started and a YouTube video plays sound through SpeakerPi with no further configuration required.

The end :smiley:

reedstonefood, 2021-01-04:
@borine thanks for your description. It's a great story and sounds so simple. However you've more or less described exactly what I've tried, and it doesn't work. I have tried oh so many deviations from it as well, nothing worked, which is why I'm here after 15+ hours of trying to get it to function.

> There are possibly thousands of different combinations of setups of audio sub-systems with linux, so it is impossible to recommend any one "best" solution from a simple exchange of comments on a forum. So instead, let me imagine a requirement which hopefully has some similarities to yours.

> There is a RPi with attached screen and keyboard on which is run a desktop (eg the latest Raspios image) and a number of different applications each requiring audio output. There is a USB DAC [...]

> There is a second RPi which is headless, and has an I2C HAT digital amplifier connected to a pair of wired speakers. This machine is to be used to play audio from bluetooth sources, and has a minimal OS installed (eg the latest Raspios Lite). It is called SpeakerPi.

All done. I updated both operating systems before proceeding. Only difference is I have Raspios full on both. Perhaps not necessary but my logic was that lite might be missing something I need so there would be fewer chances of problems if I installed the full version on both.

The bluetooth setup on SpeakerPi is working fine so far as I can tell. I can connect my phone to it and the sound is outputted through the speakers.

> On SourcePi, using the desktop audio control panel, SpeakerPi is selected as the default audio sink. 

I am unable to do this. I first pair using the bluetooth GUI. I get this message: `Paired successfully. Use the audio menu to select as output device`. I go to the audio devices and SpeakerPi is not listed.

If I try to use a regular bluetooth speaker rather than SpeakerPi, I get the same message but when I go to the audio menu, I can choose the bluetooth speaker as the output device & it plays sound from youtube or whatever through the speaker.

So in summary. SpeakerPi and SourcePi both play nicely when using bluetooth with _other_ devices, but not with each other.

The error message is essentially the same as I started this thread off with, `Protocol not available`. Running `sudo systemctl status bluetooth` I get this on SpeakerPi when I connect to it with my phone:

```
23:10:50: Authentication attempt without agent
23:10:50: Access denied: org.bluez.Error.Rejected
23:10:54: Authentication attempt without agent
23:10:54: Access denied: org.bluez.Error.Rejected
23:11:00: Endpoint registered: sender=:1.42 path/org/bluez/hci0/A2DP/SBC/Sink/3
```

And this when I try to pair SourcePi with SpeakerPi:

```
23:31:20: a2dp-source profile connect failed for aa:aa:aa:aa:aa:aa: Protocol not available
23:31:21: Endpoint registered: sender=:1.55 path/org/bluez/hci0/A2DP/SBC/Sink/3
23:31:22: /org/bluez/hci0/dev_aa_aa_aa_aa_aa_aa/fd0: fd(22) ready
```

borine, 2021-01-05:
> Only difference is I have Raspios full on both.

In that case, you need to use the latest Raspios image, and I would recommend not to use bluealsa with it. Raspios full (latest version) is designed to use pulseaudio, and you will break stuff if you remove it. [Then we are off-topic for the bluez-alsa project, of course !! ]. Older Raspios images should work on SpeakerPi (with bluealsa) but for SourcePi the lack of pulseaudio just makes the whole desktop thing too difficult - go to a Raspberry Pi forum for help if you are determined to persevere with that. 

As for the error messages, its too difficult to diagnose them without more context. Are you using bluealsa on either Pi? If so, have you completely disabled the pulseaudio bluetooth components? If using bluealsa, please post the output of this on both Pis:
```shell
ps -ef | grep bluealsa
```

Also, is bluealsa built with debug enabled? If so the bluealsa logs may contain some clues.

borine, 2021-01-06:
I've just been looking at the raspios repository, and it seems their bluealsa release is very old and its build options mean it is not compatible with the version of pulseaudio they are using. So I strongly recommend not to use bluealsa from the RaspberryPi repository, and instead build from source (or just use pulseaudio if you want full desktops on both Pis).

reedstonefood, 2021-01-06:
> Are you using bluealsa on either Pi?

To re-iterate the current state of play - as it's not be entirely clear from other posts:

`SpeakerPI` is running bluealsa as per myriad internet tutorials, it's running, works fine acting as a bluetooth audio sink if I connect it to other bluetooth sources (eg mobile phone).
`SourcePi` - after reading this thread I've stopped using bluealsa there, and am using pulseaudio instead. This works fine acting as a bluetooth audio source when pairing with a bluetooth speaker.

However making the change to Pulseaudio on `SourcePi` has yet to show any impact with my aimed-for setup - I still arrive back at the same error message which is the `Protocol not available` on `SpeakerPi` that I detailed on my last post, and the opening post.

> If so, have you completely disabled the pulseaudio bluetooth components?

This is done automatically so far as I can tell. What i mean is, if you run `sudo apt get install pulseaudio-module-bluetooth` then it informs you that to do this necessitates removing `bluealsa`. So it's not possible to have both.

> If using bluealsa, please post the output of this on both Pis:

> ps -ef | grep bluealsa

 ** to be pasted in **

I get these three lines on `SpeakerPi`. The first two are things I set up myself and work fine. I don't really understand the third. `SourcePi` returns only one line which is very similar to the first.

> Also, is bluealsa built with debug enabled? If so the bluealsa logs may contain some clues.

I'm using what came with raspbian - where can I find more logs?

borine, 2021-01-07:
> SourcePi returns only one line which is very similar to the first

If "very similar" means
```shell
/usr/bin/bluealsa ...
```
Then bluealsa _*is*_ running on SourcePi - if pulseaudio is also running, which you say it is, then immediately you will have issues with bluetooth audio.

> I'm using what came with raspbian

OK, so there is also another issue. The raspbian / raspios build of bluealsa is out-of-date, it is also not compatible with pulseaudio versions earlier than 13. Raspbian/Raspios use pulseaudio 12, or older. As mentioned [here](https://github.com/Arkq/bluez-alsa#installation) you need to build bluealsa with configure option `--disable-payloadcheck` to work around a bug in previous versions of pulseaudio. Raspbian does not build bluealsa with that flag. So as per my previous comment, you will have to build bluealsa yourself to use it in combination with Raspbian's pulseaudio.

As for the `Protocol not available` error, I'm afraid I have not seen that before, so don't know what causes it. But you did say in the post that `aplay` from SourcePi to SpeakerPi was working, so perhaps that message is a harmless warning rather than an error?

reedstonefood, 2021-01-07:
> If "very similar" means...

Sorry, it was late at night here when I made my last post, the line from `SourcePi` is (almost) the same as the _last_ one from `SpeakerPi`, not the first one:

```
pi        1691  1672  0 23:04 pts/0    00:00:00 grep --color=auto bluealsa
```

> But you did say in the post that aplay from SourcePi to SpeakerPi was working, so perhaps that message is a harmless warning rather than an error?

I interpret the situation as being that the `a2dp` part isn't working (`a2dp-source profile connect failed`), but `aplay` doesn't use `a2dp`, instead it manually sends the audio to a specific MAC address.

I'll try some of your other advice when I get time in the next few days. It's difficult to find motivation after such a long hard slog.

> I'm using what came with raspbian - where can I find more logs?

No answer for this? :(

borine, 2021-01-07:
OK, so bluealsa is not running on SourcePi

> instead it manually sends the audio to a specific MAC address

No, there is no such thing as "manually sending". The profile in use here is A2DP. The failed profile message must be from some other profile. Why a profile should fail I don't know.

>  where can I find more logs?

bluealsa logs error messages to its stderr output. Normally systemd will send them to its journal, unless told otherwise. Bluealsa is normally very "quiet" though - only serious errors are logged. It needs to be built with the `--enable-debug` flag to get verbose logging. To see what messages there are, if any, you could try:
```shell
sudo journalctl -xeu bluealsa.service
```

>  It's difficult to find motivation after such a long hard slog

Yes, its not supposed to be this hard! I have now set up a Pi with latest full RaspberryPiOS image. I have a Pi Zero running raspios lite with my own build of the latest bluealsa code. Pairing them using blutoothctl was straightforward, and connection using bluetoothctl works fine. But connection using the GUI faiis with "Connection failed - use audio menu" but I have no idea how to find the audio menu (never used raspios gui before). There are no "failed profile messages" in the logs though.

reedstonefood, 2021-01-10:
@borine thanks for trying it out yourself.

I've stuck a new SD card into SpeakerPi and started from scratch, with Raspberry pi OS lite.

- installed raspberry pi OS lite
- set a [static IP](https://pimylifeup.com/raspberry-pi-static-ip-address/) to make connecting remotely easier
- got my bluetooth dongle recognized (it's a new chipset that isn't natively supported yet so I had to follow [these steps](https://www.raspberrypi.org/forums/viewtopic.php?t=294634))
- looked at the options for compiling this repo and decided it would be a nice extra to have AAC support. So after much digging I found the required library is not available pre-compiled on Raspberry Pi OK so I compiled `fdk-aac` from source as per step 2 of [this article](https://pimylifeup.com/compiling-ffmpeg-raspberry-pi/) (took a whole night to run!)
- then compiled bluealsa from source following the [excellent instructions](https://github.com/Arkq/bluez-alsa/wiki/Installation-from-source). I used the configure options `--enable-debug --enable-aac --disable-payloadcheck` 
- disabled onboard speakers to force sound to go through my DAC (`speaker-test -c2` works)
- setup services for bluealsa and aplay ([this documentation](https://www.sigmdel.ca/michel/ha/rpi/bluetooth_n_buster_01_en.html) helped with setting these up). I had to `enable` both services so they would start on boot. The `bluealsa` refused to be enabled, giving an error message, I "fixed" this by adding `[Install] WantedBy=multi-user.service` (ie copying what was in the `aplay.service` file)
- using `bluetoothctl` I can establish a connection from my phone to `SpeakerPi` and play sound through it :+1:

Then back to `SourcePi` - here's the long description of what I did (exactly the same as before `SourcePi`'s reinstallation).

First on `SpeakerPi` I ran `bluetoothctl` then `discoverable on`. Using the "Add new Device" menu on `SourcePi`, I can see `SpeakerPi` listed. I choose to pair it, `SpeakerPi` accepts it by me typing in `pair MAC_ADDRESS` and `trust MAC_ADDRESS`. SourcePi tells me "paired successfully. Use the audio menu to select as output device." However, SpeakerPi does not show as an option in the audio menu (@borine how to do this is shown [here](https://magpi.raspberrypi.org/articles/bluetooth-audio-raspberry-pi-3) - would be interesting to know if that step works for you or not).

However, I no longer get the `a2dp-source profile connect failed: Protocol not available` message :grin:. So that's something. But the log files I get now look completely benign to me. Here's the output from `sudo journalctl -xeu bluealsa.service` t is from the point I restarted it (after which I did the above pairing).

```
-- The job identifier is 744 and the job result is done.
Jan 10 19:59:08 SpeakerPi systemd[1]: Started BluezALSA proxy.
-- Subject: A start job for unit bluealsa.service has finished successfully
-- Defined-By: systemd
-- Support: https://www.debian.org/support
--
-- A start job for unit bluealsa.service has finished successfully.
--
-- The job identifier is 744.
Jan 10 19:59:09 SpeakerPi bluealsa[1007]: /usr/bin/bluealsa: D: ../../src/bluez.c:497: Creating media endpoint object: /org/bluez/hci0/A2DP/AAC/Source/1
Jan 10 19:59:09 SpeakerPi bluealsa[1007]: /usr/bin/bluealsa: D: ../../src/bluez.c:413: Registering media endpoint: /org/bluez/hci0/A2DP/AAC/Source/1
Jan 10 19:59:09 SpeakerPi bluealsa[1007]: /usr/bin/bluealsa: D: ../../src/bluez.c:497: Creating media endpoint object: /org/bluez/hci0/A2DP/AAC/Source/2
Jan 10 19:59:09 SpeakerPi bluealsa[1007]: /usr/bin/bluealsa: D: ../../src/bluez.c:413: Registering media endpoint: /org/bluez/hci0/A2DP/AAC/Source/2
Jan 10 19:59:09 SpeakerPi bluealsa[1007]: /usr/bin/bluealsa: D: ../../src/bluez.c:497: Creating media endpoint object: /org/bluez/hci0/A2DP/AAC/Sink/1
Jan 10 19:59:09 SpeakerPi bluealsa[1007]: /usr/bin/bluealsa: D: ../../src/bluez.c:413: Registering media endpoint: /org/bluez/hci0/A2DP/AAC/Sink/1
Jan 10 19:59:09 SpeakerPi bluealsa[1007]: /usr/bin/bluealsa: D: ../../src/bluez.c:497: Creating media endpoint object: /org/bluez/hci0/A2DP/AAC/Sink/2
Jan 10 19:59:09 SpeakerPi bluealsa[1007]: /usr/bin/bluealsa: D: ../../src/bluez.c:413: Registering media endpoint: /org/bluez/hci0/A2DP/AAC/Sink/2
Jan 10 19:59:09 SpeakerPi bluealsa[1007]: /usr/bin/bluealsa: D: ../../src/bluez.c:497: Creating media endpoint object: /org/bluez/hci0/A2DP/SBC/Source/1
Jan 10 19:59:09 SpeakerPi bluealsa[1007]: /usr/bin/bluealsa: D: ../../src/bluez.c:413: Registering media endpoint: /org/bluez/hci0/A2DP/SBC/Source/1
Jan 10 19:59:09 SpeakerPi bluealsa[1007]: /usr/bin/bluealsa: D: ../../src/bluez.c:497: Creating media endpoint object: /org/bluez/hci0/A2DP/SBC/Source/2
Jan 10 19:59:09 SpeakerPi bluealsa[1007]: /usr/bin/bluealsa: D: ../../src/bluez.c:413: Registering media endpoint: /org/bluez/hci0/A2DP/SBC/Source/2
Jan 10 19:59:09 SpeakerPi bluealsa[1007]: /usr/bin/bluealsa: D: ../../src/bluez.c:497: Creating media endpoint object: /org/bluez/hci0/A2DP/SBC/Sink/1
Jan 10 19:59:09 SpeakerPi bluealsa[1007]: /usr/bin/bluealsa: D: ../../src/bluez.c:413: Registering media endpoint: /org/bluez/hci0/A2DP/SBC/Sink/1
Jan 10 19:59:09 SpeakerPi bluealsa[1007]: /usr/bin/bluealsa: D: ../../src/bluez.c:497: Creating media endpoint object: /org/bluez/hci0/A2DP/SBC/Sink/2
Jan 10 19:59:09 SpeakerPi bluealsa[1007]: /usr/bin/bluealsa: D: ../../src/bluez.c:413: Registering media endpoint: /org/bluez/hci0/A2DP/SBC/Sink/2
Jan 10 19:59:09 SpeakerPi bluealsa[1007]: /usr/bin/bluealsa: D: ../../src/main.c:390: Acquiring D-Bus service name: org.bluealsa
Jan 10 19:59:09 SpeakerPi bluealsa[1007]: /usr/bin/bluealsa: D: ../../src/main.c:395: Starting main dispatching loop
Jan 10 19:59:13 SpeakerPi bluealsa[1007]: /usr/bin/bluealsa: D: ../../src/dbus.c:59: Called: org.bluealsa.Manager1.GetPCMs() on /org/bluealsa
```

borine, 2021-01-11:
> Use the audio menu to select as output device

That's as far as I got using the GUI. I was not able to achieve a working blutooth audio connection that way. Looking at the bluealsa logs, I can see that SpeakerPi is not connected to SourcePi. I suggest you use bluetoothctl on SourcePi to make the bluetooth connection, and ignore the GUI which appears to be broken:

```shell
bluetoothctl connect xx:xx:xx:xx:xx:xx
```
where xx:xx:xx:xx:xx:xx is the address of SpeakerPi

Then check if pulseaudio on SourcePi has recognised the connection with:
```shell
pactl list sinks
```

... and what is the default sink with:
```shell
pactl info
```

You should see in the output:
```shell
Default Sink: bluez_sink.xx_xx_xx_xx_xx_xx.a2dp_sink
```

If so, then all is well on SourcePi and hopefully audio will now be sent to SpeakerPi

reedstonefood, 2021-01-11:
😁 😁 😁 It's working now, thanks so much!!!

When I ran `pactl list sinks` it showed `SpeakerPi` there. But `pactl info` did not have it as the default sink. I googled a bit and ran `pacmd set-default-sink {NAME OF SINK}` and then any audio I played in `SourcePi` came out in `SpeakerPi`!!!

To sum up - I'm still not entirely sure what the problem was. Why does Raspberry Pi OS not pair to `SpeakerPi` through the GUI when it can pair to an ordinary bluetooth speaker? What did the original error message mean? Did building the latest version of `bluez-alsa` help? I'm not sure but it's all working now, and I learned some things thanks to the kind people on this thread. Cheers for persevering with me through this 😃

borine, 2021-01-12:
> Why does Raspberry Pi OS not pair to SpeakerPi through the GUI when it can pair to an ordinary bluetooth speaker?

Pairing is nothing to do with bluez-alsa - it is a procedure implemented by the bluetooth daemon, and merely establishes an encryption key pair by which the two devices can identify each other.

Once two devices are paired, they need to be connected, and that is when the Profiles come into play and bluealsa becomes involved. For most commercial devices, the connection request originates from the speaker, whereas in the bluealsa setup described here the connection originates from SourcePi. Perhaps that is the significant difference - I don't have time to investigate further as I am not normally a GUI user. Perhaps someone on an RPi forum can help explain the workings of the Raspios bluetooth GUI.

borine, 2021-06-08:
Closing as the error appears to be with the Raspios GUI and not bluealsa. using the command line tools of pulseaudio and bluez appears to have solved the problem.

closed 2021-06-08 by borine