arkq/bluez-alsa #409: Can't connect to Blackberry 10 device

WTFThatPerson, 2021-01-03:
I have Raspberry Pis 3 & 4 with Arch Linux Arm installed 32-bit and 64-bit. Since both vers show equal manner, it's not hardware or bits dependency, I think. Bluealsa is latest and installed from aur repository. Since Blackberry declares AptX codec, I installed openaptx, included it as dependency  and compiled Bluealsa with --enable-aptx and --enable-aptx-hd. No Pulseaudio presents in system, just pure ALSA. Everything works fine except Blackberry 10 devices, I mean, I can use Pi as a source for BT loudspeakers and as a sink for Android devices, but when I try to connect to Blackberry, I can't do it. Connection tries to be established, I have that "[agent] Authorize service 0000110d-0000-1000-8000-00805f9b34fb (yes/no):" question, but it ends up here. Then connection breaks after such 3 tries. btmon log of on of tries (latter two are quit equivalent):
```
Bluetooth monitor ver 5.55
= Note: Linux version 5.4.83-4-ARCH (aarch64)                          0.930454
= Note: Bluetooth subsystem version 2.22                               0.930460
= New Index: DC:A6:32:36:7A:8D (Primary,UART,hci0)              [hci0] 0.930462
= Open Index: DC:A6:32:36:7A:8D                                 [hci0] 0.930463
= Index Info: DC:A6:32:36:7A:8D (Broadcom Corporation)          [hci0] 0.930464
@ MGMT Open: bluetoothd (privileged) version 1.14             {0x0001} 0.930466
> HCI Event: Connect Request (0x04) plen 10                 #1 [hci0] 38.043021
        Address: 94:EB:CD:0E:F5:91 (BlackBerry RTS)
        Class: 0x7a220c
          Major class: Phone (cellular, cordless, payphone, modem)
          Minor class: Smart phone
          Limited Discoverable Mode
          Networking (LAN, Ad hoc)
          Capturing (Scanner, Microphone)
          Object Transfer (v-Inbox, v-Folder)
          Audio (Speaker, Microphone, Headset)
          Telephony (Cordless telephony, Modem, Headset)
        Link type: ACL (0x01)
< HCI Command: Accept Connection Re.. (0x01|0x0009) plen 7  #2 [hci0] 38.043104
        Address: 94:EB:CD:0E:F5:91 (BlackBerry RTS)
        Role: Master (0x00)
> HCI Event: Command Status (0x0f) plen 4                   #3 [hci0] 38.043548
      Accept Connection Request (0x01|0x0009) ncmd 1
        Status: Success (0x00)
> HCI Event: Role Change (0x12) plen 8                      #4 [hci0] 38.217551
        Status: Success (0x00)
        Address: 94:EB:CD:0E:F5:91 (BlackBerry RTS)
        Role: Master (0x00)
> HCI Event: Connect Complete (0x03) plen 11                #5 [hci0] 38.226298
        Status: Success (0x00)
        Handle: 11
        Address: 94:EB:CD:0E:F5:91 (BlackBerry RTS)
        Link type: ACL (0x01)
        Encryption: Disabled (0x00)
< HCI Command: Read Remote Supporte.. (0x01|0x001b) plen 2  #6 [hci0] 38.226514
        Handle: 11
> HCI Event: Command Status (0x0f) plen 4                   #7 [hci0] 38.228497
      Read Remote Supported Features (0x01|0x001b) ncmd 1
        Status: Success (0x00)
> HCI Event: Read Remote Supported Features (0x0b) plen 11  #8 [hci0] 38.228506
        Status: Success (0x00)
        Handle: 11
        Features: 0xbf 0xfe 0x2d 0xfe 0xd8 0x79 0x79 0x87
          3 slot packets
          5 slot packets
          Encryption
          Slot offset
          Timing accuracy
          Role switch
          Sniff mode
          Power control requests
          Channel quality driven data rate (CQDDR)
          SCO link
          HV2 packets
          HV3 packets
          u-law log synchronous data
          A-law log synchronous data
          CVSD synchronous data
          Power control
          Transparent synchronous data
          Flow control lag (middle bit)
          Enhanced Data Rate ACL 2 Mbps mode
          Enhanced Data Rate ACL 3 Mbps mode
          Enhanced inquiry scan
          Interlaced inquiry scan
          Interlaced page scan
          RSSI with inquiry results
[... 242 lines cut ...]
> HCI Event: Number of Completed Packets (0x13) plen 5     #43 [hci0] 42.020190
        Num handles: 1
        Handle: 11
        Count: 2
> HCI Event: Encryption Change (0x08) plen 4               #44 [hci0] 42.040137
        Status: Success (0x00)
        Handle: 11
        Encryption: Disabled (0x00)
< HCI Command: Read Clock Offset (0x01|0x001f) plen 2      #45 [hci0] 42.040211
        Handle: 11
> HCI Event: Command Status (0x0f) plen 4                  #46 [hci0] 42.041338
      Read Clock Offset (0x01|0x001f) ncmd 1
        Status: Success (0x00)
< HCI Command: Disconnect (0x01|0x0006) plen 3             #47 [hci0] 42.041371
        Handle: 11
        Reason: Authentication Failure (0x05)
> HCI Event: Command Status (0x0f) plen 4                  #48 [hci0] 42.042100
      Disconnect (0x01|0x0006) ncmd 1
        Status: Success (0x00)
> HCI Event: Disconnect Complete (0x05) plen 4             #49 [hci0] 42.120680
        Status: Success (0x00)
        Handle: 11
        Reason: Connection Terminated By Local Host (0x16)
```
I have any idea what that thing (Bluealsa as per my understanding) means under "Authentication Failure" if it successfully reads clock offset right before.

WTFThatPerson, 2021-01-22:
Excuse me, if I'm seem to be not very polite, but to say something instead of keep total silence during 20 day is a way better, isn't it?

Now about problem itself: Blackberry 10 uses AptX as codec, but when I look into bluealsa start log, I see only endpoint pathes for
path=/org/bluez/hci0/A2DP/aptXHD/Source/1
path=/org/bluez/hci0/A2DP/aptXHD/Source/2
but there isn't any signs of sink ones. Keeping in mind I try use Raspberry Pis as sinks exactly, can it be the root of my problem?

Arkq, 2021-01-22:
Hi @WTFThatPerson, so let me ease your anxiety a little bit. I've seen your issue, and probably may others (every issue on this page is addressed at some point), but I've got no idea what might be wrong (from your first post), neither I do not own Blackberry, and nothing comes to my mind what could be checked. Since no one else has posted any comment, I assume that others do not own Blackberry either. Sorry...

But back to the issue.

> Keeping in mind I try use Raspberry Pis as sinks exactly, can it be the root of my problem?

It might be, because aptX Sink support is in master since commit b11cff0 (7 days ago), and only if you've got aptX decoder (opensource decoder is available in the latest openaptx codebase). However, if RPi does not provide aptX sink, Bluetooth source (in your case Blackberry) should select SBC, which is a mandatory codec (and you can not omit it from bluealsa other than by removing it from the source code, so I assume that your RPi has exposed at least SBC).

Maybe sniffing through HCI dump might give some clue... But, unfortunately, I've got no much time for investigating this issue. Anyway, you might post here HCI dump (preferably from Blackberry, if that's possible), maybe someone else will pick it up from here. As for myself, I don't know when I will have enough time for checking this issue, because it's not obvious what might be wrong... (it might take may hours to pinpoint the root cause).

WTFThatPerson, 2021-01-24:
Ah, thank You to remember about SBC thing although it makes me completely frustrated: it was last hope for more or less simple way to resolve that problem.

About Your suggestions:
1. Yes, I configured bluealsa for sink, moreover, I can successfully stream audio from Android device
2. At my point of view, I just put hci dump above using btmon, because hcidump has been deprecated. Maybe, I don't understand something and I should get another type of logging, say, in btsnoop format?

Just a couple words about Android vs Blackberry connection:
Everything looks very similar until event # 36 from log above:
Next Android requests omtu size 672 vs 1691 Blackberry asks, and then it sends respond for "A2CP: Configure request" without imtu field which one Blackberry respond has (see # 40 from above). Then for Android connection HCI event about successful setup QoS service appears (no any signs of such message in Blackberry log) and finally for Android connection set up, devices configure AVDTP Discover followed by configure AVDTP capabilities. Blackberry, in turn, also configures AVDTP Discover (## 41,42) and then it keeps silence until timeout, so it looks like it waiting something else from bluealsa.

Arkq, 2021-01-24:
> Next Android requests omtu size 672 vs 1691 Blackberry asks [...] AVDTP Discover followed by configure AVDTP capabilities [...] AVDTP Discover and then it keeps silence until timeout, so it looks like it waiting something else from bluealsa.

All this communication is performed by BlueZ, so I think that maybe you should seek for help in the BlueZ mailing list. BlueALSA only validates and creates capabilities, but it is relevant only after capabilities has been retrieved from the remote Bluetooth device. BlueALSA communicates with BlueZ via D-Bus only, so you can check with `dbus-monitor` whether BlueALSA is involved.

WTFThatPerson, 2021-01-24:
Thank You very match for clarification

WTFThatPerson, 2021-03-03:
Well, it's a ssp problem, not a bluealsa definitely.

closed 2021-03-03 by WTFThatPerson