arkq/bluez-alsa #410: The battery level isn't present on some Sony headsets.

alex14641, 2021-01-04:
I've tested with Sony WF-1000XM3, WM-1000XM3, WM-1000XM4: with the WM-1000XM3, the battery level is present in alsamixer; it's not present for the other two. With the patch below, the battery level is present on all three.

`diff --git a/src/ba-rfcomm.c b/src/ba-rfcomm.c
index e0fbf5b..b51f0ef 100644
--- a/src/ba-rfcomm.c
+++ b/src/ba-rfcomm.c
@@ -595,7 +595,7 @@ static int rfcomm_handler_xapl_set_cb(struct ba_rfcomm *r, const struct bt_at *a
 	d->xapl.product_id = product;
 	strcpy(d->xapl.software_version, version);
 
-	sprintf(resp, "+XAPL=BlueALSA,%u", config.hfp.xapl_features);
+	sprintf(resp, "+XAPL=iPhone,%u", config.hfp.xapl_features);
 	if (rfcomm_write_at(fd, AT_TYPE_RESP, NULL, resp) == -1)
 		return -1;
 	if (rfcomm_write_at(fd, AT_TYPE_RESP, NULL, "OK") == -1)
`
I run bluealsa like so:
/usr/bin/bluealsa -p a2dp-source -p hfp-ag -p hfp-hf

Arkq, 2021-01-04:
Ehh.... why did Sony do that....?!

To be honest I've got mixed fillings about bluealsa pretending to be an iPhone :/

alex14641, 2021-01-19:
I'm working on a patch to selectively change the identification based on the MAC address of the device: it would require supplying a list of MAC addresses to the bluealsa daemon. If the device is in the list, pretend to be an iPhone; otherwise, BlueALSA.

Arkq, 2021-01-19:
I think that the easiest approach would be to simply add `#ifdef` which will compile bluealsa with "iPhone" as an identifier for XAPL command (similar to fix RTP payload check for pulseaudio < 13.0 - `../configure --disable-payloadcheck`). If you will make it to be changed dynamically during runtime based on the MAC whitelist, the question is who will be maintaining this whitelist? Because for sure not me :)

alex14641, 2021-01-19:
True, supplying the macs isn't very convenient, but I can't see a more reliable way to make this work: you won't have access to the device names when bluealsa first starts up. Here's what I'm envisioning:
      bluealsa --emulate-iphone-addrs=addr[,addr.....] 
or 
     bluealsa --emulate-iphone-addrs-file=<file name>
where <file name> is a file containing a list of MAC addresses, one per line. I want to implement both methods. I don't want to risk making 'iPhone' default for all devices, as it might break something, especially since it's a workaround for (hopefully) a few devices.

alex14641, 2021-01-21:
Just to be clear: I was not envisioning bluealsa maintaining a MAC list, but maybe a list of brands and models, e,g Sony WH_1000XM4, that need the quirk, could be added to the README. It would be each user's responsibility to maintain their own MAC list. I'm assuming that two devices with the same brand and model could have different MAC addresses.

Arkq, 2021-01-22:
> Just to be clear: I was not envisioning bluealsa maintaining a MAC list, but maybe a list of brands and models, e,g Sony WH_1000XM4, that need the quirk, could be added to the README.

OK, maybe...

> I'm assuming that two devices with the same brand and model could have different MAC addresses.

In the ideal world, it should be like this :)

ValdikSS has created a [compatibility list](https://btcodecs.valdikss.org.ru/codec-compatibility/) with bluetooth devices. And according to this page Sony (and other manufactures), uses last two octets for device for particular model x N, where N is 1-5, which gives 65535 x1-5 unique devices. IMHO is rather smal value, which implies that this assumption might be wrong, yet still it's a good place to start. And there is no WH-1000XM4 device on that list :/ Anyway, please, check whether your MAC would fit in that list as a new entry.

alex14641, 2021-01-23:
The MAC address of the WH_1000XM3 starts with CC:98:8B:93:B8, which doesn't match what's on the page. What do you think about allowing matching on the address prefix: e.g. --emulate-iphone-addrs=CC:98:8B:93?

Arkq, 2021-01-23:
Yes, I think that matching MAC prefixes might be a good idea.

But on the other hand, I'm inclined to go with rather a simple solution for this quirk fix. And parsing MACs as a MAC prefix does not seem to be simple. Because it would be a good idea then to allow user to determine the length of the prefix in bits, e.g. `11:22:33:44:55:66/35`.  And to be honest, all this battery feature stuff is not even in the scope of audio...

I'm not sure if you will agree with me, but I think I would go with an option for setting the name which is send with this XAPL response, e.g. `--xapl-resp-name=<NAME>` where default is `BlueALSA` and user can change it to `iPhone` (for all devices). After all, all Android devices send `iPhone` in this response message. The case might be that in the Apple specification for Bluetooth Accessory Identification the name `iPhone` is typed with normal font, which might give an impression that it is a part of the specification: https://developer.apple.com/accessories/Accessory-Design-Guidelines.pdf (page 69). With such solution, Linux distributions might set `--xapl-resp-name=iPhone` as a part of a default configuration. Then this will no longer be an issue for normal users, but will give some flexibility for power users. What do you think?

alex14641, 2021-01-23:
> Yes, I think that matching MAC prefixes might be a good idea.
> 
> But on the other hand, I'm inclined to go with rather a simple solution for this quirk fix. And parsing MACs as a MAC prefix does not seem to be simple.

I already have code that does that: in ba_rfcomm.c, I convert the device MAC address to a string and compare it to the list of addresses one by one. I compare characters up to the length of the prefix.

>And to be honest, all this battery feature stuff is not even in the scope of audio...

Agreed. but I think it is a cool feature.

> I'm not sure if you will agree with me, but I think I would go with an option for setting the name which is send with this XAPL response, e.g. `--xapl-resp-name=<NAME>` where default is `BlueALSA` [...]

How about both? I would implement them such that --xapl-resp-name would have precedence over  --emulate-iphone-addrs

Arkq, 2021-01-23:
> How about both? I would implement them such that --xapl-resp-name would have precedence over --emulate-iphone-addrs. I confess to to being a bit paranoid about making it a global on/off thing.

OK, but please make it as two separate commits. About, `--xapl-resp-name` I'm positive, where for `--emulata-iphone-addrs` I'd like to see actual implementation (the option name maybe should be changed, to indicate real meaning, because it's not iPhone emulation :D). Maybe you are right that we can have both, if it does not introduce too much noise.

alex14641, 2021-01-23:
This is the emulate-iphone-addrs patch. 

[emulate-iphone-addrs.patch.txt](https://github.com/Arkq/bluez-alsa/files/5861031/emulate-iphone-addrs.patch.txt)

alex14641, 2021-01-24:
This is the xapl-resp-name patch.
[xapl.patch.txt](https://github.com/Arkq/bluez-alsa/files/5861679/xapl.patch.txt)

Arkq, 2021-01-24:
OK, thanks. It would be easier to review these patches if there were pull requests, but sitll.

And right of the bat, I've got some comments for xapl.patch:

1. variable name should reflect meaning, and it's not whole response, but only a name part. There is already `config.hfp.xapl_features`, so it would be reasonable to place name variable in the same structure.
2. please store it as `const char *`, using 32 byte array is meaningless if you do not check overflow during `sprintf`.
3. use `sptintf` with length check because right now it is required
4. set default of the name variable to `BlueALSA`
5. if `bluealsa_config_emulate_iphone(addr)` is true you might set local variable to `iPhone` (which defaults to global name variable) it will remove duplicated code

alex14641, 2021-01-24:
Updated patch. Replace the previous one with this one.
[xapl2.patch.txt](https://github.com/Arkq/bluez-alsa/files/5863333/xapl2.patch.txt)

Arkq, 2021-01-27:
I've pushed xapl2.patch to master (with a little bit of simplification). Now, I will have too review emulate-iphone-addrs.patch.txt :)

Arkq, 2021-07-31:
I think that for now for master branch the option `--xapl-resp-name` is good enough. Customization per connected device would also be nice, but I would like to be able to customize all options per device, not only XAPL. But for that some underlying work is required, and I've got no time for that now.

closed 2021-07-31 by Arkq