arkq/bluez-alsa #411: Bluez-alsa fails at make

Sindar-sudo, 2021-01-21:
Hi,

I can't get my bluez-alsa installation to "make". I tried configure with just --enable-debug and also without any options and it succeeds but then "make" fails with this.

> gcc -DHAVE_CONFIG_H -I. -I../../src -I..    -pthread -I/usr/include/gio-unix-2.0 -I/usr/include/libmount -I/usr/include/blkid -I/usr/include/glib-2.0 -I/usr/lib/x86_64-linux-gnu/glib-2.0/include -I/usr/include/glib-2.0 -I/usr/lib/x86_64-linux-gnu/glib-2.0/include  -g -O2 -MT ba-transport.o -MD -MP -MF $depbase.Tpo -c -o ba-transport.o ../../src/ba-transport.c &&\
mv -f $depbase.Tpo $depbase.Po
../../src/ba-transport.c: In function ‘transport_acquire_bt_sco’:
../../src/ba-transport.c:919:39: error: ‘BT_VOICE_CVSD_16BIT’ undeclared (first use in this function)
  919 |     t->type.codec == HFP_CODEC_CVSD ? BT_VOICE_CVSD_16BIT : BT_VOICE_TRANSPARENT) == -1) {
      |                                       ^~~~~~~~~~~~~~~~~~~
../../src/ba-transport.c:919:39: note: each undeclared identifier is reported only once for each function it appears in
../../src/ba-transport.c:919:61: error: ‘BT_VOICE_TRANSPARENT’ undeclared (first use in this function); did you mean ‘IP_TRANSPARENT’?
  919 |     t->type.codec == HFP_CODEC_CVSD ? BT_VOICE_CVSD_16BIT : BT_VOICE_TRANSPARENT) == -1) {
      |                                                             ^~~~~~~~~~~~~~~~~~~~
      |                                                             IP_TRANSPARENT
make[3]: *** [Makefile:629: ba-transport.o] Error 1
make[3]: Leaving directory '/home/user/bluez-alsa/build/src'
make[2]: *** [Makefile:686: all-recursive] Error 1
make[2]: Leaving directory '/home/user/bluez-alsa/build/src'
make[1]: *** [Makefile:444: all-recursive] Error 1
make[1]: Leaving directory '/home/user/bluez-alsa/build'
make: *** [Makefile:376: all] Error 2

I tried looking around for any similar errors but couldn't find any information. I'm running Ubuntu 20.04.
I seem to have all the dependencies installed so not sure what is this about... Any ideas what might be the issue?

borine, 2021-01-21:
Looks like you do not have libbluetooth-dev installed. Please check the dependencies on Ubuntu from the wiki guide here:
https://github.com/Arkq/bluez-alsa/wiki/Installation-from-source

Sindar-sudo, 2021-01-21:
No, that's not the case...
> sudo apt-get install libbluetooth-dev
Reading package lists... Done
Building dependency tree
Reading state information... Done
libbluetooth-dev is already the newest version (5.53-0ubuntu3).

I also tried to do everything with sudo, but the error is exactly the same.

borine, 2021-01-21:
what is the output of
```shell
grep BT_VOICE_CVSD_16BIT /usr/include/bluetooth/bluetooth.h
```

Sindar-sudo, 2021-01-21:
> grep BT_VOICE_CVSD_16BIT /usr/include/bluetooth/bluetooth.h
#define BT_VOICE_CVSD_16BIT                     0x0060

borine, 2021-01-21:
Hmm, then that's a mystery to me. I don't have Ubuntu 20.04 to test with, but there have been no other similar error reports here. Is it possible that there is some other file called bluetooth/bluetooth.h in your include search path? Other than that I can't imagine any reason why gcc would give that particular error message.

Arkq, 2021-01-21:
I've just checked with ubuntu 20.04 (desktop image from here: https://releases.ubuntu.com/20.04/) and current master bluez-alsa. I've built basic bluez-alsa configuration (`./configure` without any options) and it has succeeded.

Maybe you've got some environment variable which modifies CFLAGS or something like this. Or maybe, like @borine has said, you've got some other bluetooth.h file somewhere on include path (search paths may be also changed via environment variable).

Pleas post the line with the `configure` from the `head config.log` output.

Sindar-sudo, 2021-01-21:
You turned out to be correct - I had another bluetooth.h file in /usr/local/include/bluetooth/bluetooth.h
My installation is pretty messed up from futile attempts to make my BT speakerphone work... :) I've tried almost everything at this point, let's hope bluez-alsa helps.
Thanks a lot!

closed 2021-01-21 by Sindar-sudo