arkq/bluez-alsa #414: Help with "Couldn't get BlueALSA PCM: The name org.bluealsa was not provided by any .service files"

alpereira7, 2021-02-03:
I followed the installation guide and it seems it went well, no error message during install.
However, when I tried to play an audio file to my Bluetooth speakers (properly connected with bluetoothctl, and confirmed by the bluetooth UI), the command line is:
`aplay -D bluealsa:DEV=XX:XX:XX:XX:XX:XX,PROFILE=a2dp audio.wav`

Error: Couldn't get BlueALSA PCM: The name org.bluealsa was not provided by any .service files

borine, 2021-02-03:
> Error: Couldn't get BlueALSA PCM: The name org.bluealsa was not provided by any .service files

That indicates the bluealsa server is not running. The mention of ".service files" is from D-Bus and is not relevant unless you had intended for bluealsa to be started on demand by D-Bus clients - however as described here: https://github.com/Arkq/bluez-alsa/wiki/Systemd-integration#start-on-demand-by-a-d-bus-client that is not generally recommended.

Have you arranged for bluealsa to be started automatically by the system? If not you will need to start it manually.

alpereira7, 2021-02-04:
Thanks. I typed in another terminal `sudo bluealsa` and it fixed *this* error. There is still another error message, I'll open another topic.

closed 2021-02-04 by alpereira7