arkq/bluez-alsa #415: Paired, connected, everything seems fine, but no sound

alpereira7, 2021-02-04:
Everything seems properly set, paired, connected, short audio confirmation. But after typing 
`aplay -D bluealsa:DEV=XX:XX:XX:XX:XX:XX,PROFILE=sco audio.wav`
I have no sound output.

I have no error messages, and even have confirmation:
`Playing WAVE 'audio.wav' : Signed 16 bit Little Endian, Rate 44100 Hz, Stereo`

Is there anything I can do? What to check? Where to look?

Arkq, 2021-02-04:
Please compile bluez-alsa with debug enabled, and past the logs from the bluealsa service and from aplay.

alpereira7, 2021-02-04:
Logs from aplay
-
`pi@raspberrypi:~ $ aplay -D bluealsa:DEV=F4:4E:FD:EB:AF:04,PROFILE=sco Roland-JD-990-Windchimes.wav 
D: ../../../src/asound/bluealsa-pcm.c:1053: Getting BlueALSA PCM: PLAYBACK F4:4E:FD:EB:AF:04 sco
D: ../../../src/asound/bluealsa-pcm.c:921: /org/bluealsa/hci0/dev_F4_4E_FD_EB_AF_04/hfpag/sink: Setting constraints
Playing WAVE 'Roland-JD-990-Windchimes.wav' : Signed 16 bit Little Endian, Rate 44100 Hz, Stereo
D: ../../../src/asound/bluealsa-pcm.c:445: /org/bluealsa/hci0/dev_F4_4E_FD_EB_AF_04/hfpag/sink: Initializing HW
D: ../../../src/asound/bluealsa-pcm.c:468: /org/bluealsa/hci0/dev_F4_4E_FD_EB_AF_04/hfpag/sink: FIFO buffer size: 2048 frames
D: ../../../src/asound/bluealsa-pcm.c:476: /org/bluealsa/hci0/dev_F4_4E_FD_EB_AF_04/hfpag/sink: Selected HW buffer: 4 periods x 2000 bytes == 8000 bytes
D: ../../../src/asound/bluealsa-pcm.c:491: /org/bluealsa/hci0/dev_F4_4E_FD_EB_AF_04/hfpag/sink: Initializing SW
D: ../../../src/asound/bluealsa-pcm.c:491: /org/bluealsa/hci0/dev_F4_4E_FD_EB_AF_04/hfpag/sink: Initializing SW
D: ../../../src/asound/bluealsa-pcm.c:491: /org/bluealsa/hci0/dev_F4_4E_FD_EB_AF_04/hfpag/sink: Initializing SW
D: ../../../src/asound/bluealsa-pcm.c:491: /org/bluealsa/hci0/dev_F4_4E_FD_EB_AF_04/hfpag/sink: Initializing SW
D: ../../../src/asound/bluealsa-pcm.c:528: /org/bluealsa/hci0/dev_F4_4E_FD_EB_AF_04/hfpag/sink: Prepared
D: ../../../src/asound/bluealsa-pcm.c:491: /org/bluealsa/hci0/dev_F4_4E_FD_EB_AF_04/hfpag/sink: Initializing SW
D: ../../../src/asound/bluealsa-pcm.c:358: /org/bluealsa/hci0/dev_F4_4E_FD_EB_AF_04/hfpag/sink: Starting
D: ../../../src/asound/bluealsa-pcm.c:215: /org/bluealsa/hci0/dev_F4_4E_FD_EB_AF_04/hfpag/sink: Starting IO loop: 8
D: ../../../src/asound/bluealsa-pcm.c:491: /org/bluealsa/hci0/dev_F4_4E_FD_EB_AF_04/hfpag/sink: Initializing SW
D: ../../../src/asound/bluealsa-pcm.c:498: /org/bluealsa/hci0/dev_F4_4E_FD_EB_AF_04/hfpag/sink: Changing SW avail min: 1000 -> 1
D: ../../../src/asound/bluealsa-pcm.c:491: /org/bluealsa/hci0/dev_F4_4E_FD_EB_AF_04/hfpag/sink: Initializing SW
D: ../../../src/asound/bluealsa-pcm.c:498: /org/bluealsa/hci0/dev_F4_4E_FD_EB_AF_04/hfpag/sink: Changing SW avail min: 1 -> 1000
D: ../../../src/asound/bluealsa-pcm.c:394: /org/bluealsa/hci0/dev_F4_4E_FD_EB_AF_04/hfpag/sink: Stopping
D: ../../../src/asound/bluealsa-pcm.c:147: /org/bluealsa/hci0/dev_F4_4E_FD_EB_AF_04/hfpag/sink: IO thread cleanup
D: ../../../src/asound/bluealsa-pcm.c:394: /org/bluealsa/hci0/dev_F4_4E_FD_EB_AF_04/hfpag/sink: Stopping
D: ../../../src/asound/bluealsa-pcm.c:483: /org/bluealsa/hci0/dev_F4_4E_FD_EB_AF_04/hfpag/sink: Freeing HW
D: ../../../src/asound/bluealsa-pcm.c:433: /org/bluealsa/hci0/dev_F4_4E_FD_EB_AF_04/hfpag/sink: Closing
pi@raspberrypi:~ $ 
`

Logs in the meantime from bluealsa in another terminal
-
`pi@raspberrypi:~ $ sudo bluealsa
bluealsa: D: ../../src/bluez.c:497: Creating media endpoint object: /org/bluez/hci0/A2DP/SBC/Source/1
bluealsa: D: ../../src/bluez.c:413: Registering media endpoint: /org/bluez/hci0/A2DP/SBC/Source/1
bluealsa: D: ../../src/bluez.c:497: Creating media endpoint object: /org/bluez/hci0/A2DP/SBC/Source/2
bluealsa: D: ../../src/bluez.c:413: Registering media endpoint: /org/bluez/hci0/A2DP/SBC/Source/2
bluealsa: D: ../../src/bluez.c:792: Creating hands-free profile object: /org/bluez/HSP/AudioGateway
bluealsa: D: ../../src/bluez.c:726: Registering hands-free profile: /org/bluez/HSP/AudioGateway
bluealsa: D: ../../src/bluez.c:792: Creating hands-free profile object: /org/bluez/HFP/AudioGateway
bluealsa: D: ../../src/bluez.c:726: Registering hands-free profile: /org/bluez/HFP/AudioGateway
bluealsa: D: ../../src/main.c:396: Acquiring D-Bus service name: org.bluealsa
bluealsa: D: ../../src/main.c:401: Starting main dispatching loop
bluealsa: D: ../../src/dbus.c:59: Called: org.bluez.Profile1.NewConnection() on /org/bluez/HFP/AudioGateway
bluealsa: D: ../../src/ba-rfcomm.c:901: Starting RFCOMM loop: HFP Audio Gateway (CVSD)
bluealsa: D: ../../src/ba-rfcomm.c:1286: Created new RFCOMM thread [ba-rfcomm]: HFP Audio Gateway (CVSD)
bluealsa: D: ../../src/sco.c:201: Created SCO dispatcher [ba-sco-dispatch]: hci0
bluealsa: D: ../../src/bluez.c:618: HFP Audio Gateway (CVSD) configured for device F4:4E:FD:EB:AF:04
bluealsa: D: ../../src/ba-transport.c:644: Starting transport: HFP Audio Gateway (CVSD)
bluealsa: D: ../../src/sco.c:77: Starting SCO dispatcher loop: hci0
bluealsa: D: ../../src/ba-transport.c:994: Created new transport thread [ba-sco]: HFP Audio Gateway (CVSD)
bluealsa: D: ../../src/sco.c:245: Starting SCO loop: HFP Audio Gateway (CVSD)
bluealsa: D: ../../src/at.c:161: AT message: SET: command:+BRSF, value:109
bluealsa: D: ../../src/ba-rfcomm.c:108: Sending AT message: RESP: command:+BRSF, value:2272
bluealsa: D: ../../src/ba-rfcomm.c:108: Sending AT message: RESP: command:(null), value:OK
bluealsa: D: ../../src/ba-rfcomm.c:127: RFCOMM: HFP Audio Gateway (CVSD) state transition: 0 -> 2
bluealsa: D: ../../src/at.c:161: AT message: TEST: command:+CIND, value:(null)
bluealsa: D: ../../src/ba-rfcomm.c:108: Sending AT message: RESP: command:+CIND, value:("service",(0-1)),("call",(0,1)),("callsetup",(0-3)),("callheld",(0-2)),("signal",(0-5)),("roam",(0-1)),("battchg",(0-5))
bluealsa: D: ../../src/ba-rfcomm.c:108: Sending AT message: RESP: command:(null), value:OK
bluealsa: D: ../../src/ba-rfcomm.c:127: RFCOMM: HFP Audio Gateway (CVSD) state transition: 2 -> 5
[... cut ...]
bluealsa: D: ../../src/ba-transport.c:966: Closing PCM: 8
bluealsa: D: ../../src/sco.c:332: Releasing SCO due to PCM inactivity
bluealsa: D: ../../src/ba-transport.c:942: Closing SCO: 20
bluealsa: D: ../../src/at.c:161: AT message: CMD: command:+CLCC, value:(null)
bluealsa: W: ../../src/ba-rfcomm.c:1183: Unsupported AT message: CMD: command:+CLCC, value:(null)
bluealsa: D: ../../src/ba-rfcomm.c:108: Sending AT message: RESP: command:(null), value:ERROR
`

I am currently compiling with debug as @Arkq recommended.

alpereira7, 2021-02-04:
@Arkq after I recompiled with debug enabled (`../configure --enable-debug`), the aplay output is the same.
`bluealsa` trace is:
```
pi@raspberrypi:~ $ sudo bluealsa
bluealsa: D: ../../src/bluez.c:497: Creating media endpoint object: /org/bluez/hci0/A2DP/SBC/Source/1
bluealsa: D: ../../src/bluez.c:413: Registering media endpoint: /org/bluez/hci0/A2DP/SBC/Source/1
bluealsa: D: ../../src/bluez.c:497: Creating media endpoint object: /org/bluez/hci0/A2DP/SBC/Source/2
bluealsa: D: ../../src/bluez.c:413: Registering media endpoint: /org/bluez/hci0/A2DP/SBC/Source/2
bluealsa: D: ../../src/bluez.c:792: Creating hands-free profile object: /org/bluez/HSP/AudioGateway
bluealsa: D: ../../src/bluez.c:726: Registering hands-free profile: /org/bluez/HSP/AudioGateway
bluealsa: D: ../../src/bluez.c:792: Creating hands-free profile object: /org/bluez/HFP/AudioGateway
bluealsa: D: ../../src/bluez.c:726: Registering hands-free profile: /org/bluez/HFP/AudioGateway
bluealsa: D: ../../src/main.c:396: Acquiring D-Bus service name: org.bluealsa
bluealsa: D: ../../src/main.c:401: Starting main dispatching loop
bluealsa: D: ../../src/dbus.c:59: Called: org.bluez.Profile1.NewConnection() on /org/bluez/HFP/AudioGateway
bluealsa: D: ../../src/ba-rfcomm.c:1286: Created new RFCOMM thread [ba-rfcomm]: HFP Audio Gateway (CVSD)
bluealsa: D: ../../src/ba-rfcomm.c:901: Starting RFCOMM loop: HFP Audio Gateway (CVSD)
bluealsa: D: ../../src/sco.c:201: Created SCO dispatcher [ba-sco-dispatch]: hci0
bluealsa: D: ../../src/sco.c:77: Starting SCO dispatcher loop: hci0
bluealsa: D: ../../src/bluez.c:618: HFP Audio Gateway (CVSD) configured for device F4:4E:FD:EB:AF:04
bluealsa: D: ../../src/ba-transport.c:644: Starting transport: HFP Audio Gateway (CVSD)
bluealsa: D: ../../src/ba-transport.c:994: Created new transport thread [ba-sco]: HFP Audio Gateway (CVSD)
bluealsa: D: ../../src/sco.c:245: Starting SCO loop: HFP Audio Gateway (CVSD)
bluealsa: D: ../../src/at.c:161: AT message: SET: command:+BRSF, value:109
bluealsa: D: ../../src/ba-rfcomm.c:108: Sending AT message: RESP: command:+BRSF, value:2272
bluealsa: D: ../../src/ba-rfcomm.c:108: Sending AT message: RESP: command:(null), value:OK
bluealsa: D: ../../src/ba-rfcomm.c:127: RFCOMM: HFP Audio Gateway (CVSD) state transition: 0 -> 2
bluealsa: D: ../../src/at.c:161: AT message: TEST: command:+CIND, value:(null)
bluealsa: D: ../../src/ba-rfcomm.c:108: Sending AT message: RESP: command:+CIND, value:("service",(0-1)),("call",(0,1)),("callsetup",(0-3)),("callheld",(0-2)),("signal",(0-5)),("roam",(0-1)),("battchg",(0-5))
bluealsa: D: ../../src/ba-rfcomm.c:108: Sending AT message: RESP: command:(null), value:OK
bluealsa: D: ../../src/ba-rfcomm.c:127: RFCOMM: HFP Audio Gateway (CVSD) state transition: 2 -> 5
bluealsa: D: ../../src/at.c:161: AT message: GET: command:+CIND, value:(null)
bluealsa: D: ../../src/ba-rfcomm.c:108: Sending AT message: RESP: command:+CIND, value:0,0,0,0,0,0,5
bluealsa: D: ../../src/ba-rfcomm.c:108: Sending AT message: RESP: command:(null), value:OK
bluealsa: D: ../../src/ba-rfcomm.c:127: RFCOMM: HFP Audio Gateway (CVSD) state transition: 5 -> 7
bluealsa: D: ../../src/at.c:161: AT message: SET: command:+CMER, value:3,0,0,1
bluealsa: D: ../../src/ba-rfcomm.c:108: Sending AT message: RESP: command:(null), value:OK
bluealsa: D: ../../src/ba-rfcomm.c:127: RFCOMM: HFP Audio Gateway (CVSD) state transition: 7 -> 8
bluealsa: D: ../../src/ba-rfcomm.c:127: RFCOMM: HFP Audio Gateway (CVSD) state transition: 8 -> 9
bluealsa: D: ../../src/at.c:161: AT message: SET: command:+COPS, value:3,0
bluealsa: W: ../../src/ba-rfcomm.c:1183: Unsupported AT message: SET: command:+COPS, value:3,0
bluealsa: D: ../../src/ba-rfcomm.c:108: Sending AT message: RESP: command:(null), value:ERROR
bluealsa: D: ../../src/at.c:161: AT message: GET: command:+COPS, value:(null)
bluealsa: W: ../../src/ba-rfcomm.c:1183: Unsupported AT message: GET: command:+COPS, value:(null)
bluealsa: D: ../../src/ba-rfcomm.c:108: Sending AT message: RESP: command:(null), value:ERROR
bluealsa: D: ../../src/at.c:161: AT message: SET: command:+CMEE, value:1
bluealsa: W: ../../src/ba-rfcomm.c:1183: Unsupported AT message: SET: command:+CMEE, value:1
bluealsa: D: ../../src/ba-rfcomm.c:108: Sending AT message: RESP: command:(null), value:ERROR
bluealsa: D: ../../src/at.c:161: AT message: GET: command:+BTRH, value:(null)
bluealsa: D: ../../src/ba-rfcomm.c:108: Sending AT message: RESP: command:(null), value:OK
bluealsa: D: ../../src/at.c:161: AT message: SET: command:+CLIP, value:1
bluealsa: W: ../../src/ba-rfcomm.c:1183: Unsupported AT message: SET: command:+CLIP, value:1
bluealsa: D: ../../src/ba-rfcomm.c:108: Sending AT message: RESP: command:(null), value:ERROR
bluealsa: D: ../../src/at.c:161: AT message: SET: command:+CCWA, value:1
bluealsa: W: ../../src/ba-rfcomm.c:1183: Unsupported AT message: SET: command:+CCWA, value:1
bluealsa: D: ../../src/ba-rfcomm.c:108: Sending AT message: RESP: command:(null), value:ERROR
bluealsa: D: ../../src/at.c:161: AT message: CMD: command:+CGMI, value:(null)
bluealsa: W: ../../src/ba-rfcomm.c:1183: Unsupported AT message: CMD: command:+CGMI, value:(null)
bluealsa: D: ../../src/ba-rfcomm.c:108: Sending AT message: RESP: command:(null), value:ERROR
bluealsa: D: ../../src/at.c:161: AT message: CMD: command:+CGMM, value:(null)
bluealsa: W: ../../src/ba-rfcomm.c:1183: Unsupported AT message: CMD: command:+CGMM, value:(null)
[... cut ...]
bluealsa: D: ../../src/ba-transport.c:783: PCM dropped: 8
bluealsa: D: ../../src/ba-transport.c:966: Closing PCM: 8
bluealsa: D: ../../src/sco.c:332: Releasing SCO due to PCM inactivity
bluealsa: D: ../../src/ba-transport.c:942: Closing SCO: 20
bluealsa: D: ../../src/at.c:161: AT message: CMD: command:+CLCC, value:(null)
bluealsa: W: ../../src/ba-rfcomm.c:1183: Unsupported AT message: CMD: command:+CLCC, value:(null)
bluealsa: D: ../../src/ba-rfcomm.c:108: Sending AT message: RESP: command:(null), value:ERROR
```

Arkq, 2021-02-05:
Configure project with `--enable-hcitop`. Then, launch `hcitop` tool while playing audio, and check whether there is RX and/or TX activity. Also, try to use HSP profile instead of HFP.

alpereira7, 2021-02-10:
I'm working on it, please don't close the topic yet.

alpereira7, 2021-02-15:
I configured `sudo ../configure --enable-hcitop`. FWI, I had to do `sudo apt-get install libbsd-dev` and `sudo apt-get install ncurses-dev`. I also `make` && `make install`.

I don't know how to "launch `hcitop` tool" while playing audio. I tried to look online but found nothing... Can you explain please?

Arkq, 2021-02-16:
The hcitop tool acts like a standard UNIX top utility but for HCIs on the host - simply run it from command line, that's all.

borine, 2021-07-17:
This issue has been idle for some time now. If there are no further updates this month, I assume it is no longer relevant and can be closed.

borine, 2021-08-10:
Closing as abandoned issue.  This may be related to one of the other SCO issues (see for example #400 and #286) but it is impossible to know without more information.

closed 2021-08-10 by borine