arkq/bluez-alsa #419: bluealsa loopback

LFSveteran, 2021-02-17:
I can use aplay and arecord perfectly with bluealsa, except that my AL3+ Freedom won't work with SCO specified.
Now I want to loopback the audio from bluealsa to my soundcard with another bluetooth device (which has working SCO). 
I tried this with : 

alsaloop -C "bluealsa:DEV=XX:XX:XX:XX:XX:XX,PROFILE=sco" -P "plughw:2,0" -t 20000

But no audio...an alsaloop issue or something todo with bluealsa?

I'm running bluealsa as: bluealsa -i hci0 -p hfp-hf -p hfp-ag -p hsp-hs -p a2dp-sink -p a2dp-source &

borine, 2021-02-18:
Please can you post the version of bluealsa you are using - you need at least bluealsa 3.0.0 for use with alsaloop. Some debug info would also be useful - its not playing any audio, but what is it doing? Run alsaloop with `-v` or even `-vvvvv`, and if possible build bluealsa with debug enabled.

I would say that your latency setting of `-t 20000` (ie 20ms) is ambitious when using bluealsa - the ALSA plugin imposes a minimum delay of 10ms before we even consider delays in the bluetooth stack and alsaloop itself.

alsaloop becomes unstable if it cannot achieve the requested latency - constant rapid restarts of the capture device and infinitely enlarging the avail_min sw param of the playback device are common symptoms.

If you just want to playback from a bluetooth source to your local soundcard then I would recommend to use bluealsa-aplay instead of alsaloop.

LFSveteran, 2021-02-18:
-I'm using version v3.0.0-25-gb4cf31b
-I can rebuild it with debug enabled...
-What value would you recommend for the latency?

The goal is indeed connecting bluealsa with a local soundcard. Capture from bluetooth,send to soundcard and 
capture from soundcard, send to bluetooth (full duplex).
bluealsa-aplay would be ok, but is it capable to handle both directions? (running 2 instances propably).

LFSveteran, 2021-02-18:
The debug information captured:

Scheduler set to Round Robin with priority 99
2225.253497483: D: ../../../src/asound/bluealsa-pcm.c:1054: Getting BlueALSA PCM: CAPTURE 2C:BA:BA:54:4C:01 sco
2225.254480756: D: ../../../src/asound/bluealsa-pcm.c:922: /org/bluealsa/hci0/dev_2C_BA_BA_54_4C_01/hfpag/source: Setting constraints
2225.258476754: D: ../../../src/asound/bluealsa-pcm.c:445: /org/bluealsa/hci0/dev_2C_BA_BA_54_4C_01/hfpag/source: Initializing HW
2225.302814673: D: ../../../src/asound/bluealsa-pcm.c:468: /org/bluealsa/hci0/dev_2C_BA_BA_54_4C_01/hfpag/source: FIFO buffer size: 32768 frames
2225.302874922: D: ../../../src/asound/bluealsa-pcm.c:473: /org/bluealsa/hci0/dev_2C_BA_BA_54_4C_01/hfpag/source: Selected HW buffer: 8 periods x 160 bytes == 1280 bytes
2225.302924006: D: ../../../src/asound/bluealsa-pcm.c:491: /org/bluealsa/hci0/dev_2C_BA_BA_54_4C_01/hfpag/source: Initializing SW
2225.302961646: D: ../../../src/asound/bluealsa-pcm.c:491: /org/bluealsa/hci0/dev_2C_BA_BA_54_4C_01/hfpag/source: Initializing SW
2225.302985431: D: ../../../src/asound/bluealsa-pcm.c:491: /org/bluealsa/hci0/dev_2C_BA_BA_54_4C_01/hfpag/source: Initializing SW
2225.303035353: D: ../../../src/asound/bluealsa-pcm.c:528: /org/bluealsa/hci0/dev_2C_BA_BA_54_4C_01/hfpag/source: Prepared
2225.303058746: D: ../../../src/asound/bluealsa-pcm.c:491: /org/bluealsa/hci0/dev_2C_BA_BA_54_4C_01/hfpag/source: Initializing SW
2225.303094345: D: ../../../src/asound/bluealsa-pcm.c:528: /org/bluealsa/hci0/dev_2C_BA_BA_54_4C_01/hfpag/source: Prepared
2225.303476770: D: ../../../src/asound/bluealsa-pcm.c:358: /org/bluealsa/hci0/dev_2C_BA_BA_54_4C_01/hfpag/source: Starting
2225.303714900: D: ../../../src/asound/bluealsa-pcm.c:215: /org/bluealsa/hci0/dev_2C_BA_BA_54_4C_01/hfpag/source: Starting IO loop: 9
Plug PCM: Hardware PCM card 2 'Logitech Wireless Headset' device 0 subdevice 0
Its setup is:
  stream       : PLAYBACK
  access       : RW_INTERLEAVED
  format       : S16_LE
  subformat    : STD
  channels     : 2
  rate         : 48000
  exact rate   : 48000 (48000/1)
  msbits       : 16
  buffer_size  : 3840
  period_size  : 480
  period_time  : 10000
  tstamp_mode  : NONE
  tstamp_type  : MONOTONIC
  period_step  : 1
  avail_min    : 3120
  period_event : 0
  start_threshold  : 2147483647
  stop_threshold   : 3840
  silence_threshold: 0
  silence_size : 0
  boundary     : 8646911284551352320
  appl_ptr     : 0
  hw_ptr       : 0
Plug PCM: Route conversion PCM (sformat=S16_LE)
  Transformation table:
    0 <- 0
    1 <- 0
Its setup is:
  stream       : CAPTURE
  access       : RW_INTERLEAVED
  format       : S16_LE
  subformat    : STD
  channels     : 2
  rate         : 8000
  exact rate   : 8000 (8000/1)
  msbits       : 16
  buffer_size  : 640
  period_size  : 80
  period_time  : 10000
  tstamp_mode  : NONE
  tstamp_type  : GETTIMEOFDAY
  period_step  : 1
  avail_min    : 80
  period_event : 0
  start_threshold  : 2147483647
  stop_threshold   : 640
  silence_threshold: 0
  silence_size : 0
  boundary     : 5764607523034234880
Slave: BlueALSA PCM: /org/bluealsa/hci0/dev_2C_BA_BA_54_4C_01/hfpag/source
BlueALSA BlueZ device: /org/bluez/hci0/dev_2C_BA_BA_54_4C_01
BlueALSA Bluetooth codec: CVSD
[... 36 lines cut ...]
  silence_threshold: 0
  silence_size : 0
  boundary     : 5764607523034234880
Latency 960 frames, 20000.000us, 20.000000ms (50.0000Hz)
playback plughw:2,0/capture bluealsa:DEV=2C:BA:BA:54:4C:01,PROFILE=sco sync type: SAMPLERATE (SRC_SINC_FASTEST)
playback plughw:2,0/capture bluealsa:DEV=2C:BA:BA:54:4C:01,PROFILE=sco: Samplerate src_ratio update1: 0.16666667
New pitch for playback plughw:2,0/capture bluealsa:DEV=2C:BA:BA:54:4C:01,PROFILE=sco: 1.00000000 (min/max samples = 0/0)
playback plughw:2,0: change avail_min=3124
playback plughw:2,0: change avail_min=3128

<snip>

playback plughw:2,0: change avail_min=5184
playback plughw:2,0: change avail_min=5188
plunderrun for playback plughw:2,0
2225.343890379: D: ../../../src/asound/bluealsa-pcm.c:394: /org/bluealsa/hci0/dev_2C_BA_BA_54_4C_01/hfpag/source: Stopping
2225.344092980: D: ../../../src/asound/bluealsa-pcm.c:147: /org/bluealsa/hci0/dev_2C_BA_BA_54_4C_01/hfpag/source: IO thread cleanup
2225.344194929: D: ../../../src/asound/bluealsa-pcm.c:528: /org/bluealsa/hci0/dev_2C_BA_BA_54_4C_01/hfpag/source: Prepared
2225.344213501: D: ../../../src/asound/bluealsa-pcm.c:358: /org/bluealsa/hci0/dev_2C_BA_BA_54_4C_01/hfpag/source: Starting
2225.344282355: D: ../../../src/asound/bluealsa-pcm.c:215: /org/bluealsa/hci0/dev_2C_BA_BA_54_4C_01/hfpag/source: Starting IO loop: 9
2225.344303850: D: ../../../src/asound/bluealsa-pcm.c:394: /org/bluealsa/hci0/dev_2C_BA_BA_54_4C_01/hfpag/source: Stopping
2225.344331965: D: ../../../src/asound/bluealsa-pcm.c:147: /org/bluealsa/hci0/dev_2C_BA_BA_54_4C_01/hfpag/source: IO thread cleanup
2225.344387825: D: ../../../src/asound/bluealsa-pcm.c:528: /org/bluealsa/hci0/dev_2C_BA_BA_54_4C_01/hfpag/source: Prepared
2225.344403013: D: ../../../src/asound/bluealsa-pcm.c:358: /org/bluealsa/hci0/dev_2C_BA_BA_54_4C_01/hfpag/source: Starting
2225.344674210: D: ../../../src/asound/bluealsa-pcm.c:215: /org/bluealsa/hci0/dev_2C_BA_BA_54_4C_01/hfpag/source: Starting IO loop: 9

borine, 2021-02-18:
> bluealsa-aplay would be ok, but is it capable to handle both directions? 

no, bluealsa-aplay captures audio from bluetooth devices and does playback on an ALSA device. It cannot capture from an ALSA device.

> -What value would you recommend for the latency?

It is impossible to guess this without a lot more information. Start with a high value (say 500ms, `-t 500000`, and when you have that working try successively lower values until you find the limit. For SCO I would expect 100ms to be OK, maybe also 50ms. 20ms may work, but I just think you are getting close to the limit there.

As for the log, nothing in there shouts at me a being obviously wrong. If the last line of the log is really where output stops, it seems that everything is running smoothly from that point. If the capture device continues to repeatedly stop/prepare/start then that indicates the requested latency is unachievable.

LFSveteran, 2021-02-20:
tried with: alsaloop -C "bluealsa:DEV=2C:BA:BA:54:4C:01,PROFILE=sco" -P "plughw:2" -t 2800000 -c 1 -r24000 -fS16_LE
and now I have some sort of audio, not recognizeable but I can hear something.
So I guess alsaloop is not the way to go....maybe something with buffering

borine, 2021-02-20:
> `r24000`

That looks wrong. For HFP it should be `-r 8000` ?

LFSveteran, 2021-02-20:
OOPS.... 24000 was from other test..
now with: 
alsaloop -C "bluealsa:DEV=2C:BA:BA:54:4C:01,PROFILE=sco" -P "plughw:2" -t 100000 -c 1 -r8000 -fS16_LE
it works pretty well

borine, 2021-02-20:
Great - now try with `-t 50000` and then `-t 20000` ; and if that works `-t 10000`. Find the lowest latency that gives unbroken audio with acceptable CPU load.

borine, 2021-02-20:
... you could also try with `--sync-none` - that may reduce cpu load

LFSveteran, 2021-02-20:
the lowest I could get is 55000...
tried with the otherway around simultaneously as well but no success with that yet....

borine, 2021-02-20:
> the lowest I could get is 55000

I just remembered - the version of bluealsa you are using (commit b4cf31b, assuming the version info is reliable) shipped with a default "DELAY" correction setting of 20000. You may find you can achieve slightly lower latency from alsaloop by explicitly setting it to zero with:
``` -C "bluealsa:DEV=2C:BA:BA:54:4C:01,PROFILE=sco,DELAY=0" ```

LFSveteran, 2021-02-20:
yes with DELAY=0 added I can go as low to 15000

borine, 2021-02-20:
I recommend you edit the bluealsa alsa config file (probably `/etc/alsa/conf.d/20-bluealsa.conf`, or possibly "/usr/share/alsa/alsa.conf.d/20-bluealsa.conf " depending on your distribution. Change the default delay adjustment:

LFSveteran, 2021-02-21:
tested with:

alsaloop -P "bluealsa:DEV=2C:BA:BA:54:4C:01,PROFILE=sco,DELAY=0" -C "plughw:1" -t 15000 -c 1 -r8000 -fS16_LE

and I get scramble audio again, like I had in the begining, note that this is now from microphone to bluetooth...
If I add a second instance with:

alsaloop -C "bluealsa:DEV=2C:BA:BA:54:4C:01,PROFILE=sco,DELAY=0" -P "plughw:1" -t 15000 -c 1 -r8000 -fS16_LE 

which will work standalone, I'm now getting those loop messages again:

bluealsa: 2466.921010356: D: ../../src/ba-transport.c:751: PCM resumed: 17
2466.921041026: D: ../../../src/asound/bluealsa-pcm.c:215: /org/bluealsa/hci0/dev_2C_BA_BA_54_4C_01/hfpag/source: Starting IO loop: 9
2466.921053512: D: ../../../src/asound/bluealsa-pcm.c:394: /org/bluealsa/hci0/dev_2C_BA_BA_54_4C_01/hfpag/source: Stopping
2466.921070456: D: ../../../src/asound/bluealsa-pcm.c:147: /org/bluealsa/hci0/dev_2C_BA_BA_54_4C_01/hfpag/source: IO thread cleanup
2466.921119217: D: ../../../src/asound/bluealsa-pcm.c:528: /org/bluealsa/hci0/dev_2C_BA_BA_54_4C_01/hfpag/source: Prepared
2466.921145350: D: ../../../src/asound/bluealsa-pcm.c:358: /org/bluealsa/hci0/dev_2C_BA_BA_54_4C_01/hfpag/source: Starting

changing 15000 to 55000 again won't help.

Guess I need to get "from mic to bluetooth'" working first before I try with both directions

borine, 2021-07-26:
This issue appears to have been abandoned. Closing.

closed 2021-07-26 by borine