arkq/bluez-alsa #420: Add systemd files to repo?

gearhead, 2021-02-21:
Can you add the example systemd and config files as part of the repo as opposed to in the wiki? One way is to do it like Mike Brady does with shairport-sync:
https://github.com/mikebrady/shairport-sync where he has a build flag. Another is just to have these in a folder for repurposing when the package is built.

Arkq, 2021-02-21:
Since we've got bash completion, maybe systemd config should also be there indeed. However, I do not use systemd on daily basis, so it will be rather troublesome for me to make it "right". @borine (or maybe you @gearhead), could you provide such PR?

infirit, 2021-02-21:
Copy paste it from below. The option names are what is commonly used and what most distros expect. You guys probably only need the system part so ignore the user bits then.

https://github.com/blueman-project/blueman/blob/master/configure.ac#L222

and

https://github.com/blueman-project/blueman/blob/master/data/configs/Makefile.am#L13

borine, 2021-02-22:
The example systemd system file on the wiki was based on my experience with Ubuntu and Raspbian/RaspiOS. Other distributions probably have different conventions, I don't know. So perhaps the use of `/etc/default/bluealsa` for options or other details may need some thought.

Also site-specific choices like the service user name etc should really be in overlay files, rather than a distributed file, so that the info is not destroyed on every upgrade.

Perhaps this issue could provoke some community input on how various user groups would prefer to have systemd (or other init) configs structured and installed. FWIW I generally prefer to have such things as uninstalled examples, but perhaps I am a little old-fashioned !!

gearhead, 2021-02-22:
FWIW, with systemd, if something changes (it always does) requiring a change to the basic *.service or *.socket file, then it is pushed to /usr/lib/systemd/system (at least on Arch, this is where all 'package' service files go). If a user has any customized bits that they want to add/adjust, systemd allows for edits which are then put in /etc/systemd/system. If there is a service file in this folder, it supersedes anything 'default' installed in /usr/lib. We are adding the bluealsa package to the RuneAudio player. Any customized service files we need are put in /etc/systemd/system to ensure it works as we want it to.
IMO, the 'basic' configs and service files should provide the basic functionality of the package and the user can read further on the wiki or readme to learn how to adjust it to his/her specific case. 
In the case of Bluealsa, in my opinion, there should be a bluealsa.service and a bluealsa-aplay.service and the associated configs to provide the basic functionality of basic source and sink.  Any other 'adjustments' such as ofono or whatever be left as an exercise in reading the manual/wiki/readme ansd specific customization for the user.
In our case, we copied and pasted the service files from the bluealsa wiki and made some edits to the configs to get what we need for RuneAudio.

graham8, 2021-02-22:
> Any customized service files we need are put in /etc/systemd/system to ensure it works as we want it to.

If you are overwriting the default unit files, then why are you putting them in /usr/lib at all? AIUI, service files in /etc would cause the version in /usr/lib/ to be ignored completely, so why not just put your version in /usr/lib?  

If the default unit files are added, as suggested, it may be better to use drop-ins - any *.conf file in /etc/systemd/system/bluealsa.service.d overwrites/adds only those parameters specified in it.  Or is that what you meant?

Default location for unit files in Debian seems to be /lib/systemd/system, not /usr/lib/....

gearhead, 2021-02-22:
We are making a RPi based music player on Arch Linux and some service files are customized to work with our implementation. If a user updates the underlying OS, we want it to continue to run and the customized *.service files in /etc/systemd/system do not get overwritten with updated versions. If the 'basic default' service file overwrites a customized one, it will fail to start in some cases. I wish all customization were all in /etc/*.conf files, but not all packages work that way. My thought is that the default basic package should set up a conf file and a service so that it runs and any service customization go into /etc/systemd/system if needed. 
I believe that the default service file for bluealsa and bluealsa-aplay will go into /usr/lib (/lib on RasopiOS?) as we need no customizations for them to run. They just do not exist there, yet, so we put them in /etc/systemd/system so we do not forget to install them. That said, we do need to review our customized service versions to make sure they are still needed/desired to make a functional image.

graham8, 2021-02-22:
OK.  So I don't understand your request, then.  Why add the unit files to this repo if you're not using them?

I don't mind one way or the other - I don't think bluez-alsa is in the debian repos that we use at OSMC so there's no risk of the unit files being overwritten.  But if there was, we would use a drop-in for any customising and that would not be overwritten.

gearhead, 2021-02-22:
I will be using them. I use the ones from the wiki, now but put them in the /etc folder as they are not built with/part of the package. When they are built as part of the package, I will delete the ones I have in /etc. I'm just requesting that a 'known functional' service files be part of the repo.

infirit, 2021-02-22:
> The example systemd system file on the wiki was based on my experience with Ubuntu and Raspbian/RaspiOS. Other distributions probably have different conventions, I don't know. So perhaps the use of `/etc/default/bluealsa` for options or other details may need some thought.

If you confirmed it works properly on debian/ubuntu and whatever you are using its all good. Don't spend too much time trying to accommodate every distribution as they will patch things anyway.
 
> Perhaps this issue could provoke some community input on how various user groups would prefer to have systemd (or other init) configs structured and installed. FWIW I generally prefer to have such things as uninstalled examples, but perhaps I am a little old-fashioned !!

The biggest reason to include this in the repo/install is that certain distro's refuse to include anything that's not part of the source.

When bluealsa is packaged more booadly maintainers will come to you with patches and improvements. Or you just peak once in a while to see what they modify, when they all do the same consider including this upstream.

Arkq, 2021-07-18:
I've just added a PR for systemd integration - built-in installation of service files. However, I'm not a systemd user, so it's rather hard for me to verify/know whether these service files are OK. I've taken service files from bluez-alsa wiki and modify them slightly (maybe that was unnecessary/wrong) after comparison with service files installed on my system. One (I think) important thing which I've changed is the EnvironmentFile location. On Gentoo such files are located in `/etc/sysconfig` not `/etc/default`. The `/etc/deafult` does also exist, but I think that files there are not for systemd. So, I'd like to ask for verification of PR #473.

borine, 2021-07-18:
> On Gentoo such files are located in /etc/sysconfig not /etc/default

`/etc/default` is the Debian way of doing this
`/etc/systconfig` is used by Fedora as well as Gentoo

Neither, AFAIAA, are part of the `systemd` spec or recommended usage, they are distribition-specific. For Debian , at least, I think the choice is a hangover from the pre-systemd init.

Other distributions may choose not to use EnvironmentFile at all, and instead use a local overrride file such as `/etc/systemd/system/bluealsa.service.d/options.conf`

borine, 2021-07-18:
.. a quick web search later, I came across this interesting piece by Lennart Poettering concerning `/etc/default` and `/etc/sysconfig` with systemd. Unfortunately it doesn't give us a simple answer to this issue though. See:
http://0pointer.de/blog/projects/on-etc-sysinit.html

Arkq, 2021-07-18:
> Unfortunately it doesn't give us a simple answer to this issue though.

Yeap. But I think that the blog was mostly about shell-with-logic-like configuration files, not simple assignments for env variables. Anyway, there is one important statement:

> Unit files are simple, declarative descriptions, that usually do not consist of more than 6 lines or so. They can easily be generated, parsed without a Bourne interpreter and understood by the reader. Also, they are very easy to modify: just copy them from /lib/systemd/system to /etc/systemd/system and edit them there, where they will not be modified by the package manager. 

So, I think that we should go for that. Upstream (bluez-alsa) unit files will be more like templates with "good-enough" defaults, that all. So, I will remove EnvironmentFile from unit files.

commit 3c8b568b (2021-07-18): Integration with systemd - install service files

Fixes #420
commit 54afaf04 (2021-07-26): Integration with systemd - install service files

Fixes #420
commit d0d0483e (2021-07-27): Integration with systemd - install service files

Fixes #420
commit 5bf4148d (2021-07-27): Integration with systemd - install service files

Fixes #420
closed 2021-07-27 by Arkq