arkq/bluez-alsa #421: how to force bluealsa-aplay to run at pairing

gearhead, 2021-02-22:
We are running bluealsa in a systemd environment on an RPi. We primarily want to use bluealsa-aplay (sink) so that the RPi becomes a BT 'speaker'. We also want to be able to use it as a source for BT headphones for example. 
For our primary 'sink' scenario, we have found that if nothing is paired or connected, bluealsa-aplay will eventually die. If bluealsa-aplay is not running when a device is paired, the audio services are not advertised. After pairing, we cannot send any sound. TO fix this, we have to delete it on both ends (phone and RPi) and start over. We then manually restart the bluealsa-aplay service then pair. When we do this, the audio services show up on the paired device so they can be selected and used. Is there something we can do to make sure bluealsa-aplay is running when we pair? What is the best way to have it auto start when we are using it?

borine, 2021-02-22:
> Is there something we can do to make sure bluealsa-aplay is running when we pair?

You seem to have confused bluealsa and bluealsa-aplay. It seems likely that something else on the system is starting/stopping bluealsa and that is why audio services are not always advertised.

Please can you post the output of:
```systemctl --list-dependencies bluealsa```
``` systemctl --list-dependencies bluealsa-aplay```

Arkq, 2021-02-22:
> we have found that if nothing is paired or connected, bluealsa-aplay will eventually die

What do you mean that it will eventually die? It exists due to a signal, e.g. SIGINT or due to a crash?

With current design, `bluealsa-aplay` should be resilient to any instabilities in `bluealsa` or `bluez` or `ALSA`. So, if it exits unexpectedly (not due to one of exit signals https://github.com/Arkq/bluez-alsa/blob/master/utils/aplay/aplay.c#L816), please, try to reproduce it and report as a bug. Prior to v3.0.0 there was a bug which could kill `bluealsa-aplay` in case of D-Bus connection lost: https://github.com/Arkq/bluez-alsa/commit/a918a3fbda7d5bd6f220aea656dbb4bcc800d463

> If bluealsa-aplay is not running when a device is paired, the audio services are not advertised. After pairing, we cannot send any sound.

As far as I know, `bluealsa-aplay` does not advertise any audio service... What kind of service you have in mind?

> Is there something we can do to make sure bluealsa-aplay is running when we pair?

If it somehow does not work during BT pairing, it's a bug - I will check that.

> What is the best way to have it auto start when we are using it?

Start it at boot time. But remember that it will acquire SCO connection as soon as it becomes available in bluealsa. Also, since currently bluealsa service can support only one client per PCM, when using `bluealsa-aplay` it will be not possible to use source PCM with other clients.

gearhead, 2021-02-22:
```
# systemctl list-dependencies bluealsa
bluealsa.service
● ├─bluetooth.service
● ├─dbus.socket
● ├─system.slice
● ├─tmp.mount
● ├─var-tmp.mount
● └─sysinit.target
●   ├─dev-hugepages.mount
●   ├─dev-mqueue.mount
●   ├─haveged.service
●   ├─kmod-static-nodes.service
●   ├─ldconfig.service
●   ├─proc-sys-fs-binfmt_misc.automount
●   ├─sys-fs-fuse-connections.mount
●   ├─sys-kernel-config.mount
●   ├─sys-kernel-debug.mount
●   ├─sys-kernel-tracing.mount
●   ├─systemd-ask-password-console.path
●   ├─systemd-binfmt.service
●   ├─systemd-boot-system-token.service
●   ├─systemd-firstboot.service
●   ├─systemd-hwdb-update.service
●   ├─systemd-journal-catalog-update.service
●   ├─systemd-journal-flush.service
●   ├─systemd-journald.service
●   ├─systemd-machine-id-commit.service
●   ├─systemd-modules-load.service
●   ├─systemd-random-seed.service
●   ├─systemd-repart.service
●   ├─systemd-sysctl.service
●   ├─systemd-sysusers.service
●   ├─systemd-timesyncd.service
●   ├─systemd-tmpfiles-setup-dev.service
●   ├─systemd-tmpfiles-setup.service
●   ├─systemd-udev-trigger.service
●   ├─systemd-udevd.service
●   ├─systemd-update-done.service
●   ├─systemd-update-utmp.service
●   ├─cryptsetup.target
●   ├─local-fs.target
●   │ ├─boot.mount
●   │ ├─srv-http-tmp.mount
●   │ ├─systemd-remount-fs.service
●   │ ├─tmp.mount
●   │ ├─var-log-runeaudio.mount
●   │ ├─var-log.mount
●   │ └─var-tmp.mount
●   └─swap.target
```
bluealsa-aplay
```
# systemctl list-dependencies bluealsa-aplay
bluealsa-aplay.service
● ├─bluealsa.service
● ├─system.slice
● ├─tmp.mount
● ├─var-tmp.mount
● └─sysinit.target
●   ├─dev-hugepages.mount
●   ├─dev-mqueue.mount
●   ├─haveged.service
●   ├─kmod-static-nodes.service
●   ├─ldconfig.service
●   ├─proc-sys-fs-binfmt_misc.automount
●   ├─sys-fs-fuse-connections.mount
●   ├─sys-kernel-config.mount
●   ├─sys-kernel-debug.mount
●   ├─sys-kernel-tracing.mount
●   ├─systemd-ask-password-console.path
●   ├─systemd-binfmt.service
●   ├─systemd-boot-system-token.service
●   ├─systemd-firstboot.service
●   ├─systemd-hwdb-update.service
●   ├─systemd-journal-catalog-update.service
●   ├─systemd-journal-flush.service
●   ├─systemd-journald.service
●   ├─systemd-machine-id-commit.service
●   ├─systemd-modules-load.service
●   ├─systemd-random-seed.service
●   ├─systemd-repart.service
●   ├─systemd-sysctl.service
●   ├─systemd-sysusers.service
●   ├─systemd-timesyncd.service
●   ├─systemd-tmpfiles-setup-dev.service
●   ├─systemd-tmpfiles-setup.service
●   ├─systemd-udev-trigger.service
●   ├─systemd-udevd.service
●   ├─systemd-update-done.service
●   ├─systemd-update-utmp.service
●   ├─cryptsetup.target
●   ├─local-fs.target
●   │ ├─boot.mount
●   │ ├─srv-http-tmp.mount
●   │ ├─systemd-remount-fs.service
●   │ ├─tmp.mount
●   │ ├─var-log-runeaudio.mount
●   │ ├─var-log.mount
●   │ └─var-tmp.mount
●   └─swap.target
```
I am running bluealsa 3.0.0 and bluez-5.55. I just rebooted and tried 'from scratch' again to pair my phone as a source to document what it does. After a reboot, bluealsa-aplay is still running. I had noticed that sometmes it would not run and do not know why it was not running when I tried this before. It is set up as a systemd service and 'enabled' so it should start at boot.  Maybe it crashed before I paired? So on this attempt, the phone pairs. It does not ask to authorize any services, though. It is only after I make a connection from the phone (Connect) that bluetoothctl comes back and asks if I want to:
```
Authorize service
[agent] Authorize service 0000110d-0000-1000-8000-00805f9b34fb (yes/no): yes
```
I have to answer yes here to get "HD audio". If I had closed bluetoothctl open, I would not have caught this request and then no HD audio pathway is set up. I have paired before and I noticed that all I can do is share my phonebook... Previously, it would ask for this as soon as I paired. When is this supposed to happen? Is there some bluetooth config I can set to make this 'default' for these installs?

borine, 2021-02-22:
So it is important to know that bluealsa-aplay has nothing to do with pairing or connecting bluetooth devices - you are looking at the wrong service. Bluetooth audio is advertised by the bluealsa service and it is that which you need to have running for service discovery on your phone to see the full set of profiles available. You need to work out why the bluealsa service is not just permanently running from boot to shutdown. I see nothing unusual in the dependency output above. What is the output of:
```systemctl cat bluealsa```
and
```systemctl is-enabled bluealsa```

gearhead, 2021-02-22:
This is what I get. 
```
# systemctl cat bluealsa
# /etc/systemd/system/bluealsa.service
[Unit]
Description=Bluealsa daemon
Documentation=https://github.com/Arkq/bluez-alsa/
After=dbus-org.bluez.service
Requires=dbus-org.bluez.service

[Service]
Type=dbus
BusName=org.bluealsa
EnvironmentFile=-/etc/default/bluealsa
ExecStart=/usr/bin/bluealsa $OPTIONS
Restart=on-failure
ProtectSystem=strict
ProtectHome=true
PrivateTmp=true
PrivateDevices=true
RemoveIPC=true
RestrictAddressFamilies=AF_UNIX AF_BLUETOOTH
; Also non-privileged can user be used
; this example assumes a user and group called 'bluealsa' exist
;User=bluealsa
;Group=bluealsa
;NoNewPrivileges=true

[Install]
WantedBy=bluetooth.target

# systemctl is-enabled bluealsa
enabled
```

borine, 2021-02-22:
Hmm, so that's the wiki example with no changes or drop-ins. It is supposed to start when bluetooth starts and run until bluetooth stops. So the next question, is the bluealsa service failing? Is there any clue in the systemd journal:
```sudo journalctl -xeu bluealsa```

infirit, 2021-02-22:
I think at pair is the wrong time to do anything yet. I would instead write a small script/program that uses  dbus to watch for the "Connected" property on org.bluez to change (or whatever bluealsa exposes). Then have it spawn/kill your sink/source according.

Assuming of course you have bluealsa service running.

gearhead, 2021-02-22:
@borine, When I checked for bluealsa, it always seemd to be running. It was bluealsa-aplay which was stopped when I checked. I do not know why and the image I was using when I noted this has been updated. 
Let me back up a bit. I have been able to pair my phone as a source, but have noted that from time to time that when I pair my phone that no audio device shows up on the phone as an option to play to. I am working with another guy on this project and he had the same issue. I assumed that something was wrong as both of us have the bluealsa and bluaalsa-aplay enabled, so it 'should' start at boot and be running, whenever we try.
Previously, when I paired my phone, bluetoothctl would pair and then would ask me if I want to 'authorize' the service and I need to answer 'yes' in order to play anything. If I did not answer yes or not wait for it to prompt, I was only able to share contacts...

What we are trying to do is make a UI so a user can pair and connect from a web page and we always want the HD audio device to be enabled so it will show up on the phone. Like I said, I rebooted and paired it again and both bluealsa and bluealsa-aplay continued to run and I was able to pair,. but when I connected was when it asked to authorize. I can try removing my phone and waiting a while longer and try again to see if either service dies in the meantime and see if I can reliably get the authorize prompt.

borine, 2021-02-22:
> when I connected was when it asked to authorize

When you pair, then you need an agent to handle the authorization. There are different agents available with bluetoothctl. If you are using the default agent then the normal pairing sequence goes something like:
```
[bluetooth]# pair xx:xx:xx:xx:xx:xx
[agent]PIN code: ####
[agent]Authorize service service_uuid (yes/no): yes
[bluetooth]# trust xx:xx:xx:xx:xx:xx
```

If the connection request originates from the 'phone (which is the usual way to do this), then the bluealsa host should now accept the connection without authorization **provided the phone is trusted**. If you do `bluetoothctl info  xx:xx:xx:xx:xx:xx` do you get `Trusted: yes` in the output? If not then you need to use bluetoothctl to trust that device. If it is trusted, then I'm afraid I'm out of ideas. But you can be sure it has nothing to do with bluealsa-aplay.

infirit, 2021-02-22:
One thing to be aware of that a lot of devices are auto paired by BlueZ. For example my Sennheiser 4.40 BT has 0000 as pin but BlueZ will just handle this for me. So generally don't explicitly pair but connect and trust, if needed BlueZ will invoke the agent.

borine, 2021-06-04:
not a bluez-alsa issue.

closed 2021-06-04 by borine