arkq/bluez-alsa #422: Not able to run bluealsa properly: Couldn't acquire D-Bus name: org.bluealsa

francoisthire, 2021-02-27:
I have the following error

```bash
odroid@odroid:~$ bluealsa --device=C0:28:8D:92:18:CB
bluealsa: D: ../../src/ofono.c:638: Checking oFono service presence
bluealsa: D: ../../src/bluez.c:792: Creating hands-free profile object: /org/bluez/HSP/AudioGateway
bluealsa: D: ../../src/bluez.c:726: Registering hands-free profile: /org/bluez/HSP/AudioGateway
bluealsa: D: ../../src/bluez.c:792: Creating hands-free profile object: /org/bluez/HFP/AudioGateway
bluealsa: D: ../../src/bluez.c:726: Registering hands-free profile: /org/bluez/HFP/AudioGateway
bluealsa: D: ../../src/main.c:396: Acquiring D-Bus service name: org.bluealsa
bluealsa: D: ../../src/main.c:401: Starting main dispatching loop
bluealsa: E: ../../src/main.c:84: Couldn't acquire D-Bus name: org.bluealsa
bluealsa: D: ../../src/main.c:404: Exiting main loop
```

I have just compiled the project from sources:

```bash
odroid@odroid:~$ bluealsa --version
v3.0.0-40-g42dd5c2
```

I don't understand the error and where it might come from.

borine, 2021-02-27:
see https://github.com/Arkq/bluez-alsa#troubleshooting, item 3.

Permission for `root` to run bluealsa is granted by the file `/etc/dbus-1/system.d/bluealsa.conf` - that file should have been installed by the `make install` step. If the file is present, then possibly the system dbus needs to be restarted - a reboot appears to be the most reliable way of achieving that.

francoisthire, 2021-02-27:
Thank you. I have now another error, but I don't know if I should open a new issue:

```
odroid@odroid:~$ aplay -D bluealsa:DEV=C0:28:8D:92:18:CB,PROFILE=a2dp 
D: ../../../src/asound/bluealsa-pcm.c:1054: Getting BlueALSA PCM: PLAYBACK C0:28:8D:92:18:CB a2dp
ALSA lib ../../../src/asound/bluealsa-pcm.c:1058:(_snd_pcm_bluealsa_open) Couldn't get BlueALSA PCM: PCM not found
aplay: main:830: audio open error: No such device
```

I am paired with the device so I really don't understand the problem.

Arkq, 2021-02-28:
There is a number of reasons why this might happen:
1. You are running BlueALSA and PulseAudio at the same time (and the headset is actually connected with PulseAudio)
2. You have connected phone not headphones, hence, you've got CAPTURE capability not PLAYBACK (but you are trying to access playback device).
3. You have connected headset, but it does not suport A2DP profile, only HFP/HSP (you will have to use `PROFILE=sco`, then)
4. MAC address of your BT device is not `C0:28:8D:92:18:CB`

Post here the output of `bluealsa-aplay -L` it might give some hint what might be wrong with your setup.

francoisthire, 2021-02-28:
It is hard to understand the problem, but I think I finally managed to have something working. 

To be clear, I was not on case 2, 3 and 4. And the output of `bluealsa-aplay -L` was empty, no device were detected.

I had a hint with running `systemctl status bluetooth`:

```bash
● bluetooth.service - Bluetooth service
     Loaded: loaded (/lib/systemd/system/bluetooth.service; enabled; vendor preset: enabled)
     Active: active (running) since Sat 2021-02-27 20:01:16 UTC; 1min 34s ago
       Docs: man:bluetoothd(8)
   Main PID: 18853 (bluetoothd)
     Status: "Running"
      Tasks: 1 (limit: 3832)
     Memory: 872.0K
     CGroup: /system.slice/bluetooth.service
             └─18853 /usr/lib/bluetooth/bluetoothd

Feb 27 20:01:16 odroid bluetoothd[18853]: Starting SDP server
Feb 27 20:01:16 odroid bluetoothd[18853]: Bluetooth management interface 1.14 initialized
Feb 27 20:01:16 odroid bluetoothd[18853]: Endpoint registered: sender=:1.2107 path=/MediaEndpoint/A2DPSink/sbc
Feb 27 20:01:16 odroid bluetoothd[18853]: Endpoint registered: sender=:1.2107 path=/MediaEndpoint/A2DPSource/sbc
Feb 27 20:01:16 odroid bluetoothd[18853]: RFCOMM server failed for Headset Voice gateway: rfcomm_bind: Address already in use (98)
Feb 27 20:01:16 odroid bluetoothd[18853]: Endpoint registered: sender=:1.31 path=/MediaEndpoint/A2DPSink/sbc
Feb 27 20:01:16 odroid bluetoothd[18853]: Endpoint registered: sender=:1.31 path=/MediaEndpoint/A2DPSource/sbc
Feb 27 20:01:16 odroid bluetoothd[18853]: RFCOMM server failed for Headset Voice gateway: rfcomm_bind: Address already in use (98)
Feb 27 20:01:16 odroid bluetoothd[18853]: RFCOMM server failed for :1.2107/Profile/HSPHSProfile/00001108-0000-1000-8000-00805f9b34fb: rfcom>
```
At some point I did something (restart the Bluetooth service?), but I saw this:

```bash
Feb 27 20:01:30 odroid bluetoothd[18853]: /org/bluez/hci0/dev_C0_28_8D_92_18_CB/sep1/fd0: fd(32) ready
```

And everything started to work at this point.

closed 2021-02-28 by francoisthire

flatsiedatsie, 2022-08-10:
> `/etc/dbus-1/system.d/bluealsa.conf` - that file should have been installed by the `make install` step

For me the file existed, but it was incomplete. Added the `bluealsa` user solved the problem.

```
<!-- This configuration file specifies the required security policies
     for BlueALSA core daemon to work. -->

<!DOCTYPE busconfig PUBLIC "-//freedesktop//DTD D-BUS Bus Configuration 1.0//EN"
 "http://www.freedesktop.org/standards/dbus/1.0/busconfig.dtd">
<busconfig>

  <!-- ../system.conf have denied everything, so we just punch some holes -->

  <policy user="root">
    <allow own_prefix="org.bluealsa"/>
    <allow send_destination="org.bluealsa"/>
  </policy>

  <policy user="bluealsa">
    <allow own_prefix="org.bluealsa"/>
    <allow send_destination="org.bluealsa"/>
  </policy>

  <policy group="audio">
    <allow send_destination="org.bluealsa"/>
  </policy>

</busconfig>
```

Arkq, 2022-08-11:
> For me the file existed, but it was incomplete. Adding the bluealsa user solved the problem

That is strange... So, it seems that you are running either bluealsa system service or bluealsa client (`bluealsa-aplay`, `aplay`, etc.) as a `bluealsa` user. Original D-Bus configuration file contains policy for `audio` group. So, is should not matter what is the user name as long as it is in the `audio` group.

flatsiedatsie, 2022-08-12:
Thanks, you were right, the bluealsa user wasn't part of the audio group. I've fixed that in my install script.

flatsiedatsie, 2022-08-12:
Am I correct in understanding that the wiki doesn't really mention this?
https://github.com/Arkq/bluez-alsa/wiki/Systemd-integration

Arkq, 2022-08-12:
> Should this... ...be this?

Not likely. I think that the wiki is OK. The "bluealsa" user which runs `bluealsa` service does not need to be in the "audio" group. Also "bluealsa-aplay" user does not need to be in the "bluetooth" group. Dropping `sudo` means that you are already logged in as root. However, it is considered a bad practice to work as a root user, unless you are doing some system maintenance and it is inconvenient to use `sudo` for every command.

flatsiedatsie, 2022-08-12:
Apologies, I copied that from my install script, which runs as root. BlueAlsa itself runs as user `bluealsa`.

Earlier you said:
> So, is should not matter what is the user name as long as it is in the audio group.

Doesn't that imply that the BlueAlsa user should be added to the audio group? In the current example on the wiki the bluealsa user is never added to the audio group (as far as I can tell).

borine, 2022-08-12:
> Doesn't that imply that the BlueAlsa user should be added to the audio group?

The `audio` group is a convention on most unix-like systems that permits members to use attached sound devices. If you look at the actual device nodes for sound devices:
```
$ ls -l /dev/snd/
total 0
drwxr-xr-x  2 root root       60 Aug 12 13:14 by-path
crw-rw----+ 1 root audio 116, 11 Aug 12 13:14 controlC0
crw-rw----+ 1 root audio 116,  9 Aug  9 07:20 hwC0D0
crw-rw----+ 1 root audio 116, 10 Aug  9 07:20 hwC0D2
crw-rw----+ 1 root audio 116,  3 Aug  9 14:02 pcmC0D0c
crw-rw----+ 1 root audio 116,  2 Aug 11 15:16 pcmC0D0p
crw-rw----+ 1 root audio 116,  8 Aug 11 11:49 pcmC0D10p
crw-rw----+ 1 root audio 116,  4 Aug 12 07:59 pcmC0D3p
crw-rw----+ 1 root audio 116,  5 Aug 11 11:48 pcmC0D7p
crw-rw----+ 1 root audio 116,  6 Aug 11 11:48 pcmC0D8p
crw-rw----+ 1 root audio 116,  7 Aug 11 15:48 pcmC0D9p
crw-rw----+ 1 root audio 116,  1 Aug  9 07:20 seq
crw-rw----+ 1 root audio 116, 33 Aug  9 07:20 timer
```
You will see that they have rw permissions for group `audio` and no access for `others`.

BlueALSA extends this convention to Bluetooth sound devices (i.e. BlueALSA PCMs and CTLs) by using D-Bus to limit access to the BlueALSA service so that only `audio` group members can connect to it. The BlueALSA service itself does not need to use any "real" sound device, nor does it need permission to "talk to itself", so `audio` group membership is not required.

nasu1105, 2022-10-14:
I have the same problem.
I don't use bluezalsa by this message `bluealsa: E: ../../src/main.c:133: Couldn't acquire D-Bus name. Please check D-Bus configuration. Requested name: org.bluealsa`
Can you give me some advice?
 My raspberry pi 4 have been installed ldac and openaptx.

`osmc@osmc:~$ git clone https://github.com/Arkq/bluez-alsa.git`
`osmc@osmc:~$ cd bluez-alsa`
`osmc@osmc:~/bluez-alsa$ autoreconf --install --force`
`osmc@osmc:~/bluez-alsa$ mkdir build`
`osmc@osmc:~/bluez-alsa$ cd build`
`osmc@osmc:~/bluez-alsa/build$ ../configure  --enable-debug  --enable-aptx  --enable-aptx-hd  --enable-ldac  --enable-cli  --enable-systemd  --enable-a2dpconf`
`osmc@osmc:~/bluez-alsa/build$ make`
`osmc@osmc:~/bluez-alsa/build$ sudo make install`
`osmc@osmc:~/bluez-alsa/build$ sudo systemctl enable bluealsa.service`
`osmc@osmc:~/bluez-alsa/build$ sudo systemctl enable bluealsa-aplay.service`
`osmc@osmc:~/bluez-alsa/build$ sudo adduser --system --group --no-create-home bluealsa`
`osmc@osmc:~/bluez-alsa/build$ sudo adduser bluealsa-aplay audio`
`osmc@osmc:~/bluez-alsa/build$ systemctl daemon-reload`

`osmc@osmc:~/bluez-alsa/build$ bluealsa -p a2dp-source a2dp-sink`
`bluealsa: D: ../../src/main.c:589: Starting main dispatching loop`
`bluealsa: E: ../../src/main.c:133: Couldn't acquire D-Bus name. Please check D-Bus configuration. Requested name: org.bluealsa`

`osmc@osmc:~/bluez-alsa$ systemctl status bluealsa`
`Warning: The unit file, source configuration file or drop-ins of bluealsa.service changed on disk. Run 'systemctl daemon>`
`* bluealsa.service - BlueALSA service`
`     Loaded: loaded (/lib/systemd/system/bluealsa.service; enabled; vendor preset: enabled)`
`     Active: failed (Result: exit-code) since Fri 2022-10-14 19:16:05 JST; 1h 12min ago`
`       Docs: man:bluealsa(8)`
`    Process: 566 ExecStart=/usr/bin/bluealsa -p a2dp-source -p a2dp-sink (code=exited, status=127)`
`   Main PID: 566 (code=exited, status=127)`
`        CPU: 72ms`
`Oct 14 19:16:05 osmc systemd[1]: bluealsa.service: Scheduled restart job, restart counter is at 5.`
`Oct 14 19:16:05 osmc systemd[1]: Stopped BlueALSA service.`
`Oct 14 19:16:05 osmc systemd[1]: bluealsa.service: Start request repeated too quickly.`
`Oct 14 19:16:05 osmc systemd[1]: bluealsa.service: Failed with result 'exit-code'.`
`Oct 14 19:16:05 osmc systemd[1]: Failed to start BlueALSA service.`

Thank you.

Arkq, 2022-10-14:
@nasu1105,  when using default D-Bus configuration, only root is allowed to run BlueALSA service, so this: `osmc@osmc:~/bluez-alsa/build$ bluealsa -p a2dp-source a2dp-sink` will not work (and there is a typo in command line, it should be `bluealsa -p a2dp-source -p a2dp-sink`). In order to allow other users to run BlueALSA service you will have add such policy to "/etc/dbus-1/system.d/bluealsa.conf".

> systemctl status bluealsa

There is not much info in here, please post logs from `journalctl -u bluealsa`

nasu1105, 2022-10-14:
Thank you for your comment, Arkq.
I add such policy to "/etc/dbus-1/system.d/bluealsa.conf".

  <policy user="osmc">
    <allow own_prefix="org.bluealsa"/>
    <allow send_destination="org.bluealsa"/>
  </policy>

`osmc@osmc:~$bluealsa -p a2dp-source -p a2dp-sink` 
` bluealsa: D: ../../src/main.c:589: Starting main dispatching loop` 
` bluealsa: D: ../../src/main.c:114: Acquired D-Bus service name: org.bluealsa` 
` bluealsa: D: ../../src/bluealsa-dbus.c:368: Registering D-Bus manager: /org/bluealsa` 
` bluealsa: D: ../../src/bluez.c:679: Registering battery provider: /org/bluez/hci0/battery` 
` bluealsa: D: ../../src/bluez.c:699: BlueZ battery provider support not available` 
` bluealsa: D: ../../src/bluez.c:493: Creating media endpoint object: /org/bluez/hci0/A2DP/SBC/source/1` 
` bluealsa: D: ../../src/bluez.c:402: Registering media endpoint: /org/bluez/hci0/A2DP/SBC/source/1` 
` bluealsa: D: ../../src/bluez.c:493: Creating media endpoint object: /org/bluez/hci0/A2DP/SBC/source/2` 
` bluealsa: D: ../../src/bluez.c:402: Registering media endpoint: /org/bluez/hci0/A2DP/SBC/source/2` 
` bluealsa: D: ../../src/bluez.c:493: Creating media endpoint object: /org/bluez/hci0/A2DP/SBC/sink/1` 
` bluealsa: D: ../../src/bluez.c:402: Registering media endpoint: /org/bluez/hci0/A2DP/SBC/sink/1` 
` bluealsa: D: ../../src/bluez.c:493: Creating media endpoint object: /org/bluez/hci0/A2DP/SBC/sink/2` 
` bluealsa: D: ../../src/bluez.c:402: Registering media endpoint: /org/bluez/hci0/A2DP/SBC/sink/2` 
and freeze then
`osmc@osmc:~$bluealsa -p a2dp-source -p a2dp-sink` 
`bluealsa: D: ../../src/main.c:589: Starting main dispatching loop`
`bluealsa: E: ../../src/main.c:133: Couldn't acquire D-Bus name. Please check D-Bus configuration. Requested name: org.bluealsa`

`osmc@osmc:~$ journalctl -u bluealsa`
`Oct 14 19:16:05 osmc bluealsa[566]: /usr/bin/bluealsa: error while loading shared libraries: libx264.so.160: cannot enable executable stack as shared object requires: Operation not permitted`
`Oct 14 19:16:05 osmc systemd[1]: bluealsa.service: Main process exited, code=exited, status=127/n/a`
`Oct 14 19:16:05 osmc systemd[1]: bluealsa.service: Failed with result 'exit-code'.`
`Oct 14 19:16:05 osmc systemd[1]: Failed to start BlueALSA service.`
`Oct 14 19:16:05 osmc systemd[1]: bluealsa.service: Scheduled restart job, restart counter is at 5.`
`Oct 14 19:16:05 osmc systemd[1]: Stopped BlueALSA service.`
`Oct 14 19:16:05 osmc systemd[1]: bluealsa.service: Start request repeated too quickly.`
`Oct 14 19:16:05 osmc systemd[1]: bluealsa.service: Failed with result 'exit-code'.`
`Oct 14 19:16:05 osmc systemd[1]: Failed to start BlueALSA service.`

nasu1105, 2022-10-15:
@borine I try to Raspberry Pi OS 64-bit desktop, but I got the same error [bluealsa: E: ../../src/main.c:133: Couldn't acquire D-Bus name. Please check D-Bus configuration. Requested name: org.bluealsa]

pi@raspberrypi:~ $ journalctl -u bluealsa
Oct 15 11:21:15 raspberrypi systemd[1]: Starting BlueALSA service...
Oct 15 11:21:15 raspberrypi bluealsa[2091]: /usr/bin/bluealsa: D: ../../src/main.c:589: Starting main dispatching loop
Oct 15 11:21:15 raspberrypi bluealsa[2091]: /usr/bin/bluealsa: D: ../../src/main.c:114: Acquired D-Bus service name: org.bluealsa
Oct 15 11:21:15 raspberrypi bluealsa[2091]: /usr/bin/bluealsa: D: ../../src/bluealsa-dbus.c:368: Registering D-Bus manager: /org/bluealsa
Oct 15 11:21:15 raspberrypi systemd[1]: Started BlueALSA service.
Oct 15 11:21:15 raspberrypi bluealsa[2091]: /usr/bin/bluealsa: D: ../../src/bluez.c:679: Registering battery provider: /org/bluez/hci0/battery
Oct 15 11:21:15 raspberrypi bluealsa[2091]: /usr/bin/bluealsa: D: ../../src/bluez.c:699: BlueZ battery provider support not available
Oct 15 11:21:15 raspberrypi bluealsa[2091]: /usr/bin/bluealsa: D: ../../src/bluez.c:493: Creating media endpoint object: /org/bluez/hci0/A2DP/SBC/source/1
Oct 15 11:21:15 raspberrypi bluealsa[2091]: /usr/bin/bluealsa: D: ../../src/bluez.c:402: Registering media endpoint: /org/bluez/hci0/A2DP/SBC/source/1
Oct 15 11:21:15 raspberrypi bluealsa[2091]: /usr/bin/bluealsa: D: ../../src/bluez.c:493: Creating media endpoint object: /org/bluez/hci0/A2DP/SBC/source/2
Oct 15 11:21:15 raspberrypi bluealsa[2091]: /usr/bin/bluealsa: D: ../../src/bluez.c:402: Registering media endpoint: /org/bluez/hci0/A2DP/SBC/source/2
Oct 15 11:21:15 raspberrypi bluealsa[2091]: /usr/bin/bluealsa: D: ../../src/bluez.c:493: Creating media endpoint object: /org/bluez/hci0/A2DP/SBC/sink/1
Oct 15 11:21:15 raspberrypi bluealsa[2091]: /usr/bin/bluealsa: D: ../../src/bluez.c:402: Registering media endpoint: /org/bluez/hci0/A2DP/SBC/sink/1
Oct 15 11:21:15 raspberrypi bluealsa[2091]: /usr/bin/bluealsa: D: ../../src/bluez.c:493: Creating media endpoint object: /org/bluez/hci0/A2DP/SBC/sink/2
Oct 15 11:21:15 raspberrypi bluealsa[2091]: /usr/bin/bluealsa: D: ../../src/bluez.c:402: Registering media endpoint: /org/bluez/hci0/A2DP/SBC/sink/2

pi@raspberrypi:~ $ bluealsa-cli
Service: org.bluealsa
Version: v4.0.0-47-g0590af3
Adapters: hci0
Profiles:
  A2DP-source : SBC
  A2DP-sink   : SBC

pi@raspberrypi:~ $ bluealsa -p a2dp-source -p a2dp-sink
bluealsa: D: ../../src/main.c:589: Starting main dispatching loop
bluealsa: E: ../../src/main.c:133: Couldn't acquire D-Bus name. Please check D-Bus configuration. Requested name: org.bluealsa

Arkq, 2022-10-15:
> bluealsa: D: ../../src/bluez.c:493: Creating media endpoint object: /org/bluez/hci0/A2DP/SBC/sink/2
> bluealsa: D: ../../src/bluez.c:402: Registering media endpoint: /org/bluez/hci0/A2DP/SBC/sink/2
> and freeze then

How do you know it freeze? The output is correct. Service is running. So now you will have to connect BT audio device.

> journalctl -u bluealsa
> Oct 14 19:16:05 osmc bluealsa[566]: /usr/bin/bluealsa: error while loading shared libraries: libx264.so.160: cannot enable executable stack as shared object requires: Operation not permitted

Something is wrong with your installation. I can not help with this one, it's not related to bluealsa - executable is not even loaded/executed.

> pi@raspberrypi:~ $ bluealsa-cli
> Service: org.bluealsa
> ...
> pi@raspberrypi:~ $ bluealsa -p a2dp-source -p a2dp-sink
> bluealsa: D: ../../src/main.c:589: Starting main dispatching loop
> bluealsa: E: ../../src/main.c:133: Couldn't acquire D-Bus name. Please check D-Bus configuration. Requested name: org.bluealsa

This is correct behavior, the BlueALSA service is running properly (as verified by bluealsa-cli), so you can not run another service owning the same D-Bus name "org.bluealsa". If you really want to start more than one bluealsa service see `--dbus=NAME` in bluealsa [manual](https://github.com/Arkq/bluez-alsa/blob/master/doc/bluealsa.8.rst). If that's not the case, bluealsa service is running on your system correctly, so you should be able to connect BT audio device.

nasu1105, 2022-10-15:
@Arkq It seems to work for bluealsa, because following code.
I can't enable at least one BT profile on bluealsa and my android phone can't pair via LDAC or APTX, only SBC.

pi@raspberrypi:~ $ bluealsa
bluealsa: E: ../../src/main.c:525: It is required to enabled at least one BT profile
Try 'bluealsa --help' for more information.

pi@raspberrypi:~ $ bluealsa -p a2dp-source -p a2dp-sink
bluealsa: D: ../../src/main.c:589: Starting main dispatching loop
bluealsa: E: ../../src/main.c:133: Couldn't acquire D-Bus name. Please check D-Bus configuration. Requested name: org.bluealsa

borine, 2022-10-15:
@nasu1105 On your RaspberryPiOS machine, you have 2 mistakes.

First, the bluealsa daemon is being started (sucessfully) by systemd, as demonstrated by the output you posted from both `journalctl -u bluealsa` and `bluealsa-cli`. You can see the command-line options used for bluealsa by systemd with `systemctl status bluealsa`. My guess is that you will something like:
```
pi@raspberrypi:~ $ systemctl status bluealsa
● bluealsa.service - BlueALSA service
     Loaded: loaded (/lib/systemd/system/bluealsa.service; enabled; vendor preset: enabled)
     Active: active (running) since Fri 2022-10-14 18:30:46 BST; 16h ago
       Docs: man:bluealsa(8)
   Main PID: 3279 (bluealsa)
      Tasks: 4 (limit: 9262)
     Memory: 1.3M
        CPU: 325ms
     CGroup: /system.slice/bluealsa.service
             └─3279 /usr/bin/bluealsa -p a2dp-source -p a2dp-sink

Oct 15 11:21:15 lenovo systemd[1]: Starting BlueALSA service...
Oct 15 11:21:16 lenovo systemd[1]: Started BlueALSA service.
```

So systemd has not enabled any optional codecs for bluealsa , and that is why you can only connect with SBC.

Second, you then try to start bluealsa from the command line, **while it is already running from systemd**. That fails because only one process can register each service name with D-Bus, and `org.bluealsa` is already in use.

@Arkq - I've just re-read the bluealsa manual page, and to be fair it is not absolutely clear in the text that each instance has to have a different D-Bus service name. I'll add it to my to-do list to add some words to that effect.

@nasu1105 So you need to decide, do you want to run bluealsa from systemd? (in which case you will need to read the systemd documentation to learn how to change the command-line parameters), or do you wish to run it from the command-line? (in which case you need to stop and disable the systemd service).

On your OSMC machine, I think your own build of bluealsa is most likely clashing with OSMC's own pre-installed one. Fixing that may require help from an OSMC expert.

graham8, 2022-10-15:
@nasu1105 Let's continue in the OSMC forum.  We haven't updated bluez-alsa for a while so it's best if I tell you how to build from our git tree.
@Arkq Sorry, sent the user over here a few days ago. We're not building with potentially copyrighted codecs.