arkq/bluez-alsa #424: a stop job is running for Bluealsa daemon

graham8, 2021-03-13:
I get this with a 90 sec delay on re-booting when a BT device is connected.  Is there some systemd magic that can avoid it?

I'm running OSMC (Debian 10 based) and using the unit files recommended in here.

borine, 2021-03-20:
the bluealsa daemon should shutdown cleanly on receipt of SIGTERM, which is the default "stop job" for systemd unless the OSMC config has arranged otherwise. I have been unable to reproduce this error on Ubuntu or RaspiOS. Is it any specific BT device or device type that triggers this error?

> ... a 90 sec delay ...

That is systemd default stop job again. It sends SIGTERM to the main process ID of the service, and if all processes of the service have not terminated within 90 seconds it sends SIGKILL to the whole process group. This behaviour is configurable (see systemd documentation), but I think this default is the correct choice for bluealsa. You should debug why bluealsa is not terminating.

graham8, 2021-03-21:
Thanks. I haven't investigated further and it may have been a one-off but if we can't find anything we'll just shorten the timeout as there's no state to save or other shutdown business.

Where would we look to debug why bluealsa is not terminating?

Arkq, 2021-03-21:
> Where would we look to debug why bluealsa is not terminating?

BlueALSA service is terminated with the usage of the GLib main event loop (SIGTERM signal handler receives POSIX signal and adds new source to the main event loop). So if, for example, there is a deadlock somewhere, the termination might not occur. I think that GDB session would be enough for debugging this issue. If termination does not occur after few seconds attach gdb to the bluealsa process (e.g. `gdb -p $(pgrep -fn "bluealsa ")`) and produce backtrace for every thread.

borine, 2021-06-02:
@graham8 Have you experienced this issue again since? I have still not found any way to reproduce it on my system. If there is any more debug info available it would be good to see.

graham8, 2021-06-02:
I haven't found time to to dig deeper, yet.

graham8, 2021-06-04:
Cannot now reproduce this.  Maybe a kernel bump fixed it.

closed 2021-06-04 by graham8