arkq/bluez-alsa #427: 100% cpu busy loop with alsa rate plugin

borine, 2021-03-22:
I'm getting 100% cpu utilization when playing 44.1kHz tracks to bluetooth speakers with a transport sample rate at 48kHz. For example:
```
aplay -D bluealsa:xx:xx:xx:xx:xx:xx test-44100-16-2.wav
```
Debugging shows this to be the rate plugin failing to block when the ring buffer is full. Does anyone else see this?

`git bisect` reveals the responsible commit to be fb7b878 "ALSA PCM plug-in: use stricter period timing" comitted in October last year.  I cannot understand why initial testing did not find this issue, and why there have been no issues raised reporting this to date. Is this just an issue with my (recently relocated and rebuilt) system, or a genuine bug in the bluealsa pcm plugin?

I have a revised pcm plugin implementation that is working OK for me with the rate plugin, but I have yet to complete testing it with alsaloop (which was the main reason for commit fb7b878).

Arkq, 2021-03-22:
> Does anyone else see this?

Yes, there is indeed 100% CPU usage but only with `aplay`. With e.g. `mpv` and `cmus` everything is OK. 

> I cannot understand why initial testing did not find this issue, and why there have been no issues raised reporting this to date.

I guess that nobody was checking CPU usage with `aplay` (including me) :D

borine, 2021-03-25:
> 76efdd0657de646bce7774f60ef74bfaca75c591

I have no specific thoughts on this at present, but will certainly post any ideas when/if they occur to me,

I've just raised PR #432 which appears to address the alsa plugin busy loop, although to be honest I have not fully understood the cause.

Arkq, 2021-04-19:
Fixed by 341238b.

closed 2021-04-19 by Arkq