arkq/bluez-alsa #430: Recording/mic audio quality

dagbdagb, 2021-03-24:
Got bluealsa working yesterday. Never thought I'd be able to defeat two of my 'kryptonites' in linux (alsa and bluetooth) in one day. So: it works. Thank you for bluez-alsa.

My reason for this issue, is curiosity (and a mostly dead IRC channel :-)  :

**** List of PLAYBACK Bluetooth Devices ****
hci0: 70:26:05:XX:XX:XX [WH-1000XM2], trusted audio-card
  SCO (CVSD): S16_LE 1 channel 8000 Hz
  A2DP (LDAC): S16_LE 2 channels 96000 Hz
**** List of CAPTURE Bluetooth Devices ****
hci0: 70:26:05:XX:XX:XX [WH-1000XM2], trusted audio-card
  SCO (CVSD): S16_LE 1 channel 8000 Hz

S16_LE 1 channel 8000 Hz, is that the best audio I can via bluetooth get from the built-in microphone of this headset? Is the output above a reflection of actual hardware capabilities, or a reflection of what bluez-alsa is able to do with it at this point in time?

infirit, 2021-03-24:
There is a wideband 16khz mode using msbc, but.. How to determine if your Bluetooth adapter supports it is, let's say, problematic in the Linux kernel. Some progress has been made recently but I doubt it has reached distribution kernels.

dagbdagb, 2021-03-24:
Ah. 
I roll my own kernels. Got pointers to relevant patches, @infirit ?

dagbdagb, 2021-03-24:
Kernel 5.11.8 and bluez 5.56 here.

These appear to have integrated most of the relevant bits from the patches showing up when googling for 'wideband 16khz msbc'. 

Is there a natural next step to figure out the capabilities of the hardware? (8087:0a2b Intel Corp. Bluetooth wireless interface)
[    3.747254] Bluetooth: hci0: Bootloader revision 0.0 build 26 week 38 2015
[    3.748260] Bluetooth: hci0: Device revision is 16
[    3.748262] Bluetooth: hci0: Secure boot is enabled
[    3.748263] Bluetooth: hci0: OTP lock is enabled
[    3.748264] Bluetooth: hci0: API lock is enabled
[    3.748265] Bluetooth: hci0: Debug lock is disabled
[    3.748265] Bluetooth: hci0: Minimum firmware build 1 week 10 2014
[    3.749072] Bluetooth: hci0: Found device firmware: intel/ibt-12-16.sfi
[    5.238630] Bluetooth: hci0: Waiting for firmware download to complete
[    5.239277] Bluetooth: hci0: Firmware loaded in 1457937 usecs
[    5.239288] Bluetooth: hci0: Waiting for device to boot
[    5.251317] Bluetooth: hci0: Device booted in 11756 usecs
[    5.251416] Bluetooth: hci0: Found Intel DDC parameters: intel/ibt-12-16.ddc
[    5.254325] Bluetooth: hci0: Applying Intel DDC parameters completed
[    5.255327] Bluetooth: hci0: Reading supported features failed (-16)
[    5.256329] Bluetooth: hci0: Setting Intel telemetry ddc write event mask failed (-95)
[    5.257336] Bluetooth: hci0: Firmware revision 0.1 build 50 week 12 2019

Arkq, 2021-03-24:
> My reason for this issue, is curiosity (and a mostly dead IRC channel :-) :

Yes, recently I'm not using IRC very much, so I'm connected all the time - I'm joining #bluealsa @ freenode only when "instant" conversation is required in order to solve some issue. That's why this channel of communication is not mentioned in the README.

> S16_LE 1 channel 8000 Hz, is that the best audio I can via bluetooth get from the built-in microphone of this headset?

First of all you need bluez-alsa compiled with `--enable-msbc`, secondly you have to use HFP profile not HSP (it's hard to tell which one you are using from the output you've posted). Also, your Bluetooth adapter have to support eSCO links, but only some old Bluetooth dongles lack such support - I've got such, so that's why there is a checks for this in bluealsa code :D. And lastly, headset has to support mSBC, but I guess that your Sony headset supports that. Bluealsa, selects this codec by default, but if for some reason it was not selected, try to use `bluealsa-cli codec <path> mSBC`. And of course there is a case with HCI connection interface - UART vs USB - but if you're using custom kernel that is not an issue :).

dagbdagb, 2021-03-24:
Thank you @Arkq !

Very nice!

bluealsa -S -i hci0 --ldac-eqmid=NB --a2dp-volume -p a2dp-source -p a2dp-sink -p hfp-hf -p hfp-ag 
bluealsa: W: Unsupported AT message: GET: command:+APLSIRI, value:(null)

**** List of PLAYBACK Bluetooth Devices ****
hci0: 70:26:05:56:C0:A2 [WH-1000XM2], trusted audio-card
  SCO (mSBC): S16_LE 1 channel 16000 Hz
  A2DP (LDAC): S32_LE 2 channels 96000 Hz
**** List of CAPTURE Bluetooth Devices ****
hci0: 70:26:05:56:C0:A2 [WH-1000XM2], trusted audio-card
  SCO (mSBC): S16_LE 1 channel 16000 Hz

Recording from the mic now sounds very much better, thankyouveryverymuch!

Is it possible to use A2DP (LDAC) in one direction and SCO (mSBC) in the other, simultaneously? And if so, how do I configure my asoundrc if I want this to be the 'default'?

Arkq, 2021-03-24:
> Is it possible to use A2DP (LDAC) in one direction and SCO (mSBC) in the other, simultaneously? And if so, how do I configure my asoundrc if I want this to be the 'default'?

From the Bluetooth technical point of view it is possible. However, I doubt that there is any headset which works that way, because there would be a usability case to solve - if audio (A2DP) and voice (SCO) connection is established simultaneously, which "stream" should be in the headset speaker: audio or voice, or both (but in such case you might not hear what someone is saying if music is laud)? In reality, most headset mute A2DP when SCO connection is opened. And SCO connection is bidirectional by design, so you can not have only one stream i.e. mic. Maybe there is somewhere a headset which operates in such way that it mutes A2DP only when there is a non-silence or not at all incoming SCO stream - you can open SCO link but you may send nothing... Thea easiest way for A2DP audio and mSBC like mic is to use headset which supports FastStream codec (however, current bluez-alsa master does not have FastStream support).

dagbdagb, 2021-03-24:
Thank you for a detailed reply.
I also vaguely recall that dmix isn't an option with bluez-alsa, as dmix requires hw soundcards. So only one application can use audio at any time. Is my memory correct?

Arkq, 2021-03-24:
> So only one application can use audio at any time. Is my memory correct?

Yes, that is true, unless you want to test PR #407

borine, 2021-06-21:
This issue appears to be resolved

closed 2021-06-21 by borine