arkq/bluez-alsa #434: compatibility with bluez

brian-liu1, 2021-04-12:
I test bluez 5.54 with bluezalsa, my bt chip is rtl8822cs and the kernel is 4.9.y. when play audio with A2DP, occasionally the system will crash without any log, and when play audio as sco pcm, everytime the system will crash suddenly。i tried to use bluez without bluezalsa such as obex profile and the system won't crash，recently i found that when connect  bt headset with a2dp, occasionally the system will crash.  maybe  the cause is the compatibility between bluez and bluezalsa ?

borine, 2021-04-13:
> occasionally the system will crash

can you please describe the crash in more detail? Does the OS crash, (eg kernel panic) or just the UI or network or bluetooth system ?

I think the rtl8822cs bluetooth driver is not included in kernel 4.9.y branch, although I could be mistaken - do you compile the driver yourself, and are you certain the correct firmware is uploaded?

brian-liu1, 2021-04-15:
> > occasionally the system will crash
> 
> can you please describe the crash in more detail? Does the OS crash, (eg kernel panic) or just the UI or network or bluetooth system ?
> 
> I think the rtl8822cs bluetooth driver is not included in kernel 4.9.y branch, although I could be mistaken - do you compile the driver yourself, and are you certain the correct firmware is uploaded? In particular, I'm thinking of this commit which is not backported to the 4.9.y branch in upstream sources: [torvalds/linux@848fc61](https://github.com/torvalds/linux/commit/848fc6164158d697b70b390a2db5019663713f47)

when os crashed , the teminate and talnet are unavailable, and there are no log about kernel panic, even someabnormal log can not be found.most of the time everything stoped such as app 、lcd、tp , i found several time the teminate was still print same app log which had nothing to do with bt, at that time the input function was unavailable. i got the driver from the agent of rtl and they are sure that is the latest version, now i can not determine where the problem is , maybe the driver of chip ? bluez ?  bluezalsa? untill now, i tested the obex profile , everything is ok.

borine, 2021-04-15:
Without any logs, its impossible for anyone to diagnose problems. I'm not aware of anyone else reporting bluez-alsa problems specifically with bluez 5.54, and it seems *very unlikely* that a bug in a user-space application such as bluealsa could cause a complete OS crash. Such crashes are, 99.99% of the time, caused by buggy or incompatible kernel modules (drivers) or firmware.

You may find more people experienced with your setup by asking on forums for your OS or hardware.

borine, 2021-07-17:
This issue has had no updates for 3 months now, so I assume it has been resolved or is no longer relevant. As it seems very likely the system crash is driver-related, I am closing the issue here. If you have any updates, the issue can be re-opened if necessary should more information suggesting a bluez-alsa bug become available.

closed 2021-07-17 by borine