arkq/bluez-alsa #436: no devices shown with bluealsa-aplay -l

geoffr98, 2021-04-25:
I have installed bluealsa on a new PC and connect the bluetooth headset, but can't play audio - any suggestions on where I've gone wrong?

=============
```
geoffr@system76-pc:~$ bluetoothctl
Agent registered
[CHG] Controller 3C:58:C2:D7:17:A1 Pairable: yes
[BCS-700]# paired-devices
Device B8:F8:BE:79:40:C0 BCS-700
[BCS-700]# info  B8:F8:BE:79:40:C0
Device B8:F8:BE:79:40:C0 (public)
        Name: BCS-700
        Alias: BCS-700
        Class: 0x00240404
        Icon: audio-card
        Paired: yes
        Trusted: yes
        Blocked: no
        Connected: yes
        LegacyPairing: no
        UUID: Vendor specific           (00000000-0000-0000-0099-aabbccddeeff)
        UUID: Serial Port               (00001101-0000-1000-8000-00805f9b34fb)
        UUID: Headset                   (00001108-0000-1000-8000-00805f9b34fb)
        UUID: Audio Sink                (0000110b-0000-1000-8000-00805f9b34fb)
        UUID: A/V Remote Control Target (0000110c-0000-1000-8000-00805f9b34fb)
        UUID: A/V Remote Control        (0000110e-0000-1000-8000-00805f9b34fb)
        UUID: Handsfree                 (0000111e-0000-1000-8000-00805f9b34fb)
        UUID: PnP Information           (00001200-0000-1000-8000-00805f9b34fb)
        Modalias: bluetooth:v0094p0901d1000
[BCS-700]# exit
geoffr@system76-pc:~$ sudo bluealsa -p hfp-ag -p hsp-ag  &
[2] 2654
bluealsa: D: ../../src/ofono.c:638: Checking oFono service presence
geoffr@system76-pc:~$ bluealsa: D: ../../src/bluez.c:792: Creating hands-free profile object: /org/bluez/HSP/AudioGateway
bluealsa: D: ../../src/bluez.c:726: Registering hands-free profile: /org/bluez/HSP/AudioGateway
bluealsa: D: ../../src/bluez.c:792: Creating hands-free profile object: /org/bluez/HFP/AudioGateway
bluealsa: D: ../../src/bluez.c:726: Registering hands-free profile: /org/bluez/HFP/AudioGateway
bluealsa: D: ../../src/main.c:396: Acquiring D-Bus service name: org.bluealsa
bluealsa: D: ../../src/main.c:401: Starting main dispatching loop

geoffr@system76-pc:~$ bluealsa-aplay -l
bluealsa: D: ../../src/dbus.c:59: Called: org.bluealsa.Manager1.GetPCMs() on /org/bluealsa
**** List of PLAYBACK Bluetooth Devices ****
**** List of CAPTURE Bluetooth Devices ****
geoffr@system76-pc:~$ bluealsa-aplay --list-pcms
bluealsa: D: ../../src/dbus.c:59: Called: org.bluealsa.Manager1.GetPCMs() on /org/bluealsa
geoffr@system76-pc:~$ 

```

borine, 2021-04-25:
my guess is the device is connected - but not to bluealsa. Is pulseaudio or pipewire running perhaps?

geoffr98, 2021-04-25:
I uninstalled pulseaudio (after confirming the headset works). What you say makes sense.. how do I connect to bluealsa?

borine, 2021-04-25:
I'm confused that bluetoothctl appears to report that the device is connected __before__ bluealsa is started. That's why I think some other service is providing the connection.

geoffr98, 2021-04-25:
I tried disconnecting and reconnecting the device within bluetoothctl and it wasn't connecting correctly. So I tried the GUI tool within Ubuntu and it connected... and then bluealsa showed the device.
I have been able to play audio, but:
- it cuts the audio before it ends.. not sure why..
- I get all sorts of debug out - not sure why.

```
geoffr@system76-pc:~$ aplay -D bluealsa:B8:F8:BE:79:40:C0,sco /usr/share/sounds/alsa/Front_Center.wav
D: ../../../src/asound/bluealsa-pcm.c:1042: Getting BlueALSA PCM: PLAYBACK B8:F8:BE:79:40:C0 sco
bluealsa: D: ../../src/dbus.c:59: Called: org.bluealsa.Manager1.GetPCMs() on /org/bluealsa
D: ../../../src/asound/bluealsa-pcm.c:910: /org/bluealsa/hci0/dev_B8_F8_BE_79_40_C0/hspag/sink: Setting constraints
Playing WAVE '/usr/share/sounds/alsa/Front_Center.wav' : Signed 16 bit Little Endian, Rate 48000 Hz, Mono
D: ../../../src/asound/bluealsa-pcm.c:432: /org/bluealsa/hci0/dev_B8_F8_BE_79_40_C0/hspag/sink: Initializing HW
bluealsa: D: ../../src/dbus.c:59: Called: org.bluealsa.PCM1.Open() on /org/bluealsa/hci0/dev_B8_F8_BE_79_40_C0/hspag/sink
bluealsa: D: ../../src/ba-transport.c:965: New SCO link: B8:F8:BE:79:40:C0: 24
bluealsa: D: ../../src/hci.c:132: SCO link socket MTU: 24: 96
D: ../../../src/asound/bluealsa-pcm.c:455: /org/bluealsa/hci0/dev_B8_F8_BE_79_40_C0/hspag/sink: FIFO buffer size: 2048 frames
D: ../../../src/asound/bluealsa-pcm.c:460: /org/bluealsa/hci0/dev_B8_F8_BE_79_40_C0/hspag/sink: Selected HW buffer: 4 periods x 2000 bytes == 8000 bytes
D: ../../../src/asound/bluealsa-pcm.c:478: /org/bluealsa/hci0/dev_B8_F8_BE_79_40_C0/hspag/sink: Initializing SW
D: ../../../src/asound/bluealsa-pcm.c:478: /org/bluealsa/hci0/dev_B8_F8_BE_79_40_C0/hspag/sink: Initializing SW
D: ../../../src/asound/bluealsa-pcm.c:478: /org/bluealsa/hci0/dev_B8_F8_BE_79_40_C0/hspag/sink: Initializing SW
D: ../../../src/asound/bluealsa-pcm.c:515: /org/bluealsa/hci0/dev_B8_F8_BE_79_40_C0/hspag/sink: Prepared
D: ../../../src/asound/bluealsa-pcm.c:478: /org/bluealsa/hci0/dev_B8_F8_BE_79_40_C0/hspag/sink: Initializing SW
D: ../../../src/asound/bluealsa-pcm.c:345: /org/bluealsa/hci0/dev_B8_F8_BE_79_40_C0/hspag/sink: Starting
bluealsa: D: ../../src/ba-transport.c:776: PCM resumed: 20
D: ../../../src/asound/bluealsa-pcm.c:212: /org/bluealsa/hci0/dev_B8_F8_BE_79_40_C0/hspag/sink: Starting IO loop: 9
D: ../../../src/asound/bluealsa-pcm.c:478: /org/bluealsa/hci0/dev_B8_F8_BE_79_40_C0/hspag/sink: Initializing SW
D: ../../../src/asound/bluealsa-pcm.c:485: /org/bluealsa/hci0/dev_B8_F8_BE_79_40_C0/hspag/sink: Changing SW avail min: 1000 -> 1
D: ../../../src/asound/bluealsa-pcm.c:478: /org/bluealsa/hci0/dev_B8_F8_BE_79_40_C0/hspag/sink: Initializing SW
D: ../../../src/asound/bluealsa-pcm.c:485: /org/bluealsa/hci0/dev_B8_F8_BE_79_40_C0/hspag/sink: Changing SW avail min: 1 -> 1000
bluealsa: D: ../../src/ba-transport.c:802: PCM drained: 20
D: ../../../src/asound/bluealsa-pcm.c:381: /org/bluealsa/hci0/dev_B8_F8_BE_79_40_C0/hspag/sink: Stopping
D: ../../../src/asound/bluealsa-pcm.c:147: /org/bluealsa/hci0/dev_B8_F8_BE_79_40_C0/hspag/sink: IO thread cleanup
bluealsa: D: ../../src/ba-transport.c:808: PCM dropped: 20
D: ../../../src/asound/bluealsa-pcm.c:381: /org/bluealsa/hci0/dev_B8_F8_BE_79_40_C0/hspag/sink: Stopping
bluealsa: D: ../../src/ba-transport.c:808: PCM dropped: 20
D: ../../../src/asound/bluealsa-pcm.c:470: /org/bluealsa/hci0/dev_B8_F8_BE_79_40_C0/hspag/sink: Freeing HW
D: ../../../src/asound/bluealsa-pcm.c:420: /org/bluealsa/hci0/dev_B8_F8_BE_79_40_C0/hspag/sink: Closing
bluealsa: D: ../../src/ba-transport.c:1014: Closing PCM: 20
bluealsa: D: ../../src/sco.c:332: Releasing SCO due to PCM inactivity
bluealsa: D: ../../src/ba-transport.c:988: Closing SCO: 24
```

borine, 2021-04-25:
> it cuts the audio before it ends.. not sure why..

I guess that is issue #260

> I get all sorts of debug out - not sure why.

... because your bluealsa build was configured with `--debug` ? 

> I tried disconnecting and reconnecting the device within bluetoothctl and it wasn't connecting correctly.

That is unexpected. Please can you provide more detail on what happens with bluetoothctl? The Ubuntu desktop is built around pulseaudio, so sometimes various things break when pulse is removed. However, I use Ubuntu  (18.04 and 20.04) and have never had an issue with bluetoothctl .

geoffr98, 2021-04-25:
debug - thanks you are right - recompiled :)
Audio dropping - thanks that fixed it by changing the timer.

Still having some issues with connections..  Here is the output when trying to connect using bluetoothctl:
```
[bluetooth]# scan on
Discovery started
[CHG] Controller 3C:58:C2:D7:17:A1 Discovering: yes
[bluetooth]# bluealsa: D: ../../src/bluez.c:922: Signal: org.freedesktop.DBus.ObjectManager.InterfacesAdded()
[NEW] Device 00:1A:AE:07:EC:00 CLI-W200B-00
[bluetooth]# bluealsa: D: ../../src/bluez.c:922: Signal: org.freedesktop.DBus.ObjectManager.InterfacesAdded()
[NEW] Device 7B:D1:76:50:62:7D 7B-D1-76-50-62-7D
[bluetooth]# bluealsa: D: ../../src/bluez.c:922: Signal: org.freedesktop.DBus.ObjectManager.InterfacesAdded()
[NEW] Device B8:F8:BE:79:40:C0 BCS-700
[bluetooth]# bluealsa: D: ../../src/bluez.c:922: Signal: org.freedesktop.DBus.ObjectManager.InterfacesAdded()
[NEW] Device CC:2D:EC:DD:37:FD CC-2D-EC-DD-37-FD
[bluetooth]# bluealsa: D: ../../src/bluez.c:922: Signal: org.freedesktop.DBus.ObjectManager.InterfacesAdded()
[NEW] Device D1:51:96:B3:BB:93 SavantB3BB9
[bluetooth]# bluealsa: D: ../../src/bluez.c:922: Signal: org.freedesktop.DBus.ObjectManager.InterfacesAdded()
[NEW] Device AC:23:3F:25:55:27 kycs
[bluetooth]# scan off
Discovery stopped
[CHG] Controller 3C:58:C2:D7:17:A1 Discovering: no
[CHG] Device AC:23:3F:25:55:27 TxPower is nil
[CHG] Device AC:23:3F:25:55:27 RSSI is nil
[CHG] Device D1:51:96:B3:BB:93 RSSI is nil
[CHG] Device CC:2D:EC:DD:37:FD TxPower is nil
[CHG] Device CC:2D:EC:DD:37:FD RSSI is nil
[CHG] Device B8:F8:BE:79:40:C0 RSSI is nil
[CHG] Device 7B:D1:76:50:62:7D TxPower is nil
[CHG] Device 7B:D1:76:50:62:7D RSSI is nil
[CHG] Device 00:1A:AE:07:EC:00 RSSI is nil
[bluetooth]# pair B8:F8:BE:79:40:C0
Attempting to pair with B8:F8:BE:79:40:C0
[CHG] Device B8:F8:BE:79:40:C0 Connected: yes
[BCS-700]# bluealsa: D: ../../src/bluez.c:922: Signal: org.freedesktop.DBus.ObjectManager.InterfacesAdded()
[CHG] Device B8:F8:BE:79:40:C0 Modalias: bluetooth:v0094p0901d1000
[CHG] Device B8:F8:BE:79:40:C0 UUIDs: 00000000-0000-0000-0099-aabbccddeeff
[CHG] Device B8:F8:BE:79:40:C0 UUIDs: 00001101-0000-1000-8000-00805f9b34fb
[CHG] Device B8:F8:BE:79:40:C0 UUIDs: 00001108-0000-1000-8000-00805f9b34fb
[CHG] Device B8:F8:BE:79:40:C0 UUIDs: 0000110b-0000-1000-8000-00805f9b34fb
[CHG] Device B8:F8:BE:79:40:C0 UUIDs: 0000110c-0000-1000-8000-00805f9b34fb
[CHG] Device B8:F8:BE:79:40:C0 UUIDs: 0000110e-0000-1000-8000-00805f9b34fb
[CHG] Device B8:F8:BE:79:40:C0 UUIDs: 0000111e-0000-1000-8000-00805f9b34fb
[CHG] Device B8:F8:BE:79:40:C0 UUIDs: 00001200-0000-1000-8000-00805f9b34fb
[CHG] Device B8:F8:BE:79:40:C0 ServicesResolved: yes
[CHG] Device B8:F8:BE:79:40:C0 Paired: yes
Pairing successful
[CHG] Device B8:F8:BE:79:40:C0 ServicesResolved: no
[CHG] Device B8:F8:BE:79:40:C0 Connected: no
[bluetooth]# trust B8:F8:BE:79:40:C0
[CHG] Device B8:F8:BE:79:40:C0 Trusted: yes
Changing B8:F8:BE:79:40:C0 trust succeeded
[bluetooth]# devices
Device 00:1A:AE:07:EC:00 CLI-W200B-00
Device 7B:D1:76:50:62:7D 7B-D1-76-50-62-7D
Device B8:F8:BE:79:40:C0 BCS-700
Device CC:2D:EC:DD:37:FD CC-2D-EC-DD-37-FD
Device D1:51:96:B3:BB:93 SavantB3BB9
Device AC:23:3F:25:55:27 kycs
[bluetooth]# paired-devices
Device B8:F8:BE:79:40:C0 BCS-700
[bluetooth]# exit
geoffr@system76-pc:~$ bluealsa-aplay -l
bluealsa: D: ../../src/dbus.c:59: Called: org.bluealsa.Manager1.GetPCMs() on /org/bluealsa
**** List of PLAYBACK Bluetooth Devices ****
**** List of CAPTURE Bluetooth Devices ****

```

borine, 2021-04-25:
In the above you have both bluealsa and bluetoothctl writing to the same terminal - so the output is mixed up and somewhat confusing.

You only need to pair a device once. That sets up a private key in the bluetooth database for the device which is remembered between connections and between boots. It is used by bluez and the device so that they can securely identify each other before accepting connections.

In order to successfully connect SCO or A2DP, the bluealsa daemon must be running and its initialization completed before the connection attempt commences. The connection request may come from the device (because you have told bluez that it is "trusted", or from bluez (e.g using `bluetoothctl connect xx:xx:xx:xx:xx:xx`).

I think re-pairing a device that is already paired is the reason for the failed connection. To connect the device using bluetoothctl, simply use `bluetoothctl connect B8:F8:BE:79:40:C0` (_after_ bluealsa has started). Hopefully then all will be OK.

closed 2021-04-27 by geoffr98