arkq/bluez-alsa #438: dsnoop with bluealsa

geoffr98, 2021-04-30:
I'm pretty sure that bluealsa does not work with dnsoop (same reason as dmix is not support).

For dmix we have the documentation at : https://github.com/Arkq/bluez-alsa/wiki/Using-bluealsa-with-dmix

Is there any documentation (specifically the updates to the bmix automation) to add support for recording on multiple devices with the same process? - I'd rather not re-invent the wheel here if someone more experienced already has....

borine, 2021-04-30:
> recording on multiple devices with the same process

I don't understand how dsnoop can help with that. Do you mean record from a single device with multiple processes? If so, one possible method is described in the wiki article here: https://github.com/Arkq/bluez-alsa/wiki/Using-bluealsa%E2%80%90aplay-with-dsnoop

geoffr98, 2021-04-30:
Perfect -that is exactly what I was looking for.  (yes I meant recording from one device to multiple application/processes).

geoffr98, 2021-04-30:
Thanks - that is exactly what I was looking for. (yes I meant recording from one device to multiple application/processes).

I have now implemented both the bmix for playback and record, some things are working well.. others not so much - any suggestions on where I've gone wrong?  Below is just the playback... once that is working I'll look into why the recoding isn't working (as it's probably related...)

Here is what I'm seeing:

1. Reboot the computer & Login
2. run 'amixer -D bluealsa scontrols'
-  Gives error (not surprising as bluealsa isn't running..)
-  Couldn't get BlueALSA PCM list: The name org.bluealsa was not provided by any .service files
- **should I really have bluealsa running automatically at boot?  if so how do you suggest?**
3. run 'sudo bluealsa -p a2dp-source -p hsp-ag &'
- Seems to fail as get '[1]+  Stopped                 sudo bluealsa -p a2dp-source -p hsp-ag'
4. if above is run as 'sudo bluealsa -p a2dp-source -p hsp-ag' it just seem to hang 
5. **if I kill #4 or #3 above and re-run with & it runs fine - no idea why??**
6. Power on BT headset
- get the following error 'bluealsa: W: Unsupported AT message: SET: command:+CKPD, value:200'  But everything seems fine..
- Note: if power on BT headset before running bluealsa #7 below doesn't work - should it?
7. 'amixer -D bluealsa scontrols' provides expected output
```
Simple mixer control 'BCS-700 - A2DP',0
Simple mixer control 'BCS-700 - SCO',0
```
8. aplay -L does not show the BCS-700 as described in the instructions @ https://github.com/Arkq/bluez-alsa/wiki/Using-bluealsa-with-dmix  **What Is missing here to make this happen?**
9. **aplay -D bluetooth /usr/share/sounds/alsa/Front_Center.wav  fails, but this might be a typo as using -D bluealsa works..**.
10. **Two processes trying to play audio don't work (not using dmix?) -- the second aplay of the below command fails:
- aplay -D bluealsa Front_Center.wav & aplay -D bluealsa Front_Right.wav &**

geoffr98, 2021-05-01:
Okay making some progress...
Tried running the bmixd.bash on it's own and it failed due to missing gawk - installed it and the script now runs.

Sill seeing some problems - same issues/steps as above for 1-7.
8. aplay -L now shows BCS-700
**9. aplay -D bluetooth works partially ... for short audio files the audio is missing at the start...**
10. I can now play two audio files using -D bluetooth.

borine, 2021-05-01:
> should I really have bluealsa running automatically at boot?

You need bluealsa running before you attempt to connect any bluetooth audio devices. If your device connects automatically when powered on then you need to start bluealsa before you turn the device on - remember bluealsa needs a few seconds to register with bluez before it is useable. 

>  if so how do you suggest?

That depends on your OS. If your OS distribution uses systemd, then you could try the method from the wiki article here https://github.com/Arkq/bluez-alsa/wiki/Systemd-integration Alternatively, consult your distribution help for how to start services at boot.

> [1]+ Stopped sudo bluealsa -p a2dp-source -p hsp-ag

The problem here is that you really should not use sudo as `sudo ... &' with any program. Sudo needs to read the password from the terminal, but backgrounded processes are not allowed to do that - they will be suspended with a STOP signal. So just don't do it.

> 'sudo bluealsa -p a2dp-source -p hsp-ag' it just seem to hang

Are you sure it 'hangs', or is it just running normally, with no output? 

> if I kill #4 or #3 above and re-run with & it runs fine - no idea why??

probably you successfully entered your password at step #4, so now sudo does not ask for it again and therefore does not try to read from the terminal.

> get the following error 'bluealsa: W: Unsupported AT message: SET: command:+CKPD, value:200' But everything seems fine.

Thats just a warning (not an error) that bluealsa ignores that command from the headset, its not needed for audio playback.

> Note: if power on BT headset before running bluealsa #7 below doesn't work - should it?

Probably, your headset attempts to connect when powered on. If that fails (eg bluealsa is not running) it does not try again. You will have to manually connect using bluetoothctl as previously discussed.

> for short audio files the audio is missing at the start

That is most likely a feature of the headset - some devices do that, others not. Its not the length of the file that matters, its how much silence there is at the start. Some devices "fade in" the start of a stream and "fade out" the end. So the first fraction of a second is just too quiet to hear. This is quite common with A2DP, but rare with HSP/HFP

geoffr98, 2021-05-01:
Thanks for the systemd link - I wasn't sure if that was the best way or not due to timing issues.  The documentation for bluez is great.. I just can't seem to find it.

For the missing audio the problem is not the headset.  The audio file when correctly played is "Front Center". I get very different results with the following:
1. aplay -D bluealsa:B8:F8:BE:79:40:C0,sco /usr/share/sounds/alsa/Front_Center.wav
    - I hear "Front Center", but it take a bit of time before I hear the audio.
2. aplay -D bluetooth /usr/share/sounds/alsa/Front_Center.wav
    **- What I hear varies from nothing to "enter", so it's missing the start of the audio.**
3. aplay -D bluealsa:B8:F8:BE:79:40:C0,a2dp /usr/share/sounds/alsa/Front_Center.wav
    - works well, in fact the audio is better and comes out faster than sco.

**Background info:**
bluealsa-aplay -L
bluealsa:SRV=org.bluealsa,DEV=B8:F8:BE:79:40:C0,PROFILE=a2dp
    BCS-700, trusted audio-card, playback
    A2DP (SBC): S16_LE 2 channels 48000 Hz
bluealsa:SRV=org.bluealsa,DEV=B8:F8:BE:79:40:C0,PROFILE=sco
    BCS-700, trusted audio-card, playback
    SCO (CVSD): S16_LE 1 channel 8000 Hz
bluealsa:SRV=org.bluealsa,DEV=B8:F8:BE:79:40:C0,PROFILE=sco
    BCS-700, trusted audio-card, capture
    SCO (CVSD): S16_LE 1 channel 8000 Hz
**I think the above is saying the headset does not support a2dp recording - is this correct?**

bluealsa options:
OPTIONS="-p a2dp-source -p a2dp-sink"
**I think I'll need to add in sco for recoding later?**

geoffr98, 2021-05-01:
Update - I also added in the recoding side and now playing to A2DP doesn't working (guessing because headset is in sco mode). This also seems to cause other problems as I can then no longer play any audio until I power cycle the headset.

However -B bluetooth now is has no missing audio (although I think it's in sco mode as the quality is lower)

borine, 2021-05-01:
As for the missing start of the file, I can't check on that ATM as my only available BT speaker *always* misses the first 300ms of audio on A2DP, whether I am using the "bmix" config or not.  SCO does not usually suffer this effect, as you have found.

geoffr98, 2021-05-02:
Since I'm using SCO now (headset does not seem to support A2DP on the record side) I can't look into this to find a cause either :(

borine, 2021-07-27:
I think this issue is now abandoned - closing.

closed 2021-07-27 by borine