arkq/bluez-alsa #441: bluealsa-aplay.service not starting correctly

geoffr98, 2021-05-01:
Following the instructions at https://github.com/Arkq/bluez-alsa/wiki/Systemd-integration I added the bluealsa-aplay.service. The service does try to start , but fails. (see log below).  If I restart the service after the system has booted it works correctly 

In the service I see that it is supposed to wait for bluealsa.service - is that perhaps not enough?  I could add a wait, but I'm assuming some has seen this before and has a better solution?

```
systemctl status bluealsa-aplay
● bluealsa-aplay.service - Bluealsa audio player
     Loaded: loaded (/etc/systemd/system/bluealsa-aplay.service; enabled; vendor preset: enabled)
     Active: active (running) since Sat 2021-05-01 12:07:41 EDT; 3min 9s ago
       Docs: https://github.com/Arkq/bluez-alsa/
   Main PID: 717 (bluealsa-aplay)
      Tasks: 2 (limit: 9127)
     Memory: 1.2M
     CGroup: /system.slice/bluealsa-aplay.service
             └─717 /usr/bin/bluealsa-aplay --profile-sco --pcm=bluealsa-aplay:PROFILE=sco 00:00:00:00:00:00

May 01 12:07:41 system76-pc systemd[1]: Started Bluealsa audio player.
May 01 12:07:42 system76-pc bluealsa-aplay[717]: /usr/bin/bluealsa-aplay: W: Couldn't get BlueALSA PCM list: The name org.blueals>
```

borine, 2021-05-01:
Are you sure that it "fails"? Have tried using it? I only see a warning message in  your log ("W:"), no error messages ("E:")

geoffr98, 2021-05-02:
You are right it does work - I'm sure when I tested the first time it didn't... But works now.

closed 2021-05-02 by geoffr98