arkq/bluez-alsa #442: blueasa-aplay not showing devices with -l or -L

mr-berndt, 2021-05-10:
Hello, I am the developper of the "Aroio" audio streamer from abacus electronics in Germany.

One of Aroio's efatures is, that one can connect to it via bluetooth. I implemented this using bluealsa-aplay.

After updating bluez and bluez-alsa to 3.0 I cannot connect a phone to my raspberry pi running bluealsa-aplay anymore.
The error message I see is:
bluetoothd[321]: src/service.c:btd_service_connect() a2dp-source profile connect failed for 2C:78:0E:CF:68:E4: Protocol not available

Journalctl shows me endpoints, though, which suggest, that a2dp is kinown:
bluetoothd[321]: Endpoint registered: sender=:1.6 path=/org/bluez/hci0/A2DP/AAC/Source/1

Am I interpreting this correctly?

bluealsa-aplay is not showing me any devices with -l nor PCMs with -L. Obviously the communication with alsa is not working in my case.

I do not use a standard OS like raspbian but a self setup buildroot.

Any hints would be greatly appreciated.

Arkq, 2021-05-10:
The source profile is for sending data from device (e.g. from RPi). When
using bluealsa-aplay in most cases you need sink profile. Currently,
bluealsa does not register sink profiles by default. You need to specify
required profiles via bluealsa command line. See 'bluealsa -h' for more
info.

mr-berndt, 2021-05-10:
Ah, no, I was wrong. bluealsa was of course started beforehand, just without the new -p a2dp-sink.
It is so long ago that I build this part of the streamer that I did not remember.

Arkq, 2021-05-10:
What do you mean, that you need to restart bluealsa before bluealsa-aplay?

mr-berndt, 2021-05-10:
No, my mistake. I had to figure out, that bluealsa now needs the -p switch. It took me quite a while but now it's all running as before.

mr-berndt, 2021-05-10:
While we're at it: Any idea why some phones loose their connection and reconnect all of the time, while others work just fine.

I have a Huawei P30 lite that runs fine, for ever.
I have an IPhone SE, that disconnects and reconnects every 30 secs. or so.

I tested this with 3 BT dongles and they all give the same result.

Arkq, 2021-05-10:
> I have an IPhone SE, that disconnects and reconnects every 30 secs. or so.

Yes, I know that iPhones are very picky. It might mean., e.g. that the HFP profile is not connected, or that maybe some other profile is not connected.... so after some time (i.e. 30 seconds) there is a timeout for profile connection and iPhone restarts all other profiles in order to fix the issue with the good old "restart" trick :D.

closed 2021-05-17 by Arkq