arkq/bluez-alsa #445: After removing a paired device reconnection doesn´t possible

bawx25, 2021-05-21:


borine, 2021-05-22:
> But after canceling the pairing with the raspi

Please can you explain what you mean by that? How do you "cancel the pairing" ? Why do you ever need to "cancel" a pairing?

This looks like an issue with whatever bluetooth management application you are using on the Pi. I always use the command-line `bluetoothctl`application to pair, trust and connect my audio devices, and have no such issues. 

You should pair with a device only once, to set up the encryption keys. This info is then permanently stored on the device and the Pi so there is normally no need to do it again. After pairing, it is necessary to "connect" the device when you want to use it - connection can happen automatically or require manual intervention depending on how you have set up the system.

bawx25, 2021-05-22:
With canceling the connection I mean to disconnect the smartphone completly. You need it sometimes because when you accidently delete the connection you cant connect back and need to install everything again. Also i didnt pair the Smartphone with the `bluetoothctl` i can connect direktly without any screen. My raspi shows me the opertunity to connect and i just connect.

borine, 2021-05-22:
"cancelling the connection" is not the same thing as "cancelling the pairing". As I said above, you pair only once (usually, but not always, if pairing a mobile phone you will be asked to enter a security check number). 

Then you need to connect the phone to the Pi. It is usual to tell the Pi to "trust" the phone and then connections are initiated from the 'phone; the Pi should accept the connection request because it "trusts" the phone. Does your bluetooth management app always set the "trust" attribute for all paired devices or is there a separate dialog for you to enable that? You can see if the device is trusted by running the command:
```
bluetoothctl info XX:XX:XX:XX:XX:XX
```
where XX:XX:XX:XX:XX:XX is the bluetooth address of the 'phone.

borine, 2021-05-22:
> accidently delete the connection

hmm, perhaps you mean that you accidentally removed the *pairing* on the phone - in that case the pairing info stored on the Pi is no longer valid, and that might explain why you are getting the unexpected "password incorrect" message. In which case, you need to delete the pairing info on the Pi first, and then go back to the start and pair the phone with the Pi again. That will create new, matching, credentials on both devices and then you should be able to connect again.

I delete pairing info on my Pi by doing:
```
bluetoothctl remove XX:XX:XX:XX:XX:XX
```
No need to remove any other devices, just the 'phone where the "accident" happened.

borine, 2021-08-10:
This issue appears to be about the use of the RPi bluetooth management applet rather than bluez-alsa. Given that RaspiOS has now switched to pulseaudio for use with their desktop, and that this issue has seen no activity for a few weeks, I am closing it as no longer relevant. If new information becomes available then it can be re-opened.

closed 2021-08-10 by borine