arkq/bluez-alsa #446: Can't record from bluetooth_sco_capture

geoffr98, 2021-05-26:
Moving to a new computer and am having a problem. I can record using "arecord -D bluealsa:B8:F8:BE:79:40:C0,sco -d 10 capture.wav" but  "arecord -d bluetooth_sco_capture -d 10 capture.wav"  doesn't work - what have I missed in the set-up?

Also for some reason on this computer when I playback it misses the start of the audio, which it didn't on the other (I have installed all the scripts to allow for -D bluetooth ). Any ideas?

borine, 2021-06-02:
>  what have I missed in the set-up?

I think only you can answer that :smile: Just saying it "doesn't work" really gives no clue to anyone what might be wrong. Does arecord give any error messages at all?

> misses the start of the audio

Some bluetooth playback devices do that. Are you playing to a different speaker/headphones on this new computer than you did on the old one? If its the same speaker, then that may be a bug in bluealsa.

geoffr98, 2021-06-02:
I ended up going back through all the step and I had missed one .. what confused me is that the audio playback worked, but missed the start.. turns out that the aloop wasn't running because I forgot to tell systemctl to reload!
missing step:  sudo systemctl daemon-reload

closed 2021-06-02 by geoffr98