arkq/bluez-alsa #452: alsa integration challenges aplay -l does not show bluealsa device

gearhead, 2021-05-31:
I am on a number of 'issues' here for this repo. I think we can close them all except this one. 
One request, can you please make a release comprising either up to now or at least to last December? I have been running the latest release from git and though I can play a file via bluealsa-aplay, I get the crash unless I use the git 'HEAD' version. 

As I read the wiki, it seems that if I add the 20-bluealsa.conf file in the repo to /usr/share/alsa/alsa.conf.d, I should be able to access my newly paired and connected BT headphones as an alsa device. I cannot figure out how to make this work, I added this file and it did not show up in 'aplay -l'. After I tried this, I added a file with with this at /etc/asound,conf
```
pcm.bt-headset {
        type plug
        slave.pcm {
                type bluealsa
                device "70:BF:92:D8:82:84"
                profile "a2dp"
        }
        hint {
                show on
                description "Jabra"
        }
}
```
where the MAC is the MAC of the BT device that I get here:
```
# bluealsa-aplay -l
**** List of PLAYBACK Bluetooth Devices ****
hci0: 70:BF:92:D8:82:84 [Jabra Evolve2 65], trusted audio-card
  SCO (CVSD): S16_LE 1 channel 8000 Hz
  A2DP (SBC): S16_LE 2 channels 48000 Hz
**** List of CAPTURE Bluetooth Devices ****
hci0: 70:BF:92:D8:82:84 [Jabra Evolve2 65], trusted audio-card
  SCO (CVSD): S16_LE 1 channel 8000 Hz
```
aplay -l still does not show the output device. How do I get the device to show up when it is 'connected' in bluetooth? I can play a wav file, I want to be able to select it as an output for mpd. 

What am I missing?

Keith

Arkq, 2021-06-01:
> One request, can you please make a release comprising either up to now or at least to last December?

Yes, I think that making a release at this point is a good idea. I thought that maybe I can squeeze few "features" more, but unfortunately, I've got stuck myself on #425 (but since yesterday I think I can see a light in the tunnel).

> How do I get the device to show up when it is 'connected' in bluetooth?

I think, that it's not possible unless you will have some scripts for asoundrc automatic modifications. The problem is, that `aplay -l` uses kernel ALSA interface for listing available devices (bluealsa is not one of these) and a static configuration from asoundrc. But, there is no build-in hook (in ALSA library) for dynamic configuration from asoundrc file. I think that you will be able use the script from this wiki site https://github.com/Arkq/bluez-alsa/wiki/Using-bluealsa-as-default-ALSA-PCM and change it slightly to be able to add a device to the asoundrc only when the BT is connected.

borine, 2021-06-01:
`mpd` requires each output device to be defined explicitly in its own configuration file before it is started, so there is no way to make it aware of new output devices while it is running.

If you only want `mpd` to use the most recently connected bluetooth device, with latest bluez-alsa source you can take advantage of its new blutooth default feature. Define an output device as:
```
audio_output {
	name "bluetooth"
	type alsa
	device "bluealsa"
	mixer_type "software" # mpd has a crash bug with removable "hardware" alsa mixer
}
```

However, if you connect multiple bluetooth devices at the same time and want to select between them using an mpd client, then you will need to create an mpd config entry for each one explicitly. But then mpd has no way of knowing which ones are connected so clients will always show the full list, whether connected or not. That is a limiitation of `mpd`, it has the same issue with other removable device types such as USB soundcards.

Incidentally, you can get a list of ALSA PCM ids for connected bluealsa PCMs using
```shell
bluealsa-aplay -L
```

gearhead, 2021-06-02:
OK. I may be getting it. We already have a script that can be run that adds any new audio outputs to the mpd.conf and restarts mpd. Once restarted, we can select them from the UI. @borine with this, I think the MPD aspect may already be taken care of... The other aspect is getting the /etc/scound.conf file updated, but may be able to do that with a script and udev. We use /etc/scound.conf instead of ~./asoundrc because we are running a headless MPD w/o a login user. 

So, I will try this: 1) Connect BT device. 2) Make a udev script to look for an added BT output and run a script to extract the MAC and Description (between square brackets) and put them in the /etc/asound.conf. 3) Run our audio output script to make mpd aware of the new BT output device.

I guess I have to make sure that the 20-bluealsa.conf file is present in /usr/share/alsa/alsa.conf.d, don't I?

borine, 2021-06-02:
> I guess I have to make sure that the 20-bluealsa.conf file is present in /usr/share/alsa/alsa.conf.d, don't I?

That file defines the (parameterized) PCM called "bluealsa", so yes, if you plan to use PCM names such as "bluealsa:XX:XX:XX:XX:XX:XX" in your mpd.conf then you will need that file.

borine, 2021-07-28:
@gearhead this issue has been quiet for a while; have you made any further progress?

I have a script that I use to update the ALSA config when bluealsa PCMs connect or disconnect, but to work efficiently it requires some additional features for the bluealsa-cli utility that I have not shared upstream here yet. The reason I have hesitated to share it is that so many GUI programs (vlc, kodi, etc) struggle with PCMs that can connect or disconnect independently of a hardware sound card and I have been experimenting with various (largely unsuccesful) techniques to deal with that. However, that should not be an issue for MPD.

I am interested in your work with udev. In my experiments with it, I see link added and link removed events when bluetooth devices connect/disconnect, but no easy way to determine if the device is an audio device; also the event fires before the bluetooth daemon has notified bluealsa so no bluealsa PCM is yet available at that time. Have you had any more success than me?

With regard to /etc/asound.conf, I think that entries there are not strictly necessary for MPD. In the /etc/mpd.conf you can refer to a bluealsa A2DP PCM as `bluealsa:XX:XX:XX:XX:XX:XX` so there is no need to give it a new name via /etc/asound.conf. The wiki page https://github.com/Arkq/bluez-alsa/wiki/Using-the-bluealsa-ALSA-pcm-plugin explains the use of this pre-defined parameterized PCM - but you do have to install the file 20-bluealsa.conf to use it.

I can share my script here if it would be useful to you, as I said it uses `bluealsa-cli`, not udev, to detect new bluetooth audio devices, but I will have to raise a PR for the changes to bluealsa-cli, and get it accepted, before my script can be used.

graham8, 2021-07-28:
> I am interested in your work with udev. In my experiments with it, I see link added and link removed events when bluetooth devices connect/disconnect, but no easy way to determine if the device is [...]

FWIW, I too can't see any way udev can indentify a BT device's profile/capabilities. At the moment, we listen for a udev event to fire then have an ugly 2-3second sleep before letting Kodi enumerate devices.  Kodi doesn't list a device unless it can open it.  It seems from the above conversation mpd is different though.

gearhead, 2021-07-28:
@borine I have not had time to dedicate to it. We want to add BT audio to Rune and have it be intuitive. I was poking around trying to do something and have not gotten back to it. 
Determining if it is an audio device should be able to be done with udev as bluetooth responds with the identification value for the device. the udev rule would get a bit more complicated and may require a helper script, but my thought was that this info could be used to help configure it by keying the script/udev as to what kind of device it is.

borine, 2021-07-29:
> listen for a udev event to fire then have an ugly 2-3second sleep

@graham8 just a thought - if you are using a script to update the alsa config files, maybe it could invoke something like 
```
udevadm trigger -s bluetooth
```
after writing the file, That would generate a udev "change" event on the bluetooth subsytstem which would distinguish it from the "add" event caused by the device connection. That way you would not need the "ugly sleep" ?

graham8, 2021-07-29:
Thanks, but the problem is not sending out triggers, it's listening for them.  We can't generate a .conf before the BT device shows up on dbus. What I need is a dbus listener in Kodi that will do for BT audio devices what the existing udev listener does for hardware soundcards (USB mainly).  The boss has rejected a stand-alone listener and anyway that would have to send a message to Kodi to enumerate audio devices for which there's no plumbing.  So I'm currently on a learning curve with threads in C++.

The idea is: Kodi listens to dbus org.bluez for a parameter change for Paired, generates an alsa .conf unless one already exists and then re-enumerates audio devices.

@gearhead I had another look at what's available from udev and still can't find anything that can identify a BT device as audio.  In fact, I believe udev is only recognising my JBL Flip 3 speaker via its buttons as an input device so not as an audio device.  Sorry for hijacking your thread.

borine, 2021-09-16:
I believe the original question asked in this issue has been answered. Given that there has been several weeks without activity, I am closing this now. If any new information regarding udev events and bluetooth audio is found, please create a new issue so that a thread dedicated to that subject can be started.

closed 2021-09-16 by borine