arkq/bluez-alsa #453: Daemon is running, headphones are connected, yet alsa is not sending audio to them

czajnik2137, 2021-06-01:
I have all required services turned on and running correctly, my headphones are connected to bluetooth but I can't get them to play audio. It is directed to my laptop speakers instead. I can tell that bluealsa is functioning as I tried this command

> aplay -D bluealsa Ensoniq-ZR-76-03-Dope-85.wav

which successfully plays file in my headphones.

my .asoundrc
> \# Bluetooth headset
> defaults.bluealsa {
>     interface "hci0"            # host Bluetooth adapter
>     device "20:1B:88:97:1B:9C"  # Bluetooth headset MAC address
>     profile "a2dp"
> }

czajnik2137, 2021-06-01:
I added a band-aid solution based on some .asoundrc found on the internet, but it isn't perfect either. This script makes my bluetooth headphones the default sound device that receives output, but the default one is ignored when I unplug my headphones. Besides, when I reconnect my headphones they are cranked up to 100% which is... inconvenient to say the least.

> pcm.headphones {
>    type plug
>    slave.pcm {
>        type bluealsa
>        device "20:1B:88:97:1B:9C"
>        profile "a2dp"
>    }
>    hint {
>        show on
>        description "Headphones"
>    }
> }
>
> pcm.!default {
>    type plug
>    slave.pcm "headphones"
> }

borine, 2021-06-01:
I've written up my thoughts on using BlueALSA as the system default device, as a wiki page here: https://github.com/Arkq/bluez-alsa/wiki/Using-bluealsa-as-default-ALSA-PCM

The examples in that article require the latest bluez-alsa source code, which introduces a new utility called `bluealsa-cli` and also introduces a new feature that uses the most recently connected bluetooth device when the application does not specify which bluealsa device to use.

As you can see from that article, bluetooth audio is not suitable as default for every system - you should consider carefully your own requirements to decide if it is OK for your system.

borine, 2021-06-21:
> when I reconnect my headphones they are cranked up to 100%

Latest bluez-alsa code now has a run-time option on the `bluealsa` daemon (`--initial-volume=N`) to set the initial volume to less than 100% when a device connects. This may help in your case.

borine, 2021-07-27:
@czajnik2137 has the referenced wiki article and the new `--initial-volume` option answered your questions here?

borine, 2021-09-11:
Please see the wiki page https://github.com/Arkq/bluez-alsa/wiki/Using-bluealsa-as-default-ALSA-PCM for a discussion on using bluealsa as the system default device.

closed 2021-09-11 by borine