arkq/bluez-alsa #455: 1.4.0 bluealsa

cool-delete, 2021-06-01:
<summary> nano /etc/asound.conf
<details>
pcm.btreceiver {

    type plug
    slave.pcm {
        type bluealsa
        device "41:42:08:68:33:30"
        profile "a2dp"
    }
    hint {
        show on
        description "Bluetooth Receiver"
    }
}

pcm.!default {
 type plug
    slave.pcm "btreceiver"

}
</details></summary>
Bluetooth speakers should have been set to the default speaker, but
<summary>

cat /proc/asound/cards
<details>

 0 [Headphones     ]: bcm2835_headphonbcm2835 Headphones - bcm2835 Headphones
                      bcm2835 Headphones
</details>
There is no appearance

Execute alsamixercan only find
[image]
And execute
`
alsamixer  -D  bluealsa
`
[image]

How to modify the Bluealsa volume by simple alsamixer in the Raspberry Pi, without having to specify bluealsa each time

borine, 2021-06-02:
You set set the default pcm to be a bluealsa device, but for alsamixer you also need to set the default ctl to be bluealsa:
```
ctl.!default bluealsa
```
However, there may be some undesirable side effects of this - see my notes on using bluealsa as default ALSA device on the wiki here: https://github.com/Arkq/bluez-alsa/wiki/Using-bluealsa-as-default-ALSA-PCM

cool-delete, 2021-06-02:
> 您将默认 pcm 设置为 bluealsa 设备，但对于 alsamixer，您还需要将默认 ctl 设置为 bluealsa：
> 
> ```
> ctl.!default bluealsa
> ```
```pcm.btreceiver {
    type plug
    slave.pcm {
        type bluealsa
        device "41:42:08:68:33:30"
        profile "a2dp"
    }
    hint {
        show on
        description "Bluetooth Receiver"
    }
}

pcm.!default {
 type plug
    slave.pcm "btreceiver"

}

ctl.default {
type bluealsa
}
```

```
pi@raspberrypi:~ $ alsamixer
ALSA lib conf.c:1410:(parse_def) default is not a compound
ALSA lib conf.c:1887:(_snd_config_load_with_include) _toplevel_:20:13:Invalid argument
ALSA lib conf.c:3650:(config_file_open) /etc/asound.conf may be old or corrupted: consider to remove or fix it
ALSA lib conf.c:3572:(snd_config_hooks_call) function snd_config_hook_load returned error: Invalid argument
ALSA lib conf.c:4026:(snd_config_update_r) hooks failed, removing configuration
cannot open mixer: Invalid argument
```
I automatically link the speaker by setting the 20 seconds after power on, but runs `alsamixer` to get these errors.

borine, 2021-06-02:
`ctl.default` is wrong. Should be `ctl.!default`

cool-delete, 2021-06-02:
> `ctl.default` is wrong. Should be `ctl.!default`
```
ctl.!default {
type bluealsa
}
```

Thank you very much

cool-delete, 2021-06-02:
@borine  hi 
But after rebooting, the adjusted volume will return to 100. isn't that a problem with bluealsa?

borine, 2021-06-02:
> But after rebooting, the adjusted volume will return to 100. isn't that a problem with bluealsa?

Not really. `alsactl` only works with hardware soundcards, so does not save the state of bluealsa controls, so that is a problem with ALSA.

cool-delete, 2021-06-02:
Okay, a little regret

closed 2021-06-02 by cool-delete