arkq/bluez-alsa #457: Unable to auto authorize service

dandanio, 2021-06-06:
Hey guys!
Pi zero W + HiFi Berry Zero, Rasbpian - all patched, fresh install. Trying to build a bluetooth a2db speaker.
Everything is working except that each time I have to allow the service upon connect from a new device which defeats the purpose (in bluetoothctl interactive):
```
[NEW] Device xx:xx:xx:xx:xx:xx MacBook Pro
[CHG] Device xx:xx:xx:xx:xx:xx Modalias: bluetooth:v004Cp4A00d1011
[CHG] Device xx:xx:xx:xx:xx:xx UUIDs: 00001101-0000-1000-8000-00805f9b34fb
[CHG] Device xx:xx:xx:xx:xx:xx UUIDs: 0000110a-0000-1000-8000-00805f9b34fb
[CHG] Device xx:xx:xx:xx:xx:xx UUIDs: 0000110c-0000-1000-8000-00805f9b34fb
[CHG] Device xx:xx:xx:xx:xx:xx UUIDs: 0000110e-0000-1000-8000-00805f9b34fb
[CHG] Device xx:xx:xx:xx:xx:xx UUIDs: 00001112-0000-1000-8000-00805f9b34fb
[CHG] Device xx:xx:xx:xx:xx:xx UUIDs: 00001117-0000-1000-8000-00805f9b34fb
[CHG] Device xx:xx:xx:xx:xx:xx UUIDs: 0000111f-0000-1000-8000-00805f9b34fb
[CHG] Device xx:xx:xx:xx:xx:xx UUIDs: 00001200-0000-1000-8000-00805f9b34fb
[CHG] Device xx:xx:xx:xx:xx:xx UUIDs: 02030302-1d19-415f-86f2-22a2106a0a77
[CHG] Device xx:xx:xx:xx:xx:xx ServicesResolved: yes
[CHG] Device xx:xx:xx:xx:xx:xx Paired: yes
Authorize service
[agent] Authorize service 0000110d-0000-1000-8000-00805f9b34fb (yes/no):

[CHG] Device xx:xx:xx:xx:xx:xx ServicesResolved: no
[CHG] Device xx:xx:xx:xx:xx:xx Connected: no
Request canceled
```
I followed a few guides, it seems like I am really that close, but I need this one hurdle to overcome and I am golden. My devices pair without passcode, everything plays as expected if I approve, but I would like to have the speaker start playing without that approval step. I run aplay from systemd as:
```
/usr/bin/bluealsa-aplay -vv --profile-a2dp --pcm-buffer-time=24000 --pcm-period-time=6000 00:00:00:00:00:00
```
and bluealsa:
```
/usr/bin/bluealsa --profile=a2dp-sink
```
I have been writing a howto (that I will share) but I can't make it work.

Cześć i dzięki! (Thanks!)

Daniel

Arkq, 2021-06-06:
Is your device trusted?
```sh
bluetoothctl trust xx:xx:xx:xx:xx:xx
```

dandanio, 2021-06-06:
> Is your device trusted?
> 
> ```shell
> bluetoothctl trust xx:xx:xx:xx:xx:xx
> ```
I have configured the NoInputNoOutput plugin in bluetoothctl and I do not have to trust my devices, but still the service needs to be authorized. Weird, right?

Basically I want a hands off a2dp sink so I can stream from my backyard when friends come over. And if they want to stream their Spotify list, they can find my RPi0, pair with it and start streaming, without me logging to a ssh console and authorizing or pairing anything. Is it possible at all?

infirit, 2021-06-06:
> > Is your device trusted?
> > ```shell
> > bluetoothctl trust xx:xx:xx:xx:xx:xx
> > ```
> 
> I have configured the NoInputNoOutput plugin in bluetoothctl and I do not have to trust my devices, but still the service needs to be authorized. Weird, right?

This is how bluetoothctl works.

> Basically I want a hands off a2dp sink so I can stream from my backyard when friends come over. And if they want to stream their Spotify list, they can find my RPi0, pair with it and start streaming, without me logging to a ssh console and authorizing or pairing anything. Is it possible at all?

Sure, but not with bluetoothctl's agent. Instead use ``/usr/bin/bt-agent -c NoInputNoOutput``.

dandanio, 2021-06-06:
> Sure, but not with bluetoothctl's agent. Instead use `/usr/bin/bt-agent -c NoInputNoOutput`.
It works! Thanks a bunch! I will edit to link to my howto and close!

Thx!

dandanio, 2021-06-07:
As promised. Thanks guys for your help. I have finished ver. 0.99 of my [quick howto](https://gist.github.com/dandanio/705447323c621969894b9a2cd27f7fc8). Check it out and all comments are welcome.
One question I had: It seems I did not need to install python-dbus, am I missing something from functionality?

Thanks again!

closed 2021-06-07 by dandanio