arkq/bluez-alsa #464: Possibility to control external volume unit in sink mode

jahudka, 2021-06-28:
Hi, I'd like to ask if it's possible (or if it could be made possible, hopefully without too much work) to get notified of volume changes sent by the source _instead of_ applying them via AlsaMixer. I'm building a bluetooth receiver on top of a Raspberry Pi for an amplifier which has motorised volume control - basically you can control the volume using a remote and the volume knob physically turns to the correct position. I'd like to keep using that feature, because the amp has multiple inputs and the bluetooth receiver should be just one of them, but I'd also like to integrate it with bluetooth volume control, so what I'm looking for is a way to specify e.g. an executable path which `bluealsa-aplay` would call when it receives volume commands instead of passing the commands to AlsaMixer. Is something like this possible?

Arkq, 2021-06-28:
Currently there is no code for forwarding volume level to hardware mixer via `bluealsa-aplay`. There is a PR #390 (currently stale, but it's still in the back of my head) which is a work in progress, which should handle such scenario. With current master code, however, you could use native volume control in bluealsa service `bluealsa --a2dp-volume -p a2dp-sink` which will disable software volume. Then, you can monitor bluealsa ALSA mixer volume knob (the same way the `alsamixer -D bluealsa` is monitoring it) and use it with your hardware volume application.

borine, 2021-06-30:
> instead of passing the commands to AlsaMixer

I think it is important to clarify here that `bluealsa-aplay` does not send any commands to `alsamixer`, nor does it communicate with `alsamixer` in any other way.

The `bluealsa` daemon receives the volume change event from the bluez `bluetoothd` daemon via D-Bus. `alsamixer -D bluealsa` then fetches the event from `bluealsa` also via D-Bus, using the bluez-alsa CTL plugin.

So you would need an application that essentially does what alsamixer is doing, but uses the event notification to operate your motor instead updating a screen widget. You could  make that an ALSA application, and use `alsamixer` as a model for your code, or you could make it a D-Dus application, and subscribe to `PropertiesChanged` signals for the org.bluealsa.PCM1 interface - see src/asound/bluealsa-ctl.c for example code. The D-Bus approach is possibly simpler, as it could be scripted using Python, or shell script using a tool such as `dbus-monitor` or `gdbus` etc.

Arkq, 2021-07-28:
PR #390 has been merged into the master, so right now it is possible to forward volume information to ALSA mixer (I know that that is not what you want, but it might help). So, in order to be compatible with upstream bluez-alsa you can create a software control plugin for ALSA which will forward volume to your hardware device. Then you can use bluealsa-aplay as follows:
```
bluealsa-aplay --mixer=my-plugin --mixer-name=my-volume-element
```

Other option is to modify the code which is responsible for forwarding volume to ALSA mixer in bluealsa-aplay and instead of using ALSA API launch your custom control application.

I'm closing this issue, because right now it should possible to achieve what you want with built-in options (there is no need for any extra feature in bluez-alsa).

closed 2021-07-28 by Arkq