arkq/bluez-alsa #467: Build warning: overflow in conversion

skidoo23, 2021-07-01:
Building bluez-alsa with Debian testing (i386, gcc version 10.2.1 20210110 (Debian 10.2.1-6)) results in a warning.

```
../../../src/asound/bluealsa-pcm.c: In function ‘io_thread’:
../../../src/asound/bluealsa-pcm.c:251:21: warning: overflow in conversion from ‘snd_pcm_uframes_t’ {aka ‘long unsigned int’} to ‘snd_pcm_sframes_t’ {aka ‘volatile long int’} changes value from ‘io_hw_ptr = 4294967295’ to ‘-1’ [-Woverflow]
  251 |    pcm->io_hw_ptr = io_hw_ptr = -1;
      |                     ^~~~~~~~~
```

Something to worry about?

Arkq, 2021-07-02:
No, this implicit conversion is there since some time, but gcc 10 reports it (there is more warnings in bluez-alsa which are reported by gcc >= 10). Anyway, it will be nice to add an explicit conversion. Thanks for being vigilant :)

borine, 2021-07-21:
This patch gets rid of that specific warning:
```diff
index 6e42634..7882290 100644
--- a/src/asound/bluealsa-pcm.c
+++ b/src/asound/bluealsa-pcm.c
@@ -207,7 +207,7 @@ static void *io_thread(snd_pcm_ioplug_t *io) {
        /* We update pcm->io_hw_ptr (i.e. the value seen by ioplug) only when
         * a period has been completed. We use a temporary copy during the
         * transfer procedure. */
-       snd_pcm_uframes_t io_hw_ptr = pcm->io_hw_ptr;
+       snd_pcm_sframes_t io_hw_ptr = pcm->io_hw_ptr;
 
        debug2("Starting IO loop: %d", pcm->ba_pcm_fd);
        for (;;) {
@@ -246,7 +246,7 @@ static void *io_thread(snd_pcm_ioplug_t *io) {
                 * zero: XRUN or drained final samples; we set the HW pointer to
                 * -1 to indicate we have no work to do. */
                snd_pcm_uframes_t avail;
-               if ((avail = snd_pcm_ioplug_hw_avail(io, io_hw_ptr, io->appl_ptr)) == 0) {
+               if ((avail = snd_pcm_ioplug_hw_avail(io, (snd_pcm_uframes_t)io_hw_ptr, io->appl_ptr)) == 0) {
                        io_thread_update_delay(pcm, 0);
                        pcm->io_hw_ptr = io_hw_ptr = -1;
                        eventfd_write(pcm->event_fd, 1);
```

commit 1b3e47f2 (2021-08-10): Get rid of GCC warning about integer overflow

Fixes #467
closed 2021-08-10 by Arkq