arkq/bluez-alsa #474: Clicks in a2dp-aac playback

paul-1, 2021-07-18:


Arkq, 2021-07-18:
Hmm, the question is whether your headset correctly understands payload fragmentation. Can you try with some other headset? Because I haven't seen any clicks since I've fixed fragmentation code (few releases ago). I'm using Jabra MOVE headset, and fragmented RTP packets are played correctly. But I suspect that not all sinks (e.g. headsets) understand payload fragmentation, because Apple (i.e. iPhone) sends incorrect RTP packets.... (bluealsa has quirk correction for such packets) and Apple is the major "supporter" for AAC codec.

paul-1, 2021-07-18:
I was a bit surprised when aac was used.  This headset did not advertise aac support.  But I just enabled aac for this release package, previously I was only using sbc.

Anyway to know when a iPhone sends a fragmented packet?   I can try this headset with my iPhone.   The other thing is that the amount of fragmentation is a bit random. Sometimes it goes a few minutes with no fragmenting, the I get several in a row.

paul-1, 2021-07-19:
I misread your post, Upon re-read, BA is correcting the RTP packet, but Apple just sends the incorrect (probably too large) packet.   I tried a different set of earbuds, same problem.   I'm testing the one set of earbuds playing from my iphone, with no issues.

I've been meaning to get a full headset, The Jabra Move is only $30US, so I picked up a set.  Will be here in a couple days.

Arkq, 2021-07-19:
So, you can play fragmented AAC with your earbuds, which normally you are
using with your phone. Do I read you correctly?

And regarding iPhone. It sends not too big packets, but incorrect ones (not
conformant with RTP spec). *RTP* packet has a fragmentation bit (markbit),
which has to be set for not fragmented packets or the last part of
fragmented packet. IPhone never sets this bit, so according to spec, iPhone
sends one big *RTP* packet fragmented over entire transfer :D So, if iPhone
sends such packets *I* assume that there might be headsets which do not
check this markbit. And if I remember correctly, iPhone uses CBR, with such
a bit rate which does not exceed *MTU*, so there is no need for
fragmentation.

paul-1, 2021-07-19:
Yes, the earbuds play fine connected to the iphone.

paul-1, 2021-07-21:
Well I struck out with the Jabra Move.  The set I got only connects with SBC.  (But they sound good for a $30 headset)  So I guess I'm stuck to test with what I have.

The MTU size does appear to be dependent on the devices, both sets of earbuds will pop, but one worse than the other.
A2DP socket MTU: 28: R:672 W:1020
A different set of earbuds that I see more fragmentation.
A2DP socket MTU: 37: R:672 W:675

I guess the smaller write MTU make sense why it's fragmenting more.

Arkq, 2021-07-21:
> Well I struck out with the Jabra Move. The set I got only connects with SBC.

Ohh... So, I guess you've got a new model of Jabra MOVE now called "Style Edition". Actually, I've got two sets of this headset. One older "Jabra MOVE v2.3.0" and one newer "Jabra MOVE v2.5.0". And also to my surprise the v2.5.0 has only SBC... I guess Jabra was trying to sell more of its products based on good reviews (I bought this model because of wirecutter recommendation for budget headset from 5-6 years ago) but now without paying fees for AAC codec license :(.

> I guess the smaller write MTU make sense why it's fragmenting more.

Yes, that's definitely the reason. Try to use bluealsa with `--aac-vbr-mode=0` and in case of constant fragmentation, set lower bitrate in here: https://github.com/Arkq/bluez-alsa/blob/master/src/a2dp.c#L191 (this defines the max bitrate supported by BlueALSA, however, in case where headset/earbuds supports lower bitrate, lower will be selected).

paul-1, 2021-07-21:
Yes, it is the Move v2.5.0  But it was sold as the older Model.  The "Style" edition was $100.   Probably for the same device in a different box.   Oh well.

I did play with the write buffer, https://github.com/Arkq/bluez-alsa/blob/819c47dd573248d5df1c0852ab32db82569d9105/src/ba-transport.c#L559  Increasing it led to a much smaller glitch when there was fragmentation, but there was still an audible click/glitch.   I'll play with the bitrates and see what happens.

paul-1, 2021-07-21:
--acc-vbr-mode=0, appear to fragment every single packet
```
./bluealsa: D: ../../src/a2dp-aac.c:244: Payload fragmentation: extra 40 bytes
./bluealsa: D: ../../src/a2dp-aac.c:244: Payload fragmentation: extra 41 bytes
./bluealsa: D: ../../src/a2dp-aac.c:244: Payload fragmentation: extra 40 bytes
./bluealsa: D: ../../src/a2dp-aac.c:244: Payload fragmentation: extra 41 bytes
./bluealsa: D: ../../src/a2dp-aac.c:244: Payload fragmentation: extra 40 bytes
./bluealsa: D: ../../src/a2dp-aac.c:244: Payload fragmentation: extra 41 bytes
./bluealsa: D: ../../src/a2dp-aac.c:244: Payload fragmentation: extra 41 bytes
./bluealsa: D: ../../src/a2dp-aac.c:244: Payload fragmentation: extra 40 bytes
./bluealsa: D: ../../src/a2dp-aac.c:244: Payload fragmentation: extra 51 bytes
./bluealsa: D: ../../src/a2dp-aac.c:244: Payload fragmentation: extra 30 bytes
./bluealsa: D: ../../src/a2dp-aac.c:244: Payload fragmentation: extra 41 bytes
./bluealsa: D: ../../src/a2dp-aac.c:244: Payload fragmentation: extra 41 bytes
./bluealsa: D: ../../src/a2dp-aac.c:244: Payload fragmentation: extra 40 bytes
./bluealsa: D: ../../src/a2dp-aac.c:244: Payload fragmentation: extra 41 bytes
./bluealsa: D: ../../src/a2dp-aac.c:244: Payload fragmentation: extra 40 bytes
```

Changing AAC_INIT_BITRATE as low as 8000 (I just kept cutting the value in half)  never had any affect.

Arkq, 2021-07-22:
> --acc-vbr-mode=0, appear to fragment every single packet

That's expected. The mode=0 disables variable bit rate.

> Changing AAC_INIT_BITRATE as low as 8000 (I just kept cutting the value in half) never had any affect.

That's not expected. It seems that setting this to a lower value does not decrease encoding bitrate. I will have to investigate that.

Arkq, 2021-07-22:
Yeap, there is a glitch in the code which is responsible for selecting AAC bittrate - headset can select bigger bitrate than bluealsa supports. I will fix that.

Arkq, 2021-07-23:
I've fixed bitrate selection, so now you can edit this line: https://github.com/Arkq/bluez-alsa/blob/master/src/a2dp.c#L323 in order to lower the bitrate for AAC encoder. Also I'd like to see one debug line. When connecting headset (you have to connect from bluealsa side, if headset will connect automatically, there will be no such log line) there should be a line like this:
```
D: ../../src/bluez.c:296: A2DP selected configuration blob [len=6]: 4000848186a0
```
Please post the configuration blob here. If you like you can decode it with `a2dpconf -x <blob>`. The a2dpconf is provided by bluez-alsa when configured with `--enable-a2dpconf`. I'd like to see what max bitrate your headset supports.

paul-1, 2021-07-24:
```
/usr/local/bin/bluealsa: D: ../../src/bluez.c:296: A2DP selected configuration blob [len=6]: 80008484e200

$ a2dpconf -x 80008484e200
MPEG-2,4 AAC <hex:80008484e200> {
  object-type:8 = MPGE2-LC
  sampling-frequency:12 = 48000
  channel-mode:2 = Stereo
  <reserved>:2
  vbr:1 = true
  bit-rate:23 = 320000
}
```
I was still getting clicks with packet fragmenting, so I recompiled with a bitrate of 160000
This time when I connect, I get a blob of
```
/usr/local/bin/bluealsa: D: ../../src/bluez.c:296: A2DP selected configuration blob [len=6]: 800084827100

$ a2dpconf -x 800084827100
MPEG-2,4 AAC <hex:800084827100> {
  object-type:8 = MPGE2-LC
  sampling-frequency:12 = 48000
  channel-mode:2 = Stereo
  <reserved>:2
  vbr:1 = true
  bit-rate:23 = 160000
}
```
I thought it was interesting the bit rate dropped in capabilities.  But that was teh configured rate.  Are you sure you did not want the blob reported on line 150 of bluez.c, I think this is the device capabilities.
```
/usr/local/bin/bluealsa: D: ../../src/bluez.c:150: A2DP peer capabilities blob [len=6]: 80ff84860000

$ a2dpconf -x 80ff84860000
MPEG-2,4 AAC <hex:80ff84860000> {
  object-type:8 = MPGE2-LC
  sampling-frequency:12 = 48000 44100 32000 24000 22050 16000 12000 11025 8000
  channel-mode:2 = Stereo
  <reserved>:2
  vbr:1 = true
  bit-rate:23 = 393216
}
```

Then I ran bluealsa with --aac-vbr-mode=0, and I get no packet fragmentation with this headset.

I grabbed a different set of earbuds, with bluealsa running in the same mode.  These are the ones with the smaller MTU
MTU: 37: R:672 W:675, they do not fragment either at 160000 bit rate.   I then recompiled with a bit rate of 256000, but that caused fragmentation with this headset.

Arkq, 2021-07-24:
> Are you sure you did not want the blob reported on line 150 of bluez.c

Yes, you are right, I've copied the wrong line as an example.

> These are the ones with the smaller MTU MTU: 37: R:672 W:675, they do not fragment either at 160000 bit rate. I then recompiled with a bit rate of 256000, but that caused fragmentation with this headset.

OK, that's great. Right now I'm working on changing AAC bitrate via command line option. I hope it will be ready in few days.

paul-1, 2021-07-24:
That's a reasonable option, but it would be nice to be able to set dynamically per device.  Just looking for easier ways to allow users to adjust based on their hardware.   IMHO AAC does sounds a bit better than SBC, so just looking how to optimize for the user experience.

Arkq, 2021-07-24:
>  but it would be nice to be able to set dynamically per device.

Yeah, that would be nice, but I'm afraid it's not possible (at least not possible in a deterministic way). The problem is that the A2DP capabilities for particular endpoint (codec) are kept in BlueZ. When remote device connects it asks BlueZ about its capabilities. BlueZ sends that to the device and the device selects configuration which is within BlueZ capabilities. BlueALSA is not involved. When BlueZ wants to connect with remote device it asks remote device for its capabilities and then asks BlueALSA to select configuration. So, only in the latter case we can select bitrate per device, because we control what will be selected. For former case its not up to BlueALSA what will be select. BlueALSA only needs to tell what capabilities it supports and such capability blob will be advertised to any remote device.

> IMHO AAC does sounds a bit better than SBC

That's my impression as well. But to be honest SBC can sound good as well. All depends on selected configuration. Anyway, I'm using AAC because it supports VBR which greatly reduce overall transmission bitrate so there is less dropped frames in case of bad connection (transmitter and receiver are far apart).

paul-1, 2021-07-24:
Understood.

Arkq, 2021-07-24:
You may try to check PR https://github.com/Arkq/bluez-alsa/pull/477 I've fixed a little bit AAC bitrate handling. From now on by default BlueALSA will use AAC CBR mode with bitrate up to 220 kbps. This should produce RTP frames of 651 bytes. This PR is not ready yet - I have to update manuals for new options :)

paul-1, 2021-07-25:
here is the a2dp config for the default state.
```
a2dpconf -x 800084035b60
MPEG-2,4 AAC <hex:800084035b60> {
  object-type:8 = MPGE2-LC
  sampling-frequency:12 = 48000
  channel-mode:2 = Stereo
  <reserved>:2
  vbr:1 = false
  bit-rate:23 = 220000
}
```

The new command line options appear to do what they are supposed to.
```
a2dpconf -x 80008483e800
MPEG-2,4 AAC <hex:80008483e800> {
  object-type:8 = MPGE2-LC
  sampling-frequency:12 = 48000
  channel-mode:2 = Stereo
  <reserved>:2
  vbr:1 = true
  bit-rate:23 = 256000
}
```

I'll do some listening with various devices to see how it works/sounds.

On a side note, I'm also going to give users the ability to set the codec.    Do I need to validate anything, or a certain amount of delay before I call method org.bluealsa.PCM1 - SelectCodec

Arkq, 2021-07-25:
> Do I need to validate anything, or a certain amount of delay before I call method org.bluealsa.PCM1 - SelectCodec

You don't need to but if the configuration is invalid you might end up with A2DP profile being disconnected. As for a delay I'm not sure, you will have to test that. I would say that you can call SelectCoded just after receiving PCMAdded signal.

Arkq, 2021-07-25:
For the record, I've also checked AAC fragmentation with my Sony WF-1000X earbuds and everything works correctly. This earbuds reports MTU like this: `A2DP socket MTU: 18: R:672 W:672`

paul-1, 2021-07-25:
I think we're good now.  I'll get some more feedback once I get the package updates distributed to users.

commit 2c2605be (2021-07-25): Fix A2DP specification conformance for AAC codec

According to the A2DP spec, AAC bitrate field in the A2DP configuration
shall control CBR bit rate and the max peak bit rate for VBR mode. This
commit fixes previous incorrect behavior and removes the --aac-vbr-mode
command line option in favor of --aac-bitrate and --aac-vbr options.

Fixes #474
commit 721995a8 (2021-07-26): Fix A2DP specification conformance for AAC codec

According to the A2DP spec, AAC bitrate field in the A2DP configuration
shall control CBR bit rate and the max peak bit rate for VBR mode. This
commit fixes previous incorrect behavior and removes the --aac-vbr-mode
command line option in favor of --aac-bitrate and --aac-vbr options.

Fixes #474
PR #477 (2021-07-26): Fix A2DP specification conformance for AAC codec
According to the A2DP spec, AAC bitrate field in the A2DP configuration
shall control CBR bit rate and the max peak bit rate for VBR mode. This
commit fixes previous incorrect behavior and removes the --aac-vbr-mode
command line option in favor of --aac-bitrate and --aac-vbr options.

Fixes #474
closed 2021-07-26 by Arkq