arkq/bluez-alsa #475: Compile error missing bsd/sys/time.h include

hrc-amilcar, 2021-07-19:
Hi all,

I'm getting the following error when attempting to compile for my platform:
```
../../src/ba-transport.c:27:10: fatal error: bsd/sys/time.h: No such file or directory
   27 | #include <bsd/sys/time.h>
      |          ^~~~~~~~~~~~~~~~
```

My platform does not have libbsd available during configuration:
```
checking for libbsd >= 0.8... no
```

Removing line 27 from `ba-transport.c` fixes the issue.

Arkq, 2021-07-19:
Upss... my bad :/ I will remove it :)

hrc-amilcar, 2021-07-19:
No worries! :)

As an unrelated side note. I'm also getting these as warnings in a few places:
```
bluez.c:996:6: warning: ‘g_memdup’ is deprecated: Use 'g_memdup2' instead [-Wdeprecated-declarations]
  996 |      g_memdup(&addr, sizeof(addr)), seps = g_array_new(FALSE, FALSE, sizeof(sep)));
      |      ^~~~~~~~
```
```
bluez.c:147:2: warning: ‘g_memdup’ is deprecated: Use 'g_memdup2' instead [-Wdeprecated-declarations]
  147 |  capabilities = g_memdup(data, size);
      |  ^~~~~~~~~~~~
```

Arkq, 2021-07-19:
Yes, I've updated glibc on my machine few days ago, and I've also noticed
these  warnings. However, I'm not sure how to get rid of them. Replacing
memdup with memdup2 is not possible, because that would force glibc>=x.y.
*I* thought about some macro which defines memdup as memdup2 if glibc is
new enough. But I'm not sure...

closed 2021-07-20 by Arkq

hrc-amilcar, 2021-08-11:
Hey @Arkq, there seem to be a few `g_memdup` left in the latest master 7d2a440c

One in `dbus.c`:
```
src/dbus.c:73:4: warning: ‘g_memdup’ is deprecated: Use 'g_memdup2' instead [-Wdeprecated-declarations]
   73 |    void *userdata = g_memdup(&data, sizeof(data));
      |    ^~~~
```
And a couple in `bluez.c`:
```
src/bluez.c:951:6: warning: ‘g_memdup’ is deprecated: Use 'g_memdup2' instead [-Wdeprecated-declarations]
  951 |      sep.capabilities = g_memdup(data, sep.capabilities_size);
      |      ^~~

src/bluez.c:982:6: warning: ‘g_memdup’ is deprecated: Use 'g_memdup2' instead [-Wdeprecated-declarations]
  982 |      g_memdup(&addr, sizeof(addr)), seps = g_array_new(FALSE, FALSE, sizeof(sep)));
      |      ^~~~~~~~
```

Arkq, 2021-08-11:
Yes, I know :) I will refactor the code around these memdup calls, but it's not a priority. Right now it's possible that BlueALSA will report duplicated SEPs, so while resolving this issue I will also get rid of these warnings as well. Anyway, these warnings are not harmful, there is more serious bug in there (memory leak), but it's not reported by GCC...

hrc-amilcar, 2021-08-11:
No worries! Memory leak is certainly a bigger deal.

commit 56d0f33d (2021-07-20): Store A2DP configuration as a union of all codecs

Also this commit fixes build break caused by missing libbsd header
file - left over from 076c837.

Fixes #475
commit 819c47dd (2021-07-20): Store A2DP configuration as a union of all codecs

Also this commit fixes build break caused by missing libbsd header
file - left over from 076c837.

Fixes #475
PR #476 (2021-07-20): Store A2DP configuration as a union of all codecs
Also this commit fixes build break caused by missing libbsd header
file - left over from 076c837.

Fixes #475