arkq/bluez-alsa #478: a2dp Breaks Connectability of Other Bluetooth Device

Fredrum, 2021-07-26:
Hello!!
I am still learning about all this bluetooth and sound stuff so apologies if this is not the right place to ask this question!

The basic problem is that I can get my bt headphones (Apple AirPods) to work and I can get my bt game controller (Sony DS4) to work but I can't get them both to work at the same time.

I'm using a RaspberryPi4, RaspbianOS 32bit, Kernel 5.4 and 5.10 tested.
Using bluealsa from standard apt install.  PulseAudio is un-installed.

I believe I have narrowed it down to being something to do with a a2dp plugin.

I have read and followed a lot of help pages on the web but nothing has made a difference to this problem. I have re-flashed my RPi4 sd card twice to start over.

The following BREAKS Sony game controller bluetooth connectability, but make the Bluetooth Audio Headphones connect correctly.

When I do the following it BREAKS Sony game controller bluetooth connectability. 
But make the Bluetooth Audio Headphones connect fine.

`sudo nano /etc/systemd/system/bluetooth.target.wants/bluetooth.service`
to....
`ExecStart=/usr/lib/bluetooth/bluetoothd --noplugin=sap --plugin=a2dp`

And the following BREAKS the Bluetooth Audio Headphones connectability.
But makes the Sony game controller connect correctly.

`sudo nano /etc/systemd/system/bluetooth.target.wants/bluetooth.service`
to....
`ExecStart=/usr/lib/bluetooth/bluetoothd --noplugin=sap`

Does anyone know what this might mean and how I could get both devices to work at the same time?

Cheers!

borine, 2021-07-26:
> Using bluealsa from standard apt install.

The Raspbian/RaspiOS bluealsa package is *very* old - you would need to contact The RaspiOS team to get support for it. So many changes have been made here since then, that realistically the best advice I can give to fix issues with it is to build from source. Having said that, I am surprised that you would add `--plugin=a2dp` to the bluetoothd command line. That tells bluetoothd to load **only** the a2dp plugin -its unlikely to be what you really want. Bluetoothd supports A2DP automatically, provided some service such as bluealsa has registered itself to implement it. Sounds to me like you have bigger issues than just an out-of-date bluealsa package. So perhaps you should first report the issue to a Raspberry Pi forum to see if there is some specific issue with their bluetooth package or configuration. If they cannot help, then perhaps build bluealsa from source with debug enabled, follow the guidance on the wiki here, and report back here any issues with debug logs.

BTW If you are using the Desktop or Full image that includes pulseaudio, it might be worth testing with pulse first, as that is how their system is now designed. If pulse works correctly then at least you know that the bluetooth stack is set up correctly before beginning to investigate bluealsa.

Fredrum, 2021-07-26:
Hi and thanks for the pointers! I'm using this with a 'Retropie' image and they say that they only support sound over ALSA so I abandoned my pulseaudio exploration.  I also posted on the Raspberry forums but there has been no reply yet. There's many posts about BT problems so not sure if they have people with the right experience.
It sounds like I want to try to build a new version then. As if I remove the '--plugin=a2dp' flag then bt headphones won't do the Connect step.
Not sure when I'll be able to get to the rebuild as there seems to be quite a few dependencies needed but I'll post back here with results if I do.

borine, 2021-07-26:
> if I remove the '--plugin=a2dp' flag then bt headphones won't do the Connect step.

That surely means something is wrong. That's not how it is supposed to work with Bluez 5. I would also recommend to remove the pairing info (bluetoothctl remove) before doing any new test just to make sure that pairing has happened with your current build of bluealsa and install of bluetooth, and that any errors caused by previous setups are not remaining in the bluetoothd database.

borine, 2021-08-10:
Thinking again about this issue, the symptoms you report could result if the DS4 driver or service registers itself as an A2DP endpoint, because that would then prevent bluealsa from doing so. Starting bluetoothd with `--plugin=a2dp` disables support for bluetooth input devices, so the DS4 service fails to start, allowing bluealsa to become the A2DP endpoint.

Fredrum, 2021-08-11:
Hm interesting even if I don't fully foloow yet I'll have to re-read some more times I think.
I stumbled on something that made it work for me without trying to build bluealsa from source. You can see my personal wip step-by-step here and see if it matches your theory.
https://github.com/Fredrum/retropiescripts/wiki/Bluetooth---Both-Controllers-and-Audio

The DS4 controller actually does transmit Audio over Bluetooth when used with the actual PS4. Io driver I'm aware of have managed to do his on any other OS/system. I believe Sony uses a custom protocol for the audio stream, something like that.

You can check out the Linux driver (by Sony) that I run for my DS4 controller,
https://github.com/torvalds/linux/blob/master/drivers/hid/hid-sony.c

borine, 2021-08-11:
> I believe Sony uses a custom protocol for the audio stream

Please can you post the output of
```
bluetoothctl info xx:xx:xx:xx:xx:xx | grep UUID
```
where xx:xx:xx:xx:xx:xx is the bluetooth address of the DS4. That will tell which profiles are supported (and can be connected). We are looking for `UUID: Audio Sink      (0000110b-0000-1000-8000-00805f9b34fb)` which tells us that bluez has recognized it as an A2DP sink.

Fredrum, 2021-08-12:
I only get these two,

UUID: Human Interface Device... (00001124-0000-1000-8000-00805f9b34fb)
UUID: PnP Information           (00001200-0000-1000-8000-00805f9b34fb)