arkq/bluez-alsa #480: Can't get audio to and from two Bluetooth Headsets simulataneously

tristonn, 2021-07-29:
I have tried using the dmix and dsnoop plugins to go about doing this similar to what is shown in the following post: ALSA Api: How to play two wave files simultaneously? And have looked at the following sources: https://www.alsa-project.org/main/index.php/Asoundrc#Software_mixing, and https://github.com/Arkq/bluez-alsa/wiki/Using-the-bluealsa-ALSA-pcm-plugin. All I want to do is use the arecord and aplay commands to input and output to these two Bluetooth devices at the same time. 

When I ran: aplay -D mix_stream /home/pi/Desktop/noise.wav
Error: ALSA lib pcm_direct.c:1809:(_snd_pcm_direct_get_slave_ipc_offset) Invalid type 'bluealsa' for slave PCM
aplay: main:828: audio open error: Invalid argument

Current asoundrc file:
`pcm.mix_stream {
    type plug
    slave.pcm dmix_stream
}

pcm.dmix_stream {
    type dmix
    ipc_key 1234
    slave.pcm y30
    slave.pcm seshevo
}

pcm.y30 {
    type plug
    slave.pcm {
            type bluealsa
            device "FA:D8:78:FB:57:36"
            profile "a2dp"
    }
}

pcm.seshevo {
    type plug
    slave.pcm {
            type bluealsa
            device "5C:44:3E:54:E0:01"
            profile "a2dp"
    }
}`

graham8, 2021-07-30:
dmix and dsnoop allow more than one application to share the same alsa device.  It looks like what you want is the multi plug-in which allows one application to output to play to two devices. There's a lot of advice around on that.  I haven't looked into the equivalent for combining input devices.

dmix and dsnoop only work with hardware devices.  They don't work with bluez-alsa.

borine, 2021-07-30:
> There's a lot of advice around on that.

see, for example:
https://github.com/Arkq/bluez-alsa/issues/379#issuecomment-707309026
https://github.com/Arkq/bluez-alsa/issues/309#issuecomment-609427129

tristonn, 2021-07-31:
Okay, thanks that enabled me to play a wave file and output it to my two Bluetooth headsets. But do you have any suggestions as far as using arecord to record those two Bluetooth devices? My current . asoundrc file I will post below which I referenced off of @Arkq. When I run the following command: arecord -D bth-multi -c 2 -r 44100 -f s16_le test.wav. It outputs the following error: ALSA lib bluealsa-pcm.c:763:(_snd_pcm_bluealsa_open) Couldn't get BlueALSA PCM: PCM not found
arecord: main:828: audio open error: No such device. 

'''

pcm.bth-1 {
    type plug
    slave.pcm {
            type bluealsa
            device "FA:D8:78:FB:57:36"
            profile "a2dp"
    }
}

pcm.bth-2 {
    type plug
    slave.pcm {
            type bluealsa
            device "5C:44:3E:54:E0:01"
            profile "a2dp"
    }
}

pcm.bth-multi {
    type plug;
    slave.pcm {
         type multi;
         slaves.a.pcm "bth-1";
         slaves.a.channels 2;
         slaves.b.pcm "bth-2";
         slaves.b.channels 2;
         bindings.0 { slave a; channel 0; }
         bindings.1 { slave a; channel 1; }
         bindings.2 { slave b; channel 0; }
         bindings.3 { slave b; channel 1; }
         master 1
    }
    ttable.0.0 1
    ttable.1.1 1
    ttable.0.2 1
    ttable.1.3 1
}

'''

borine, 2021-08-10:
> do you have any suggestions as far as using arecord to record those two Bluetooth devices?

Are you sure that the two devices actually have A2DP from their microphones? Very often bluetooth headsets use only SCO (HFP or HSP) for the microphone link, designed for use only when making 'phone calls - their A2DP support in that case is only for music playback.

borine, 2021-11-25:
@tristonn is this issue still of interest to you? There has been no activity here for several weeks now, so I would like to close it if no-one has anything else to add. I think that recording from two bluetooth devices is the same as recording from two soundcard devices - you just need 2 instances of arecord, one recording from each device. Have you tried that?

borine, 2021-12-14:
This issue appears to be abandoned. Closing.

closed 2021-12-14 by borine